fwupd/fwupd #7574: Regression -- UEFI update failed (Thinkpad t14s) -- Failed to find 'HD(1,GPT...)...cap' DevicePath

gdonval, 2024-08-06:
**Describe the bug**

Re #6023 (maybe not the same cause though). Back then, we figured assuming 512b block sizes was wrong and caused the problem.

Here fwupd 1.9.22 fails to update:

    WARNING: Failed to find 'HD(1,GPT,0C8FC2F7-EF45-472E-BCAA-D5691BD44E83)/\EFI\arch\fw\fwupd-<...>.cap' DevicePath
    WARNING: Could not locate device handle: Unsupported

(`<...>` is an actual guid).

**Steps to Reproduce**

* fwupdmgr refresh
* fwupdmgr get-updates
* fwupdmgr update
* Reboot when prompted.

**Expected behavior**

The cap files should be found and the system should be updated.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.22
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.10.2-arch1-1
runtime   org.freedesktop.fwupd         1.9.22
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

LENOVO 21F9S1JQ00
│
├─USB3.1 Hub:
│ │   Device ID:          48cdf390f0a1768e3fb0bcb10af8fcfb38e5d89e
│ │   Summary:            USB 3.x hub
│ │   Current version:    13.24
│ │   Vendor:             VIA Labs, Inc. (USB:0x17EF)
│ │   Install Duration:   15 seconds
│ │   GUIDs:              227d0f95-a764-598e-85de-3d2bd0a7372f
│ │                       c6381f16-fe6d-5c45-a4ad-b619af4f0f0e ← USB\VID_17EF&PID_A391
│ │                       fc6cfb35-d4e9-5467-a014-8863de104969 ← USB\VID_17EF&PID_A391&HUB_000A
│ │                       ef65710a-3fa0-5330-9a61-5ced5e976be1 ← USB\VID_17EF&PID_A391&SPI_C220
│ │                       7ae3100f-9b01-571c-9fc2-20cd4544123a ← USB\VID_17EF&PID_A391&SPI_C220&REV_0D24
│ │   Device Flags:       • Updatable
│ │                       • Cryptographic hash verification is available
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
[... 354 lines cut ...]
│   │   └── BOOTX64.EFI
│   ├── Linux
│   │   ├── archlinux.efi
│   │   ├── archlinux-fallback.efi
│   │   ├── arch-linux-lts.efi
│   │   └── arch-linux-lts-fallback.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   └── fwupd.conf
    ├── entries.srel
    ├── loader.conf
    └── random-seed

```

**Additional questions**

- Operating system and version: Arch (rolling)
- Have you tried rebooting? Yep
- Is this a regression? Yep
- Are you using an NVMe disk? Yep
- Is secure boot enabled? Yep
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Nope

hughsie, 2024-08-06:
Can you get the `efibootmgr -v` when using the old version of fwupd that does work please?

gdonval, 2024-08-06:
Good news! This might have been a archlinux problem: I had the problem on Friday and still yesterday (after update, reboot, etc.). 

The problem was still present this morning but I updated right after I reported the bug and that did the trick.

The latest version, 1.9.23-1, released late yesterday, fixed it.

Thanks again for your work and super fast response.

closed 2024-08-06 by gdonval