fwupd/fwupd #7592: Translations not updated on Transifex

haaninjo, 2024-08-10:
**Describe the bug**
Translations for fwupd have not been updated for a few releases.

On https://app.transifex.com/freedesktop/fwupd/main/ we can see that the English strings for translation there were last updated Dec 31st 2023, 15:40

Statistics on https://l10n.gnome.org/module/fwupd/ that updates automatically from the source code show that there are about 30 strings untranslated for all languages.

hughsie, 2024-08-11:
I've pushed them now; longer term we should do this as a CI action I suppose.

haaninjo, 2024-08-11:
Thanks!

Saw a typo immediately policy/org.freedesktop.fwupd.policy.in:135  

"reest daemon configuration" -> "reset"

As for an automatic translation solution, l10n.gnome.org now supports pushing translations to github repos as well if [gnomesysadmins](https://github.com/orgs/GNOME/people/gnomesysadmins) is added as a developer.

Here is a report when it was added for udisks: https://gitlab.gnome.org/Infrastructure/damned-lies/-/issues/383

hughsie, 2024-08-11:
I don't know how well l10n.gnome.org+Transifex would work -- I fear the gnome supplied ones would be overwritten by `tx pull`

haaninjo, 2024-08-11:
In that case it would be skipping Transifex altogether. Translation teams can send translations directly from l10n.gnome.org now for e.g udisks.

hughsie, 2024-08-11:
But... we have been using transifex since ~2017... I'm not sure what problem is trying to be solved here?

haaninjo, 2024-08-12:
Any automated solution sounds good to me :+1: . Just mentioned it in case you were tired of transifex, I've seen other projects that have had some problems.

closed 2024-11-11 by hughsie