fwupd/fwupd #7593: [redfish plugin] potential issue that be unable to exit 'fwupd' install excution

vincent-tang1, 2024-08-12:
**Describe the bug**
Sometimes, after fwupd finished firmware writing via redfish protocol, then entered 'wait' status, but for very long time, it still be unable to exit this excution. 

**Steps to Reproduce**
1. Upgrade BMC firmware via redfish protocol, then can see that it was being written
```
(fwupd) root@vincent:/home/vincent/fwupd# fwupdmgr reinstall
WARNING: This package has not been validated, it may not work properly.
Selected device: FWA6172 Platform Management
╔══════════════════════════════════════════════════════════════════════════════╗
║ Reinstall FWA6172 Platform Management to 1.07.0000?                          ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This unstable release fixes the following issues:                            ║
║                                                                              ║
║ • Fix bug for test 1                                                         ║
║ • Improve performance for test 1                                             ║
║                                                                              ║
║ Some new functionality has also been added:                                  ║
║                                                                              ║
║ • Add support for test 1                                                     ║
║                                                                              ║
║ FWA6172 Platform Management and all connected devices may not be usable      ║
║ while updating.                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: Y
Writing…                 [********************************       ] Less than one minute remaining…
```
2. After finished writing firmware, then entered to 'Restarting device' status and then to 'Waiting' finnally, but for very long time, it still be unable to exit this excution. 
```
Restarting device…       [***************************************] Less than one minute remaining…
Waiting…                 [***************************************] Less than one minute remaining…
```

**Expected behavior**
fwupd install excution can close normally with reasonalbe time

**Root cause analysis**
After finished the firmware writing, then BMC will reboot. During BMC boot, 'fwupd' will try connect BMC and wait BMC come back. When BMC finished booting, becacuse 'fwupd' got some problem to lead to unable correctly connect into BMC. As result 'fwup' keep retying same callback function such that can not exit the excuation. For more detail reason, please see below:

1. In x86 host, can see the network interface 'usb4bmc' which is used to communicate to BMC Redfish service (eno1 & eno2 for out of band usage, usb4bmc for in-band usage)
```
(fwupd) root@vincent:/home/vincent/fwupd# ifconfig
eno1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.17.10.74  netmask 255.255.254.0  broadcast 172.17.11.255
        inet6 fe80::c600:adff:feb8:be36  prefixlen 64  scopeid 0x20<link>
        ether c4:00:ad:b8:be:36  txqueuelen 1000  (Ethernet)
        RX packets 327267  bytes 39696766 (39.6 MB)
        RX errors 0  dropped 145391  overruns 0  frame 0
        TX packets 14806  bytes 3141812 (3.1 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0x96100000-9617ffff

eno2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.17.10.73  netmask 255.255.254.0  broadcast 172.17.11.255
        inet6 fe80::c600:adff:feb8:be37  prefixlen 64  scopeid 0x20<link>
        ether c4:00:ad:b8:be:37  txqueuelen 1000  (Ethernet)
        RX packets 14055  bytes 1110803 (1.1 MB)
        RX errors 0  dropped 3830  overruns 0  frame 0
        TX packets 73  bytes 6520 (6.5 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0x96200000-9627ffff

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 241  bytes 21713 (21.7 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 241  bytes 21713 (21.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
[... 156 lines cut ...]
│     Device ID:          62c1cd95692c5225826cf8568a460427ea3b1827
│     Summary:            Redfish host bus adapter
│     Current version:    1.07.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              10598c6b-6f24-518d-9065-bb948baa4907 ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-BMC
│                         69ec5e09-0466-5679-83f5-0dc75e3ff4d5 ← REDFISH\VENDOR_Advantech&ID_BMC
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│
├─NVRAM:
│     Device ID:          e62d2a3a9c875839f7b57843190f695bca7f222f
│     Summary:            Redfish host bus adapter
│     Current version:    3.00.0000
│     Vendor:             Advantech (REDFISH:ADVANTECH)
│     GUIDs:              30a77683-3cfe-5fe8-b2ed-b89f0ba2a16d ← REDFISH\VENDOR_Advantech&SOFTWAREID_894feb65-2ca8-4378-a5d3-41c78ff5b59f-NVRAM
│                         afe3785d-abe4-546c-8889-1d3fbe16e6c1 ← REDFISH\VENDOR_Advantech&ID_NVRAM
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
...
```
**Additional questions**
- Operating system and version: ubuntu 24.04 LTS

vincent-tang1, 2024-08-16:
I think it should be common issue.
not sure if above information is enough, or do you want to know more other furthermore information? thanks a lot ^^

superm1, 2024-08-16:
> I think it should be common issue.
> not sure if above information is enough, or do you want to know more other furthermore information? thanks a lot ^^

Can you please upgrade to current tip and see if it still reproduces? Richard had some clean ups the last week or so that I think it's worth double checking.

closed 2024-08-18 by hughsie

vincent-tang1, 2024-08-20:
> Can you please upgrade to current tip and see if it still reproduces? Richard had some clean ups the last week or so that I think it's worth double checking.

Yes, and after applied Hughsie patch, this issue already solved

vincent-tang1, 2024-08-20:
thanks for the commit!
And not sure if there is a opportunity that this solution can also can merged into Ubuntu 24.04 LTS distrubution ?

hughsie, 2024-08-20:
> this solution can also can merged into Ubuntu 24.04 LTS distrubution

No idea about when Ubuntu would rebase, but I've backported it into 1_9_X for the next release.

vincent-tang1, 2024-08-20:
not sure if my understanding is correct.

1. So next release fwupd v1.9.24 will included this solution, right ?
2. If Ubuntu 24.04 can rebase in the future, it can change into new fwupd version 1.9.16-2 from current 1.9.16-1 and include the solution, correct ?

Many thanks

hughsie, 2024-08-20:
> So next release fwupd v1.9.24 will included this solution, right

Right.

> If Ubuntu 24.04 can rebase in the future, it can change into new fwupd version

Ubuntu would need to rebase to v1.9.24 -- or backport the patch that fixes the issue. Either way you should file a bug report with them in launchpad.

vincent-tang1, 2024-08-20:
Got thanks! From below [table](https://wiki.ubuntu.com/Releases),
1. Not sure if new Ubuntu 24.10 version will use fwupd v1.9.24 or above version ?
2. Will Ubuntu 24.04.1 LTS version rebase in 8/29/2024 ?

[image]

hughsie, 2024-08-20:
No idea! I work for Red Hat not Canonical :)

vincent-tang1, 2024-08-20:
> Ubuntu would need to rebase to v1.9.24 -- or backport the patch that fixes the issue. Either way you should file a bug report with them in launchpad.

Created the request into launchpad
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2077411

vincent-tang1, 2024-08-23:
>Created the request into launchpad
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/2077411

@superm1 thanks your update in the Launchpad ticket
Seems like ubuntu 24.10 (Oracular) already rebased into fwupd v1.9.24, right ? 
However, ubuntu 24.04 (Noble) have not rebased into the v1.9.24 yet because official site still be reviewing it, correct ? thanks

[image]

superm1, 2024-08-23:
Canonical sru team will decide whether to accept 1.9.24. I can't force them.

vincent-tang1, 2024-08-23:
got it, many thanks :)

superm1, 2024-08-26:
@vincent-tang1 they accepted it, but it needs to be validated.  Can you please test it and then leave the comment on the bug if it works?

vincent-tang1, 2024-08-27:
@superm1 thanks for the reminder
Following their test proposed package, verified result as work  and left the comment on the bug