fwupd/fwupd #7598: Ux-capsule is not appended with firmware  capsule

jebasimmanuvel07, 2024-08-13:
**Describe the bug**

In **plugins/uefi-capsule/fu-uefi-common.c**

```
gboolean
fu_uefi_get_framebuffer_size(guint32 *width, guint32 *height, GError **error)
{
	guint32 height_tmp;
	guint32 width_tmp;
	g_autofree gchar *sysfsdriverdir = NULL;
	g_autofree gchar *fbdir = NULL;

	sysfsdriverdir = fu_path_from_kind(FU_PATH_KIND_SYSFSDIR_DRIVERS);
	fbdir = g_build_filename(sysfsdriverdir, "efi-framebuffer", "efi-framebuffer.0", NULL);
	if (!g_file_test(fbdir, G_FILE_TEST_EXISTS)) {
		g_set_error_literal(error,
				    FWUPD_ERROR,
				    FWUPD_ERROR_INVALID_DATA,
				    "EFI framebuffer not found");
		return FALSE;
	}
```
Framebuffer's width, height are taken from  **efi-framebuffer.0** . code is searching efi-framebuffer.0 in sys/bus/platform/drivers/efi-framebuffer folder.

If **efi-framebuffer.0** is not present on that folder, code returns "EFI framebuffer not found" which cause Ux-capsule is not appended with firmware capsule.

I have tested firmware update in ubuntu 22.04.  In that, Ux capsule was not appended with firmware capsule because of following error.  

```

Scheduling…              [  -                                    ]12:11:15:0303 FuPluginUefiCapsule  failed to upload UEFI UX capsule text: EFI framebuffer not found
Decompressing…           [  -                                    ]12:11:15:0303 FuDevice             installing onto a45df35ac0e948ee180fe216a5f703f32dda163f:
<firmware>
  <data size="0xa01000">..&lt;J#w.H.=W....M</data>
</firmware>
```

Have checked and  there is no **efi-framebuffer.0** in sys/bus/platform/drivers/efi-framebuffer folder but /dev/fb0 is present. 

Checked in kernal logs , EFI Framebuffer had been initialized successfully.  

```
[    0.508054] efifb: probing for efifb
[    0.508071] efifb: showing boot graphics
[    0.508870] efifb: framebuffer at 0xa0000000, using 8100k, total 8100k
[    0.508871] efifb: mode is 1920x1080x32, linelength=7680, pages=1
[    0.508873] efifb: scrolling: redraw
[    0.508873] efifb: Truecolor: size=8:8:8:8, shift=24:16:8:0
```
 It indicates that the EFI framebuffer is likely being initialized, but the system is not naming it explicitly as **efi-framebuffer.0**. Instead, it's being registered under the more generic name of **/dev/fb0** (EFI framebuffer is present but not explicitly named as efi-framebuffer.0) 
 
Even EFI framebuffer is present but if not explicitly named as efi-framebuffer.0, ux-capsule won't be appended with firmware capsule.

**Steps to Reproduce**
Try firmware update in ubuntu 22.04

**Expected behavior**
Ux capsule have to be appended with firmware capsule

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.5.0-44-generic
[... 328 lines cut ...]
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-6.5.0-44-generic
├── initrd.img-6.5.0-41-generic
├── initrd.img-6.5.0-44-generic
├── initrd.img.old -> initrd.img-6.5.0-41-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-6.5.0-41-generic
├── System.map-6.5.0-44-generic
├── vmlinuz -> vmlinuz-6.5.0-44-generic
├── vmlinuz-6.5.0-41-generic
├── vmlinuz-6.5.0-44-generic
└── vmlinuz.old -> vmlinuz-6.5.0-41-generic
**Additional questions**

- Operating system and version: Ubuntu 22.04
- Have you tried rebooting? Yes
- Is this a regression?
- Are you using an NVMe disk?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

hughsie, 2024-08-13:
Do you know why the attributes are not being stored in `efi-framebuffer.0`? I took a quick look at `drivers/video/fbdev/efifb.c` but couldn't see anything immediately fishy. @martinezjavier do you know why this might be? I somewhat hoped this kind of attribute in `/sys` was kernel ABI.

jebasimmanuvel07, 2024-08-16:
Hi @hughsie   

I don't know about why the attributes are not being stored in **efi-framebuffer.0** . I think , in latest kernal (may be above v5.15) 
efi-framebuffer.0 is a not a generic place to attributes. 

@martinezjavier If you have any suggestion/comment on this, could you please  post here

Thanks!!!

hughsie, 2024-08-16:
@superm1 any ideas on where else we should be getting the efi framebuffer size from? I was slightly surprised to see it disappear...

superm1, 2024-08-16:
> @superm1 any ideas on where else we should be getting the efi framebuffer size from? I was slightly surprised to see it disappear...

I suspect this is related to changes for using simpledrm.  Or it's an NVIDIA system that doesn't behave like everyone else.  How about?
`/sys/class/graphics/fb0/virtual_size`

hughsie, 2024-08-16:
I've got intel here -- and I also don't see anything in `/sys/class/graphics/*/virtual_size` :(

superm1, 2024-08-16:
How about `/sys/class/graphics/fb0/modes`?  For reference on my side this is everything I see:
```
❯ grep -v /sys/class/graphics/fb0/*
/sys/class/graphics/fb0/dev:29:0
grep: /sys/class/graphics/fb0/device: Is a directory
/sys/class/graphics/fb0/modes:U:2880x1920p-0
/sys/class/graphics/fb0/name:amdgpudrmfb
/sys/class/graphics/fb0/pan:0,0
grep: /sys/class/graphics/fb0/power: Is a directory
/sys/class/graphics/fb0/rotate:0
/sys/class/graphics/fb0/state:0
/sys/class/graphics/fb0/stride:11520
grep: /sys/class/graphics/fb0/subsystem: Is a directory
/sys/class/graphics/fb0/uevent:MAJOR=29
/sys/class/graphics/fb0/uevent:MINOR=0
/sys/class/graphics/fb0/uevent:DEVNAME=fb0
/sys/class/graphics/fb0/virtual_size:2880,1920
```

hughsie, 2024-08-16:
Nope :(

    $ cat /sys/class/graphics/fbcon/
    cursor_blink  power/        rotate        rotate_all    subsystem/    uevent

superm1, 2024-08-16:
Ok, as another idea, can we just use the preferred mode advertised in the edid on the internal panel (if present)?

superm1, 2024-08-16:
Example, in `drm_info` can find the primary one and the preferred modes

```
Node: /dev/dri/card0
├───Driver: amdgpu (AMD GPU) version 3.58.0 (0)
│   ├───DRM_CLIENT_CAP_STEREO_3D supported
│   ├───DRM_CLIENT_CAP_UNIVERSAL_PLANES supported
│   ├───DRM_CLIENT_CAP_ATOMIC supported
│   ├───DRM_CLIENT_CAP_ASPECT_RATIO supported
│   ├───DRM_CLIENT_CAP_WRITEBACK_CONNECTORS supported
│   ├───DRM_CAP_DUMB_BUFFER = 1
│   ├───DRM_CAP_VBLANK_HIGH_CRTC = 1
│   ├───DRM_CAP_DUMB_PREFERRED_DEPTH = 24
│   ├───DRM_CAP_DUMB_PREFER_SHADOW = 1
│   ├───DRM_CAP_PRIME = 3
│   ├───DRM_CAP_TIMESTAMP_MONOTONIC = 1
│   ├───DRM_CAP_ASYNC_PAGE_FLIP = 1
│   ├───DRM_CAP_CURSOR_WIDTH = 256
│   ├───DRM_CAP_CURSOR_HEIGHT = 256
│   ├───DRM_CAP_ADDFB2_MODIFIERS = 1
│   ├───DRM_CAP_PAGE_FLIP_TARGET = 0
│   ├───DRM_CAP_CRTC_IN_VBLANK_EVENT = 1
│   ├───DRM_CAP_SYNCOBJ = 1
│   └───DRM_CAP_SYNCOBJ_TIMELINE = 1
├───Device: PCI 1002:15bf Advanced Micro Devices, Inc. [AMD/ATI] Phoenix1
│   └───Available nodes: primary, render
├───Framebuffer size
│   ├───Width: [0, 16384]
│   └───Height: [0, 16384]
├───Connectors
│   ├───Connector 0
│   │   ├───Object ID: 93
│   │   ├───Type: eDP
│   │   ├───Status: connected
│   │   ├───Physical size: 290x190 mm
│   │   ├───Subpixel: unknown
│   │   ├───Encoders: {0}
│   │   ├───Modes
│   │   │   ├───2880x1920@120.00 preferred driver nhsync nvsync 
│   │   │   ├───2880x1920@60.00 preferred driver phsync nvsync 
```

superm1, 2024-08-16:
> Example, in `drm_info` can find the primary one and the preferred modes
>
> ```
> Node: /dev/dri/card0
> ├───Driver: amdgpu (AMD GPU) version 3.58.0 (0)
> │   ├───DRM_CLIENT_CAP_STEREO_3D supported
> │ [...]

We're already linking with libdrm too when amdgpu plugin is enabled.
https://github.com/fwupd/fwupd/blob/main/meson.build#L579

So _surely_ we could get that information from a DRM API.

jebasimmanuvel07, 2024-08-20:
@superm1 
Thanks for Quick responce and analysis.

ChristianAtSIE, 2024-08-21:
Hey all,
I encountered this issue too, thanks for report.
I tried to an workaround by changing 
``` c
gboolean
fu_uefi_get_framebuffer_size(guint32 *width, guint32 *height, GError **error)
{
	guint32 height_tmp;
	guint32 width_tmp;
	g_autofree gchar *sysfsdriverdir = NULL;
	g_autofree gchar *fbdir = NULL;

	
	sysfsdriverdir = fu_path_from_kind(FU_PATH_KIND_SYSFSDIR_DRIVERS);
	fbdir = g_build_filename(sysfsdriverdir, "efi-framebuffer", "efi-framebuffer.0", NULL);
	if (!g_file_test(fbdir, G_FILE_TEST_EXISTS)) {
		g_set_error_literal(error,
				    FWUPD_ERROR,
				    FWUPD_ERROR_INVALID_DATA,
				    "EFI framebuffer not found");
		return FALSE;
	}
	height_tmp = fu_uefi_read_file_as_uint64(fbdir, "height");
	width_tmp = fu_uefi_read_file_as_uint64(fbdir, "width");
	if (width_tmp == 0 || height_tmp == 0) {
		g_set_error(error,
			    FWUPD_ERROR,
			    FWUPD_ERROR_INVALID_DATA,
			    "EFI framebuffer has invalid size "
			    "%" G_GUINT32_FORMAT "x%" G_GUINT32_FORMAT,
			    width_tmp,
			    height_tmp);
		return FALSE;
	}
	if (width != NULL)
		*width = width_tmp;
	if (height != NULL)
		*height = height_tmp;
	return TRUE;
}
```
to
``` c
gboolean
fu_uefi_get_framebuffer_size(guint32 *width, guint32 *height, GError **error)
{
	guint32 height_tmp;
	guint32 width_tmp;
	g_autofree gchar *sysfsdriverdir = NULL;
	g_autofree gchar *fbdir = NULL;

	//KCHCH
	//sysfsdriverdir = fu_path_from_kind(FU_PATH_KIND_SYSFSDIR_DRIVERS);
	//fbdir = g_build_filename(sysfsdriverdir, "efi-framebuffer", "efi-framebuffer.0", NULL);
	//if (!g_file_test(fbdir, G_FILE_TEST_EXISTS)) {
	//	g_set_error_literal(error,
	//			    FWUPD_ERROR,
	//			    FWUPD_ERROR_INVALID_DATA,
	//			    "EFI framebuffer not found");
	//	return FALSE;
	//}
	//height_tmp = fu_uefi_read_file_as_uint64(fbdir, "height");
	//width_tmp = fu_uefi_read_file_as_uint64(fbdir, "width");
	//if (width_tmp == 0 || height_tmp == 0) {
	//	g_set_error(error,
	//		    FWUPD_ERROR,
	//		    FWUPD_ERROR_INVALID_DATA,
	//		    "EFI framebuffer has invalid size "
	//		    "%" G_GUINT32_FORMAT "x%" G_GUINT32_FORMAT,
	//		    width_tmp,
	//		    height_tmp);
	//	return FALSE;
	//}
	//if (width != NULL)
	//	*width = width_tmp;
	//if (height != NULL)
	//	*height = height_tmp;
	*width = 800; //KHCH
	*height = 600; //KHCH
	return TRUE;
}
```
The resolution I got from the kernel log `sudo dmesg`
```shell
[    0.504166] efifb: probing for efifb
[    0.504186] efifb: framebuffer at 0x70000000, using 1876k, total 1875k
[    0.504190] efifb: mode is 800x600x32, linelength=3200, pages=1
[    0.504193] efifb: scrolling: redraw
[    0.504194] efifb: Truecolor: size=8:8:8:8, shift=24:16:8:0
[    0.504284] Console: switching to colour frame buffer device 100x37
[    0.504673] fb0: EFI VGA frame buffer device
```

This works, but I encounter another issue:
```shell
16:21:39.233 FuPluginUefiCapsule  framebuffer size 800 x600
16:21:39.237 FuArchive            adding fwupd-ca-640-480.bmp [32262]
...Extracting many more .bmp files...
16:21:39.774 FuArchive            adding fwupd-en-7680-4320.bmp [1325394]
16:21:39.776 FuPluginUefiCapsule  no fwupd-C-800-600.bmp found
16:21:39.776 FuPluginUefiCapsule  no fwupd-C-800-600.bmp found
16:21:39.787 FuPluginUefiCapsule  failed to upload UEFI UX capsule text: failed to get splash file for C.UTF-8,C,C in /home/phoenix/fwupd/venv/bin/../dist/share/fwupd
```
The issue is that my systems environment var is set to 
```shell
echo $LANG
C.UTF-8
```
setting it to `export LANG=en.UTF-8` resolves this issue, but would it not be nicer to set English as default, if no compatible language is set?

As @hughsie mentioned before that he is using Intel:
I am also using Intel with onboard graphics 
`hwinfo` information
6.150.1 "Intel Atom(R) x6425E Processor @ 2.00GHz"

jebasimmanuvel07, 2024-09-05:
Hi @hughsie   

Sorry for bothering you.  

Could you please check @superm1 suggestion and help fix this issue.

martinezjavier, 2024-09-05:
@hughsie sorry for missing your ping. I was on vacation for a few weeks and just came back recently. 

As others pointed out, this change is due distros migrating from the old `efifb` fbdev driver to the new `simpledrm` DRM driver for early graphics. I wonder if maybe could be possible to instead look at what was the size and offsets of the [Boot Graphics Record Table (BGRT)](https://wiki.osdev.org/BGRT)

```
$ file /sys/firmware/acpi/bgrt/image
/sys/firmware/acpi/bgrt/image: PC bitmap, Windows 3.x format, 768 x 256 x 24, image size 589824, cbSize 589878, bits offset 54

$ cat /sys/firmware/acpi/bgrt/{x,y}offset
576
284
```

tdz, 2024-09-05:
@jebasimmanuvel07  @hughsie 

Kernel-side maintainer here. efi-framebuffer.0 is the platform device that represented the firmware's display. When the native graphics driver takes over the graphics hardware, it removes the platform device. So efifb is long gone whne the system runs.

Screen resolutions aren't stable either and depend on hardware and whatever the BIOS/UEFI does. I think Javier's advice of looking at the BGRT is sound.

Alternatively, you could read display modes from the running DRM driver. IIRC, the ioctl is DRM_IOCTL_MODE_GETCONNECTOR. In libdrm, there's struct _drmModeConnector that returns the information. Each connector has a list of modes (struct _drmModeModeInfo). There's likely one mode with DRM_MODE_TYPE_PREFERRED set in mode->type.  That's the one that the firmware will most likely boot up with.

martinezjavier, 2024-09-05:
> Alternatively, you could read display modes from the running DRM driver. IIRC, the ioctl is DRM_IOCTL_MODE_GETCONNECTOR. In libdrm, there's struct _drmModeConnector that returns the information. Each connector has a list of modes (struct _drmModeModeInfo). There's likely one mode with DRM_MODE_TYPE_PREFERRED set in mode->type. That's the one that the firmware will most likely boot up with.
> 
> EDIT: The preferred mode comes from the display's EDID data and usually contains the display's native resolution. The UEFI/BIOS much likely picks up that mode for boot-up output.

I agree with Thomas that is quite likely that both the firmware / EFI-GOP and native DRM driver will pick the same preferred mode from the same EDID data. That seems like the most robust approach.

martinezjavier, 2024-09-05:
In fact, a combination of both methods probably is preferred to also know the vendor logo size and offset, and determine where to place the help text correctly.

521like, 2024-09-12:
HI @hughsie,

Sorry for bothering you.
May I get your current status about this issue?  （”Ux capsule was not appended with firmware capsule“）
Regarding this issue, is there a WA solution that can be provided first?

hughsie, 2024-09-12:
> May I get your current status about this issue

It's on my list of stuff to be fixed -- if you'd like to submit a pull request with the fix I'd happily review it.

memfissoftware, 2024-09-25:
@hughsie for a question, ux-capsule is a must to update bios? because i can see a Flags=no-ux-capsule
(Do not use the additional UX capsule.) here: https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/README.md

what happens if we do not use ux-capsule?

hughsie, 2024-09-25:
> ux-capsule is a must to update bios

In 99% of cases, it's entirely optional.

> what happens if we do not use ux-capsule

The UX capsule is actually just the BMP text along the bottom of the update that says something like "Updating firmware, please don't power-off" rendered in the users language.

ChristianAtSIE, 2024-09-26:
@memfissoftware 
> what happens if we do not use ux-capsule?
From my experience with an AMI based UEFI, if no Ux-Capsule is attached, the BIOS menu is rendered in the before invisible tab 'Recovery' with selection options for 'Override NVRAM', 'Override BIOS setup' and 'Start update'. So manual interaction is needed. By attaching the Ux-Capsule, this step is skipped and the update runs through without interaction.

But I do not have years of experience with UEFI, so I do not know if this is normal behavior. In fact, currently I am currently stuck at the issue, that the NVRAM region is overriden by default, what is not wanted. If someone knows if there is a setting for this, I would be happy to know.

hughsie, 2024-09-26:
I'm looking at the DRM approach now -- But I'm a bit unsure whether to use the preferred _CRTC 0 Mode_ or _Connector 0 preferred mode_. I'm guessing they should be the same for a eDP panel, but advice welcome.

> quite likely that both the firmware / EFI-GOP and native DRM driver will pick the same preferred mode from the EDID data

I'm not 100% sure about that -- although it'll work okay for laptops with eDP panels, I'm somewhat sceptical that the GOP driver is going to choose the exact same resolution as the Linux kernel. I could totally understand the firmware choosing 800x600 to show the custom "tweak your CPU voltage" GUI and then Linux actually uses the correct resolution.

jwrdegoede, 2024-10-02:
I became aware of this issue because of #7793.

This is all pretty tricky stuff actually. In my experience the efifb resolution may differ depending on how the system was booted. E.g. If grub does not show its menu (because of e.g. https://fedoraproject.org/wiki/Changes/HiddenGrubMenu) then the efifb may very well run at the native resolution of the panel when the kernel starts, e.g. 1920x1080. Where as when grub does show its menu for whatever reason, the call grub makes to initialize the EFI text console may have changed the resolution to e.g. 1024x768 and that is what the kernel will see then as efifb resolution.

So the old way of accessing the efifb resolution was not necessarily reliable.

In this sense the new drm-info based method might be better. I wonder what happens if the BMP in the ux-capsule is too big for what the efib can show ?

jebasimmanuvel07, 2024-10-09:
Hi @hughsie @superm1   

One quary. will Ux-capsule be passed in capsule-on-disk method also?? If yes means what will be the name of the ux-capsule file name??

hughsie, 2024-10-09:
> One quary. will Ux-capsule be passed in capsule-on-disk method also

No, I don't think the Capsule-on-disk has any guidance on this. If we **are** supposed to deliver one, please let me know and we can make that happen.

jebasimmanuvel07, 2024-11-06:
Hi @hughsie   

As you know, the Ux capsule is used to display any info/message/guidelines to users when a firmware update is ongoing. I believe it would be useful if we send the Ux capsule in capsule-on-disk method as well. Please share your thoughts.

hughsie, 2024-11-06:
> I believe it would be useful if we send the Ux capsule in capsule-on-disk method as well.

How?