fwupd/fwupd #7639: Failed to reboot and update firmware [Dell Inc. OptiPlex 3060]

escape0707, 2024-08-22:
**Describe the bug**
I was attempting to install updates on my Dell Inc. OptiPlex 3060.

**Steps to Reproduce**

    fwupdmgr get-updates
    fwupdmgr update
    Reboot

**Expected behavior**
Update success.

**fwupd version information**

```shell
$ fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.24
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.10.6-arch1-1
runtime   org.freedesktop.fwupd         1.9.24
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): 

The machine's OS is ArchLinux. And I'm using `pacman`.

<details>

<summary>**fwupd device information**</summary>

```shell
$ fwupdmgr get-devices --show-all-devices
Dell Inc. OptiPlex 3060
│
├─CoffeeLake-S GT2 [UHD Graphics 630]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              36d6f0eb-c9e4-50ce-a5ae-88e528c3180b ← PCI\VEN_8086&DEV_3E92
│                         23ec3061-192e-5dbf-99b6-d31b3d2990b5 ← PCI\VEN_8086&DEV_3E92&SUBSYS_1028085C
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i5-8500T CPU @ 2.10GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f8
│     Vendor:             Intel
│     GUIDs:              809a0b93-8a12-5338-a571-ad5583acf896 ← CPUID\PRO_0&FAM_06&MOD_9E
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543 ← CPUID\PRO_0&FAM_06&MOD_9E&STP_A
│     Device Flags:       • Internal device
│   
├─Internal SPI Controller:
│     Device ID:          b04e387fb80d2b91f37a4d0c7b21461c451775e1
│     Summary:            Memory Technology Device
│     Vendor:             DMI:Dell Inc.
│     GUIDs:              5f93d7e7-e282-59b9-b663-0146e382f8f6 ← MTD\NAME_0000:00:1f.5
│                         b858f197-3e26-5077-a45c-69bb2ddd6a9e ← MTD\VENDOR_Dell-Inc.&NAME_0000:00:1f.5
│                         57f35dfe-1dce-5a06-a062-fa82c608d469 ← MTD\VENDOR_Dell-Inc.&PRODUCT_OptiPlex-3060&NAME_0000:00:1f.5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│   
[... 254 lines cut ...]
│   │       ├── diags_current.xml
│   │       └── diags_previous.xml
│   ├── Linux
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   └── arch.conf
│   ├── entries.srel
│   ├── loader.conf
│   └── random-seed
└── vmlinuz-linux

```

**Additional questions**

- Operating system and version: ArchLinux
- Have you tried rebooting? Yes
- Is this a regression? I don't know. This is the first time I run fwupd on a machine with UEFI update available.
- Are you using an NVMe disk? No
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

hughsie, 2024-08-22:
What do you see on reboot? Does it work if secure boot is turned off?

escape0707, 2024-08-22:
That machine is currently headless. I usually connect to it through ssh. Do you mean for me to connect it to a monitor and see the video output?

Secure boot is already off since day 1 when I installed Arch on it.

hughsie, 2024-08-22:
> That machine is currently headless

That could be the problem. Some firmware updaters check for a display before allowing the process to continue. Could you insert one of those dummy hdmi plugs (so the computer thinks it has a display connected) and try again?

escape0707, 2024-08-23:
@hughsie Thank you for your answer. The issue is indeed solved by connecting to my monitor and then boot the device. The update process went smoothly, and the device is now updated!

closed 2024-08-23 by escape0707

hughsie, 2024-08-23:
Hey, we've got a quirk for that -- so we show an error until the display device is connected. @CragW looking at https://fwupd.org/lvfs/firmware/84929 it's got a terrible percentage success -- and it looks like quite a few of those failures are "didn't apply because headless" -- e.g. look for `Display State: Unknown` in https://fwupd.org/lvfs/firmware/84929/analytics/reports/3

@crag would you like me to add the `display-required` flag to all the firmwares with GUID `603297b5-9cc4-49fa-be2a-c14f6e8c1fb1`? I think that should also be set on upload, e.g. https://lvfs.readthedocs.io/en/latest/metainfo.html#device-flags

CragW, 2024-08-23:
> @crag would you like me to add the `display-required` flag to all the firmwares with GUID `603297b5-9cc4-49fa-be2a-c14f6e8c1fb1`?

Yeah that makes sensible, thank you @hughsie .

hughsie, 2024-08-23:
@CragW ahh, the problem with this is, as the LVFS tells us:

     The requirement of org.freedesktop.fwupd >= 1.9.1 is missing to force fwupd to use device flags specified in the metadata. 

Do you want me to add the `org.freedesktop.fwupd` requirement? 1.9.1 might be a bit new for your Ubuntu preload. I think at the moment it's okay as I added the flag to https://fwupd.org/lvfs/firmware/14963 -- so that has a "problem" but the flag is shared by everything with the same Appstream ID -- which I think is what we want - -i.e. if fwupd is new enough we warn the user, otherwise we do as before (and work with older fwupd versions).