fwupd/fwupd #7664: fwupdmgr refresh failes with "Failed to download metadata for lvfs: Failed to download, server response was 404"

pgrunewald, 2024-08-29:
**Describe the bug**
I try upgrade my firmware of my T480s, but when running "fwupdmgr refresh" it fails for lvfs. See the screenshot.

[image]

**Steps to Reproduce**
1. fwupdmgr refresh

**Expected behavior**
Downloading the latest versions that enables me to update later on. 

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.23
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.4.16-5-MANJARO
runtime   org.freedesktop.fwupd         1.9.23
```

I have installed it with pacman.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
LENOVO 20L8S29T00
│
├─Unknown Device:
│     Device ID:          96928709c665f953e18829015803502c7fb9a1ef
│     Serial Number:      1144009866
│     GUID:               e52b578d-cee7-5621-b76d-775ff48bc2cf ← DRM\VEN_ACR&DEV_0413
│   
├─Unknown Device:
│     Device ID:          f56ccce5ec5fd3e3443b9aa43f7d4fb9bbd708e5
│     Serial Number:      1144009870
│     GUID:               e52b578d-cee7-5621-b76d-775ff48bc2cf ← DRM\VEN_ACR&DEV_0413
│   
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f6
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─LENSE30512GMSP34MEAT3TA:
│     Device ID:          c82980cd3fe66f7d0a8bb83b353cbc271177efd4
│     Summary:            NVM Express solid state drive
│     Current version:    2.5.0412
│     Vendor:             Lenovo (NVME:0x17AA)
│     Serial Number:      FBBS19071340000472
│     GUIDs:              70cc283d-bc6c-5b87-9977-17c5f95c3168 ← NVME\VEN_17AA&DEV_0006
│                         9b4e3e73-ffd8-5440-b673-9558c98af843 ← NVME\VEN_17AA&DEV_0006&SUBSYS_17AA1006
│                         0a2bdb12-23de-5728-829f-b1674e12dde0 ← LENSE30512GMSP34MEAT3TA
│     Device Flags:       • Internal device
[... 722 lines cut ...]
├── vmlinuz-5.10-x86_64
├── vmlinuz-5.15-x86_64
├── vmlinuz-5.4-x86_64
├── vmlinuz-6.10-rt-x86_64
├── vmlinuz-6.10-x86_64
├── vmlinuz-6.1-rt-x86_64
├── vmlinuz-6.1-x86_64
├── vmlinuz-6.4-rt-x86_64
├── vmlinuz-6.4-x86_64
├── vmlinuz-6.6-rt-x86_64
├── vmlinuz-6.6-x86_64
├── vmlinuz-6.9-rt-x86_64
└── vmlinuz-6.9-x86_64

12 directories, 416 files
```

**Additional questions**

- Operating system and version: 6.4.16-5-MANJARO
- Have you tried rebooting? Yes
- Is this a regression? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? not sure

pgrunewald, 2024-08-29:
No problem. The output is: 
```
✖1 ❯ fwupdmgr refresh -vv --force
(pkttyagent:76180): GLib-GIO-DEBUG: 17:48:10.067: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:76178): GLib-GIO-DEBUG: 17:48:10.079: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.079: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.080: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:76178): GLib-GIO-DEBUG: 17:48:10.081: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.084: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.084: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.084: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.085: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:76178): dconf-DEBUG: 17:48:10.085: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:76178): GLib-GIO-DEBUG: 17:48:10.087: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087: px_config_kde_set_config_file: Could not read file /home/paul/.config/kioslaverc
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087: px_config_kde_set_config_file: Could not read file /home/paul/.config/kioslaverc
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087: px_config_sysconfig_set_config_file: Could not read file /etc/sysconfig/proxy
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087: Active config plugins:
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087:  - config-env
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087:  - config-xdp
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087:  - config-kde
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087:  - config-gnome
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.087:  - config-sysconfig
(fwupdmgr:76178): GLib-GIO-DEBUG: 17:48:10.091: Failed to initialize portal (GNetworkMonitorPortal) for gio-network-monitor: Not using portals
(fwupdmgr:76178): GLib-GIO-DEBUG: 17:48:10.091: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:76178): GLib-GIO-DEBUG: 17:48:10.094: _g_io_module_get_default: Found default implementation networkmanager (GNetworkMonitorNM) for ‘gio-network-monitor’
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:10.094: px_manager_constructed: Up and running
(fwupdmgr:76178): GLib-GIO-DEBUG: 17:48:10.094: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:76178): Fwupd-DEBUG: 17:48:13.651: Emitting ::status-changed() [idle]
Updating lvfs
Fwupd-INFO: 17:48:13.661: downloading https://cdn.fwupd.org/downloads/firmware.xml.zst.asc
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:13.661: px_manager_get_proxies_sync: url=https://cdn.fwupd.org/downloads/firmware.xml.zst.asc online=1
(fwupdmgr:76178): pxbackend-DEBUG: 17:48:13.661: px_manager_get_proxies_sync: Proxy[0] = direct://
(fwupdmgr:76178): Fwupd-DEBUG: 17:48:13.661: Emitting ::status-changed() [downloading]
Downloading…             [                -                      ]Fwupd-INFO: 17:48:14.301: download progress: 43%
Downloading…             [****************                       ]Fwupd-INFO: 17:48:14.314: download progress: 100%
(fwupdmgr:76178): Fwupd-DEBUG: 17:48:14.314: Emitting ::status-changed() [idle]
Fwupd-INFO: 17:48:14.314: status-code was 404

Failed to download metadata for lvfs: Failed to download, server response was 404: 

<!DOCTYPE html>
<!-- Copyright 2015 Richard Hughes <richard@hughsie.com>
     SPDX-License-Identifier: GPL-2.0-or-later -->
...
```

hughsie, 2024-08-29:
> https://cdn.fwupd.org/downloads/firmware.xml.zst.asc

That won't exist. What have you got in `/etc/fwupd/remotes.d/lvfs.conf` please

pgrunewald, 2024-08-30:
```
❯ cat /etc/fwupd/remotes.d/lvfs.conf
[fwupd Remote]

# this remote provides metadata and firmware marked as 'stable' from the LVFS
Enabled=true
Title=Linux Vendor Firmware Service
Keyring=gpg
MetadataURI=https://cdn.fwupd.org/downloads/firmware.xml.gz
ReportURI=
OrderBefore=fwupd
AutomaticReports=false
ApprovalRequired=false

```

hughsie, 2024-08-30:
That's really old. You want to remove `Keyring=gpg` and make it `MetadataURI=https://cdn.fwupd.org/downloads/firmware.xml.zst`

pgrunewald, 2024-08-30:
Now it works as expected. Thank you very much going through each step:

```
❯ fwupdmgr refresh
Updating lvfs
Downloading…             [*************************************  ]
Successfully downloaded new metadata: 6 local devices supported
```

pgrunewald, 2024-09-04:
I'm gonna close this issue. Thanks for the help again.

closed 2024-09-04 by pgrunewald