fwupd/fwupd #7670: Firmware available in fwupdmgr, but not on website.

vzsze, 2024-08-31:
**Describe the bug**
Firmware available for Dell XPS 13 9310 but no info on the website. 

**Steps to Reproduce**
fwupdmgr shows Firmware version 2.25.0. The website is still on 2.23.0 since a coupe of weeks. https://fwupd.org/lvfs/devices/com.dell.uefiecfae66b.firmware

**Expected behavior**
The website should be showing the latest available firmware.

**fwupd version information**
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.8.0-40-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>
It was preinstalled on Ubuntu 22.04.

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

XPS 13 9310
│
├─11th Gen Intel Core™ i7-1185G7 @ 3.00GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000b8
│     Vendor:             Intel
│     GUID:               b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba
│                         1df99df5-22f4-5505-98cc-22dcc53b132a
│     Device Flags:       • Internal device
│   
├─KXG70PNV2T04 NVMe KIOXIA 2048GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
│     Summary:            NVM Express solid state drive
│     Current version:    10904107
│     Vendor:             KIOXIA Corporation (NVME:0x1E0F)
│     Update Error:       ^[[31m^[[1mCannot install update when not on AC power^[[0m
│     GUID:               d326d254-bee7-532f-8ffe-7f43a208484f
│                         d74bf9aa-40d9-11ea-b77f-2e728ce88125
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│                         • Signed Payload
│   
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    3.25.0
│ │   Minimum Version:    3.25.0
│ │   Vendor:             Dell (DMI:Dell Inc.)
│ │   Update State:       Success
│ │   Update Error:       ^[[31m^[[1mCannot install update when not on AC power^[[0m
│ │   GUID:               cfb08d7c-8d3d-41b1-846b-48f634dfac9d
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │   Device Flags:       • Internal device
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │                       • Updatable
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  267 
│       Minimum Version:  267 
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUID:             00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Only version upgrades are allowed
│                         • Signed Payload
│    
├─TPM:
│     Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│     Current version:    1.257.0.0
│     Vendor:             ST Microelectronics (TPM:STM)
│     GUID:               ff71992e-52f7-5eea-94ef-883e56e034c6
│                         84df3581-f896-54d2-bd1a-372602f04c32
│                         bfaed10a-bbc1-525b-a329-35da2f63e918
│                         70b7b833-7e1a-550a-a291-b94a12d0f319
│                         06f005e9-cb62-5d1a-82d9-13c534c53c48
│     Device Flags:       • Internal device
│   
├─TPM 2.0:
│     Device ID:          a3487e128cf1413519bce8e9a1ab3f5981e61458
│     Summary:            UEFI ESRT device
│     Current version:    0.1.1.1
│     Vendor:             Dell Inc. (PCI:0x1028)
│     Update State:       Success
│     Update Error:       ^[[31m^[[1mCannot install update when not on AC power^[[0m
│     GUID:               1c510e19-0b20-5a48-aa73-cba9daabdead
│                         ff71992e-52f7-5eea-94ef-883e56e034c6
│                         73730635-f6c2-53da-9df2-948bb5ac1022
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Updatable
│   
├─TigerLake-LP GT2 [Iris Xe Graphics]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    01  
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               7ad6854a-8a22-543c-a31d-9d8b5c66bd2b
│                         ce09e928-da40-5d4b-b881-986f41632a37
│                         3c11d563-4f83-5bd1-8369-75359ea6b6f6
│                         54ff3ed7-1ea3-5279-975f-93c13bf2aa53
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─UEFI Device Firmware:
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    184 
│     Minimum Version:    184 
│     Vendor:             DMI:Dell Inc.
│     Update State:       Success
│     Update Error:       ^[[31m^[[1mCannot install update when not on AC power^[[0m
│     GUID:               176e090d-0ddb-495e-8173-bc998ccfecd0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─UEFI Device Firmware:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    277889287
│     Minimum Version:    277889287
│     Vendor:             DMI:Dell Inc.
│     Update State:       Success
│     Update Error:       ^[[31m^[[1mCannot install update when not on AC power^[[0m
│     GUID:               d74bf9aa-40d9-11ea-b77f-2e728ce88125
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Updatable
│   
├─USB4 host controller:
│     Device ID:          2f9254cbed76204c7830f9696798f74b00c9fcf6
│     Summary:            Unmatched performance for high-speed I/O 
│     Update Error:       ^[[31m^[[1mCannot install update when not on AC power^[[0m
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Signed Payload
│
└─USB4 host controller:
      Device ID:          4778bb6744c9a5ed546af2a9375646cd9823bb64
      Summary:            Unmatched performance for high-speed I/O
      Update Error:       ^[[31m^[[1mCannot install update when not on AC power^[[0m
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c
      Device Flags:       • Internal device
                          • System requires external power source
                          • Signed Payload

</details>

**Additional questions**

- Operating system and version: Ubuntu 22.04
- Have you tried rebooting? n.a.
- Is this a regression? n.a.

hughsie, 2024-08-31:
Are you sure that's your Dell XPS 13 9310? There seem to be a few different variants:

[image]

vzsze, 2024-08-31:
I'm not sure. It was always in sync till now. How can I check which model I have?

vzsze, 2024-09-18:
Closing it due to mismatched machine type.

closed 2024-09-18 by vzsze