fwupd/fwupd #7712: Can't sign updates via SSH

Sean-StarLabs, 2024-09-06:
Very minor bug:

[Screencast from 2024-09-06 14-25-21.webm](https://github.com/user-attachments/assets/339a94dd-1686-4fe6-be33-e940c4ae3907)

superm1, 2024-09-06:
Are you logged in graphically as well?

Sean-StarLabs, 2024-09-06:
No

superm1, 2024-09-06:
> No

This seems like it's an issue with how  polkit is interacting then.

For now as a workaround you can launch fwupdmgr as root using sudo.

hughsie, 2024-09-06:
Yuk. Is this a regression from 1.9.x to main?

Sean-StarLabs, 2024-09-06:
> For now as a workaround you can launch fwupdmgr as root using sudo.

I plugged in a monitor :)

>  Yuk. Is this a regression from 1.9.x to main?

Does it on both

Sean-StarLabs, 2026-03-02:
Completely forgot this - discovered that it is actually waiting for the password, and will accept it - just doesn't say "Enter password:..."