fwupd/fwupd #7753: Can't install update on Thinkstation P5

FilipLaurentiu, 2024-09-19:
**Describe the bug**
I can't install the updates for a Lenovo Thinkstation P5. After the update is installed I need to reboot, the reboot process is normal, nothing different and when I log in the update is not installed 

**Steps to Reproduce**
I can either press "Update" from Debian Software application ori I can use the cli to update, same behavior.
`sudo fwupdmgr update`
I even tried to download the updates from lenovo website and install it with `fwupdmgr`

**Expected behavior**
Install updates

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

compile   org.freedesktop.fwupd         1.8.12
compile   com.hughsie.libxmlb           0.3.10
compile   com.hughsie.libjcat           0.1.9
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    6.1.0-25-amd64
runtime   org.freedesktop.fwupd         1.8.12

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

LENOVO 30GA000LGE
│
├─Bolt Receiver:
│ │   Device ID:          2d687c9105d12706ab306d71285f4e3cdbed39aa
│ │   Summary:            Miniaturised USB wireless receiver
│ │   Current version:    MPR05.01_B0010
│ │   Bootloader Version: BOT30.01_B0010
│ │   Vendor:             Logitech (HIDRAW:0x046D, USB:0x046D)
│ │   Install Duration:   30 seconds
│ │   GUIDs:              02d049be-c3f5-51c4-a312-b2284b43415a
│ │                       f70d9241-2656-546b-a8e4-211bac348667 ← HIDRAW\VEN_046D&DEV_C548
│ │                       af1404c4-f038-5b3e-92b0-09bf4aa84f1c ← HIDRAW\VEN_046D&DEV_C548&REV_00
│ │   Device Flags:       • Updatable
│ │                       • Signed Payload
│ │ 
│ ├─LIFT:
│ │ │   Device ID:        11b8f527a637c40075f90b9fedadbb8c176146c9
│ │ │   Summary:          Unifying Mouse
│ │ │   Current version:  RBM21.01_B0010
│ │ │   Bootloader Version:BL156.01_B0010
│ │ │   Vendor:           Logitech (USB:0x046D)
│ │ │   Install Duration: 4 minutes
│ │ │   Battery:          30% (threshold 20%)
│ │ │   GUIDs:            6241aea0-cebb-53fe-b81c-d15e598771b7 ← UFY\VID_0000&PID_0000
│ │ │                     981d7d58-7f37-50c8-9bda-1da4d52437c6 ← HIDRAW\VEN_046D&MOD_B03100000000
│ │ │   Device Flags:     • Updatable
│ │ │                     • Signed Payload
│ │ │ 
│ │ └─Radio:
│ │       Device ID:      cfd420c3961eeefcb58ad26018c56afcfca60e75
[... 566 lines cut ...]
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img-6.1.0-23-amd64
├── initrd.img-6.1.0-25-amd64
├── System.map-6.1.0-23-amd64
├── System.map-6.1.0-25-amd64
├── vmlinuz-6.1.0-23-amd64
└── vmlinuz-6.1.0-25-amd64

7 directories, 331 files
```

**Additional questions**

- Operating system and version:  Debian 12
- Have you tried rebooting? Sure
- Is this a regression?
- Are you using an NVMe disk? yes
- Is secure boot enabled? no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I will check

FilipLaurentiu, 2024-09-19:
[exact model](https://pcsupport.lenovo.com/us/en/products/workstations/thinkstation-p-series-workstations/thinkstation-p5/30ga/30ga000lge/gm09d000)

FilipLaurentiu, 2024-09-19:
I successfully upgrade my bios with a USB but the problem persist for other system updates 
[image]

```
sudo dmidecode -s bios-version
S0CKT19A

```

lluki, 2025-05-23:
Can confirm on Ubuntu 22.04 LTS