fwupd/fwupd #7759: plugins: nordic_hid: Update error - No vendor ID set

MarekPieta, 2024-09-20:
**Describe the bug**
`nordic-hid` plugin reports `No vendor ID set` update error for nRF Desktop devices.

**Steps to Reproduce**
1. Connect nRF52840 Development Kit running [nRF Desktop](https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/applications/nrf_desktop/description.html) application over USB to host computer.
2. Get list of devices using fwupdtool: `sudo ./build/src/fwupdtool get-devices --plugin-whitelist=nordic-hid`.
3. See that the connected nRF Desktop device is marked with the `No vendor ID set` update error.

**Expected behavior**
The update error should not appear.

**fwupd version information**
Issue replicated with fwupdtool built from source from main.
The issue replicates since  f0e54446f3c2f37c57181f5406a908de94243bcb.

**Additional questions**
Replicated with Ubuntu 24.04.1 LTS release.
The issue seems like a regression in fwupd.

MarekPieta, 2024-09-20:
Hi @hughsie,
Could you advise why the linked commit results in update errors in our plugin?

hughsie, 2024-09-20:
Ahh, I think that's my fault, sorry. Were you expecting a `HIDRAW:something` vendor ID? Patch incoming.

MarekPieta, 2024-09-20:
Please let me know as soon as the patch is ready so I can test locally.

closed 2024-09-20 by hughsie