fwupd/fwupd #7763: WARNING: UEFI ESP partition not detected or configured

rsm-gh, 2024-09-20:
Hello,

Despite all my attempts, reading of [PluginFlag:esp not found](https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found) and some other links, I still having the `WARNING: UEFI ESP partition not detected or configured` error.

Any help on this would be very appreciated, I need to update my firmware :)
Thank you,

----

- UDisks is up and running:
```
● udisks2.service - Disk Manager
     Loaded: loaded (/usr/lib/systemd/system/udisks2.service; disabled; preset: disabled)
     Active: active (running) since Fri 2024-09-20 12:20:20 CEST; 8min ago
 Invocation: 30de534b2ffb477db658a6d685256584
       Docs: man:udisks(8)
   Main PID: 1627 (udisksd)
      Tasks: 6 (limit: 18854)
     Memory: 7M (peak: 8M)
        CPU: 69ms
     CGroup: /system.slice/udisks2.service
             └─1627 /usr/lib/udisks2/udisksd

Sep 20 12:20:20 CookieMonster systemd[1]: Starting Disk Manager...
Sep 20 12:20:20 CookieMonster udisksd[1627]: udisks daemon version 2.10.1 starting
Sep 20 12:20:20 CookieMonster systemd[1]: Started Disk Manager.
```
- The efi partition looks good, except of the `esp` flag:
```
$lsblk
NAME          MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
nvme0n1       259:0    0 931.5G  0 disk  
├─nvme0n1p1   259:1    0     1G  0 part  /boot
└─nvme0n1p2   259:2    0 930.5G  0 part  
  └─cryptroot 254:0    0 930.5G  0 crypt /

$tree /boot
/boot
├── efi
│   └── Arch
│       └── grubx64.efi
├── grub
│   ├── fonts
│   │   └── unicode.pf2
│   ├── grub.cfg
│   ├── grubenv
│   ├── locale
│   │   ├──..MO
│   ├── themes
│   │   └── starfield
│   │       ├── etc..
│   └── x86_64-efi
│       ├── ... MOD
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux
```
[image]

And actually I tried by adding the `esp` flag, but system stop booting and I had to re-create it.. I don't know if it is important, but I used the following command `grub-install --target=x86_64-efi --efi-directory=/boot --bootloader-id=ARSM`, and I also tried by setting the target to `efi` but it didn't work.

- I updated the configuration file `/etc/fwupd/fwupd.conf`:
```
[fwupd]
EspLocation=/boot/efi
```
and restarted the systemd service. Also tried by rebooting..

- I do not seem to have the dbus error:
```
[... 425 lines cut ...]
10:42:17.770 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0x83, internal: 1, fs: vfat
10:42:17.774 FuVolume             Looking for volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
10:42:17.775 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0x83, internal: 1, fs: crypto_LUKS
10:42:17.777 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0x83, internal: 1, fs: vfat

No ESP or BDP found

```

```
$fwupdmgr --version
compile   org.freedesktop.fwupd         1.9.24
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.freedesktop.Passim        0.1.8
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.10.10-arch1-1
runtime   org.freedesktop.fwupd         1.9.24

```

hughsie, 2024-09-20:
> I tried by adding the esp flag, but system stop booting

You need the ESP flag if you want the firmware update to work correctly. Otherwise it's just another fat32 partition rather than the "blessed" ESP.

superm1, 2024-09-20:
> I updated the configuration file /etc/fwupd/fwupd.conf:

You shouldn't _need_ to specify the path.  

>>I tried by adding the esp flag, but system stop booting
> You need the ESP flag if you want the firmware update to work correctly. Otherwise it's just another fat32 partition rather than the "blessed" ESP.

Hmm, but isn't that only to decide between two vfat partitions?  I thought with only one vfat we should still pick it.

Can you please rerun with `-vv` and without that conffile?  `sudo fwupdtool esp-list -vv`?

Also please share the `udisksctl dump` output in case it's still not obvious what's wrong.

hughsie, 2024-09-20:
> Hmm, but isn't that only to decide between two vfat partitions

rsm-gh, 2024-09-20:
The weird thing is that I used the same install instructions some months ago, and the software worked okay... and to be honest, if the `esp` flag is required, I'll add it, but I wonder why `EspLocation=/boot/efi` is not enough...

@superm1  here you have the logs:
[udisk.txt](https://github.com/user-attachments/files/17077858/udisk.txt)
[esp-list.txt](https://github.com/user-attachments/files/17077859/esp-list.txt)

Thank you

superm1, 2024-09-20:
Ummm, is this an MBR disk layout not GPT?  Because it sure looks like it..

superm1, 2024-09-20:
Even if it's MBR( which it looks like) we normally handle that.

Yeah so you have the wrong filesystem type set on your disk.
```
16:03:57.957 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: 0x83, internal: 1, fs: vfat
```

You can see that's type 0x83.

[image]
https://en.wikipedia.org/wiki/Partition_type

I think it should be filesystem type `0xB` or `0xC`:
[image]

I don't think we can safely handle the case that you have vfat in a filesystem declared as Linux filesystem.

superm1, 2024-09-20:
The correct action for you is to change that partition type.
Please also work with any installer or install instructions that created this invalid configuration to get it fixed.

I've added your case to https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-found so others find it.

closed 2024-09-20 by superm1

rsm-gh, 2024-09-20:
@superm1 , yes you're right, when I formatted the disk with `cfdisk`, it said linux, and it was weird for me, but in the next steps I had `mkfs.fat -F32 /dev/nvme0n1p1` so I tough in this step I was correcting it to FAT.

Thank you very much !!