fwupd/fwupd #7776: Unable to detect devices with steelseries plugin after 56d2e703

d4s, 2024-09-23:
**Describe the bug**
Commit 56d2e703f1d3e6777a1951f5493f2a1eae59cf95 is preventing SteelSeries plugin from devices detection.
Found with bisect, reverting on top of main restore the proper behavior.

Testing with Aerox 3 dongle and Nova 5 headset.

See the attached log from `fwupdtool --plugins steelseries get-devices -vv`  
[ss-not_detected-56d2e703.txt](https://github.com/user-attachments/files/17103966/ss-not_detected-56d2e703.txt)

**Steps to Reproduce**
1. git checkout 56d2e703f1d3e6777a1951f5493f2a1eae59cf95
2. build-fwupd
3. fwupdtool --plugins steelseries get-devices -vv

**Expected behavior**
Devices should be detetected.
See the attached log with reverted commit 56d2e703f1d3e6777a1951f5493f2a1eae59cf95 
[ss-detected-revert_56d2e703.txt](https://github.com/user-attachments/files/17103917/ss-detected-revert_56d2e703.txt)

<details>

<summary>**detected devices**</summary>

```
$ LANGUAGE=C fwupdtool --plugins steelseries get-devices 
Intel(R) Client Systems NUC8i7HVK
│
├─Aerox 3 Wireless USB Receiver:
│ │   Device ID:          813b4757e78d708603a80ead0a2b6ef1364b7790
│ │   Current version:    1.11.4
│ │   Vendor:             SteelSeries ApS (USB:0x1038)
│ │   Install Duration:   13 seconds
│ │   GUID:               291b0582-8d08-5714-88db-986911c744d9 ← USB\VID_1038&PID_1838
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Cryptographic hash verification is available
│ │                       • Unsigned Payload
│ │ 
│ └─Aerox 3 Wireless Mouse via USB Receiver:
│       Device ID:        22c1abc42c63290513f78f84bc0a6f89851b485c
│       Vendor:           SteelSeries ApS (USB:0x1038)
│       Install Duration: 37 seconds
│       Problems:         • Device is unreachable, or out of wireless range
│       GUID:             541b2713-d367-5240-a443-bed07f09ffbf ← STEELSERIES\VID_1038&PID_1838&PROTOCOL_FIZZ_TUNNEL
│       Device Flags:     • Cryptographic hash verification is available
│                         • Updatable
│                         • Device is unreachable
│                         • Unsigned Payload
│     
└─Arctis Nova 5 USB Receiver:
  │   Device ID:          25e83b27177b70aa34d43f80eb01613f91e13e91
  │   Current version:    1.6.0
  │   Vendor:             SteelSeries ApS (USB:0x1038)
  │   Install Duration:   2 minutes
  │   GUID:               ba9a40e9-9992-5673-8cdb-c60a5c1e3ca2 ← USB\VID_1038&PID_2232
  │   Device Flags:       • Updatable
  │                       • Supported on remote server
  │                       • Cryptographic hash verification is available
  │                       • Unsigned Payload
  │ 
  └─Arctis Nova 5 Headset via USB Receiver:
        Device ID:        d4746c28759b680f56b995ace2e9c234b179ccf8
        Current version:  1.6.0
        Vendor:           SteelSeries ApS (USB:0x1038)
        Serial Number:    616703980403243136
        Battery:          93% (threshold 20%)
        GUID:             a0e5f3d2-6c1c-5478-8ea5-9e118951c07b ← STEELSERIES\VID_1038&PID_2232&PROTOCOL_FIZZ_TUNNEL
        Device Flags:     • Updatable
                          • Supported on remote server
                          • Cryptographic hash verification is available
                          • Unsigned Payload   
```
</details>

**fwupd version information**

```shell
# fwupdtool --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.0
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libjcat           0.2.1
runtime   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.6.48-un-def-alt0.p11.1
runtime   org.freedesktop.fwupd         2.0.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
  Yes

hughsie, 2024-09-24:
@d4s out of interest, can you find out what `streamsz` is both during `fu_usb_device_parse_bos_descriptor` and also `fu_usb_device_parse_descriptor`

I've also pushed `hughsie/descriptor-fd-weirdness` which has a workaround, but I'd like to understand a bit more what's going on first.

d4s, 2024-09-24:
Added the debug line and got for all devices I have in both places with the patch 56d2e703f1d3e6777a1951f5493f2a1eae59cf95: 
```
FuUsbDevice          XXXXX: fu_usb_device_parse_descriptor:1745: stream size for Device (null)((null)) = 65553
```
again, for all devices on the host.

But with reverted patch the size is varying for each device (36-1151 in my case).
For SteelSeries devices in particular:
```
FuUsbDevice          XXXXX: fu_usb_device_parse_descriptor:1745: stream size for DeviceId (null)((null)) = 152
FuUsbDevice          XXXXX: fu_usb_device_parse_descriptor:1745: stream size for DeviceId (null)((null)) = 282
```

Interesting -- no debug lines for `fu_usb_device_parse_bos_descriptor()`

d4s, 2024-09-24:
I cherry-picked the `hughsie/descriptor-fd-weirdness` branch -- doesn't help