fwupd/fwupd #7783: Firmware update expects shim at EFI/aeon instead of EFI/systemd

cyril279, 2024-09-24:
**Summary** 
 fwupd seems to be misinterpreting the directory that the OS stores the shim.efi

**Steps to reproduce**  
When I try to apply the firmware updates, I receive the following error message:

> Secure boot is enabled, but shim isn't installed to EFI/aeon/shim.efi

This is true, the **shim is installed to EFI/systemd/shim.efi.**

The EFI mount point seems to be successfully identified, but not the efi-directory-name that the OS is using.  

**Expected behavior**  
Whether from gui or command-line, the update should complete successfully.
 ...or have ability to override the specific OS-EFI-directory [ overrideEFIDir ] in fwupd.conf

**fwupd version information**  
Please provide the version of the daemon and client.

```shell
cyril@localhost:~> fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.24
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.gusb          0.4.9
runtime   org.kernel                    6.10.9-1-default
runtime   org.freedesktop.fwupd         1.9.24
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Pre-installed with the OS

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
cyril@localhost:~> fwupdmgr get-devices --show-all-devices
LENOVO 20KH002RUS
│
├─B140QAN02.3:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     GUID:               d640763b-fe58-5d0a-9c42-5ea4fbcd80f4 ← DRM\VEN_AUO&DEV_2336
│     Device Flags:       • Internal device
│   
├─Core™ i7-8650U CPU @ 1.90GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f4
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          a54e6e63dc271f06218a877e5bf661c28ae95c64
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
[... 227 lines cut ...]
│   │       ├── grub.efi
│   │       ├── installed_by_sdbootutil
│   │       ├── MokManager.efi
│   │       └── shim.efi
│   └── loader
│       ├── entries  [11 entries exceeds filelimit, not opening dir]
│       ├── entries.srel
│       ├── loader.conf
│       └── random-seed
└── mbrid
```

**Additional questions**

- Have you tried rebooting? YES
- Is this a regression? NO
- Are you using an NVMe disk? YES
- Is secure boot enabled? YES
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? This IS a lenovo, but NOT boot-order-lock

Operating System: Aeon  
CPE OS Name: cpe:/o:opensuse:aeon:20240912  
Desktop environment: Gnome  
Kernel: Linux 6.10.9-1-default  
Boot manager: systemd-boot

superm1, 2024-09-24:
This _should_ be a packaging problem.  Your distro hasn't configured `-Defi_os_dir` properly at build time to reflect where binaries used for the boot process are to be stored:

https://github.com/fwupd/fwupd/blob/main/meson_options.txt#L46

If not configured at build time then fwupd will try to use `/etc/lsb-release` to figure out what to store it in.

https://github.com/fwupd/fwupd/blob/fee8b14d7aa4a20a5be55b93b4a3b20e3029c459/plugins/uefi-capsule/fu-uefi-common.c#L178

That being said @hughsie how would you feel about falling "back" to the directory `systemd` if nothing found using lsb-release?  Seems logical to me.

superm1, 2024-09-24:
Or even the other way around - try **systemd** _first_.

cyril279, 2024-09-24:
To cover both predictable AND unique naming scenarios, perhaps:
Include systemd as a fallback for the ID_LIKE fallback (or systemd *first* as mentioned)  
and ALSO provide an override for the OS-EFI-directory.

hughsie, 2024-09-24:
> how would you feel about falling "back" to the directory systemd

Honestly, I think it's adding more complexity in already complicated flow. I think -Defi_os_dir needs to be set properly.

superm1, 2024-09-24:
Well it's not too bad really if it's _first_.  Here's a PR to decide.

https://github.com/fwupd/fwupd/pull/7784

aplanas, 2024-09-24:
I like @superm1 approach, as a distro can have multiple bootloader options to install. `systemd-boot` will always use `EFI/systemd` but GRUB2 will use `EFI/$ID_LIKE`, for example, so there is no single `-Defi_os_dir` for this distribution.

sysrich, 2024-09-25:
Exactly, like @aplanas says, bootloader can be a choice in some distributions (eg. openSUSE Tumbleweed, possibly future SUSE products)

And in Aeon's case, we typically prefer to use Tumbleweed binaries where we can, so I can't easily define `-Defi_os_dir` at build time

There needs be some logic somewhere for handling distro-agnostic bootloaders like systemd-boot somehow, and the suggestion in #7784 seems like an option that minimises the obviously unwanted logic complexity

Shorrer, 2024-10-01:
I don't understand why it's impossible to just give the path to shim manually or in config. Seems like the most logical and simplest solution to me, supporting the ability of normal EFI firmwares to load .efi binary from whenever.

aplanas, 2024-10-01:
> I don't understand why it's impossible to just give the path to shim manually or in config.

Sure, this can work. The only issue that I see is that if the user changes bootloader, some components needs to update the configuration file. In some distributions this can be tricky, as a package should not touch the user configuration files.

Shorrer, 2024-10-01:
> ... a package should not touch the user configuration files. 
I think this is the answer. Package managers handle this: Debian shows "config updated" if it was ever touched by user, Arch gives a warning about new package config file, etc.
Also configuration option is for someone who knows what he is doing so it's safe to place the burden of making sure what his current bootloader is at such person in my opinion.

superm1, 2024-10-01:
I've merged the code that will look for the systemd path as well to the upcoming 1.9.26 and 2.0.0 releases.

closed 2024-10-01 by superm1