fwupd/fwupd #7791: after capsule creation and restart UEFI update does not start.

memfissoftware, 2024-09-26:
**Describe the bug**
Fw update creates a cap file then move it to /boot/efi but on restart update never starts. 

**Steps to Reproduce**
 sudo fwupdtool install-blob ../files/firmwares/ET88.bin

Writing…                 [************                           ]
0.	Cancel
1.	ace1a4af09df31a320f87dffcfc605fce0c5de6a (LITEON CV3-CE256-HP)
2.	c71ddeb9b9b7cfbad92693ab22f52cb3b6e57392 (System Firmware)
3.	cc997b76c83152570335880a682ff0b6d34b837e (UEFI Device Firmware)
4.	739341b26ecccba80817f07db9b94a40ce1dedfd (UEFI Device Firmware)
5.	997303909cf6add879d623391963206b6983d59b (UEFI Device Firmware)
6.	b8dc8fe916acd1622fab5a826f9497cc43ee2d49 (UEFI Device Firmware)
7.	a53554484fa5805fd602986a88f049117af8dedb (UEFI Device Firmware)
8.	68e75f5a598755712017b52c8f710e53aa26eab2 (UEFI Device Firmware)
9.	30ea49fe76e21e476923957f37afe2729b9c7b33 (UEFI Device Firmware)
10.	27aeff3756fd9eef56d39af69455eb2f8e41e476 (UEFI Device Firmware)
11.	8ba44eb9760e4c2a1e41edb9fb303fc276344150 (UEFI Device Firmware)
12.	a8c64758b967679856c7cbcf16c476927d760d61 (UEFI Device Firmware)
13.	29ab676e258448d22871bd5b4538bb38a90056f0 (UEFI Device Firmware)
Choose device [0-13]: 2
Waiting…                 [***************************************]
An update requires a reboot to complete. Restart now? [y|N]: N

before restart:
[image]

after restart:
[image]

**Expected behavior**
On restart uefi update should start

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
```
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
compile   org.freedesktop.fwupd         1.9.24
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.kernel                    6.8.0-45-generic
runtime   org.freedesktop.fwupd         1.9.24

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
24.04 ubuntu with apt

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
├─Unknown Device:
│     Device ID:          57e35fbedc32c4e5e69c55e3d7810f7ae6f1427a
│     Serial Number:      YVAR007748
│     GUID:               77be30ea-aca8-5afb-9f99-c26f40a525bd ← DRM\VEN_FUS&DEV_088E
│   
├─Atom x6413E Processor @ 1.50GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000019
│     Vendor:             Intel
[... 564 lines cut ...]
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfs.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-6.8.0-45-generic
├── initrd.img-6.8.0-45-generic
├── initrd.img.old -> initrd.img-6.8.0-45-generic
├── memtest86+ia32.bin
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── vmlinuz -> vmlinuz-6.8.0-45-generic
├── vmlinuz-6.8.0-45-generic
└── vmlinuz.old -> vmlinuz-6.8.0-45-generic

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
- Are you using an NVMe disk?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

superm1, 2024-09-26:
What is the BIOS vendor?

> │ Summary: UEFI System Resource Table device (Updated via capsule-on-disk)

Try forcing it to use NVRAM instead.  See https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#uefi_capsule-parameters for how to disable CoD.

memfissoftware, 2024-09-26:
> BIOS vendor
is AMI

superm1, 2024-09-26:
This is the first time I've ever seen COD on AMI. Maybe they have a bug.

hughsie, 2024-09-27:
What hardware is this please? e.g. What's the make and model of the desktop or laptop?

superm1, 2024-10-02:
> it is one of atom board from ibase.

You should contact them to debug this issue.  There is definitely at least one bug if CoD doesn't work.  It works for other vendors.

> Is there anyway to add ux capsule to NVRAM?

UX capsule _should_ be sent by default.  Can you launch fwupd daemon with `-vv` or `fwupdtool` with `-vv` and share log?

memfissoftware, 2024-10-08:
@superm1  
> UX capsule _should_ be sent by default. Can you launch fwupd daemon with `-vv` or `fwupdtool` with `-vv` and share log?
here it is:
`Writing…                 [************                           ]
0.	Cancel
1.	b179505511d8070c538a892385d0a3ed9d2c765c (LITEON CV3-CE256-HP)
2.	5dbeb140337f610d54913c0f43d15fdaa0eafb51 (SATADOM-SL 3ME3 V2)
3.	c71ddeb9b9b7cfbad92693ab22f52cb3b6e57392 (System Firmware)
4.	cc997b76c83152570335880a682ff0b6d34b837e (UEFI Device Firmware)
5.	739341b26ecccba80817f07db9b94a40ce1dedfd (UEFI Device Firmware)
6.	997303909cf6add879d623391963206b6983d59b (UEFI Device Firmware)
7.	b8dc8fe916acd1622fab5a826f9497cc43ee2d49 (UEFI Device Firmware)
8.	a53554484fa5805fd602986a88f049117af8dedb (UEFI Device Firmware)
9.	68e75f5a598755712017b52c8f710e53aa26eab2 (UEFI Device Firmware)
10.	30ea49fe76e21e476923957f37afe2729b9c7b33 (UEFI Device Firmware)
11.	27aeff3756fd9eef56d39af69455eb2f8e41e476 (UEFI Device Firmware)
12.	8ba44eb9760e4c2a1e41edb9fb303fc276344150 (UEFI Device Firmware)
13.	a8c64758b967679856c7cbcf16c476927d760d61 (UEFI Device Firmware)
14.	29ab676e258448d22871bd5b4538bb38a90056f0 (UEFI Device Firmware)
Choose device [0-14]: 3
Waiting…                 [************                           ]20:45:46.177 FuEngine             install phase now prepare
20:45:46.178 FuDeviceList         no replug or re-enumerate required
20:45:46.178 FuEngine             prepare -> FuUefiNvramDevice:
  DeviceId:             c71ddeb9b9b7cfbad92693ab22f52cb3b6e57392
  Name:                 System Firmware
  Status:               device-busy
  Guid:                 17fd627c-fad7-4911-b72b-57043da89017
  Summary:              UEFI System Resource Table device (updated via NVRAM)
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|require-ac|registered|needs-reboot|can-verify|usable-during-update|updatable-hidden
  Problems:             update-in-progress
  RequestFlags:         allow-generic-message
  Checksum:             SHA256(b29cfb4a11b39e7c04659663e75229387363b23d8a10b574bb3a9d03c16c8411)
  Vendor:               secunet Security Networks AG
  VendorId:             DMI:iBase
  Version:              236
  VersionLowest:        236
  VersionFormat:        number
  VersionRaw:           0x000000ec
  VersionLowestRaw:     0x000000ec
  Icon:                 computer
  Created:              2024-10-08
  Modified:             2024-10-08
  UpdateState:          success
  UpdateError:          An update is in progress
  PhysicalId:           /sys/firmware/efi/esrt
  LogicalId:            17fd627c-fad7-4911-b72b-57043da89017
  BackendId:            /sys/firmware/efi/esrt/entries/entry0
  AcquiesceDelay:       50
  Order:                0
  InternalFlags:        md-set-verfmt|md-set-icon|md-set-vendor|md-set-signed|md-set-flags|host-firmware
  PrivateFlags:         use-shim-for-sb
  Inhibit:              [update-in-progress] An update is in progress
  Kind:                 system-firmware
  FwClass:              17fd627c-fad7-4911-b72b-57043da89017
  CapsuleFlags:         0x50000
  FwVersion:            0xec
  FwVersionLowest:      0xec
  LastAttemptStatus:    success
  LastAttemptVersion:   0xec
  EspId:                /org/freedesktop/UDisks2/block_devices/sdb2
  EspPath:              /boot/efi
  EspKind:              c12a7328-f81f-11d2-ba4b-00a0c93ec93b
  RequireESPFreeSpace:  0

20:45:46.179 FuPlugin             prepare(gpio)
20:45:46.179 FuPlugin             prepare(logind)
20:45:46.183 FuPluginLogind       opened logind fd 18
20:45:46.183 FuDeviceList         no replug or re-enumerate required
[... 438 lines cut ...]
  Order:                0
  InternalFlags:        md-set-verfmt|md-set-icon|md-set-vendor|md-set-signed|md-set-flags|host-firmware
  PrivateFlags:         use-shim-for-sb
  Kind:                 system-firmware
  FwClass:              17fd627c-fad7-4911-b72b-57043da89017
  CapsuleFlags:         0x50000
  FwVersion:            0xec
  FwVersionLowest:      0xec
  LastAttemptStatus:    success
  LastAttemptVersion:   0xec
  EspId:                /org/freedesktop/UDisks2/block_devices/sdb2
  EspPath:              /boot/efi
  EspKind:              c12a7328-f81f-11d2-ba4b-00a0c93ec93b
  RequireESPFreeSpace:  0

20:46:00.511 FuPlugin             cleanup(gpio)
20:46:00.511 FuPlugin             cleanup(jabra)
20:46:00.511 FuPlugin             cleanup(logind)
20:46:00.511 FuPluginLogind       closed logind fd 18
20:46:00.511 FuDeviceList         no replug or re-enumerate required
Waiting…                 [***************************************]20:46:00.511 FuEngine             Updating System Firmware took 14.334097 seconds

An update requires a reboot to complete. Restart now? [y|N]: 

`

hughsie, 2024-10-11:
> failed to upload UEFI UX capsule text: EFI framebuffer not found

That interface doesn't exist any more -- we have to add a workaround in main to recreate the data using libdrm. Would you be able to compile the main branch and try fwupdtool again please?

memfissoftware, 2024-10-16:
I have cloned via git and compiled.
still i get same error.

```
(fwupd) vb@vb:~/workspace/fwupd$ git status
On branch main
Your branch is up to date with 'origin/main'.

```

```
Scheduling…              [************                           ]12:49:34.948 FuPluginUefiCapsule  failed to upload UEFI UX capsule text: EFI framebuffer not found
Decompressing…           [************                           ]12:49:34.948 FuPluginUefiCapsule  required ESP free space is not configured, using 2 x 13MB + 20MB

```

```
12:49:30.901 FuDevice             changing verfmt for (null): plain->triplet
12:49:30.901 FuEngine             android_boot ignoring: device does not expose its size
12:49:30.902 FuEngine             ata ignoring: has no ID_ATA_DOWNLOAD_MICROCODE
12:49:30.902 FuEngine             emmc ignoring: no MMC parent
12:49:30.902 FuEngine             scsi ignoring: has no ID_SCSI
12:49:30.902 FuEngine             uf2 ignoring: is not correct ID_BUS=(null), expected usb
12:49:30.902 FuEngine             /sys/devices/virtual/block/loop9 added FuUdevDevice:
  Flags:                none
  BackendId:            /sys/devices/virtual/block/loop9
  AcquiesceDelay:       2500
  PossiblePlugin:       android_boot
  PossiblePlugin:       ata
  PossiblePlugin:       emmc
  PossiblePlugin:       scsi
  PossiblePlugin:       uf2
  Subsystem:            block
  DeviceFile:           /dev/loop9
  SysfsPath:            /sys/devices/virtual/block/loop9

12:49:30.902 FuEngine             /sys/devices/virtual/block/loop9 added FuUdevDevice:
  Flags:                none
  Guid[quirk]:          0b18d9bf-3b6a-5a32-832a-d7b0bc4d34b0 ← BLOCK\TYPE_DISK
  BackendId:            /sys/devices/virtual/block/loop9
  AcquiesceDelay:       2500
  PossiblePlugin:       android_boot
  PossiblePlugin:       ata
  PossiblePlugin:       emmc
  PossiblePlugin:       scsi
  PossiblePlugin:       uf2
  Subsystem:            block
  DeviceFile:           /dev/loop9
  SysfsPath:            /sys/devices/virtual/block/loop9

12:49:30.902 FuDevice             changing verfmt for (null): plain->triplet
12:49:30.902 FuEngine             android_boot ignoring: device does not expose its size
12:49:30.902 FuEngine             ata ignoring: has no ID_ATA_DOWNLOAD_MICROCODE
12:49:30.902 FuEngine             emmc ignoring: no MMC parent
12:49:30.902 FuEngine             scsi ignoring: has no ID_SCSI
12:49:30.902 FuEngine             uf2 ignoring: is not correct ID_BUS=(null), expected usb
12:49:30.902 FuEngine             /sys/devices/virtual/iommu/dmar0 added FuUdevDevice:
  Flags:                none
  BackendId:            /sys/devices/virtual/iommu/dmar0
  AcquiesceDelay:       2500
  PossiblePlugin:       iommu
  Subsystem:            iommu
  SysfsPath:            /sys/devices/virtual/iommu/dmar0

12:49:30.902 FuEngine             /sys/devices/virtual/iommu/dmar0 added FuUdevDevice:
  Flags:                none
  BackendId:            /sys/devices/virtual/iommu/dmar0
  AcquiesceDelay:       2500
  PossiblePlugin:       iommu
  Subsystem:            iommu
[... 922 lines cut ...]
  EspId:                /org/freedesktop/UDisks2/block_devices/sdb2
  EspPath:              /boot/efi
  EspKind:              c12a7328-f81f-11d2-ba4b-00a0c93ec93b
  RequireESPFreeSpace:  0

12:49:43.362 FuPlugin             cleanup(gpio)
12:49:43.362 FuPlugin             cleanup(jabra)
12:49:43.362 FuPlugin             cleanup(logind)
12:49:43.363 FuPluginLogind       closed logind fd 18
12:49:43.363 FuDeviceList         no replug or re-enumerate required
Waiting…                 [***************************************]12:49:43.363 FuEngine             Updating System Firmware took 8.421868 seconds

An update requires a reboot to complete. Restart now? [y|N]: n
../src/fu-tool.c:957 [14524.27ms]:
    ../src/fu-engine.c:3455 [12448.36ms]:
        ../plugins/uefi-capsule/fu-uefi-device.c:682 [8415.57ms]:
            :write [8414.53ms]

12:51:22.126 FuPlugin             finalize(modem_manager)
12:51:22.126 FuPlugin             finalize(flashrom)
12:51:22.126 FuPluginTest         destroy
12:51:22.127 FuProgress           raw timing data was { 0.002, 0.000, 0.000, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 91 5 1 3 ] at ../libfwupdplugin/fu-context.c:986
12:51:22.127 FuProgress           raw timing data was { 0.118, 0.000, 0.023, 0.002, 0.005, 0.002 } -- steps were set as [ 63 1 8 1 26 1 ] but should have been [ 79 0 15 1 3 1 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1128
(fwupd) vb@vb:~/workspace/fwupd$ 
```

hughsie, 2024-11-11:
> FuPluginUefiCapsule  framebuffer size 0 x0

That seems... suboptimal. Can you grab the logs from `sudo fwupdtool --plugins uefi_capsule -vv install THE_CAB_FILE_HERE.cab` and `drm_info` please. Thanks!

eliasp, 2025-05-10:
Seeing the same problem here on a Lenovo Thinkpad A485.

Before the reboot, I have `/boot/EFI/nixos/fw/fwupd-ff8bc5c6-2169-42b7-907b-c726a9c2dab2.cap` (18M size, sha256 `2d2e62377642516584a3ae8081f31affc4ddf2f652879b79378f7a7fad52e505`), after a reboot it's gone.

When manually selecting the `Linux-Firmware-Update` entry at boot, the screen goes black for a split second and then goes back to the selection menu.

Executing `fwupdtool --plugins uefi_capsule -vv install /boot/EFI/nixos/fw/fwupd-ff8bc5c6-2169-42b7-907b-c726a9c2dab2.cap` ends with (full log below):
```
16:37:14.067 FuEngine             service restarted, but no reboot has taken place
16:37:14.067 FuEngine             inheriting needs-reboot for System Firmware
Loading…                 [*******************                    ]16:37:14.068 FuEngine             resetting update motd timeout
Decompressing…           [ -                                     ]did not find magic

16:38:24.466 FuProgress           raw timing data was { 0.004, 0.000, 0.002, 0.084, 0.120 } -- steps were set as [ 1 1 3 1 94 ] but should have been [ 2 0 1 40 57 ] at ../libfwupdplugin/fu-context.c:1101
16:38:24.469 FuProgress           raw timing data was { 0.002, 0.000, 0.002, 0.494, 0.002 } -- steps were set as [ 1 8 64 26 1 ] but should have been [ 0 0 0 99 0 ] at ../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:1029
```
Output of `fwupdmgr --version`:
```
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.3
compile   org.freedesktop.fwupd         2.0.7
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.12.21
runtime   org.freedesktop.fwupd         2.0.7
```
Output of `efivar -l | grep -i fw`:
```
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ff8bc5c6-2169-42b7-907b-c726a9c2dab2-0
```
Output of `fwupdmgr get-devices --show-all-devices`:
[fwupdmgr-get-devices.log](https://github.com/user-attachments/files/20140586/fwupdmgr-get-devices.log)
Output of `efibootmgr -v`:
[efibootmgr-v.log](https://github.com/user-attachments/files/20140589/efibootmgr-v.log)
Output of `drm_info`:
[drm_info.log](https://github.com/user-attachments/files/20140583/drm_info.log)

hughsie, 2025-05-10:
@eliasp does it work when not using nixos? We do zero testing on that I'm afraid.

eliasp, 2025-05-10:
@hughsie it used to work just fine using NixOS before on other devices, but I'll try to get my hands on some live USB of Fedora or so and will give it a try.