fwupd/fwupd #7816: UX capsule size calculation doesn't work unless screen is on

superm1, 2024-10-01:
I noticed on a few systems I usually use remotely the following DRM error pops up in the logs:

```sh
$ fwupdtool get-devices
Loading…                 [***************                        ]17:29:42.394 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:42.395 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:42.396 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:42.397 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:42.398 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
Loading…                 [*****************                      ]17:29:42.482 FuPluginUefiCapsule  failed to get screen size: failed to setup: failed to get DRM resources
```

However if I turn the screen (for example press keyboard key) then I'll get no errors from DRM:
```sh
$ fwupdtool get-devices
Loading…                 [***************                        ]17:29:59.279 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:59.281 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:59.282 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:59.283 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
17:29:59.284 FuUsbDevice          failed to parse platform BOS descriptor: failed to parse: no supported platform version: did not find magic
Loading…                 [************************************** ]
```

So this has me wondering, should there be a different fallback to use in this case?  I notice that the modes are all listed in sysfs..
```sh
$ cat /sys/class/drm/card0-eDP-1/modes
2560x1440
2560x1440
1920x1200
1920x1080
1600x1200
1680x1050
1280x1024
1440x900
1280x800
1280x720
1024x768
800x600
640x480
```

As is the EDID.

```sh
$ edid-decode /sys/class/drm/card0-eDP-1/edid
edid-decode (hex):

00 ff ff ff ff ff ff 00 06 af 8c e6 00 00 00 00
00 1d 01 04 a5 1f 11 78 03 f4 f5 a4 54 4d 9c 27
0f 50 54 00 00 00 01 01 01 01 01 01 01 01 01 01
01 01 01 01 01 01 e6 5f 00 a0 a0 a0 40 50 30 20
35 00 35 ae 10 00 00 18 b8 4c 00 a0 a0 a0 40 50
30 20 35 00 35 ae 10 00 00 18 00 00 00 fd 00 28
3c 5b 5b 19 01 0a 20 20 20 20 20 20 00 00 00 fe
00 42 31 34 30 51 41 4e 30 32 2e 33 20 0a 00 d3

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.4
  Vendor & Product Identification:
    Manufacturer: AUO
    Model: 59020
    Made in: 2019
  Basic Display Parameters & Features:
    Digital display
    Bits per primary color channel: 8
    DisplayPort interface
    Maximum image size: 31 cm x 17 cm
    Gamma: 2.20
    Supported color formats: RGB 4:4:4
    First detailed timing includes the native pixel format and preferred refresh rate
    Display is continuous frequency
  Color Characteristics:
    Red  : 0.6435, 0.3310
    Green: 0.3017, 0.6093
    Blue : 0.1552, 0.0615
    White: 0.3134, 0.3291
  Established Timings I & II: none
  Standard Timings: none
  Detailed Timing Descriptors:
    DTD 1:  2560x1440   60.011538 Hz  16:9     90.257 kHz    245.500000 MHz (309 mm x 174 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol N
                 Vfront    3 Vsync   5 Vback   56 Vpol N
    DTD 2:  2560x1440   48.009230 Hz  16:9     72.206 kHz    196.400000 MHz (309 mm x 174 mm)
                 Hfront   48 Hsync  32 Hback   80 Hpol N
                 Vfront    3 Vsync   5 Vback   56 Vpol N
    Display Range Limits:
      Monitor ranges (Bare Limits): 40-60 Hz V, 91-91 kHz H, max dotclock 250 MHz
    Alphanumeric Data String: 'B140QAN02.3 '
Checksum: 0xd3
```

hughsie, 2024-10-01:
> /sys/class/drm/card0-eDP-1/modes

But we don't know which /sys/class/drm/* device is default -- both the name and number isn't suitable for sorting to find "the default". Can you grab a drm_info from a machine with a powered down screen (e.g. using ssh) -- I'm wondering if the connector is going to be better than the crtc in that case.

superm1, 2024-10-01:
[drm_info.txt](https://github.com/user-attachments/files/17214323/drm_info.txt)

superm1, 2024-10-01:
Yeah the connector has the important info too.
> │   │   │   ├───2560x1440@60.01 preferred driver nhsync nvsync

hughsie, 2024-10-01:
Ohh hmm, we're getting `failed to get DRM resources` -- which is the call to `drmModeGetResources(fd)` -- I don't understand how drm_info is getting the data then -- as that's using basically the same things. Will ponder.

superm1, 2024-10-01:
Well AFAICT `errno` is getting set too; to `-EACCES`.

superm1, 2024-10-01:
Maybe it's getting opened RW and because a compositor is running that's now allowed.  So open the fd read only?

closed 2024-10-02 by superm1