fwupd/fwupd #7842: optionrom plugins overwrites device from amd_gpu plugin

superm1, 2024-10-04:
This is unique to 2.0.0.  When the daemon is launched with `--plugins amd-gpu` the device is enumerated properly.  but when launched with `--plugins optionrom --plugins amd-gpu` (or no args for that matter) then the `optionrom` ends up overwriting the amd-gpu device.

IE just amdgpu:
```
└─AMD Radeon RX 7900 XT:
      Device ID:          5cebed789f449e29b2b77ee5561740074c9c734c
      Summary:            AMD D704_XT_A0_20GB_MBA
      Current version:    022.001.002.008.000001
      Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
      Install Duration:   1 minute
      GUID:               8ed32270-e736-52e2-87d8-acef1dd2a946 ← AMD\113-D70401
      Device Flags:       • Updatable
                          • Needs a reboot after installation
                          • Cryptographic hash verification is available
                          • Device stages updates
                          • Device can recover flash failures
                          • Device is usable for the duration of the update
                          • Signed Payload
```

both:
```
└─Unknown Device:
      Device ID:          5cebed789f449e29b2b77ee5561740074c9c734c
      Current version:    113-D70401-00
      Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
      GUID:               d6b3dd44-3823-5271-a913-1bddb6f1f06b ← PCI\VEN_1002&DEV_744C
      Device Flags:       • Internal device
                          • Cryptographic hash verification is available
                          • Signed Payload
```

[amdgpu.log](https://github.com/user-attachments/files/17260122/amdgpu.log)
[both.log](https://github.com/user-attachments/files/17260124/both.log)

superm1, 2024-10-04:
I wonder if maybe we want to get rid of the optionrom plugin - it's just for dumps.  How about we just merge that functionality with the PCI device?

hughsie, 2024-10-04:
Hmm, so they have the same device ID? Can you use `fu_plugin_add_rule(plugin, FU_PLUGIN_RULE_BETTER_THAN, "optionrom");` in the amd-gpu plugin?

hughsie, 2024-10-04:
Can you grab me the -vv output for both plugins please -- I'll try to reproduce with the bcm57 I have here tomorrow that does the same trick.

superm1, 2024-10-04:
> Can you grab me the -vv output for both plugins please -- I'll try to reproduce with the bcm57 I have here tomorrow that does the same trick.

Yeah it's in first post

superm1, 2024-10-07:
this is fixed in cee6429

commit cee64297 (2024-10-04): Fix amdgpu probing

set priority above optionrom

Fixes: #7842
closed 2024-10-07 by superm1