fwupd/fwupd #7848: Compiling from fwupd-2.0.0.tar.xz fails du to SOURCE_VERSION being undefined

hardfalcon, 2024-10-05:
Compiling from the 2.0.0 release tarball [`fwupd-2.0.0.tar.xz`](https://github.com/fwupd/fwupd/releases/download/2.0.0/fwupd-2.0.0.tar.xz) fails because in [line 3968 of `fu-engine.c`](https://github.com/fwupd/fwupd/blob/2.0.0/src/fu-engine.c#L3968), `SOURCE_VERSION` is used without having been defined beforehand. I've resolved the issue with the following patch on my machine, but since I don't have a lot of experience with meson, I'm not sure if that's really the "correct" solution, and hence I've refrained from opening a pull request.

```patch
diff --git a/meson.build b/meson.build
index d8c4456fb0cc..cb9df0a3dd38 100644
--- a/meson.build
+++ b/meson.build
@@ -27,6 +27,8 @@ if git.found()
   endif
   tag = run_command([git, 'describe', '--exact-match'], check: false).returncode() == 0
   conf.set_quoted('SOURCE_VERSION', source_version)
+else
+  conf.set_quoted('SOURCE_VERSION', fwupd_version)
 endif
 
 # libtool versioning - this applies to libfwupd
```

superm1, 2024-10-05:
Thanks for reporting it. A fix went in similar to your proposal.

https://github.com/fwupd/fwupd/commit/9637c35339111d92add461fc3ad2a032e90abc9c

hardfalcon, 2024-10-06:
Ohh, thanks for the hint, and sorry for having overlooked that commit 😅

closed 2024-10-06 by hardfalcon