fwupd/fwupd #7874: parade-usbhub plugin causing system to reboot in 1.9.25

rishabh-chromium, 2024-10-08:
**Describe the bug**
The system is rebooting once a parade-usb hub is attached and detached with this error:
```
FuDeviceList         ::added cf294bf55b333004beb7c41f952c1838c23e1f4a [USB2.0 Hub]
FuEngine             failed to probe device usb:02:00:01:03: failed to get interfaces: USB error on device 04b4:6502 : No such device (it may have been disconnected) [-4]
FuEngine             failed to probe device usb:02:00:01:03:01: failed to get interfaces: USB error on device 0bda:8153 : No such device (it may have been disconnected) [-4]
FuEngine             failed to probe device usb:02:00:01:05: failed to get interfaces: USB error on device 0489:e102 : No such device (it may have been disconnected) [-4]
FuEngine             failed to probe device usb:02:00:01:06: failed to get interfaces: USB error on device 1da0:551e : No such device (it may have been disconnected) [-4]
FuEngine             failed to probe device usb:02:00:01:07: failed to get interfaces: USB error on device 1da0:551f : No such device (it may have been disconnected) [-4]
```

There is a kernel panic with this log:
```
   47.015955] Bluetooth: btmtk_usb_hci_wmt_sync() hci0: Execution of wmt command timed out
[   47.024120] Bluetooth: btmtk_usb_setup() hci0: Failed to send wmt func ctrl (-110)
[   47.031727] Bluetooth: hci_dev_setup_sync() hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[   47.044805] Unable to handle kernel paging request at virtual address ffffffffffffffd8
[   47.052730] Mem abort info:
[   47.055518]   ESR = 0x0000000096000005
[   47.059269]   EC = 0x25: DABT (current EL), IL = 32 bits
[   47.064584]   SET = 0, FnV = 0
[   47.067634]   EA = 0, S1PTW = 0
[   47.070768]   FSC = 0x05: level 1 translation fault
[   47.075644] Data abort info:
[   47.078521]   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
[   47.084004]   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
[   47.089053]   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
[   47.094366] swapper pgtable: 4k pages, 39-bit VAs, pgdp=0000000086900000
[   47.101063] [ffffffffffffffd8] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[   47.109756] Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
[   47.116022] cpufreq stop DVFS log done
[   47.119764] Modules linked in: xt_nat 8021q zram zsmalloc zstd_compress lz4_compress uinput r8153_ecm cdc_ether usbnet btusb btmtk btbcm btintel r8152 btrtl mii mt8196_mt6681 mtk_jpev
[   47.188916] CPU: 1 PID: 85 Comm: kworker/u17:0 Not tainted 6.6.53-05684-g7becd7b69aff #1 82063f7ec383f5ff2ca1268969acc9a8a468160d
[   47.200575] Hardware name: Google Rauru board (DT)
[   47.205370] Workqueue: hci0 hci_cmd_sync_work [bluetooth]
[   47.210939] pstate: 834000c9 (Nzcv daIF +PAN -UAO +TCO +DIT -SSBS BTYPE=--)
[   47.217903] pc : usb_kill_anchored_urbs+0x60/0x168
[   47.222703] lr : usb_kill_anchored_urbs+0x44/0x168
[   47.227496] sp : ffffffc080d2bca0
[   47.230810] x29: ffffffc080d2bca0 x28: ffffff80812dbe00 x27: ffffff808796e040
[   47.237958] x26: dead000000000100 x25: ffffffee16379f69 x24: ffffff808796e4a8
[   47.245101] x23: 0000000000000000 x22: 00000000ffffffff x21: 0000000000000001
[   47.252244] x20: ffffffffffffffd8 x19: ffffff808796f560 x18: 0000000000000249
[   47.259387] x17: ffffff808796e490 x16: ffffffee8608c880 x15: ffffff8080d53c28
[   47.266529] x14: 0000000000000000 x13: 0000000000011ec2 x12: 0000000000000018
[   47.273672] x11: 000000004cec4ec5 x10: 002d6b3280396750 x9 : 0000000000000001
[   47.280816] x8 : 0000000000000000 x7 : 0000000080808080 x6 : 8080808000000000
[   47.287959] x5 : ffffffc080d2bbb0 x4 : 0000000000000000 x3 : 0000000000000000
[   47.295102] x2 : 0000000000000000 x1 : 0000000000000000 x0 : ffffff808796f588
[   47.302244] Call trace:
[   47.304691]  usb_kill_anchored_urbs+0x60/0x168
[   47.309138]  btusb_mtk_release_iso_intf+0x2c/0xd8 [btusb d57c30d4227bd4c44f22eef7d8f15020516d6fde]
[   47.318126]  btusb_mtk_reset+0x5c/0x470 [btusb d57c30d4227bd4c44f22eef7d8f15020516d6fde]
[   47.326209]  hci_cmd_sync_work+0x10c/0x188 [bluetooth 2764b3717521b9e36357f8819b5d9baca9c211a5]
[   47.334965]  process_scheduled_works+0x250/0x4a0
[   47.339590]  worker_thread+0x330/0x428
[   47.343352]  kthread+0x108/0x1d0
[   47.346597]  ret_from_fork+0x10/0x20
[   47.350191] Code: f9400677 f100a2f4 54fffea0 d503201f (b8350288) 
[   47.356285] ---[ end trace 0000000000000000 ]---
[   47.385633] Kernel panic - not syncing: Oops: Fatal exception
[   47.391381] SMP: stopping secondary CPUs
[   47.395722] Kernel Offset: 0x2e05600000 from 0xffffffc080000000
[   47.401642] PHYS_OFFSET: 0x80000000
[   47.405129] CPU features: 0x1,00000000,d063cd43,90fe7607
[   47.410443] Memory Limit: none
[DL] 00000000 00000000 010701
```

If we disable BT in kernel and connect USB to DUT directly, the kernel won't crash. 
Pretty sure this CL is causing it: https://github.com/fwupd/fwupd/commit/4ead6cd9f1acfc1aeb52c5dfd91b3a569da177b7

```shell
fwupdmgr --version
```
1.9.25

**Additional questions**

- Operating system and version: chromeos
- Have you tried rebooting? Yes
- Is this a regression? Yes

hiwang123, 2024-10-08:
Prior to the parade plugin CL, usb devices (including BT usb) can be detected. When fwupd is run with this plugin, we saw usb disconnected. If device is boot from USB, there will be another kernel crash
```
[   33.209141] usb 2-1.3: USB disconnect, device number 3
[   33.209157] r8152-cfgselector 2-1.3.1: USB disconnect, device number 6
[   33.209397] r8152 2-1.3.1:1.0 eth0: Stop submitting intr, status -108
[   33.209488] sd 1:0:0:0: [sdb] tag#0 UNKNOWN(0x2003) Result: hostbyte=0x07 driverbyte=DRIVER_OK cmd_age=0s
[   33.209502] sd 1:0:0:0: [sdb] tag#0 CDB: opcode=0x28 28 00 00 51 b7 20 00 00 48 00
[   33.209506] I/O error, dev sdb, sector 5355296 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 2
[   33.209511] device-mapper: verity: verification failure occurred: block failure (transient)
[   33.209517] Kernel panic - not syncing: dm-verity failure: device:8:19 status:10 block:610532 message:block
[   33.209522] CPU: 3 PID: 178 Comm: usb-storage Not tainted 6.6.52-CL7032071-v164-05297-g3d1a6a82a249-dirty #1 39429b85c18a7ca7006af678436bb7c1b981141a
[   35.179277] Hardware name: Google Rauru board (DT)
[   35.179281] Call trace:
[   35.179284]  dump_backtrace+0xec/0x130
[   35.179295]  show_stack+0x1c/0x28
[   35.179300]  dump_stack_lvl+0x80/0xa8
[   35.179311]  dump_stack+0x14/0x28
[   35.179316]  panic+0x140/0x398
[   35.179321]  verity_finish_io+0x1d8/0x1e0
[   35.179336]  verity_end_io+0x20/0x70
[   35.179341]  bio_endio+0x1ac/0x1d0
[   35.179351]  blk_update_request+0x178/0x3b8
[   35.179366]  scsi_end_request+0x34/0x2d8
[   35.179380]  scsi_io_completion_action+0x3a0/0x400
[   35.179386]  scsi_io_completion+0xd0/0x138
[   35.179390]  scsi_finish_command+0xa4/0xc0
[   35.179395]  scsi_complete+0x74/0x190
[   35.179403]  scsi_done_internal+0x64/0x138
[   35.179408]  scsi_done_direct+0x18/0x28
[   35.179413]  usb_stor_control_thread+0x198/0x258
[   35.179425]  kthread+0x108/0x1d0
[   35.179438]  ret_from_fork+0x10/0x20
[   35.179449] SMP: stopping secondary CPUs
[   35.179571] cpufreq stop DVFS log done
[   35.179577] Kernel Offset: 0x23c3000000 from 0xffffffc080000000
[   35.179581] PHYS_OFFSET: 0x80000000
[   35.179584] CPU features: 0x1,00000000,d063cd43,90fe7607
[   35.179589] Memory Limit: none
```

Both the BT usb crash or this crash are due to USB disconnected after the plugin is run

superm1, 2024-10-09:
CC @hublin2024 @jimmytu5167 @andychu5168

hughsie, 2024-10-09:
We're only matching

	# PS5512 devboard
	[USB\VID_1DA0&PID_551A]
	Plugin = parade_usbhub
	[USB\VID_1DA0&PID_551D]
	Plugin = parade_usbhub

Those should be specific to the devboards -- is that what you're plugging in?

hiwang123, 2024-10-09:
we're using  PS5512/PS5511

hughsie, 2024-10-09:
What USB vid pid is your device tho? Surely not the same as the devboard?!

hiwang123, 2024-10-10:
On our devboard:

Bus 002 Device 002: ID 1da0:551d Parade Technologies, Inc. USB2.0 Hub

I think this one matches 

[USB\VID_1DA0&PID_551D]
Plugin = parade_usbhub

hughsie, 2024-10-10:
I suspect that your devboard is supposed to have a different VID/PID to the Parade one. :)

The only thing we're doing at startup is doing 4 u8 mmio reads, which I'd be very surprised if caused problems. I wonder if you're running a very old parade firmware version or something. 

Could you grab us the `sudo fwupdtool --plugins parade_usbhub get-devices -vv` output please.

superm1, 2024-10-10:
@hughsie I wonder if we want a flag on devices that are known EVB/DVB IDs to require an extra nudge to do stuff to avoid this?

hughsie, 2024-10-10:
Right; the thing is it's 4 control transfers that trigger mmio reads -- I'm a bit surprised it's making the device explode. My suspicion is that older firmware (i.e. keyed from the REV) doesn't handle REQUEST_READ with a NAK, but instead **explodes** but the Parade team might know more.

superm1, 2024-10-10:
Yeah; but this is probably not the last of those.  Maybe something like `EnableDeveloperBoards=true` in `/etc/fwupd/fwupd.conf`? And then we can flag anything that's an EVB with a quirk for `developer-board`.

hiwang123, 2024-10-10:
fwupdtool --plugins parade_usbhub get-devices -vv outout 
[output.txt](https://github.com/user-attachments/files/17333977/output.txt)

hughsie, 2024-10-10:
So this:

	20:33:33.945 FuEngine             usb:02:00:01 added FuUsbDevice:
	  Guid:                 92ac3e32-0949-593d-8fa4-7e9b1561836f <- USB\VID_1DA0&PID_551D ?
	  Flags:                none
	  VendorId:             USB:0x1DA0
	  Version:              1.2
	  VersionFormat:        bcd
	  VersionRaw:           0x00000102
	  Guid[quirk]:          8ff24b17-c4fb-5dd4-a94c-d0d7c0529212 <- USB\VID_1DA0
	  Guid[quirk]:          49318c48-6d84-5963-92e5-453212356f7c <- USB\CLASS_09
	  Guid[quirk]:          c059f9a6-f12a-5960-8e74-ca66ea22b934 <- USB\CLASS_09&SUBCLASS_00
	  Guid[quirk]:          317baf53-b789-59f5-b08e-aebcb14fac6f <- USB\CLASS_09&SUBCLASS_00&PROT_01
	  Guid[quirk]:          7bac9b06-d7af-521f-8331-cb2c8995add0 <- USB\CLASS_09&SUBCLASS_00&PROT_02
	  PhysicalId:           usb:02:00:01
	  AcquiesceDelay:       2500
	  PossiblePlugin:       parade_usbhub
	  ParentPhysicalIds:    usb:02:00,usb:02
	  UsbDeviceClass:       hub

I think we need some help from the Parade people now. Help!

hughsie, 2024-10-10:
Random question for @hiwang123 -- do you get identical behaviour with the same devboard on any random Linux machine? Or another _type_ of ChromeBook?

hiwang123, 2024-10-10:
I currently don't have other boards (linux machine or chromebook) using the same devboard though.

hublin2024, 2024-11-20:
I found an issue here:
Should be PS5512, not PS551A.
[image]

closed 2025-05-03 by hughsie