fwupd/fwupd #7879: test-snap CI job is failing

superm1, 2024-10-10:
This seems to be caused by something outside of fwupd control, but our CI started failing the test snap job.  The only changes to fwupd between these were to a contrib/ python script not run during CI.

Working commit: https://github.com/fwupd/fwupd/actions/runs/11254189339/job/31291297158
Failing commit: https://github.com/fwupd/fwupd/actions/runs/11256125590/job/31297660415

Here is the failure that occurs now:
```
Getting the list of remotes...
Failed to connect to daemon: GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: An AppArmor policy prevents this sender from sending this message to this recipient; type="method_call", sender=":1.23" (uid=0 pid=3021 comm="/snap/fwupd/x1/bin/fwupdmgr get-remotes" label="snap.fwupd.fwupdmgr (enforce)") interface="org.freedesktop.fwupd" member="SetHints" error name="(unset)" requested_reply="0" destination=":1.24" (uid=0 pid=3064 comm="/usr/libexec/fwupd/fwupd" label="unconfined")
Oct 09 13:42:10 fv-az104[6](https://github.com/fwupd/fwupd/actions/runs/11254189339/job/31291297158#step:4:7)-0 systemd[1]: Starting fwupd.service - Firmware update daemon...
Oct 09 13:42:11 fv-az1046-0 fwupd[3064]: 13:42:11.180 FuMain               Daemon ready for requests (locale C.UTF-8)
Oct 09 13:42:11 fv-az1046-0 systemd[1]: Started fwupd.service - Firmware update daemon.
```

CC @valentindavid @bboozzoo for any comments

bboozzoo, 2024-10-10:
If I'm reading the log correctly, `fwupdmgr get-remotes` was running with an apparmor label, so likely executing from a snap. The fwupd dbus service label was `uconfined`, so it's likely it wasn't a snap process, but rather something from fwupd distribution package. This should be allowed if the `fwupd` snap is connected to `fwupd` slot on the system (as in `snap connect fwupd:fwupdmgr :fwupd`). Is that the case in the test?

superm1, 2024-10-10:
I tried to add the extra connection and now it looks like it's definitely reaching out to the daemon outside of the snap.

https://github.com/fwupd/fwupd/actions/runs/11276166932/job/31359753649?pr=7882

superm1, 2024-10-10:
Looking more closely between the pass and fail it looks like this is tied to a flip with the underlying runner version...

Pass:
```
Ubuntu
  22.0[4](https://github.com/fwupd/fwupd/actions/runs/11254189339/job/31291297158#step:1:4).5
  LTS
Runner Image
  Image: ubuntu-22.04
  Version: 2024100[6](https://github.com/fwupd/fwupd/actions/runs/11254189339/job/31291297158#step:1:7).1.0
  Included Software: https://github.com/actions/runner-images/blob/ubuntu22/2024[10](https://github.com/fwupd/fwupd/actions/runs/11254189339/job/31291297158#step:1:11)06.1/images/ubuntu/Ubuntu2204-Readme.md
  Image Release: https://github.com/actions/runner-images/releases/tag/ubuntu22%2F20241006.1
```

Fail:
```
Ubuntu
  2[4](https://github.com/fwupd/fwupd/actions/runs/11256125590/job/31297660415#step:1:4).04.1
  LTS
Runner Image
  Image: ubuntu-24.04
  Version: 20240[9](https://github.com/fwupd/fwupd/actions/runs/11256125590/job/31297660415#step:1:10)22.1.0
  Included Software: https://github.com/actions/runner-images/blob/ubuntu24/20240922.1/images/ubuntu/Ubuntu2404-Readme.md
  Image Release: https://github.com/actions/runner-images/releases/tag/ubuntu24%2F20240922.1
```

superm1, 2024-10-10:
So I guess fwupd distro package is installed in the rootfs starting for 24.04.    Removing that during the job fixes it.  I'll merge that.

superm1, 2024-10-10:
closed with https://github.com/fwupd/fwupd/commit/6662076f269fb21b4efdf35cd6e191b2cf536da5

closed 2024-10-10 by superm1

bboozzoo, 2024-10-14:
Glad you were able to sort it out. Yes, it's likely that fwupd is now seeded in the desktop images.