fwupd/fwupd #7894: Unclear activation message

bulletmark, 2024-10-14:
I updated my Arch laptop today  and got update `fwupd 1.9.25-1 -> 2.0.0-1`. See  the annotated comments on the commands below:

```
### Do a refresh/update:
$ sudo fwupdmgr refresh
Metadata is up to date; use --force to refresh again.

$ sudo fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [  -                                    ]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: Updates have been published for 8 of 12 local devices
WD19TB is pending activation; use fwupdmgr activate to complete the update.

### Never seen that WD19TB message before. Try something intuitive:
$ sudo fwupdmgr activate WD19TB
failed to find WD19TB

### Damn, was hoping that earlier message gave a bit of a hint how to resolve this.
### Must need something else so list devices to get an ID:
$ sudo fwupdmgr get-devices
Dell Inc. XPS 13 9300
│
├─WD19TB:
│ │   Device ID:          4940a095e87a030ab4507472901d8fe8cc7b9785
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.07
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      4DN5523/3159128927644594
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │
│ ├─Package level of Dell dock:
│ │     Device ID:        dfca826f643852980d61dde0e4b644e89e4a08e8
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.36.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUIDs:            d8927ff5-a5b2-5618-848b-8e8bfb75b66f
│ │                       8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        12e83c17bab7a5fd17ce1cd8b314c3403415e237
│ │     Summary:          USB 3.1 Generation 1 Hub
│ │     Current version:  01.22
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 14 seconds
│ │     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │
│ ├─RTS5487 in Dell dock:
│ │     Device ID:        13b657afad936c93b8fa18bfc6cd602e046b89ed
│ │     Summary:          USB 3.1 Generation 2 Hub
[... 158 lines cut ...]

### Now do the activate using device ID:
$ sudo fwupdmgr activate 4940a095e87a030ab4507472901d8fe8cc7b9785
No firmware to activate

### Run the refresh again to see that "error" message is still present.
### Note message does not appear unless I add `--force`.
$ sudo fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [  -                                    ]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: Updates have been published for 8 of 12 local devices
WD19TB is pending activation; use fwupdmgr activate to complete the update.

### Message is still there but NFI how to address it.
```

Basically, a message is being presented to me but gives me no idea what it means or how to address it?

Also, and I realise this will be a completely different issue, I thought I'd run `gnome-firmware` (current Arch version is 46.0-1) to see if the fancy GNOME interface would give me better info or guide me over this but it just does:
```
$ gnome-firmware 
gnome-firmware: error while loading shared libraries: libfwupd.so.2: cannot open shared object file: No such file or directory
```

superm1, 2024-10-14:
Wd19 does a delayed update. You can trigger it by either unplugging the dock or running the command suggested above.

bulletmark, 2024-10-14:
Sorry, but what "command suggested above"?

bulletmark, 2024-10-14:
Also, I unplugged the dock, waited, and plugged it back in but no difference. I also power reset the dock, and then reset my laptop, again with no difference. I always get that "WD19TB is pending activation" message, whatever that means.

superm1, 2024-10-14:
When you say unplugged dock did you mean from wall or laptop? I meant laptop.

CC @CragW

bulletmark, 2024-10-14:
Of course I unplugged it from the laptop. That is why I next said "I also power reset the dock".

superm1, 2024-10-14:
> Of course I unplugged it from the laptop. That is why I next said "I also power reset the dock".

How much time between these steps?

bulletmark, 2024-10-14:
Doesn't seem to matter how long I wait. I've had the laptop disconnected for hours, power reset both laptop and dock many times, etc. Every time I run the command I get the `pending activation` message. I get this message whether the dock is connected or not.
```
$ sudo fwupdmgr refresh
Metadata is up to date; use --force to refresh again.

$ sudo fwupdmgr refresh --force
Updating lvfs-testing
Downloading…             [  -                                    ]
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: Updates have been published for 0 of 4 local devices
WD19TB is pending activation; use fwupdmgr activate to complete the update.
```

I raised this issue simply because the message is unclear. Are you saying that the message should have gone by now so there must be a software bug preventing that?

superm1, 2024-10-15:
> Does this help to remove the message? 
> ```
> $ sudo mv /var/lib/fwupd/pending.db /tmp/
> $ fwupdmgr refresh --force
> ```

And if so; share that pending.db before rebooting.

bulletmark, 2024-10-15:
Yes, if I remove that file the message goes away on the next `sudo fwupdmgr refresh --force`.

Here is the file https://app.box.com/s/1tg80axqibxcipbixc8uu2ufablbx7ms

CragW, 2024-10-22:
> Yes, if I remove that file the message goes away on the next `sudo fwupdmgr refresh --force`.

Good to know it did help.

> Here is the file https://app.box.com/s/1tg80axqibxcipbixc8uu2ufablbx7ms

No idea why a staged update status persist in the `db`, while the device list looks okay.  Let's close issue for now, please reopen or new an issue if you run into this issue again.

closed 2024-10-22 by CragW

superm1, 2024-10-28:
I just happened to try to update a WD22 and reproduced this exact same issue with https://github.com/fwupd/fwupd/commit/5fe5a9e685f62763a610a16dfb15a5edc9e1b67b

[activation.txt](https://github.com/user-attachments/files/17547593/activation.txt)

At least in this case it seems to be from the USB4 controller not updating properly.
```
18:36:54.957 FuDevice             setting parent of USB4 controller in Dell dock [486e29f297ec9cd67b9b4c8fac5fed3e24aa9b3f] to be WD22TB4-upgraded [8a59d433aee8a8b410eaedee1b0a623f878aa7f4]
18:36:54.964 FuEngine             inheriting needs-activation for USB4 controller in Dell dock as version 38.0 != 41.81
```

superm1, 2024-10-28:
Yeah so the message is coming from
```
fu_device_add_private_flag(FU_DEVICE(self), FU_DEVICE_PRIVATE_FLAG_INHERIT_ACTIVATION);
```

But it seems like the intel-usb4 plugin is totally not working.

Zenkibou, 2024-11-10:
Hello, I have the same issue with my WD19TBS, I would be happy to provide my logs if needed.

fardog, 2024-11-27:
I've got the same issue with a WD19S;

```
➜  ~ fwupdmgr refresh --force
Updating lvfs
Downloading…             [    \                                  ]
Successfully downloaded new metadata: Updates have been published for 7 of 8 local devices
WD19S is pending activation; use fwupdmgr activate to complete the update.
➜  ~ fwupdmgr activate
No firmware to activate
```

Unplugging the laptop from the dock (even for hours/days) nor resetting the dock (by unplugging power) solves this issue; it's been happening for actual months at this point through all the usage scenarios you might imagine. Let me know if there's any additional information you'd like me to provide, thanks!

devurandom, 2024-12-11:
Similar here:
```
❯ sudo fwupdmgr refresh --force
Updating lvfs
Downloading…             [    \                                  ]
Successfully downloaded new metadata: Updates have been published for 5 of 11 local devices
WD22TB4 is pending activation; use fwupdmgr activate to complete the update.
WD22TB4 is pending activation; use fwupdmgr activate to complete the update.

❯ sudo fwupdmgr activate
No firmware to activate

❯ sudo fwupdmgr get-devices
[REDACTED]

❯ sudo fwupdmgr activate [REDACTED DEVICE ID]
No firmware to activate

❯ sudo fwupdmgr activate [REDACTED GUID]
Selected device: WD22TB4
No firmware to activate
```

devurandom, 2024-12-12:
> still same issue with WD19
> 
> WD19:
>   │   Device ID:          d2141f8b95426f7c80924585c9966e48dbe3324f
>   │   Previous version:   01.01.00.10
>   │   Install Duration:   3 seconds
>   │   Update State:       Success
>   │   Update Error:       Device has been removed, Pending activation
>   │   Last modified:      2024-12-05 14:47

Seeing this reminds me: During the update of my Dell dock I was told "update will start when device is unplugged" (or something to that effect).

* Did y'all also see that message?
* Did y'all, like me, also follow that instruction and unplugged the dock from the computer?
* Is that, from a fwupd developer's perspective, actually the correct action, or should I not have done that?

lemmyg, 2024-12-13:
yes, 
I got that message and I believe after disconnect the the thunderbolt cable the dock station got updated.
however the error message in history is confusing.
Update Error: Device has been removed, Pending activation

kriive, 2025-01-22:
The solution, for me, was to re-install the update (maybe I rolled back to a previous version and re-did the upgrade, don't remember well) and *not remove the Dock*. Instead, I launched `fwupdmgr activate` and it managed to actually update the firmware. No more pending activations. :)

superm1, 2025-01-22:
> The solution, for me, was to re-install the update (maybe I rolled back to a previous version and re-did the upgrade, don't remember well) and *not remove the Dock*. Instead, I launched `fwupdmgr activate` and it managed to actually update the firmware. No more pending activations. :)

That points to a firmware bug with delayed activation. @CragW does it help with investigation perhaps?

CragW, 2025-01-23:
> That points to a firmware bug with delayed activation. [@CragW](https://github.com/CragW) does it help with investigation perhaps?

hmm.. that sounds like something around disconnect update. Will check.

eoli3n, 2025-02-14:
I face that same problem, any idea how to fix it ?

```
~ 
➜ sudo LANG=C.utf8 fwupdmgr refresh --force
Updating lvfs
Downloading…             [   -                                   ]
Successfully downloaded new metadata: Updates have been published for 7 of 11 local devices
WD19 is pending activation; use fwupdmgr activate to complete the update.

~ 
➜ sudo LANG=C.utf8 fwupdmgr get-devices
Dell Inc. Latitude 5420
│
├─WD19:
│ │   Device ID:          b16e5d2fe8ef7c8b6b8ab3b1b553ed687e6cf13a
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.10
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      HSZJ883/3118489924401789
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        d9a15c80c62781efab2a76e75ef936937bfa9e26
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.44.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUIDs:            d8927ff5-a5b2-5618-848b-8e8bfb75b66f
│ │                       8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        897209047b125330149327f81d7802751699f801
│ │     Summary:          USB 3.1 Generation 1 Hub
│ │     Current version:  01.23
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 14 seconds
│ │     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   
│ ├─RTS5487 in Dell dock:
│ │     Device ID:        a0ac5add13ce7e5536b77487c6e5676b0a2d39ca
│ │     Summary:          USB 3.1 Generation 2 Hub
│ │     Current version:  01.59
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 3 seconds
│ │     GUIDs:            acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
│ │                       568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
[... 109 lines cut ...]
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    554774546
│     Minimum Version:    554774546
│     Vendor:             DMI:Dell Inc.
│     Update State:       Success
│     GUID:               b0e59891-8629-46c3-8151-cbf85ab13ae0
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Dell Inc. Platform Key
      Vendor:             Dell Inc.
      GUID:               5f7cc4ea-9f7a-5c5f-b535-dfbcc73f5ba6 ← UEFI\CRT_2493B118EABD632FD3BA49C333A8329B1982E872
    

~ 
➜ sudo LANG=C.utf8 fwupdmgr activate cd357cf1-40b2-5d87-b8df-bb2dd82774aa
Selected device: WD19
No firmware to activate
```

bulletmark, 2025-02-14:
I still get this message every time I do a refresh.

DorBreger, 2025-03-12:
Any  news on this? I still have this problem too

DorBreger, 2025-03-12:
I think this problem is due to #8368  
What happens is the upgrade fails, and then the dock is waiting for activation ?? or something to be updated again. The fix for this is to wait for fwupd 2.0.7 to be released with the fix, then run 
```
fwupdmgr clear-results
fwupdtool clear-history
```
And upgrade as usual.

bulletmark, 2025-03-26:
FYI, `fwupd` 2.0.7 released on Arch as of today.

DorBreger, 2025-03-27:
> FYI, `fwupd` 2.0.7 released on Arch as of today.

I can tell. This allowed me to upgrade the dock, but unfortunately it still goes to sleep unexpectedly when used in thenderbolt mode. So I see no difference

CragW, 2025-03-27:
> > FYI, `fwupd` 2.0.7 released on Arch as of today.
> 
> I can tell. This allowed me to upgrade the dock

What is your thunderbolt version before and after the upgrade?

> it still goes to sleep unexpectedly when used in thenderbolt mode. So I see no difference

Can you eleborate more about the sleep?

DorBreger, 2025-03-28:
>> What is your thunderbolt version before and after the upgrade?

How do I check? Thudnerbolt of what?

> Can you eleborate more about the sleep?

Sure. The dock only works in thunderbolt mode when I connect it to a turned off laptop and turn it on. Otherwise it appears as turned off in `boltctl list`. The updates notes in fwupd said something about fixing a problem with going to sleep when plugging and unplugging laptop.

iyanmv, 2025-04-23:
As mentioned in [this comment](https://github.com/fwupd/fwupd/issues/8706#issuecomment-2823331613), I experience this issue on latest version after updating WD22TB4.

```
fwupdmgr --version
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.8
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.9
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.3-arch1-1
runtime   org.freedesktop.fwupd         2.0.8
```

rosti-il, 2025-05-20:
Same issue in Fedora 42

```
Name            : fwupd
Epoch           : 0
Version         : 2.0.9
Release         : 1.fc42
Architecture    : x86_64
Installed size  : 8.8 MiB
Source          : fwupd-2.0.9-1.fc42.src.rpm
From repository : updates
Summary         : Firmware update daemon
URL             : https://github.com/fwupd/fwupd
License         : LGPL-2.1-or-later
Description     : fwupd is a daemon to allow session software to update device firmware.
Vendor          : Fedora Project
```
```
$ fwupdmgr --version
compile   info.libusb                   1.0.28
compile   org.freedesktop.fwupd         2.0.9
compile   com.hughsie.libxmlb           0.3.22
compile   org.freedesktop.Passim        0.1.10
compile   com.hughsie.libjcat           0.2.3
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.10
runtime   com.hughsie.libxmlb           0.3.22
runtime   com.hughsie.libjcat           0.2.3
runtime   org.kernel                    6.14.6-300.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.9

$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [***************************************]
Successfully downloaded new metadata: Updates have been published for 8 of 21 local devices
WD19 is pending activation; use fwupdmgr activate to complete the update.

$ fwupdmgr get-updates | tail -n 0
No updates available
```
related part of the `fwupdmgr get-devices` output
```
├─WD19:
│ │   Device ID:          b16e5d2fe8ef7c8b6b8ab3b1b553ed687e6cf13a
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.13
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      4HGZBB3/3126229594688973
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        d9a15c80c62781efab2a76e75ef936937bfa9e26
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.47.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUID:             8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        897209047b125330149327f81d7802751699f801
│ │     Summary:          USB 3.1 Generation 1 Hub
│ │     Current version:  01.23
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 14 seconds
│ │     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   
│ ├─RTS5487 in Dell dock:
│ │     Device ID:        a0ac5add13ce7e5536b77487c6e5676b0a2d39ca
│ │     Summary:          USB 3.1 Generation 2 Hub
│ │     Current version:  01.62
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 3 seconds
│ │     Serial Number:    2000
│ │     GUIDs:            acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
│ │                       568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
│ │   
│ └─VMM5331 in Dell dock:
│       Device ID:        98109c061a84a70a3ac0f6a19d90b9983877c53b
│       Summary:          Multi Stream Transport controller
│       Current version:  05.07.08
│       Vendor:           Dell Inc. (USB:0x413C)
│       Install Duration: 6 minutes
│       GUID:             89fec0b6-6b76-5008-b82c-5e5c6c164007 ← MST-panamera-vmm5331-259
│       Device Flags:     • Updatable
│                         • Supported on remote server
│                         • Device stages updates
│                         • Device is usable for the duration of the update
│                         • Unsigned Payload
```

The `/var/lib/fwupd/pending.db` file was deleted but it didn't help.
The 01.01.00.13 version is the latest now.

What else can I do?

rosti-il, 2025-05-20:
Found the solution (workaround?). I just removed and installed again `fwupd` using DNF. After this the `/var/lib/fwupd/pending.db` file was created again (I deleted in manually previously). Probably some other data file(s) were recreated.

```
$ fwupdmgr refresh --force
Updating lvfs
Downloading…             [         |                             ]
Successfully downloaded new metadata: Updates have been published for 8 of 20 local devices
```
If you want investigate it further you can check the source code of the `fwupd` package in Fedora there: https://src.fedoraproject.org/rpms/fwupd/tree/f42

bulletmark, 2025-06-04:
Package `fwupd` updated from 2.0.10-1 -> 2.0.11-1 on Arch Linux today but this bug still exists. Always get `WD19TB is pending activation; use fwupdmgr activate to complete the update.` after a refresh and doing `fwupdmgr activate <whatever>` always says `No firmware to activate`.

CragW, 2025-07-03:
It seems a new flow has been introduced at some time? If the device has added `needs-activation` flag when a firmware has installed and is staged, and we don't use `activate` for the activation to take effect immediately, then every time run `fwupdmgr refresh --force` the client side pops up a notification say the device is waiting for `activation`. However, `get-devices` doesn't show `needs-activation` flag at all.

For example, a new firmware has installed on WD22TB4, and a `needs-activation` has been added to it but we go `disconnect-update` flow so the `activate` step has intended to skip. I can see the following message every time refreshes the metadata: 
```
WD22TB4 is pending activation; use /home/crabo/mywork/fwupd/venv/bin/../dist/bin/fwupdmgr activate to complete the update.
```

superm1, 2025-07-04:
Can you trace it?  Is `FWUPD_DEVICE_FLAG_NEEDS_ACTIVATION` coming from the history database donor device?  I think it should be cleared when the real device shows up.

CragW, 2025-07-04:
I'm seeing two issues:

1. The output of `get-history` and `get-devices` do not have correct relationship for a device to its new version. For example:
```
 • Package level of Dell dock (01.00.44.01 → 06.05.06)
 • RTS5487 in Dell dock (01.59 → 06.05.06)
 • VMM5331 in Dell dock (05.07.07 → 06.05.06)
 • WD22TB4 (01.01.00.10 → 06.05.06)
```
2. The output of `refresh` always remind users there is a pending update to be activated even the `needs-activation` flag already cleared (refer to the output of `get-devices`).
```
WD22TB4 is pending activation; use /home/crlab/mywork/fwupd/venv/bin/../dist/bin/fwupdmgr activate to complete the update.
```

[pending.db.txt](https://github.com/user-attachments/files/21049698/pending.db.txt)

The history  for WD22TB4 says (after device is re-connected to the host system):
```
Device has been removed, Pending activation
```

<details>

```
(fwupd) crlab@crlab-Precision-7670:~/mywork/fwupd$ fwupdmgr get-devices
WARNING: This package has not been validated, it may not work properly.
Dell Inc. Precision 7670
│
├─WD22TB4:
│ │   Device ID:          d956921368fea25d1dd4fdc9bf52c5659070a6e2
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.14
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      7T2X0L3/3198720181961507
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │
│ ├─Package level of Dell dock:
│ │     Device ID:        a24dd6545a582ee8ccc008cf9f836623eff25e8e
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.01.00.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 5 seconds
│ │     GUID:             7fb217aa-2312-5c99-a853-56d865c76815 ← USB\VID_413C&PID_B06E&hub&salomon_mlk_status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        afcf38cb620bab2b9589d3a857f00c43e552b301
│ │     Summary:          USB 3.1 Generation 1 Hub
│ │     Current version:  01.23
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 14 seconds
│ │     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │                       • Can tag for emulation
[... 240 lines cut ...]
                          • System requires external power source
                          • Device stages updates
                          • Signed Payload
                          • Can tag for emulation

────────────────────────────────────────────────
Devices that have been updated successfully:
 • Package level of Dell dock (01.00.44.01 → 06.05.06)
 • RTS5487 in Dell dock (01.59 → 06.05.06)
 • VMM5331 in Dell dock (05.07.07 → 06.05.06)
 • WD22TB4 (01.01.00.10 → 06.05.06)
Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Review and upload report now? (Requires internet connection) [Y|n]: n
Do you want to disable this feature for future updates? [y|N]: n
Declined upload

(fwupd) crlab@crlab-Precision-7670:~/mywork/fwupd$ fwupdmgr refresh --force
WARNING: This package has not been validated, it may not work properly.
Updating lvfs
Downloading…             [                        |              ]
Successfully downloaded new metadata: Updates have been published for 12 of 22 local devices
WD22TB4 is pending activation; use /home/crlab/mywork/fwupd/venv/bin/../dist/bin/fwupdmgr activate to complete the update.

```
</details>

superm1, 2025-07-04:
Possible theory; `fu_engine_update_history_database()` doesn't run again to clear needs activation because the update state is specifically `FWUPD_UPDATE_STATE_SUCCESS`.

superm1, 2025-07-04:
My theory is the problem is the history database doesn't get updated again with the state of the device.   You can double check with a debugger as it walks through `fu_engine_update_history_database()`

CragW, 2025-07-04:
> My theory is the problem is the history database doesn't get updated again with the state of the device. You can double check with a debugger as it walks through `fu_engine_update_history_database()`

Yes, you're correct. The debugger doesn't go to `fu_engine_update_history_database()` when the device is re-connected to the system, so the history doesn't get updated. And `needs-activation` in the history persists to the device..

superm1, 2025-07-04:
OK - please take a stab at plumbing it.

dataCobra, 2025-07-31:
Hello, I've got the same problem today with my WD22TB4 on archlinux with the latest fwupd version.

Since this issue is still open, what is the correct workaround to make sure everything is correct and working?

I don't want brocken firmware or something.

dataCobra, 2025-07-31:
@bulletmark I just tried fwupdmgr reinstall and then fwupdmgr activate and this worked and the message is also gone.

This I find a bit more comfortable because I know the firmware is not broken.

bulletmark, 2025-07-31:
@dataCobra reinstalling deletes that file but, as I said, the message will come back after the next new refresh (i.e. when it downloads new metadata).

dataCobra, 2025-07-31:
@bulletmark I'm sorry but this is not true for me. I did a fwupdmgr refresh with force afterwards and there is no message coming up again.

bulletmark, 2025-07-31:
@dataCobra well I can make the message go away also but it comes back. Maybe I am not sure why/when it comes back but I suspect you will soon see it again.

usuallycwdillon, 2025-08-29:
Looks like it was 6.16.1-1.0.2

superm1, 2025-08-29:
Nope then if wasn't my commit that did it.

devurandom, 2025-10-24:
I first tried `fwupdmgr activate`, but only got "No firmware to activate".  Then I did as you suggested, @DragosMocrii, but it did not help, either.  Then I activated all devices (`sudo fwupdmgr get-devices --json | from json | get Devices | each {get DeviceId} | each {fwupdmgr activate $in}`), but also only got "No firmware to activate".  But still, `sudo fwupdmgr refresh` reports "Embedded Controller is pending activation; use fwupdmgr activate to complete the update."

rosti-il, 2025-10-26:
Previously I just deleted the `/var/lib/fwupd/pending.db` file but today after a newer firmware update of WD19S was installed I successfully tried a different solution:

1. Disconnect Type-C cable of the WD19S from the laptop
2. Long press on the power button on the WD19S
3. Power cycle
4. Connect Type-C cable of the WD19s back to the laptop

Then `fwupdmgr refresh --force` stopped printing that `WD19S is pending activation; use fwupdmgr activate to complete the update` message. No reboot of the laptop itself was needed. OS: Fedora 42

iyanmv, 2025-10-28:
This is still an issue. Yes, there are workarounds here and there, but every time the docking station gets updates I have to remember what I did last time. And in my opinion, this is a fwupd UX bug.

```console
WD22TB4 is pending activation; use fwupdmgr activate to complete the update.
$ fwupdmgr activate 
No firmware to activate
```

linuxabides, 2026-03-29:
I fixed my WD22TB4 (Not sure which of the following did it). Hopefully this helps someone.

## Purge pending DB
sudo mv /var/lib/fwupd/pending.db /tmp/
sudo systemctl restart fwupd

## Force re-install of firmware even if it is same version
DEVICE_ID=$(fwupdmgr get-devices --json | jq -r '.Devices.[] | select(.Name == "WD22TB4")' | jq -r '.DeviceId')
sudo fwupdmgr reinstall --assume-yes  --allow-reinstall --force ${DEVICE_ID}
sudo fwupdmgr activate ${DEVICE_ID}

## No pending activations or Updates needed
sudo fwupdmgr refresh --force
sudo fwupdmgr update

genodeftest, 2026-11-16:
> Previously I could `sudo rm /var/lib/fwupd/pending.db` and the activation message would go away temporarily but now it doesn't make any difference. The message remains. Unplugging the dock for N secs, etc also makes no difference.

For me to get the message away, I need to also restart fwupd. I.e., as root, execute:
```
mv /var/lib/fwupd/pending.db /tmp
systemctl restart fwupd.service
```
it seems like fwupd otherwise keeps the content of the _pending_ database in RAM.

bulletmark, 2026-11-16:
@genodeftest Yes sorry, you are right. Also need a `sudo systemctl restart fwupd.service` to make the message go away. Though it has always come back after an update so let's see next time ...