fwupd/fwupd #7912: Raspberry Pico RP2040 not detected as UF2 device

zeha, 2024-10-16:
I'm trying to add a new plugin, which will reboot a device into an rp2040 bootrom. The reboot works, but then the rp2040 is not detected. I think this might be a general problem with the uf2 plugin.

**Describe the bug**

I have a device using an rp2040. I've put it into bootrom/bootloader mode, and `fwupdtool get-devices` now shows the following.
However I would have expected it to show `UF2\FAMILY_E48BFF56` or similar.

```
├─RP2:
│     Device ID:          cf58b1152e8c843a94cffab96490478c4f177081
│     Summary:            SCSI device
│     Current version:    3
│     Vendor:             RPI (USB:0x2E8A, SCSI:RPI)
│     GUIDs:              874a17a9-dd06-5e8b-9007-09bc9be85b74 ← BLOCK\VEN_2E8A&DEV_0003
│                         9ffc390f-db75-569e-8013-2a8bef3fc891 ← SCSI\VEN_RPI     &DEV_RP2
│                         29a6d769-ef4f-5048-ab4f-aee5b5d514e1 ← SCSI\VEN_RPI     &DEV_RP2     &REV_3
│     Device Flags:       • Can tag for emulation

```

**Steps to Reproduce**

Plug an rp2040 in and run `picotool reboot -u`, then `fwupdtool get-devices`.

**Expected behavior**

Unfortunately unclear, but I'd expect to see a hwid of `UF2\FAMILY_E48BFF56` and maybe some indication of the UF2 plugin having setup the device.

**fwupd version information**
git 405f0e280e317a501e2681e5d6baa73993265b04

```shell
% fwupdtool --version
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
runtime   org.freedesktop.fwupd         2.0.2
runtime   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    6.11.2-mnt-reform-arm64
compile   com.hughsie.libjcat           0.2.0
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.2
compile   org.freedesktop.fwupd.source  2.0.1-2-g405f0e280
```

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdtool get-devices --show-all-devices

mntre MNT Pocket Reform with i.MX8MP Module
│
├─Unknown Device:
│     Device ID:          810b2559ffc547398d373f708fcf20bbfece1f1f
│     Current version:    01
│     Vendor:             Synopsys, Inc. (PCI:0x16C3)
│     GUID:               f7748de6-70da-5676-ad8e-de49cfde8935 ← PCI\VEN_16C3&DEV_ABCD
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│
├─RP2:
│     Device ID:          cf58b1152e8c843a94cffab96490478c4f177081
│     Summary:            SCSI device
│     Current version:    3
│     Vendor:             RPI (USB:0x2E8A, SCSI:RPI)
│     GUIDs:              874a17a9-dd06-5e8b-9007-09bc9be85b74 ← BLOCK\VEN_2E8A&DEV_0003
│                         9ffc390f-db75-569e-8013-2a8bef3fc891 ← SCSI\VEN_RPI     &DEV_RP2
│                         29a6d769-ef4f-5048-ab4f-aee5b5d514e1 ← SCSI\VEN_RPI     &DEV_RP2     &REV_3
│     Device Flags:       • Can tag for emulation
│
└─SSD 1TB:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    W0613A3
      Vendor:             Silicon Motion, Inc. (PCI:0x126F)
      Serial Number:      ALJK2023112100000520
      Problems:           • System power is too low to perform the update (0%, requires 10%)
      GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ← NVME\VEN_126F&DEV_2263
                          165a89d1-acc2-51c7-92b9-300207a5409c ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
                          ea9ce501-0633-5767-8264-0c3a51b59f72 ← SSD 1TB
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
                          • Can tag for emulation

```

</details>

**Additional questions**

- Operating system and version: Debian unstable, aarch64
- Have you tried rebooting? no
- Is this a regression? unsure

zeha, 2024-10-16:
Here is my branch with the new plugin:
https://github.com/fwupd/fwupd/compare/main...zeha:fwupd:zeha-mnt-reform

Maybe it's something really obvious and the get-devices output is just misleading me somewhere...

Device info: this is a system controller. In normal operation it shows up as USB dev VID=1209 PID=6d07. The plugin sends a simple reset command, after re-plug it becomes an rp2040 bootrom (VID=2e8a PID=0003).

```
├─Pocket Reform SysCtl 1.0:
│     Device ID:          3ead601f43028c79656941fc3832e6b67412e2b8
│     Current version:    0.0.0
│     Vendor:             Generic (USB:0x1209)
│     Serial Number:      DE63544193253234
│     GUID:               df0c9fc8-88d6-5100-ba66-0f5c436fc522 ← USB\VID_1209&PID_6D07
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Unsigned Payload
│                         • Can tag for emulation
```

zeha, 2024-10-16:
fwupdtool install log, showing weirdness after re-plug

```
00:48:21.630 FuDebug              verbose to info (on console 0)
00:48:21.631 FuEngine             starting fwupd 2.0.2...
Loading…: 0%
00:48:21.636 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/lvfs-testing.conf
00:48:21.637 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/lvfs.conf
00:48:21.638 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/vendor-directory.conf
00:48:21.639 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…: 1%
Loading…: 2%
00:48:21.714 FuQuirks             loading quirks from /home/ch/fwupd/venv/bin/../dist/share/fwupd/quirks.d
00:48:21.714 FuQuirks             loading quirks from /home/ch/fwupd/venv/bin/../dist/lib/fwupd/quirks.d
00:48:21.722 FuContext            failed to load smbios: SMBIOS tables not found at /sys/firmware/dmi/tables
00:48:21.755 FuContext            failed to load dmi: no /sys/class/dmi/id
00:48:21.757 FuContext            added udev subsystem watch of firmware-attributes
Loading…: 3%
00:48:21.757 FuEngine             loading metadata for remote 'vendor-directory'
00:48:21.757 FuEngine             ignoring: /home/ch/fwupd/venv/bin/../dist/share/fwupd/remotes.d/vendor/firmware/README.md
00:48:21.758 FuEngine             ignoring: Error opening directory "/home/ch/fwupd/venv/bin/../dist/lib/fwupd/local.d": No such file or directory
00:48:21.758 FuEngine             ignoring: Error opening directory "/home/ch/fwupd/venv/bin/../dist/share/fwupd/local.d": No such file or directory
00:48:21.759 FuContext            added udev subsystem watch of usb
Loading…: 4%
00:48:21.772 FuContext            added udev subsystem watch of tty
00:48:21.772 FuContext            added udev subsystem watch of usbmisc
00:48:21.772 FuContext            added udev subsystem watch of wwan
00:48:21.772 FuContext            added udev subsystem watch of drm
00:48:21.773 FuContext            added udev subsystem watch of drm_dp_aux_dev
00:48:21.773 FuContext            added udev subsystem watch of block for plugin algoltek_usbcr
00:48:21.773 FuContext            added udev subsystem watch of i2c
00:48:21.773 FuContext            added udev subsystem watch of pci
00:48:21.775 FuContext            added udev subsystem watch of i2c-dev
00:48:21.775 FuContext            added udev subsystem watch of hidraw
00:48:21.776 FuContext            added udev subsystem watch of gpio for plugin gpio
00:48:21.777 FuContext            added udev subsystem watch of mei
00:48:21.779 FuContext            added udev subsystem watch of video4linux
00:48:21.780 FuContext            added udev subsystem watch of mtd for plugin mtd
00:48:21.780 FuContext            added udev subsystem watch of nvme for plugin nvme
00:48:21.782 FuContext            added udev subsystem watch of serio
00:48:21.784 FuEngine             plugins disabled: test, test_ble
00:48:21.784 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
00:48:21.784 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
00:48:21.784 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
00:48:21.784 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
00:48:21.784 FuPluginList         cannot find plugin 'bios' referenced by 'uefi_esrt'
Loading…: 5%
00:48:21.786 FuContext            battery threshold now 10
Loading…: 0%
00:48:21.790 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
00:48:21.790 FuEngine             disabling plugin because: failed to startup using lenovo_thinklmi: unsupported manufacturer
00:48:21.799 FuEngine             disabling plugin because: failed to startup using powerd: no service that owns the name for org.chromium.PowerManager
00:48:21.799 FuEngine             disabling plugin because: failed to startup using redfish: no username and password specified, and no vendor quirk for 'ipmi-create-user'
00:48:21.800 FuEngine             disabling plugin because: failed to startup using uefi_sbat: SecureBoot is not available
00:48:21.812 FuContext            power state now battery-discharging
00:48:21.813 FuContext            battery level now 87
00:48:21.813 FuEngine             disabling plugin because: failed to startup using uefi_capsule: SMBIOS not supported
00:48:21.813 FuEngine             disabling plugin because: failed to startup using dell: firmware updating not supported: no structures with type de
00:48:21.813 FuEngine             disabling plugin because: failed to startup using uefi_recovery: kernel efivars support missing: /sys/firmware/efi/efivars
00:48:21.815 FuEngine             disabling plugin because: failed to coldplug using acpi_phat: Failed to open file "/sys/firmware/acpi/tables/PHAT": No such file or directory
Loading…: 6%
00:48:21.818 FuEngine             disabling plugin because: failed to coldplug using uefi_dbx: Error opening file /sys/firmware/efi/efivars/KEK-8be4df61-93ca-11d2-aa0d-00e098032b8c: No such file or directory
00:48:21.818 FuEngine             disabling plugin because: failed to coldplug using vbe: no image id fwupd found in firmware
Loading…: 7%
[... cut ...]
00:48:25.373 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: SG_IO INQUIRY_CMD data error: cf58b1152e8c843a94cffab96490478c4f177081 [RP2] has not been opened
00:48:32.822 FuDeviceList         no replug or re-enumerate required
failed to detach: failed to wait for detach replug: device 3ead601f43028c79656941fc3832e6b67412e2b8 did not come back
00:48:32.823 FuMain               failed to detach: failed to wait for detach replug: device 3ead601f43028c79656941fc3832e6b67412e2b8 did not come back

```

zeha, 2024-10-16:
fwupdtool get-devices output from fwupd tag 1.9.26 shows extra GUIDs `UF2\BOARD_RPI-RP2` and `UF2\FAMILY_E48BFF56` as expected:

```
├─RP2:
│     Device ID:          6b9f3d05d9a779b1dd3da6f1c8e30ed82871d466
│     Current version:    03
│     Vendor:             Raspberry Pi (BLOCK:0x2E8A, USB:0x2E8A)
│     Serial Number:      E0C9125B0D9B
│     GUIDs:              874a17a9-dd06-5e8b-9007-09bc9be85b74 ← BLOCK\VEN_2E8A&DEV_0003
│                         6dd89c00-e9c8-5add-a69e-9f00fb08f52f ← UF2\FAMILY_E48BFF56
│                         08022771-a1e2-55b9-b682-a156b1b4b40d ← USB\VID_2E8A&PID_0003
│                         abb9adbc-77dd-5c23-9ffd-b3039abccb4e ← UF2\BOARD_RPI-RP2
│     Device Flags:       • Updatable
│                         • Device will not re-appear after update completes
│                         • Cryptographic hash verification is available
│                         • Unsigned Payload
```

zeha, 2024-10-16:
I've rebased my branch on top of d089d8aeeb2b81c3f8223c48fc2f85bcc4a8ec59, and can observe two things:

1) get-devices now presents the expected GUIDs on the bootrom, but also shows it once as a SCSI device and once as USB:

```
% fwupdtool get-devices
Loading…                 [*******************                    ]14:51:54.041 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: failed to read version: command fail with status 2, senseKey 0x05, asc 0x20, ascq 0x00
Loading…                 [************************************** ]
WARNING: This package has not been validated, it may not work properly.
mntre MNT Pocket Reform with i.MX8MP Module
│
├─Unknown Device:
│     Device ID:          810b2559ffc547398d373f708fcf20bbfece1f1f
│     Current version:    01
│     Vendor:             Synopsys, Inc. (PCI:0x16C3)
│     GUID:               f7748de6-70da-5676-ad8e-de49cfde8935 ← PCI\VEN_16C3&DEV_ABCD
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│
├─RP2:
│     Device ID:          cf58b1152e8c843a94cffab96490478c4f177081
│     Summary:            SCSI device
│     Current version:    3
│     Vendor:             RPI (USB:0x2E8A, SCSI:RPI)
│     GUIDs:              874a17a9-dd06-5e8b-9007-09bc9be85b74 ← BLOCK\VEN_2E8A&DEV_0003
│                         9ffc390f-db75-569e-8013-2a8bef3fc891 ← SCSI\VEN_RPI     &DEV_RP2
│                         29a6d769-ef4f-5048-ab4f-aee5b5d514e1 ← SCSI\VEN_RPI     &DEV_RP2     &REV_3
│     Device Flags:       • Can tag for emulation
│
├─RP2:
│     Device ID:          979d82b10e5a5f49da18e40cd1d34756c4fa8b30
│     Vendor:             Raspberry Pi (USB:0x2E8A)
│     Problems:           • System power is too low to perform the update (0%, requires 10%)
│     GUIDs:              874a17a9-dd06-5e8b-9007-09bc9be85b74 ← BLOCK\VEN_2E8A&DEV_0003
│                         6dd89c00-e9c8-5add-a69e-9f00fb08f52f ← UF2\FAMILY_E48BFF56
│                         08022771-a1e2-55b9-b682-a156b1b4b40d ← USB\VID_2E8A&PID_0003
│                         abb9adbc-77dd-5c23-9ffd-b3039abccb4e ← UF2\BOARD_RPI-RP2
│     Device Flags:       • Device will not re-appear after update completes
│                         • Cryptographic hash verification is available
│                         • Updatable
│                         • Unsigned Payload
│                         • Can tag for emulation
│
└─SSD 1TB:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    W0613A3
      Vendor:             Silicon Motion, Inc. (PCI:0x126F)
      Serial Number:      ALJK2023112100000520
      Problems:           • System power is too low to perform the update (0%, requires 10%)
      GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ← NVME\VEN_126F&DEV_2263
                          165a89d1-acc2-51c7-92b9-300207a5409c ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
                          ea9ce501-0633-5767-8264-0c3a51b59f72 ← SSD 1TB
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
                          • Can tag for emulation
```

2) updating still fails:

```
...
14:48:18.403 FuDeviceList         waiting 10000ms for replug
14:48:18.417 FuEngine             auto-removing backend device Pocket Reform SysCtl 1.0 [3ead601f43028c79656941fc3832e6b67412e2b8]
14:48:18.423 FuDeviceList         waiting 10000ms for Pocket Reform SysCtl 1.0 device removal
14:48:18.423 FuEngine             resetting update motd timeout
[... 82 lines cut ...]
│     Serial Number:      ALJK2023112100000520
│     Problems:           • System power is too low to perform the update (0%, requires 10%)
│     GUIDs:              eb4c6074-9dc2-57ae-bd43-1119ad6080f5 ← NVME\VEN_126F&DEV_2263
│                         165a89d1-acc2-51c7-92b9-300207a5409c ← NVME\VEN_126F&DEV_2263&SUBSYS_126F2263
│                         ea9ce501-0633-5767-8264-0c3a51b59f72 ← SSD 1TB
│     Device Flags:       • System requires external power source
│                         • Needs a reboot after installation
│                         • Updatable
│
└─X29128:
      Device ID:          95881bc907dddb08d41ee1fc40887711ba95d9a8
      Current version:    0x8a00000000000000
      Vendor:             Kingston (EMMC:0x000070)
      Serial Number:      0x512c813e
      Problems:           • System power is too low to perform the update (0%, requires 10%)
      GUIDs:              53118fd4-627a-560e-a92f-9f8c1c299459 ← EMMC\NAME_X29128
                          ce07e22b-d8e6-5b21-8cf8-5d84ef80df21 ← EMMC\MAN_0070&OEM_0100&NAME_X29128
                          579386f5-b86e-55e8-84b6-c51c97ab46b0 ← EMMC\MAN_0070&NAME_X29128&REV_0x8a00000000000000
                          caf347d3-1d61-51b8-b6b2-09a38829be42 ← EMMC\MAN_0070&OEM_0100&NAME_X29128&REV_0x8a00000000000000
                          1c434c3f-0b1d-5a48-8cde-e50f7bd000a3 ← EMMC\0112&0256&X29128
      Device Flags:       • Internal device
                          • Updatable
```

Let me know if any other info would be useful!

hughsie, 2024-10-16:
> fu_volume_new_by_device: assertion 'device != NULL' failed

Oof, I can see that too -- one sec.

hughsie, 2024-10-16:
@zeha can you rebase again pls.

zeha, 2024-10-16:
Rebased on top of f1238aac1a97a689ffb483661ea3bd4465e91244 (+ 527224123d8d3ca6742937699ccfdf4552e52e04 from master), getting further, but not quite:

```
20:53:26.159 FuDeviceList         waiting 10000ms for replug
20:53:26.174 FuEngine             auto-removing backend device Pocket Reform System Controller 1.0 [3ead601f43028c79656941fc3832e6b67412e2b8]
20:53:26.180 FuDeviceList         waiting 10000ms for Pocket Reform System Controller 1.0 device removal
20:53:26.180 FuEngine             resetting update motd timeout
20:53:28.651 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: failed to read version: 13644190a0eb9fc2d16c0b58456fd601de07bfae [(null)] has not been opened
20:53:28.717 FuPluginScsi         found ufshci controller at /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto
20:53:28.722 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: SG_IO INQUIRY_CMD data error: cf58b1152e8c843a94cffab96490478c4f177081 [RP2] has not been opened
20:53:29.809 FuDevice             failed on try 1 of 20: is not mounted
20:53:29.947 FuDevice             failed on try 2 of 20: is not mounted
20:53:30.086 FuDevice             failed on try 3 of 20: is not mounted
20:53:30.223 FuDevice             failed on try 4 of 20: is not mounted
20:53:30.359 FuDevice             failed on try 5 of 20: is not mounted
20:53:30.495 FuDevice             failed on try 6 of 20: is not mounted
20:53:30.632 FuDevice             failed on try 7 of 20: is not mounted
20:53:30.768 FuDevice             failed on try 8 of 20: is not mounted
20:53:30.905 FuDevice             failed on try 9 of 20: is not mounted
20:53:31.042 FuDevice             failed on try 10 of 20: is not mounted
20:53:31.179 FuDevice             failed on try 11 of 20: is not mounted
20:53:31.314 FuDevice             failed on try 12 of 20: is not mounted
20:53:31.450 FuDevice             failed on try 13 of 20: is not mounted
20:53:31.587 FuDevice             failed on try 14 of 20: is not mounted
20:53:31.724 FuDevice             failed on try 15 of 20: is not mounted
20:53:31.861 FuDevice             failed on try 16 of 20: is not mounted
20:53:31.998 FuDevice             failed on try 17 of 20: is not mounted
20:53:32.137 FuDevice             failed on try 18 of 20: is not mounted
20:53:32.274 FuDevice             failed on try 19 of 20: is not mounted
20:53:32.410 FuVolume             fu_volume_new_by_device: assertion 'device != NULL' failed
/home/ch/fwupd/venv/bin/fwupdtool: line 45:  9137 Segmentation fault      ${SUDO} ${ENV} ${DEBUG} ${EXE} "$@"
```

zeha, 2024-10-16:
It ends up in `fu_uf2_device_volume_mount`, which uses `fu_udev_device_get_device_file`, and that returns NULL.

zeha, 2024-10-16:
I tried this patch, and while it doesn't crash, it also does not work:

```diff
diff --git a/plugins/uf2/fu-uf2-device.c b/plugins/uf2/fu-uf2-device.c
index 3a0ea9718..db29cb21f 100644
--- a/plugins/uf2/fu-uf2-device.c
+++ b/plugins/uf2/fu-uf2-device.c
@@ -198,7 +198,15 @@ fu_uf2_device_read_firmware(FuDevice *device, FuProgress *progress, GError **err
 static gboolean
 fu_uf2_device_volume_mount(FuUf2Device *self, GError **error)
 {
-       const gchar *devfile = fu_udev_device_get_device_file(FU_UDEV_DEVICE(self));
+       g_autofree gchar *devfile = NULL;
+       g_autofree gchar *prop_devname = NULL;
+       g_autoptr(FuVolume) volume = NULL;
+
+       /* build the mountpoint we are going to look for */
+       prop_devname = fu_udev_device_read_property(FU_UDEV_DEVICE(self), "DEVNAME", error);
+       if (prop_devname == NULL)
+               return FALSE;
+       devfile = g_strdup_printf("/dev/%s", prop_devname);

        /* mount volume if required */
        self->volume = fu_volume_new_by_device(devfile, error);

```

```
21:11:44.277 FuDeviceList         waiting 10000ms for replug
21:11:44.291 FuEngine             auto-removing backend device Pocket Reform System Controller 1.0 [3ead601f43028c79656941fc3832e6b67412e2b8]
21:11:44.296 FuDeviceList         waiting 10000ms for Pocket Reform System Controller 1.0 device removal
21:11:44.296 FuEngine             resetting update motd timeout
21:11:45.004 FuContext            battery level now 55
21:11:46.790 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: failed to read version: 13644190a0eb9fc2d16c0b58456fd601de07bfae [(null)] has not been opened
21:11:46.854 FuPluginScsi         found ufshci controller at /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto
21:11:46.858 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: SG_IO INQUIRY_CMD data error: cf58b1152e8c843a94cffab96490478c4f177081 [RP2] has not been opened
21:11:53.563 FuDevice             failed on try 1 of 20: is not mounted
21:12:16.780 FuDevice             failed on try 2 of 20: is not mounted
21:12:16.918 FuDevice             failed on try 3 of 20: is not mounted
21:12:17.057 FuDevice             failed on try 4 of 20: is not mounted
21:12:17.194 FuDevice             failed on try 5 of 20: is not mounted
21:12:17.330 FuDevice             failed on try 6 of 20: is not mounted
21:12:17.467 FuDevice             failed on try 7 of 20: is not mounted
21:12:17.604 FuDevice             failed on try 8 of 20: is not mounted
21:12:17.740 FuDevice             failed on try 9 of 20: is not mounted
21:12:17.876 FuDevice             failed on try 10 of 20: is not mounted
21:12:18.013 FuDevice             failed on try 11 of 20: is not mounted
21:12:18.151 FuDevice             failed on try 12 of 20: is not mounted
21:12:18.286 FuDevice             failed on try 13 of 20: is not mounted
21:12:18.423 FuDevice             failed on try 14 of 20: is not mounted
21:12:18.559 FuDevice             failed on try 15 of 20: is not mounted
21:12:18.695 FuDevice             failed on try 16 of 20: is not mounted
21:12:18.831 FuDevice             failed on try 17 of 20: is not mounted
21:12:18.968 FuDevice             failed on try 18 of 20: is not mounted
21:12:19.106 FuDevice             failed on try 19 of 20: is not mounted
21:12:58.466 FuDeviceList         no replug or re-enumerate required

failed to detach: failed to wait for detach replug: device 3ead601f43028c79656941fc3832e6b67412e2b8 did not come back
21:12:58.467 FuMain               failed to detach: failed to wait for detach replug: device 3ead601f43028c79656941fc3832e6b67412e2b8 did not come back
```

zeha, 2024-10-16:
@hughsie d3cb341b5c6fa5bbe3e304a06c11385b4c9127c8 makes it work. Feel free to pick / rework that. Not sure it fits your design goal.

zeha, 2024-10-16:
Update log with d3cb341 below. There is some `failed to reload` stuff in the end looking like a bug (probably my patch is not correct).

```
21:52:08.919 FuDeviceList         waiting 10000ms for replug
21:52:08.934 FuEngine             auto-removing backend device Pocket Reform System Controller 1.0 [3ead601f43028c79656941fc3832e6b67412e2b8]
21:52:08.940 FuDeviceList         waiting 10000ms for Pocket Reform System Controller 1.0 device removal
21:52:08.940 FuEngine             resetting update motd timeout
21:52:11.436 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: failed to read version: 13644190a0eb9fc2d16c0b58456fd601de07bfae [(null)] has not been opened
21:52:11.501 FuPluginScsi         found ufshci controller at /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto
21:52:11.505 FuEngine             failed to add device /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda: failed to setup: SG_IO INQUIRY_CMD data error: cf58b1152e8c843a94cffab96490478c4f177081 [RP2] has not been opened
21:53:08.196 FuDevice             failed on try 1 of 20: is not mounted
21:53:08.334 FuDevice             failed on try 2 of 20: is not mounted
21:53:08.471 FuDevice             failed on try 3 of 20: is not mounted
21:53:08.610 FuDevice             failed on try 4 of 20: is not mounted
21:53:08.747 FuDevice             failed on try 5 of 20: is not mounted
21:53:08.883 FuDevice             failed on try 6 of 20: is not mounted
21:53:09.019 FuDevice             failed on try 7 of 20: is not mounted
21:53:09.156 FuDevice             failed on try 8 of 20: is not mounted
21:53:09.292 FuDevice             failed on try 9 of 20: is not mounted
21:53:09.430 FuDevice             failed on try 10 of 20: is not mounted
21:53:09.568 FuDevice             failed on try 11 of 20: is not mounted
21:53:09.703 FuDevice             failed on try 12 of 20: is not mounted
21:53:09.840 FuDevice             failed on try 13 of 20: is not mounted
21:53:09.984 FuDevice             failed on try 14 of 20: is not mounted
21:53:10.122 FuDevice             failed on try 15 of 20: is not mounted
21:53:10.258 FuDevice             failed on try 16 of 20: is not mounted
21:53:10.394 FuDevice             failed on try 17 of 20: is not mounted
21:53:10.532 FuDevice             failed on try 18 of 20: is not mounted
21:53:10.668 FuDevice             failed on try 19 of 20: is not mounted
21:54:41.733 FuDeviceList         found compatible device 3ead601f43028c79656941fc3832e6b67412e2b8 recently removed, reusing item from plugin mntre for plugin uf2
21:54:41.733 FuDeviceList         not adding GUID df0c9fc8-88d6-5100-ba66-0f5c436fc522 to device, use FU_DEVICE_PRIVATE_FLAG_ADD_COUNTERPART_GUIDS if required
21:54:41.734 FuDeviceList         copying old version 0 to new device
21:54:41.734 FuDeviceList         ::changed 979d82b10e5a5f49da18e40cd1d34756c4fa8b30 [RP2]
21:54:41.734 FuDeviceList         3ead601f43028c79656941fc3832e6b67412e2b8 old device came back, clearing flag
21:54:41.734 FuEngine             resetting update motd timeout
21:54:42.785 FuDeviceList         waited for replug
Writing…                 [*******************                    ]21:54:42.785 FuEngine             install phase now install
21:54:42.785 FuDeviceList         no replug or re-enumerate required
21:54:42.786 FuEngine             update -> FuUf2Device:
  DeviceId:             979d82b10e5a5f49da18e40cd1d34756c4fa8b30
  Name:                 RP2
  Guid:                 874a17a9-dd06-5e8b-9007-09bc9be85b74 ← BLOCK\VEN_2E8A&DEV_0003
  Guid:                 6dd89c00-e9c8-5add-a69e-9f00fb08f52f ← UF2\FAMILY_E48BFF56
  Guid:                 08022771-a1e2-55b9-b682-a156b1b4b40d ← USB\VID_2E8A&PID_0003
  Guid:                 abb9adbc-77dd-5c23-9ffd-b3039abccb4e ← UF2\BOARD_RPI-RP2
  Plugin:               uf2
  Protocol:             com.microsoft.uf2
  Flags:                updatable|can-verify|can-verify-image|unsigned-payload|can-emulation-tag
  Vendor:               Raspberry Pi
  VendorId:             USB:0x2E8A
  VendorId:             USB:0x1209
  Version:              0
  VersionFormat:        number
  VersionRaw:           0x00000100
  Created:              2024-10-16
  UpdateError:          Device has been removed, An update is in progress
  Guid[quirk]:          b802107d-bf76-5eed-ae55-641a5028c109 ← BLOCK\VEN_2E8A
  Guid[quirk]:          e3ac56d5-08ce-55c7-b842-120023e685f7 ← USB\VID_2E8A
  PhysicalId:           DEVPATH=/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1
  BackendId:            /sys/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1/3-1:1.0/host0/target0:0:0/0:0:0:0/block/sda/sda1
  Vid:                  0x2e8a
  Pid:                  0x3
  RemoveDelay:          10000
  AcquiesceDelay:       2500
  CustomFlags:          updatable,will-disappear
  Order:                0
  PossiblePlugin:       android_boot
  PossiblePlugin:       uf2
  PrivateFlags:         replug-match-guid,registered
  Events:
[... 95 lines cut ...]
  Guid[quirk]:          e3ac56d5-08ce-55c7-b842-120023e685f7 ← USB\VID_2E8A
  PhysicalId:           DEVPATH=/devices/platform/soc@0/32f10108.usb/38200000.usb/xhci-hcd.1.auto/usb3/3-1
[... cut ...]
  Subsystem:            block
  DeviceFile:           /dev/sda1
  OpenFlags:            read,nonblock
  FsType:               vfat
  FsLabel:              RPI-RP2

21:54:51.392 FuDeviceList         no replug or re-enumerate required

failed to reload: failed to reload device: failed to subclass open: failed to open /dev/sda1: No such file or directory
21:54:51.395 FuMain               failed to reload: failed to reload device: failed to subclass open: failed to open /dev/sda1: No such file or directory

```

hughsie, 2024-10-17:
> Not sure it fits your design goal

So it's a workaround :) The root issue is fixed in https://github.com/fwupd/fwupd/pull/7922 although I've forced pushed the hughsie/subsystem_devtype branch with all the bits you should need.

zeha, 2024-10-17:
> The root issue is fixed in #7922 although I've forced pushed the hughsie/subsystem_devtype branch with all the bits you should need.

With `hughsie/subsystem_devtype` it indeed works, but it still ends with "failed to reload ..." and exit code 3. I think that means from a fwupd perspective the update was not successful?

Log:

```
16:08:17.518 FuDebug              verbose to info (on console 0)
16:08:17.519 FuEngine             starting fwupd 2.0.2...
Loading…: 0%
16:08:17.524 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/lvfs-testing.conf
16:08:17.525 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/lvfs.conf
16:08:17.526 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/vendor-directory.conf
16:08:17.527 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…: 1%
Loading…: 2%
16:08:17.602 FuQuirks             loading quirks from /home/ch/fwupd/venv/bin/../dist/share/fwupd/quirks.d
16:08:17.603 FuQuirks             loading quirks from /home/ch/fwupd/venv/bin/../dist/lib/fwupd/quirks.d
16:08:17.610 FuContext            failed to load smbios: SMBIOS tables not found at /sys/firmware/dmi/tables
16:08:17.643 FuContext            failed to load dmi: no /sys/class/dmi/id
16:08:17.645 FuContext            added udev subsystem watch of firmware-attributes
Loading…: 3%
16:08:17.645 FuEngine             loading metadata for remote 'vendor-directory'
16:08:17.645 FuEngine             ignoring: /home/ch/fwupd/venv/bin/../dist/share/fwupd/remotes.d/vendor/firmware/README.md
16:08:17.645 FuEngine             ignoring: Error opening directory "/home/ch/fwupd/venv/bin/../dist/lib/fwupd/local.d": No such file or directory
16:08:17.646 FuEngine             ignoring: Error opening directory "/home/ch/fwupd/venv/bin/../dist/share/fwupd/local.d": No such file or directory
16:08:17.647 FuContext            added udev subsystem watch of usb
Loading…: 4%
16:08:17.660 FuContext            added udev subsystem watch of tty
16:08:17.660 FuContext            added udev subsystem watch of usbmisc
16:08:17.660 FuContext            added udev subsystem watch of wwan
16:08:17.660 FuContext            added udev subsystem watch of drm
16:08:17.660 FuContext            added udev subsystem watch of drm_dp_aux_dev
16:08:17.661 FuContext            added udev subsystem watch of block
16:08:17.661 FuContext            added udev subsystem watch of block:disk for plugin algoltek_usbcr
16:08:17.661 FuContext            added udev subsystem watch of i2c
16:08:17.661 FuContext            added udev subsystem watch of pci
16:08:17.661 FuContext            added udev subsystem watch of block:partition for plugin android_boot
16:08:17.663 FuContext            added udev subsystem watch of i2c-dev
16:08:17.663 FuContext            added udev subsystem watch of hidraw
16:08:17.664 FuContext            added udev subsystem watch of gpio for plugin gpio
16:08:17.665 FuContext            added udev subsystem watch of mei
16:08:17.667 FuContext            added udev subsystem watch of video4linux
16:08:17.667 FuContext            added udev subsystem watch of mtd for plugin mtd
16:08:17.668 FuContext            added udev subsystem watch of nvme for plugin nvme
16:08:17.669 FuContext            added udev subsystem watch of serio
16:08:17.671 FuEngine             plugins disabled: test, test_ble
16:08:17.672 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
16:08:17.672 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
16:08:17.672 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
16:08:17.672 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
16:08:17.672 FuPluginList         cannot find plugin 'bios' referenced by 'uefi_esrt'
Loading…: 5%
16:08:17.674 FuContext            battery threshold now 10
Loading…: 0%
16:08:17.677 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
16:08:17.678 FuEngine             disabling plugin because: failed to startup using lenovo_thinklmi: unsupported manufacturer
16:08:17.687 FuEngine             disabling plugin because: failed to startup using powerd: no service that owns the name for org.chromium.PowerManager
16:08:17.687 FuEngine             disabling plugin because: failed to startup using redfish: no username and password specified, and no vendor quirk for 'ipmi-create-user'
16:08:17.688 FuEngine             disabling plugin because: failed to startup using uefi_sbat: SecureBoot is not available
16:08:17.700 FuContext            power state now ac-charging
16:08:17.701 FuContext            battery level now 100
16:08:17.701 FuEngine             disabling plugin because: failed to startup using uefi_capsule: SMBIOS not supported
16:08:17.701 FuEngine             disabling plugin because: failed to startup using dell: firmware updating not supported: no structures with type de
16:08:17.701 FuEngine             disabling plugin because: failed to startup using uefi_recovery: kernel efivars support missing: /sys/firmware/efi/efivars
16:08:17.703 FuEngine             disabling plugin because: failed to coldplug using acpi_phat: Failed to open file "/sys/firmware/acpi/tables/PHAT": No such file or directory
Loading…: 6%
[... cut ...]
  Subsystem:            block
  DeviceFile:           /dev/sda1
  OpenFlags:            read,nonblock
  FsType:               vfat
  FsLabel:              RPI-RP2

16:08:33.362 FuDeviceList         no replug or re-enumerate required
failed to reload: failed to reload device: failed to subclass open: failed to open /dev/sda1: No such file or directory
16:08:33.363 FuMain               failed to reload: failed to reload device: failed to subclass open: failed to open /dev/sda1: No such file or directory

```

hughsie, 2024-10-17:
@zeha I guess that's closer! The wrote looks successful, and it looks like we're attaching the device back into runtime mode. It says "no replug or re-enumerate required" -- are you setting WAIT_FOR_REPLUG in attach? If you can push your branch again and point me at your firmware .cab I can give it a go myself. I'll give you ACLs too, so you can push to origin -- it'll be easier to work together that way. Just don't push to main directly :)

zeha, 2024-10-17:
#7926 fixes the "no replug or re-enumerate required" for me. I think the change should be correct for all uf2 devices.

New log below, looking better IMO. Attach is now handled by FuMntreDevice, and not FuUf2Device. There's some problem with the version numbers, which I'll investigate tomorrow.

Firmware cab I'm using for testing: https://per.namespace.at/~ch/sysctl.cab

Branch: https://github.com/fwupd/fwupd/tree/zeha/mntre

```
22:02:48.580 FuDebug              verbose to info (on console 0)
22:02:48.582 FuEngine             starting fwupd 2.0.2...
Loading…: 0%
22:02:48.587 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/lvfs-testing.conf
22:02:48.588 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/lvfs.conf
22:02:48.589 FuRemoteList         loading remote from /home/ch/fwupd/venv/bin/../dist/etc/fwupd/remotes.d/vendor-directory.conf
22:02:48.590 FuRemoteList         enabled remotes: vendor-directory[1000], lvfs[0]
Loading…: 1%
Loading…: 2%
22:02:48.667 FuQuirks             loading quirks from /home/ch/fwupd/venv/bin/../dist/share/fwupd/quirks.d
22:02:48.668 FuQuirks             loading quirks from /home/ch/fwupd/venv/bin/../dist/lib/fwupd/quirks.d
22:02:48.675 FuContext            failed to load smbios: SMBIOS tables not found at /sys/firmware/dmi/tables
22:02:48.709 FuContext            failed to load dmi: no /sys/class/dmi/id
22:02:48.710 FuContext            added udev subsystem watch of firmware-attributes
Loading…: 3%
22:02:48.711 FuEngine             loading metadata for remote 'vendor-directory'
22:02:48.711 FuEngine             ignoring: /home/ch/fwupd/venv/bin/../dist/share/fwupd/remotes.d/vendor/firmware/README.md
22:02:48.711 FuEngine             ignoring: Error opening directory "/home/ch/fwupd/venv/bin/../dist/lib/fwupd/local.d": No such file or directory
22:02:48.712 FuEngine             ignoring: Error opening directory "/home/ch/fwupd/venv/bin/../dist/share/fwupd/local.d": No such file or directory
22:02:48.713 FuContext            added udev subsystem watch of usb
Loading…: 4%
22:02:48.726 FuContext            added udev subsystem watch of tty
22:02:48.726 FuContext            added udev subsystem watch of usbmisc
22:02:48.726 FuContext            added udev subsystem watch of wwan
22:02:48.726 FuContext            added udev subsystem watch of drm
22:02:48.727 FuContext            added udev subsystem watch of drm_dp_aux_dev
22:02:48.727 FuContext            added udev subsystem watch of block
22:02:48.727 FuContext            added udev subsystem watch of block:disk for plugin algoltek_usbcr
22:02:48.727 FuContext            added udev subsystem watch of i2c
22:02:48.727 FuContext            added udev subsystem watch of pci
22:02:48.728 FuContext            added udev subsystem watch of block:partition for plugin android_boot
22:02:48.729 FuContext            added udev subsystem watch of i2c-dev
22:02:48.730 FuContext            added udev subsystem watch of hidraw
22:02:48.731 FuContext            added udev subsystem watch of gpio for plugin gpio
22:02:48.731 FuContext            added udev subsystem watch of mei
22:02:48.734 FuContext            added udev subsystem watch of video4linux
22:02:48.734 FuContext            added udev subsystem watch of mtd for plugin mtd
22:02:48.734 FuContext            added udev subsystem watch of nvme for plugin nvme
22:02:48.736 FuContext            added udev subsystem watch of serio
22:02:48.738 FuEngine             plugins disabled: test, test_ble
22:02:48.738 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
22:02:48.739 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
22:02:48.739 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
22:02:48.739 FuPluginList         cannot find plugin 'thunderbolt' requested by 'intel_usb4'
22:02:48.739 FuPluginList         cannot find plugin 'bios' referenced by 'uefi_esrt'
Loading…: 5%
22:02:48.741 FuContext            battery threshold now 10
Loading…: 0%
22:02:48.744 FuEngine             disabling plugin because: failed to startup using flashrom: no HwIDs found
22:02:48.745 FuEngine             disabling plugin because: failed to startup using lenovo_thinklmi: unsupported manufacturer
22:02:48.754 FuEngine             disabling plugin because: failed to startup using powerd: no service that owns the name for org.chromium.PowerManager
22:02:48.754 FuEngine             disabling plugin because: failed to startup using redfish: no username and password specified, and no vendor quirk for 'ipmi-create-user'
22:02:48.755 FuEngine             disabling plugin because: failed to startup using uefi_sbat: SecureBoot is not available
22:02:49.471 FuContext            power state now ac-charging
22:02:49.471 FuContext            battery level now 100
22:02:49.471 FuEngine             disabling plugin because: failed to startup using uefi_capsule: SMBIOS not supported
22:02:49.472 FuEngine             disabling plugin because: failed to startup using dell: firmware updating not supported: no structures with type de
22:02:49.472 FuEngine             disabling plugin because: failed to startup using uefi_recovery: kernel efivars support missing: /sys/firmware/efi/efivars
[... cut ...]
    FuUsbConfigDescriptor:
        ConfigurationValue: 0x1

22:03:05.708 FuDeviceList         no replug or re-enumerate required
Waiting…: 100%
22:03:05.708 FuEngine             Updating Pocket Reform System Controller 1.0 took 15.022604 seconds
22:03:08.220 FuEngine             system acquiesced after 2500ms
22:03:08.220 FuEngine             install phase now composite-cleanup
22:03:08.220 FuDeviceList         no replug or re-enumerate required
device version not updated on success, 1017 != 1
```

closed 2024-10-19 by hughsie