fwupd/fwupd #7918: missing firmware-updates since upgrading to 1.9.26

slartibart70, 2024-10-17:
On two systems 
- lenovo t460s
- lenovo p14s-gen4

i was lazy updating the firmware, although for both machines updates were available. (namely uefi-dbx certificate updates, update of uefi-bios, not other devices)

After update to 1.9.26, those firmware updates are gone. A 'refresh --force' only indicates 
`Successfully downloaded new metadata: 0 local devices supported`

A downgrade to 1.9.25 does not bring 'em back, though.

Smells like a bug to me, either in the software or in the database?
Any hints on this?

```
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
compile   org.freedesktop.Passim        0.1.7
compile   com.hughsie.libjcat           0.2.1
compile   org.freedesktop.fwupd         1.9.26
runtime   org.freedesktop.fwupd-efi     1.6
compile   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libxmlb           0.3.20
runtime   org.freedesktop.Passim        0.1.7
runtime   org.freedesktop.gusb          0.4.9
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.11.3-200.fc40.x86_64
runtime   org.freedesktop.fwupd         1.9.26
```

``` 
Operating System: Fedora Linux 40
KDE Plasma Version: 6.2.1
KDE Frameworks Version: 6.7.0
Qt Version: 6.7.2
Kernel Version: 6.11.3-200.fc40.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics
Memory: 58,5 GiB of RAM
Graphics Processor: AMD Radeon 780M
Manufacturer: LENOVO
Product Name: 21K5000JGE
System Version: ThinkPad P14s Gen 4
```

superm1, 2024-10-17:
Try upgrading to libxmlb 0.3.21 please.

slartibart70, 2024-10-18:
the installed libs are on 3.21 already
```
libxmlb-0.3.21-1.fc40.x86_64
```

slartibart70, 2024-10-18:
but, in the version info of fwupd there is obviously
```
fwupdmgr --version
compile   com.hughsie.libxmlb           0.3.19
```

Sounds like the culprit is there?

slartibart70, 2024-10-24:
some os-updates later, this seems resolved?
At least the 'missing' firmware updates are shown to me...closing this ticket

closed 2024-10-24 by slartibart70