fwupd/fwupd #7929: fwupd daemon prevents start of X

Polynomial-C, 2024-10-18:
**Describe the bug**
After upgrading fwupd from 1.9.26 to 2.0.1, havig the fwupd daemon running while trying to start X results in X crashing immediately. The only kind of "error message" I can see in the `Xorg.0.log` file is this:

```log
[    38.562] (II) AMDGPU: Driver for AMD Radeon:
        All GPUs supported by the amdgpu kernel driver
[    38.563] (EE) AMDGPU(0): [drm] device is not DRM master.
[    38.563] (EE) AMDGPU(G0): [drm] device is not DRM master.
[    38.565] (II) AMDGPU(1): [KMS] Kernel modesetting enabled.
[    38.567] (EE) AMDGPU(1): [drm] device is not DRM master.
[    38.567] (II) AMDGPU(2): [KMS] Kernel modesetting enabled.
[    38.568] (EE) AMDGPU(2): [drm] device is not DRM master.
[    38.568] (EE) AMDGPU(G1): [drm] device is not DRM master.
[    38.568] (EE) AMDGPU(G2): [drm] device is not DRM master.
[    38.568] (EE) Screen 0 deleted because of no matching config section.
[    38.568] (II) UnloadModule: "amdgpu"
[    38.568] (EE) Screen 0 deleted because of no matching config section.
[    38.568] (II) UnloadModule: "amdgpu"
[    38.568] (EE) Screen 0 deleted because of no matching config section.
[    38.568] (II) UnloadModule: "amdgpu"
[    38.568] (EE) Device(s) detected, but none match those in the config file.
[    38.568] (EE)
Fatal server error:
[    38.568] (EE) no screens found(EE)
[    38.568] (EE)
```

**Steps to Reproduce**
* Upgrade to `fwupd-2.0.1` (didn't check 2.0.0 version)
* run `/usr/libexec/fwupd/fwupd`
* Try to (re)start your display manager and find X not starting up

**Expected behavior**
No messing with X 

**fwupd version information**
Please provide the version of the daemon and client.

```shell
# fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.1
runtime   org.freedesktop.fwupd-efi     1.7
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.6.57
runtime   org.freedesktop.fwupd         2.0.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

source

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
# fwupdmgr get-devices --show-all-devices
Micro-Star International Co., Ltd. Bravo 17 A4DDR
│
├─AMD Radeon RX 5500M:
│     Device ID:          594baa45f6742120d71dd82166e762f1529c2a8a
│     Summary:            MSI 17FK NAVI14-XTM GDDR6 4GB RNHG
│     Current version:    017.002.000.000.056561
│     Vendor:             PCI:0x1002
│     GUID:               d2179188-7922-5c23-ab9e-6e2c2b40e301 ← AMD\SWBRT56561
│     Device Flags:       • Cryptographic hash verification is available
[... 125 lines cut ...]
│       GUIDs:            d07ff664-b0e1-5f4e-a723-d7fbcbfcb94f ← UEFI\CRT_3CD3F0309EDAE228767A976DD40D9F4AFFC4FBD5218F2E8CC3C9DD97E8AC6F9D&ARCH_X64
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            MSI NB - 2013 PK
      GUID:               7fb4c062-4d9d-5e50-908e-822c6c661a2c ← UEFI\CRT_3D4AA9598C139AA53872F37F67FEF1084262EDBE
```

</details>

**Additional questions**

- Operating system and version:
  - Gentoo Linux
- Have you tried rebooting?
  - yes
- Is this a regression?
  - Compared to 1.9.x versions, yes.

hughsie, 2024-10-18:
I don't think anyone has tested with Xorg in a very long time. I suspect you want to compile fwupd with `-Dlibdrm=disabled` if you still want to use the X server without debugging things yourself.

superm1, 2024-10-18:
Yeah I'm really surprised anyone is using X still in 2024. We're going to need your debugging help to understand this.

I'm guessing what's happening is the daemon starts before X does for some reason on your system and claims DRM master.

I think we need to understand whether it's the daemon causing this with it's integrated drm code or it's the amdgpu plugin with it's integrated libdrm_amdgpu code. I think it's "more" likely the daemon code is doing it.

Both are new to 2.0.0 though.

Here's the 4 things I think make most sense for you to try to understand this:

1. Starting daemon after X starts, no fwupd changes.
2. Compile with libdrm off plugin amdgpu on. Start fwupd before X.
3. Compile with libdrm on plugin amdgpu off. Start fwupd before X.
4. Compile with libdrm off plugin amdgpu off. Start fwupd before X.

pschyska, 2024-12-02:
@hughsie You closed this issue, but the problem still persists for me.

```
$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.2
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.1
runtime   org.freedesktop.fwupd         2.0.2
```

As long as fwupd is running, Xorg can't start. When stopping fwupd from console, I can start X. Then I can start fwupd again without further apparent issues.
Re-building with `-Dlibdrm=disabled` didn't help.

Xorg log:
```
[   495.659]
X.Org X Server 1.21.1.14
X Protocol Version 11, Revision 0
[   495.659] Current Operating System: Linux leonardo 6.12.1 #1-NixOS SMP PREEMPT_DYNAMIC Fri Nov 22 14:30:26 UTC 2024 x86_64
[   495.659] Kernel command line: initrd=\EFI\nixos\b5sdv61vgrv1dm02vmx4ciw00ivcas2d-initrd-linux-6.12.1-initrd.efi init=/nix/store/k2yfqq925nsw7s6x6fw36n7d051y1sj7-nixos-system-leonardo-genesis/init console=tty1 amd_pstate=active loglevel=4 vt.default_red=0x2b,0xbf,0xa3,0xeb,0x8f,0xb4,0x96,0xc0,0x65,0xbf,0xa3,0xeb,0x8f,0xb4,0x96,0xef vt.default_grn=0x30,0x61,0xbe,0xcb,0xa1,0x8e,0xb5,0xc5,0x73,0x61,0xbe,0xcb,0xa1,0x8e,0xb5,0xf1 vt.default_blu=0x3b,0x6a,0x8c,0x8b,0xb3,0xad,0xb4,0xce,0x7e,0x6a,0x8c,0x8b,0xb3,0xad,0xb4,0xf5
[   495.659]
[   495.659] Current version of pixman: 0.43.4
[   495.659] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   495.659] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   495.659] (==) Log file: "/home/p/.local/share/xorg/Xorg.0.log", Time: Mon Dec  2 16:34:55 2024
[   495.659] (++) Using config file: "/nix/store/j338kh35sjzds4za6251dwa2nqrm5kf4-xserver.conf"
[   495.659] (==) Using config directory: "/etc/X11/xorg.conf.d"
[   495.659] (==) Using system config directory "/nix/store/cwa9ihrzv9zcg6rci26g5kxj7lgm3110-xorg-server-21.1.14/share/X11/xorg.conf.d"
[   495.660] (==) ServerLayout "Layout[all]"
[   495.660] (**) |-->Screen "Screen-amdgpu[0]" (0)
[   495.660] (**) |   |-->Monitor "Monitor[0]"
[   495.660] (**) |   |-->Device "Device-amdgpu[0]"
[   495.660] (**) Option "DontZap" "off"
[   495.660] (**) Option "AllowMouseOpenFail" "on"
[   495.660] (**) Option "BlankTime" "0"
[   495.660] (**) Allowing byte-swapped clients
[   495.660] (==) Automatically adding devices
[   495.660] (==) Automatically enabling devices
[   495.660] (==) Automatically adding GPU devices
[   495.660] (==) Automatically binding GPU devices
[   495.660] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   495.660] (**) FontPath set to:
	/nix/store/vr6ycm45cppkjp6an07y1r79fpjxipws-font-cursor-misc-1.0.4/lib/X11/fonts/misc,
	/nix/store/bw6mdh23wckwd78sjxsiyl5vlimjs63k-font-misc-misc-1.1.3/lib/X11/fonts/misc,
	/nix/store/xn9d7w44krbnx9gkk5kvb4lidzwzf6cn-unifont-16.0.01/share/fonts,
	/nix/store/87gk2jxsd5l6lcl7n6bk66db3a9hrggf-font-adobe-100dpi-1.0.4/lib/X11/fonts/100dpi,
	/nix/store/yc1by5p1acgs9xwyqpxmljxf7g29chm1-font-adobe-75dpi-1.0.4/lib/X11/fonts/75dpi,
	/nix/store/0q5r0r9z7rakhhawkchixr5vnkh57p4w-X11-fonts/share/X11/fonts
[   495.660] (**) ModulePath set to "/nix/store/4wf8749br6ygnvh3gyz4nv601k0xzk7c-xf86-video-amdgpu-23.0.0/lib/xorg/modules,/nix/store/cwa9ihrzv9zcg6rci26g5kxj7lgm3110-xorg-server-21.1.14/lib/xorg/modules,/nix/store/f3bl0b1d9p5s801nzfd841lz4vvs8d6c-xf86-input-evdev-2.11.0/lib/xorg/modules,/nix/store/q64rw2c0sm2gr69s9f0xmrip5s8cyp75-xf86-input-libinput-1.5.0/lib/xorg/modules"
[   495.660] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   495.660] (II) Loader magic: 0x64ee40
[   495.660] (II) Module ABI versions:
[   495.660] 	X.Org ANSI C Emulation: 0.4
[   495.660] 	X.Org Video Driver: 25.2
[   495.660] 	X.Org XInput driver : 24.4
[   495.660] 	X.Org Server Extension : 10.0
[   495.660] (++) using VT number 2

[   495.663] (II) systemd-logind: took control of session /org/freedesktop/login1/session/_39
[   495.664] (II) xfree86: Adding drm device (/dev/dri/card1)
[   495.664] (II) Platform probe for /sys/devices/pci0000:00/0000:00:08.1/0000:05:00.0/drm/card1
[... 13 lines cut ...]
[   495.685] 	Module class: X.Org Video Driver
[   495.685] 	ABI class: X.Org Video Driver, version 25.2
[   495.685] (II) AMDGPU: Driver for AMD Radeon:
	All GPUs supported by the amdgpu kernel driver
[   495.686] (EE) AMDGPU(0): [drm] device is not DRM master.
[   495.686] (II) AMDGPU(1): [KMS] Kernel modesetting enabled.
[   495.686] (EE) AMDGPU(1): [drm] device is not DRM master.
[   495.686] (EE) AMDGPU(G0): [drm] device is not DRM master.
[   495.686] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[   495.686] (EE) Screen 0 deleted because of no matching config section.
[   495.686] (II) UnloadModule: "amdgpu"
[   495.686] (EE) Screen 0 deleted because of no matching config section.
[   495.686] (II) UnloadModule: "amdgpu"
[   495.686] (EE) Device(s) detected, but none match those in the config file.
[   495.686] (EE)
Fatal server error:
[   495.686] (EE) no screens found(EE)
[   495.686] (EE)
Please consult the The X.Org Foundation support
	 at http://wiki.x.org
 for help.
[   495.686] (EE) Please also check the log file at "/home/p/.local/share/xorg/Xorg.0.log" for additional information.
[   495.686] (EE)
[   495.701] (EE) Server terminated with error (1). Closing log file.
```

pschyska, 2024-12-02:
Does *not* occur with

```
[fwupd]
# …
DisabledPlugins=amd_gpu
# …
```
(and no `-Dlibdrm=disabled` either)

So it might be the amd_gpu plugin.

NixOS: `services.fwupd.daemonSettings.DisabledPlugins = [ "amd_gpu" ];`

superm1, 2024-12-02:
> @hughsie You closed this issue, but the problem still persists for me.

It was closed because no one came to the table to help debugging it.  See the [above comments](https://github.com/fwupd/fwupd/issues/7929#issuecomment-2422227221).  I'll re-open it.

I think what we need to do is query whether we're master during probe and drop master at the end of probe.
https://dri.freedesktop.org/docs/drm/gpu/drm-uapi.html#primary-nodes-drm-master-and-authentication

@pschyska could you work on a PR perhaps to try this?

pschyska, 2024-12-02:
> > @hughsie You closed this issue, but the problem still persists for me.
> 
> It was closed because no one came to the table to help debugging it. See the [above comments](https://github.com/fwupd/fwupd/issues/7929#issuecomment-2422227221). I'll re-open it.

Fair, I thought there might be some activity elsewhere that wasn't linked into here, so I asked

> 
> I think what we need to do is query whether we're master during setup() callback and drop master at the end of setup() callback. https://dri.freedesktop.org/docs/drm/gpu/drm-uapi.html#primary-nodes-drm-master-and-authentication
> 
> @pschyska could you work on a PR perhaps to try this?

C is not a language I commonly use, so probably not. I barely read it 🙂 .
I'm happy to try experimental patches if someone else could step up.

superm1, 2024-12-02:
I looked at the helpers and maybe you can try and see if this works: https://github.com/fwupd/fwupd/pull/8190

closed 2024-12-02 by superm1

pschyska, 2024-12-05:
@superm1 Thanks! With your patch, I don't need `DisabledPlugins=amd_gpu` any longer — Xorg can now start again even if fwupd is currently running

superm1, 2024-12-05:
> @superm1 Thanks! With your patch, I don't need `DisabledPlugins=amd_gpu` any longer — Xorg can now start again even if fwupd is currently running

Thanks, glad to hear.