fwupd/fwupd #7934: fwupd 2.0.1: device ID was not unique

Haewelmann, 2024-10-18:
**Describe the bug**
I have a Lenovo L14 Gen-5 notebook and a connected Universal Thunderbolt 4 Docking Station. 
The system has no internet connection. A local path for the update files is configured in the 
vendor-directory.conf file. There is a local update for the docking station available (10.16 -> 10.18). 

When I call the GetUpgrades method with fwupd 2.0.1 via dbus, I get an error that the device-id 
is not unique. 

busctl call org.freedesktop.fwupd / org.freedesktop.fwupd GetUpgrades s 481f4eef35467530fca9b1f64d81d7b14973afea

...
‣ Type=error  Endian=l  Flags=1  Version=1 Cookie=1162  ReplyCookie=2  Timestamp="Fri 2024-10-18 16:13:09.109374 UTC"
  Sender=:1.11  Destination=:1.25
  ErrorName=org.freedesktop.fwupd.NotSupported  ErrorMessage="device ID 481f4eef35467530fca9b1f64d81d7b14973afea was not unique"
  UniqueName=:1.11
  MESSAGE "s" {
          STRING "device ID 481f4eef35467530fca9b1f64d81d7b14973afea was not unique";
  };
...

But the id is only available once. 

With fwupd version 1.9.26 I get this result:

...
‣ .Type=error  Endian=l  Flags=1  Version=1 Cookie=1008  ReplyCookie=2  Timestamp="Fri 2024-10-18 18:11:52.008323 UTC"
  Sender=:1.5  Destination=:1.15
  ErrorName=org.freedesktop.fwupd.NothingToDo  ErrorMessage="Device is not updatable"
  UniqueName=:1.5
  MESSAGE "s" {
          STRING "Device is not updatable";
  };
...

**Steps to Reproduce**
busctl call org.freedesktop.fwupd / org.freedesktop.fwupd GetUpgrades s 481f4eef35467530fca9b1f64d81d7b14973afea

**Expected behavior**
ErrorName=org.freedesktop.fwupd.NothingToDo
  MESSAGE "s" {
          STRING "Device is not updatable";
  };

**fwupd version information**
compile   info.libusb                   1.0.26
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.1
runtime   org.freedesktop.fwupd-efi     1.6
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd.source  2.0.1-1-g3aac369a6
runtime   org.kernel                    6.8.12-grsec+
runtime   org.freedesktop.fwupd         2.0.1

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source

**fwupd device information**
LENOVO 21L2S06800
│
├─ELAN06D8:00 04F3:3195:
│     Device ID:          65dd55ecb739ecf3d71334790e364396063db673
│     Summary:            Touchpad
│     Current version:    0x0005
│     Bootloader Version: 0x0002
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              4e8cda57-37ba-5dc6-b114-7610e38714e2 ← HIDRAW\VEN_04F3&DEV_3195
│                         cc456594-6f7e-5d1b-82a6-b4558a813152 ← HIDRAW\VEN_04F3&DEV_3195&MOD_0006
│                         d15e804c-570f-511b-803a-b38a4eddae29 ← ELANTP\ICTYPE_14&MOD_0006
│                         296e64f7-9e08-5717-bff2-651e41b0477e ← ELANTP\ICTYPE_14&MOD_0006&DRIVER_HID
[... 447 lines cut ...]
│     Current version:    23.81
│     Vendor:             Intel (THUNDERBOLT:0x8087)
│     GUIDs:              74fb1b87-53e6-5115-bd02-a9fefd06ed9a ← THUNDERBOLT\VEN_8087&DEV_0D9C
│                         dac6f95d-7c5b-52b5-8c79-a0436a5ef784 ← TBT-80870d9c-retimer0-0:1.1
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Device stages updates
│                         • Unsigned Payload
│                         • Can tag for emulation
│   
└─USB4 host controller:
      Device ID:          7bfee19a4024946c3e7f5ecce5a77d2a490fd806
      Summary:            Unmatched performance for high-speed I/O
      Current version:    41.81
      Vendor:             Lenovo (THUNDERBOLT:0x0108, TBT:0x0108)
      Problems:           [31m[1m• Device is lower priority than an equivalent device[0m
      GUIDs:              dc3a1b4e-176f-589a-949c-d058324bb6a7 ← THUNDERBOLT\VEN_0108&DEV_2031
                          350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
                          3e36d06f-3e44-5eb8-87c7-de0b723b1595 ← TBT-01082031-controller0-1
      Device Flags:       • Internal device
                          • System requires external power source
                          • Device stages updates
                          • Updatable
                          • Signed Payload
                          • Can tag for emulation

superm1, 2024-10-19:
It is a bit weird to be calling this on a device that isn't updatable, but yeah this does look like a bug very likely.
Can you please share the output for `sudo fwupdtool get-devices -vv`?

Haewelmann, 2024-10-21:
[fwupd-get-devices-vv-1.9.26.log](https://github.com/user-attachments/files/17457824/fwupd-get-devices-vv-1.9.26.log)
[fwupd-get-devices-vv-2.0.1.log](https://github.com/user-attachments/files/17457825/fwupd-get-devices-vv-2.0.1.log)

Haewelmann, 2024-10-21:
Thank you for your awnser.
Here is the output of fwupdtool get-devices -vv

[fwupdtool-get-devices-vv-2.0.1.log](https://github.com/user-attachments/files/17458647/fwupdtool-get-devices-vv-2.0.1.log)

About the background: I use the dbus-interface, the 3 functions are called: GetDevices (out: id) -> GetUpgrades (in: id) -> Install

hughsie, 2024-10-21:
Ahh, so that device isn't actually updatable:

	FuUsiDockChildDevice:
	  DeviceId:             481f4eef35467530fca9b1f64d81d7b14973afea
	  Plugin:               usi_dock
	  Flags:                none

You need to update via `7073f7228b6ee3d493dfaa222ede7b68832d676e`, i.e. `FuUsiDockMcuDevice`.

I think the `device ID 481f4eef35467530fca9b1f64d81d7b14973afea was not unique` error message is misleading. I can try to find that bug when I'm back at my desk tomorrow, unless someone beats me.

hughsie, 2024-10-22:
So I debugged this a bit just now. The multiple messages thing is actually correct, the two devices are:

	14:35:04.221 FuDeviceList         adding FuThunderboltController:
	  DeviceId:             2294959ce69ba71b10dada894868479e10d0b2eb
	  Name:                 USB4 host controller
	  Guid:                 dc3a1b4e-176f-589a-949c-d058324bb6a7 ← THUNDERBOLT\VEN_0108&DEV_2031
	  Guid:                 350ec5bb-25c0-543e-8d4c-72c30d2b8df0 ← TBT-01082031
	  Guid:                 3e36d06f-3e44-5eb8-87c7-de0b723b1595 ← TBT-01082031-controller0-1
	  Summary:              Unmatched performance for high-speed I/O
	  Plugin:               thunderbolt
	  Protocol:             com.intel.thunderbolt
	  Flags:                internal|require-ac|dual-image|updatable-hidden|signed-payload|can-emulation-tag
	  Problems:             lower-priority
	  Vendor:               Lenovo
	  VendorId:             THUNDERBOLT:0x0108
	  VendorId:             TBT:0x0108
	  Version:              41.81
	  Icon:                 thunderbolt
	  Created:              2024-10-22
	  UpdateError:          Device is lower priority than an equivalent device
	  Guid[quirk]:          847dea9d-1c72-5892-bb93-708bb4f2940b ← THUNDERBOLT\VEN_0108
	  Guid[quirk]:          f70b02b8-cac4-5138-aef2-83a4b2d3afef ← THUNDERBOLT\TYPE_THUNDERBOLT_DEVICE
	  EquivalentId:         f2d46aa175ab9e23c81b4a0108332b093956f8bf
	  PhysicalId:           00486068-1132-8780-ffff-ffffffffffff
	  BackendId:            /sys/devices/pci0000:00/0000:00:1c.0/0000:04:00.0/0000:05:00.0/0000:06:00.0/domain0/0-0/0-1
	  Vid:                  0x108
	  Pid:                  0x2031
	  AcquiesceDelay:       2500
	  GType:                FuThunderboltController
	  Order:                0
	  PossiblePlugin:       thunderbolt
	  PrivateFlags:         replug-match-guid,registered
	  Inhibit:              [lower-priority] Device is lower priority than an equivalent device
	  Events:
	  Number:               0x1
	  Subsystem:            thunderbolt
	  OpenFlags:            none
	  AuthMethod:           nvm_authenticate
	  DeviceType:           USB4 host controller
	  SafeMode:             false
	  NativeMode:           false
	  Generation:           4

	14:35:04.222 FuDeviceList         adding FuUsiDockChildDevice:
	  DeviceId:             f2d46aa175ab9e23c81b4a0108332b093956f8bf
	  CompositeId:          e1c158392fda6698c66decf536d42c5b775ba0d6
	  Name:                 Thunderbolt 4 Controller
	  Guid:                 d6408693-09a5-5f0f-b492-275b5ed16d10 ← USB\VID_17EF&PID_30B4&CID_TBT4
	  Plugin:               usi_dock
	  Flags:                none
	  Vendor:               Lenovo
	  VendorId:             USB:0x17EF
	  Version:              41.81.06
	  VersionFormat:        triplet
	  Icon:                 thunderbolt
	  Created:              2024-10-22
	  EquivalentId:         2294959ce69ba71b10dada894868479e10d0b2eb
	  PhysicalId:           5-1.1
	  LogicalId:            TBT4
	  BackendId:            /sys/devices/pci0000:00/0000:00:1c.0/0000:04:00.0/0000:05:01.0/0000:07:00.0/0000:08:00.0/0000:09:00.0/usb5/5-1/5-1.1
	  RemoveDelay:          10000
	  AcquiesceDelay:       2500
	  Order:                -1
	  Priority:             1
	  PrivateFlags:         use-parent-for-open,registered
	  ChipIdx:              tbt4

I'm not sure what the proper fix is here; they're both the same device. The entire idea of `EquivalentId` was to signify "_the same device_" although in this case it's not terribly helpful.

superm1, 2024-10-22:
> I'm not sure what the proper fix is here; they're both the same device. 

How about the daemon skips inhibited devices when processing `GetUpgrades`?

closed 2024-10-28 by superm1

Haewelmann, 2024-11-04:
Thank you very much, it worked for me. 

Side note:
I have seen that the path to the cab file is only saved in ‘Locations’. ‘Uri’ no longer seems to exist. But that's okay so far.