fwupd/fwupd #7935: Unexpected "Me FW Downgrade - Request MeSpiLock Failed"

Lj907, 2024-10-19:
**Describe the question**
I've had an odd issue, twice, with a 3rd-gen Lenovo Thinkpad X1 Carbon unning UEFI BIOS version N14ET25W (1.03) from 2014-12-01 and ME Firmware Version 10.0.29.1000.  OS is Ubuntu 22.04.

On two separate dates, a few months apart, I've had errors on boot - very similar to the "Me FW Downgrade - Request MeSpiLock Failed" error described here: https://github.com/fwupd/firmware-lenovo/issues/199 - though it wasn't displayed long enough for me to get a photo of it and capture all the text.

The first day it happened, it occurred a few times in a row - I had to power off/on the laptop a few times before it would boot normally.

But I've never made any attempt to update my firmware.

When I run this command, it shows no attempts to update my firmware by fwupd:

```
$ fwupdmgr get-history
WARNING: UEFI capsule updates not available or enabled in firmware setup
See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
No history
```

Running fwupdmgr get-devices just shows my SSD.  I have not been able to spot anything unusual in syslog that seems to relate to fwupd and is specific to the dates on which this occurred (though nor am I sure what to look for).

What could be happening here?  Should I suspect some kind of firmware attack - or is there a reasonable explanation?

Any input from the learned people here would be appreciated.

**fwupd version information**
```
$ wupdmgr --version
runtime   org. freedesktop. fwupd      1.7.9
runtime   com. dell. libsmbios         2.4
compile   org. freedesktop.gusb        0.3.10
compile   com. hughsie. libjcat        0.1.9
runtime   org. kernel                  6.8.0-47-generic
compile   org. freedesktop. fwupd      1.7.9
runtime   org. freedesktop.gusb        0.3.10
```

hughsie, 2024-11-11:
I think this is *a* bug, but I don't think it's our bug unfortunately. I would suggest contacting Lenovo support.

closed 2024-11-11 by hughsie

hu606, 2024-11-17:
Is there some output/string in syslog (or some other log) that could be searched for to see if fwupd did indeed attempt a bios/uefi firmware update that corresponds with these incidents?

hughsie, 2024-11-17:
> fwupd did indeed attempt a bios/uefi firmware update

fwupd doesn't do anything automatically; the daemon doesn't even have network access. If you asked it to do an upgrade it would be in the history database.

> WARNING: UEFI capsule updates not available or enabled in firmware setup

This tells me that fwupd wouldn't be able to update the firmware even if you wanted to. It's literally incapable of updating firmware on your system in it's current BIOS configuration.

hu606, 2024-11-18:
Would fwupd not be able to update my firmware because it is too old?  (I do see in syslog from FuPluginPciMei that "ME family not supported for 0:10.0.29.1000").

Any wild guesses as to what might cause the "Request MeSpiLock Failed" in this case?  I've had this laptop many years and never seen this before this year.

hu606, 2024-11-19:
Thanks - really appreciate your responses!