fwupd/fwupd #7938: failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load Boot0009: A maximum read size must be specified

Latnook, 2024-10-20:
Hello everyone,
I get this message when I try to run fwupdmgr update: `failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load Boot0009: A maximum read size must be specified`

I tried to check the wiki for the Blocked executable in the ESP but there is no mention of the exact error I get.

`$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd         2.0.1
runtime   org.kernel                    6.11.4-arch1-1
`

I installed fwupd using pacman.

`$ fwupdmgr get-devices --show-all-devices
LENOVO 21K70045IV
│
├─ELAN0688:00 04F3:320B:
│     Device ID:          787612607316474f295c267228ac08e06922c409
│     Summary:            Touchpad
│     Current version:    0x0005
│     Bootloader Version: 0x0003
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              7fd09f45-2b14-5100-9334-b061ff620984 ← HIDRAW\VEN_04F3&DEV_320B
│                         6392df21-879e-5db8-8910-d23e0fab4f7c ← HIDRAW\VEN_04F3&DEV_320B&MOD_003C
│                         922006d1-d20b-50f1-90e1-3a32f9fb70ac ← ELANTP\ICTYPE_14&MOD_003C
│                         524197f9-106d-53f5-bb98-10882bb7d123 ← ELANTP\ICTYPE_14&MOD_003C&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Can tag for emulation
│   
├─AMD Ryzen 7 PRO 7840U w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a704107
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              2a1df9de-fe02-59c8-8af1-32c8bb011f94 ← CPUID\PRO_0&FAM_19&MOD_74
│ │                       5a4bca32-6547-584f-bb36-920d9adede25 ← CPUID\PRO_0&FAM_19&MOD_74&STP_1
│ │   Device Flags:       • Internal device
│ │ 
│ ├─AMD Radeon 780M:
│ │ │   Device ID:        75dbd857f37f0f1aa3d5bf2fea25b876288fb415
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  022.012.000.027.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             4a1501b7-b500-5255-9d9c-41d652a4d5bc ← AMD\113-PHXGEN
│ │ │   Device Flags:     • Internal device
│ │ │                     • Can tag for emulation
│ │ │ 
│ │ └─ATNA60YV04-0:
│ │       Device ID:      aec1a869eb0df71b7cea6b3ac71d39b830faf164
│ │       Vendor:         PNP:SDC
│ │       GUID:           fa4ccc78-6ad6-5e94-9b9d-2dfae7cfb53d ← DRM\VEN_SDC&DEV_4165
│ │       Device Flags:   • Internal device
│ │                       • Can tag for emulation
│ │     
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.00.7d
│ │     Bootloader Version: 00.2d.00.7d
│ │     Vendor:           Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│ │     GUID:             9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │     Device Flags:     • Internal device
│ │                       • Can tag for emulation
[... 643 lines cut ...]
│       ├── video.mod
│       ├── videotest_checksum.mod
│       ├── videotest.mod
│       ├── wrmsr.mod
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initramfs-linux-fallback.img
├── initramfs-linux.img
└── vmlinuz-linux

9 directories, 363 files
`

OS: Arch Linux x86_64 
I have tried rebooting
First time I try this on this OS and on this machine.
I am using an NVME disk.
Secure boot is not enabled.

Latnook, 2024-10-21:
Sorry, there is no such file.

hughsie, 2024-10-21:
Can you attach the output from `sudo fwupdtool efivar-boot -vv` pls.

lucaspar, 2024-10-21:
@hughsie not OP, but I'm seeing a similar message (`failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: failed to load Boot0007: A maximum read size must be specified`).

```
sudo fwupdtool efivar-boot -vv
```

Produces:

<pre>
23:22:59.854 FuDebug              verbose to debug (on console 0)
23:22:59.854 FuEngine             starting fwupd 2.0.1...
23:22:59.858 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x74
23:22:59.858 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x1
  length: 0x2a
23:22:59.858 FuStruct             FuStructEfiHardDriveDevicePath:
  subtype: 0x1 [hard-drive]
  partition_number: 0x1
  partition_start: 0x22
  partition_size: 0x2007de
  partition_signature: eed9c4d4-d93a-4c57-8bf1-9e86d9596495
  partition_format: 0x2 [guid-partition-table]
  signature_type: 0x2 [guid]
23:22:59.858 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x4
  length: 0x46
23:22:59.858 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x4
  length: 0x46
23:22:59.858 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
23:22:59.860 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
23:22:59.860 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.860 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.860 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
23:22:59.861 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
23:22:59.861 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.861 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.861 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
23:22:59.862 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x29
23:22:59.862 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x25
23:22:59.862 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x25
23:22:59.862 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
23:22:59.863 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
23:22:59.863 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.863 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.863 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
23:22:59.864 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
23:22:59.864 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.864 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.864 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
23:22:59.865 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x63
23:22:59.865 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.865 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.865 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x3b
23:22:59.865 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x3b
23:22:59.865 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
23:22:59.867 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x2c
23:22:59.867 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.867 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
23:22:59.867 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x4
23:22:59.867 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x4
failed to load Boot0007: A maximum read size must be specified
</pre>

Latnook, 2024-10-22:
`06:57:36.061 FuDebug              verbose to debug (on console 1)
06:57:36.062 FuEngine             starting fwupd 2.0.1…
06:57:36.063 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
06:57:36.064 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x5e
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x1
  length: 0x2a
06:57:36.064 FuStruct             FuStructEfiHardDriveDevicePath:
  subtype: 0x1 [hard-drive]
  partition_number: 0x1
  partition_start: 0x800
  partition_size: 0x400000
  partition_signature: 8e3f0b4c-989e-43c8-ac74-47a1d4272766
  partition_format: 0x2 [guid-partition-table]
  signature_type: 0x2 [guid]
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x4
  length: 0x30
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x4
  length: 0x30
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.064 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x74
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x1
  length: 0x2a
06:57:36.064 FuStruct             FuStructEfiHardDriveDevicePath:
  subtype: 0x1 [hard-drive]
  partition_number: 0x1
  partition_start: 0x800
  partition_size: 0x32000
  partition_signature: e1e8ca0d-f6be-4168-b2c9-35c3993987bc
  partition_format: 0x2 [guid-partition-table]
  signature_type: 0x2 [guid]
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x4
  length: 0x46
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x4 [media]
  subtype: 0x4
  length: 0x46
06:57:36.064 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.064 FuEfiLoadOption      not a shim hive, ignoring: constant FuStructShimHive.magic was not valid
06:57:36.064 FuEfiLoadOption      not a path, saving as raw blob: not ASCII data: 䥗䑎坏S
06:57:36.065 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.065 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.065 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x29
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x25
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x25
06:57:36.065 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.066 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x29
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x25
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x25
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.066 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.066 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x28
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.066 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.067 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x63
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x3b
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x3b
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x7f [end]
  subtype: 0xff
  length: 0x4
06:57:36.067 FuStruct             FuStructEfiLoadOption:
  attrs: 0x1 [active]
  dp_size: 0x2c
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0xa
  length: 0x24
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x4
06:57:36.067 FuStruct             FuStructEfiDevicePath:
  type: 0x3 [message]
  subtype: 0x18
  length: 0x4
failed to load Boot0009: A maximum read size must be specified
`

hughsie, 2024-10-22:
@lucaspar do you have a `/sys/firmware/efi/efivars/Boot0007-8be4df61-93ca-11d2-aa0d-00e098032b8c`? If so, can you attach please?

hughsie, 2024-10-22:
@lucaspar @Latnook can you try the branch in https://github.com/fwupd/fwupd/pull/7944 please.

Latnook, 2024-10-22:
How can I do it? I installed it via pacman?

lucaspar, 2024-10-22:
> @lucaspar do you have a `/sys/firmware/efi/efivars/Boot0007-8be4df61-93ca-11d2-aa0d-00e098032b8c`? If so, can you attach please?

No such file, or any other starting with `Boot0007`.

Latnook, 2024-10-22:
No file starting with Boot0007.

closed 2024-10-22 by hughsie

Latnook, 2024-11-17:
I just updated fwupd and I still get the same error message.

hughsie, 2024-11-17:
> I just updated fwupd and I still get the same error message.

Updated to what?

Latnook, 2024-11-17:
2.0.1-2

hughsie, 2024-11-17:
2.0.1 doesn't contain the fix, it was merged after 2.0.1 was released. You need 2.0.2 (due to be released on Tuesday) or to backport the patch, or to build a git snapshot.

Latnook, 2024-12-01:
I upgraded to 2.0.2 and it still has the same error.

```
$ sudo fwupdmgr upgrade
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 220 to 371?                                            ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Insecure versions of the Microsoft Windows boot manager affected by Black    ║
║ Lotus were added to the list of forbidden signatures due to a discovered     ║
║ security problem.This updates the dbx to the latest release from Microsoft.  ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.Applying this update may also cause     ║
║ some Windows install media to not start correctly.                           ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝
Perform operation? [Y|n]: y   
Verifying…               [                                       ]
failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: could not find EFI DP
$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.2
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd         2.0.2
runtime   org.kernel                    6.12.1-arch1-1

```