fwupd/fwupd #7939: Mark UEFI dbx updates as FWUPD_DEVICE_FLAG_AFFECTS_FDE

kyl191, 2024-10-21:
**Is your feature request related to a problem? Please describe.**
tl;dr - An UEFI dbx update (from version 312 to 377) appears to have tripped the Bitlocker Boot Policy protection.

I'm dual booting a laptop (Lenovo X13) between Fedora 40 and Windows 10 with Bitlocker enabled and the key in the TPM.

I ran `fwupdmgr update` from Fedora and it prompted me to install 2 updates: a BIOS update and a UEFI dbx update.

The BIOS update had the full disk encryption warning (and linked to https://github.com/fwupd/fwupd/wiki/Full-Disk-Encryption-Detected) and I rejected the update (entered `n` at the prompt), but the dbx update didn't have a similar warning so I allowed that to continue.

When I tried to reboot into Windows, I got a Bitlocker message that the secure boot policy had changed and I needed to enter the recovery key:

[image]

Given the timing, I looked into how a dbx update could change the boot policy. I think the comment in https://github.com/fwupd/fwupd/blob/main/plugins/uefi-capsule/fu-uefi-capsule-plugin.c#L1039 is not valid any more - specifically that only a BIOS update touches the PCR registers.

Microsoft has had issues with dbx updates triggering BitLocker protections in the past: 
* https://support.microsoft.com/en-us/topic/kb5012170-security-update-for-secure-boot-dbx-72ff5eed-25b4-47c7-be28-c42bd211bb15 (this mentions PCR7 specifically)
* https://www.theregister.com/2022/08/15/bitlocker_microsoft/

**Describe the solution you'd like**
I'd like to apply the same FDE flag to the dbx update that the BIOS update already has to at least warn the user about the updates.

I _think_ it might be as simple as copying the FDE flag code from the BIOS updater to `plugins/uefi-dbx/fu-uefi-dbx-device.c:fu_uefi_dbx_device_init`, something like
```
if (!fu_common_check_full_disk_encryption(&error_fde)) {
		g_debug("FDE in use, set flag: %s", error_fde->message);
		fu_device_add_flag(FU_DEVICE(self), FWUPD_DEVICE_FLAG_AFFECTS_FDE);
	}			
```

This is a guess based on the common `fu_device_add_flag`, but I'm not familiar enough with the fwupd codebase to say if there's other locations that need to be updated. (Also, not familiar with C.)

**Describe alternatives you've considered**
Dump the existing dbx list somewhere on disk before updating it.

This is focused specifically on recovering from a dbx update, one thing I tried was resetting the signature list back to the version in the BIOS, then allowing Windows Update (on a separate install) to update and install dbx updates.

This specific attempt failed (ended up on version 310 instead of 312, so there was still a mismatch), but it's plausible I could have restored the original dbx list if I had a dump on disk.

I don't think it's really viable because the potential procedure is messy to run - and if someone is warned about the update breaking Bitlocker they wouldn't perform the update.

Alternatively, if a BIOS update is canceled, then the dbx update should automatically be canceled.

I don't think this makes sense since updating the dbx without a BIOS update should be possible, and if added to fwupd it would start having dependencies between the install order that would add unnecessary complexity.

**Additional context**
The previous "version" doesn't match any of the official UEFI revocation list releases - somehow 312 signatures were in the list before upgrading to version 377.

```
$ fwupdmgr get-history
LENOVO 20UFCTO1WW
│
└─UEFI dbx:
  │   Device ID:          362301da643102b9f38477387e2193e57abaa590
  │   Previous version:   312
  │   Update State:       Success
  │   Last modified:      2024-10-15 02:50
  │   GUID:               5971a208-da00-5fce-b5f5-1234342f9cf7
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Device is usable for the duration of the update
  │                       • Only version upgrades are allowed
  │                       • Signed Payload
  │ 
  └─Secure Boot dbx Configuration Update:
        New version:      371
        Remote ID:        lvfs
        Release ID:       35287
        Summary:          UEFI Secure Boot Forbidden Signature Database
        Variant:          x64
        License:          Proprietary
        Size:             21.2 kB
        Created:          2023-05-09
        Urgency:          High
```

I'm assuming the version weirdness is down to Microsoft not releasing official updates through the UEFI revocation list, or maybe using their own telemetry to decide when to block individual file hashes.

hughsie, 2024-10-21:
Ideologically, I don't think a dbx update is supposed to affect sealing against any of the PCRs -- i.e. it feels like a bug somewhere.  @flanfly can you sanity check my assertion please.

kyl191, 2024-10-27:
@hughsie I agree - this does feel like a bug somewhere, potentially in some defaults that got chosen by Windows, or how Bitlocker interacted with settings chosen by Lenovo, or some box I checked somewhere during the install.

But Microsoft has KB articles mentioning "known compatibility issues with TPM measurements" (admittedly calling out Windows Server instead of Windows 10): https://support.microsoft.com/en-us/topic/kb5025885-how-to-manage-the-windows-boot-manager-revocations-for-secure-boot-changes-associated-with-cve-2023-24932-41a975df-beb2-40c1-99a3-b3ff139f832d

And CVE bug fix releases were halted with 
> When customers applied the fix for this vulnerability to their devices, we received feedback about firmware incompatibility issues that were causing BitLocker to go into recovery mode on some devices. As a result, with the release of the August 2024 security updates we are disabling this fix.
https://msrc.microsoft.com/update-guide/vulnerability/CVE-2024-38058

There's been reasonably wide spread issues in 2022 and 2024 seemingly tied to dbx updates:
* https://www.theregister.com/2022/08/15/bitlocker_microsoft/
* https://www.theregister.com/2024/07/24/windows_update_bitlocker/

-----

Whatever the bug is, I do not believe it is fwupd's responsibility to fix. The Secure Boot failure shouldn't have happened, but it did. My goal here is to try and prevent another person from experiencing the same issue. 

The easiest path I see is that dbx updates get a warning like BIOS updates get a warning once Bitlocker/FDE is detected. Would you prefer another alternative?

Wack0, 2024-11-02:
With Secure Boot enabled, bitlocker seals keys to PCRs 7 and 11, of which 11 is capped after key derivation.

PCR 7, of course, has db and dbx extended into it by firmware.

hughsie, 2024-11-02:
@superm1 what do you think?

superm1, 2024-11-02:
Huh, I wonder if we're not parsing with `fwupdtpmevlog` properly or there is a bug in the implementation on my machine.

I didn't see any `EV_EFI_VARIABLE_AUTHORITY` on PCR7 on the event log for my machine.  We're only finding `EV_EFI_VARIABLE_DRIVER_CONFIG`.

Here's the output of `sudo ./venv/build/plugins/tpm/fwupdtpmevlog --pcr 7 > ~/pcr7.txt` from fwupd dev env on my machine (Framework 13 AMD).
[pcr7.txt](https://github.com/user-attachments/files/17608316/pcr7.txt)

kyl191, 2024-11-02:
Using PCR 7 matches with what I see in `manage-bde` output:
```
PS C:\WINDOWS\system32> manage-bde -protectors -get C:
BitLocker Drive Encryption: Configuration Tool version 10.0.19041
Copyright (C) 2013 Microsoft Corporation. All rights reserved.

Volume C: []
All Key Protectors

    TPM:
      ID: {265F491D-7AE0-4B50-B58F-489059DBF627}
      PCR Validation Profile:
        7, 11
        (Uses Secure Boot for integrity validation)

    Numerical Password:
      ID: {6A6BA07A-6236-4533-BC67-4688A31E316D}
      Password:
        lol-no-redacted
```

And the group policy editor has this description for "Configure TPM platform validation profile for native UEFI firmware configurations":
> On PCs that lack Secure Boot State (PCR 7) support, the default platform validation profile secures the encryption key against changes to the core system firmware executable code (PCR 0), extended or pluggable executable code (PCR 2), boot manager (PCR 4), and the BitLocker access control (PCR 11).

> When Secure Boot State (PCR7) support is available, the default platform validation profile secures the encryption key using Secure Boot State (PCR 7) and the BitLocker access control (PCR 11). 

I'll reboot into Fedora and try using the command @superm1 just ran, see what output I get for PCR 7.

kyl191, 2024-11-02:
I see 3 instances of `ev-efi-variable-authority` in my PCR 7: [pcr7.txt](https://github.com/user-attachments/files/17608398/pcr7.txt)

superm1, 2024-11-02:
In that case it sounds like a pretty solid indication we should be setting this flag, thanks for sharing it.

kyl191, 2024-11-02:
No worries - also happy to run additional commands if it might help.

It sounds like a desired solution would be to flag if EV_EFI_VARIABLE_AUTHORITY is found in PCR 7 instead of my original blanket "flag dbx updates as impacting FDE" so I won't submit a PR with the dbx flagging.

superm1, 2024-11-04:
It is a bit complicated to plumb through.  Furthermore this flag [isn't assertive](https://fwupd.github.io/libfwupd/flags.DeviceFlags.html#affects_fde).

> Updating the firmware on this device may invalidate secrets used to decrypt the volume, and the recovery key may be required.

I'm tending to think we might as well just turn it on for the dbx plugin.  Can you do that PR?

flanfly, 2024-11-04:
The behavior is correct. All variables that influence Secure Boot need to be measured into PCR[07]. This allows you for example to seal against PCR 0,1,2 and 7 to express a policy that allows all EFI applications to boot that have valid Secure Boot signatures given the seal-against db and dbx.

superm1, 2024-11-04:
Right, I'm saying it's not worth collecting the data and passing between plugins whether PCR 7 has any relevant entries. I think it's going be fine to just set that for UEFI dbx device for everyone.

kyl191, 2024-11-05:
Submitted https://github.com/fwupd/fwupd/pull/8033

Ran it locally and included the output showing the FDE prompt is now shown when updating the DBX

kyl191, 2024-11-08:
#8033 is committed, and now I'm trying to address the point that was brought up by @superm1 in the comments:

> for how the codebase works "today" this makes sense, but I have to question if we should be changing semantics of `fu_common_check_full_disk_encryption` as part of this PR.
...
> Then return `TRUE` if the function "was checked" successfully, `FALSE` if there was a disk problem.  And `*result` can be set based upon whether FDE was found or not.

_Originally posted by @superm1 in https://github.com/fwupd/fwupd/pull/8033#discussion_r1830926355_

1. Should I create a new issue to track this, or is it fine to keep in this issue?
2. It's not immediately clear how to handle the "failed to probe" case.  It's straightforward to replace the "check return value" with "check result" for the FDE was detected case, would we want to keep the current behaviour of flagging FDE or introduce a new `DEVICE_FLAG` flag (or `DEVICE_PROBLEM`?).

```
probe_success = fu_common_check_full_disk_encryption(&has_fde, &error_fde);
if (!probe_success) {
	g_debug("Failed to probe block devices for FDE: %s", error_fde->message);
	fu_device_add_flag(device, FWUPD_DEVICE_FLAG_AFFECTS_FDE); // ?
}
if (has_fde) (
	fu_device_add_flag(device, FWUPD_DEVICE_FLAG_AFFECTS_FDE);
}
```

superm1, 2025-02-07:
We changed the semantics of the flag recently where the only failure is a failed to detect block devices.  If the block devices are not detected we won't be able to check for binaries on the ESP, and thus won't do the update.  I feel this is sufficiently solved now.

https://github.com/fwupd/fwupd/commit/515e8ce0272ae43d899acb0e5e249658ba1f5cca

closed 2025-02-07 by superm1

kyl191, 2025-02-09:
I concur that it sounds sufficiently solved - the failed to probe part looks like it gets handled as an error now, which I hadn't considered.

Thanks!