fwupd/fwupd #7953: Verification failed: (0x1A) Security Violation - Odd efibootmgr entry when Secure Boot is active

ChristianAtSIE, 2024-10-23:
**Describe the bug**
I am currently testing to do UEFI Capsule update via fwupdx64.efi, when Secure Boot is enabled. 

Running the command `fwupdtool install-blob systemFirmware.bin` with Secure Boot disabled works fine.

Running the same command with Secure Boot enabled results in a blue error screen noting 'Verification failed: (0x1A) Security Violation', with the option to install MOK keys. As far I know this comes from SHIM. Every Bootfile is singed with the one db entry, so I do not know why it is failing here.

Investigating the output of `sudo efibootmgr --verbose`, the automated generated boot entry (Boot0000) seems odd to me:
``` shell
(fwupd) phoenix@phoenix:/boot/efi$ sudo efibootmgr --verbose
BootNext: 0000
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0002,0001
Boot0000* Linux Firmware Updater	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\ubuntu\shimx64.efi\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001* ubuntu	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\UBUNTU\SHIMX64.EFI..BO
Boot0002* UEFI: Corsair Voyager SliderX1000A, Partition 1	PciRoot(0x0)/Pci(0x14,0x0)/USB(12,0)/USB(2,0)/HD(1,MBR,0xb3c93838,0x800,0x39d1800)..BO
Boot0003* KHCH	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\ubuntu\fwupdx64.efi
```
The odd thing I see is `\EFI\ubuntu\shimx64.efi\.f.w.u.p.d.x.6.4...e.f.i...` which look to me, it asumes 'shimx64.efi' is a directory. Also the . in between the filename is confusing to me.
When Secure Boot is disabled the boot entry looks like Boot0003, what I manually created and this works.

I am not sure, whether this is a bug, it also could be not proper use of the SHIM mechanics.

**Steps to Reproduce**

- Have SecureBoot enabled.
- Have *.efi files signed with installed db-key in /boot/efi/*
- Have fwupdx64.efi signed.
- Install firmware `fwupdtool install-blob firmware.bin`
-- Select 'System Firmware' entry

**Expected behavior**

- Rebooting
- Installing UEFI Capsule update with attached UX-Capsule
- Back to Linux

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

compile info.libusb 1.0.26
compile org.freedesktop.fwupd 2.0.0
compile com.hughsie.libxmlb 0.3.18
compile com.hughsie.libjcat 0.2.1
runtime org.freedesktiop.fwupd-efi 1.7
runtime com.hughsie.libxmlb 0.3.x
runtime com.hughsie.libjcat 0.2.1
runtime org.kernel 6.5.0.-44-generic
runtime org.freedesktiop.fwupd 2.0.0
```

hughsie, 2024-10-23:
Weird; can you grab us the log from `sudo fwupdtool install-blob systemFirmware.bin -vv --plugins uefi_capsule` please.

ChristianAtSIE, 2024-10-24:
Commands executed:

- Activate environment to use self compiled fwupd (only change is the workaround for UX-capsule)
- Verbose efibootmgr output before running system firmware update
- mokutil to show Secureboot enabled
- efi-readvar to show that keys are installed
- system firmware update like it was requested
  - The question at the end for rebooting was denied to do other commands
- Verbose efibootmgr output after running system firmware update

```shell
phoenix@phoenix:~$ cd fwupd
phoenix@phoenix:~/fwupd$ source venv/bin/activate
To build or rebuild fwupd within development environment run:

# build-fwupd

To run the test suite run:

# test-fwupd

To run any tool under gdbserver add DEBUG=1 to env, for example:

# DEBUG=1 fwupdtool get-devices

To leave fwupd development environment run:

# deactivate
(fwupd) phoenix@phoenix:~/fwupd$ sudo efibootmgr --verbose
[sudo] password for phoenix: 
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0002,0001
Boot0001* ubuntu	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\UBUNTU\SHIMX64.EFI..BO
Boot0002* UEFI: Corsair Voyager SliderX1000A, Partition 1	PciRoot(0x0)/Pci(0x14,0x0)/USB(12,0)/USB(2,0)/HD(1,MBR,0xb3c93838,0x800,0x39d1800)..BO
(fwupd) phoenix@phoenix:~/fwupd$ sudo mokutil --sb-state 
SecureBoot enabled
(fwupd) phoenix@phoenix:~/fwupd$ sudo efi-readvar 
Variable PK, length 829
PK: List 0, type X509
    Signature 0, size 801, owner 26dc4851-195f-4ae1-9a19-fbf883bbb35e
        Subject:
            CN=SIE_Test_PK
        Issuer:
            CN=SIE_Test_PK
Variable KEK, length 831
KEK: List 0, type X509
    Signature 0, size 803, owner 26dc4851-195f-4ae1-9a19-fbf883bbb35e
        Subject:
            CN=SIE_Test_KEK
        Issuer:
            CN=SIE_Test_KEK
Variable db, length 829
db: List 0, type X509
    Signature 0, size 801, owner 26dc4851-195f-4ae1-9a19-fbf883bbb35e
        Subject:
            CN=SIE_Test_DB
        Issuer:
            CN=SIE_Test_DB
Variable dbx, length -4
Variable MokList has no entries
(fwupd) phoenix@phoenix:~/fwupd$ LANG=en.UTF-8 fwupdtool --verbose --verbose install-blob systemFirmware.bin --plugins uefi_capsule
15:03:19.181 FuDebug              verbose to debug (on console 1)
15:03:19.186 FuEngine             starting fwupd 2.0.0?
Loading?                 [ -                                     ]15:03:19.189 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ?gio-vfs?
Loading?                 [                                       ]15:03:19.198 FuMain               locked /run/lock/fwupdtool
15:03:19.199 FuConfig             loading config /home/phoenix/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
15:03:19.199 FuConfig             loading config /home/phoenix/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
15:03:19.199 FuConfig             trying to load config values from /home/phoenix/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf
15:03:19.199 FuCommon             mapped file /home/phoenix/fwupd/venv/bin/../dist/etc/fwupd/fwupd.conf of size 0x62
[... 2313 lines cut ...]
  EspId:                /org/freedesktop/UDisks2/block_devices/sda1
  EspPath:              /boot/efi
  EspKind:              c12a7328-f81f-11d2-ba4b-00a0c93ec93b

15:03:27.856 FuDeviceList         no replug or re-enumerate required
Waiting?                 [***************************************]15:03:27.856 FuEngine             Updating System Firmware took 1.132196 seconds

An update requires a reboot to complete. Restart now? [y|N]: n
../../src/fu-tool.c:938 [8675.87ms]:
    ../../src/fu-engine.c:3534 [6849.18ms]:

15:03:29.892 FuPluginTest         destroy
15:03:29.894 FuProgress           raw timing data was { 0.002, 0.000, 0.003, 0.000 } -- steps were set as [ 1 1 3 95 ] but should have been [ 41 2 55 2 ] at ../../libfwupdplugin/fu-context.c:1010
15:03:29.894 FuProgress           raw timing data was { 0.000, 0.030, 0.003, 0.168, 0.001 } -- steps were set as [ 1 8 64 26 1 ] but should have been [ 0 15 2 83 0 ] at ../../plugins/uefi-capsule/fu-uefi-capsule-plugin.c:917
(fwupd) phoenix@phoenix:~/fwupd$ sudo efibootmgr --verbose
BootNext: 0000
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0002,0001
Boot0000* Linux Firmware Updater	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\ubuntu\shimx64.efi\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001* ubuntu	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\UBUNTU\SHIMX64.EFI..BO
Boot0002* UEFI: Corsair Voyager SliderX1000A, Partition 1	PciRoot(0x0)/Pci(0x14,0x0)/USB(12,0)/USB(2,0)/HD(1,MBR,0xb3c93838,0x800,0x39d1800)..BO
(fwupd) phoenix@phoenix:~/fwupd$ 

```

hughsie, 2024-12-04:
> Boot0000* Linux Firmware Updater	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\ubuntu\shimx64.efi\.f.w.u.p.d.x.6.4...e.f.i...

That looks completely fine to me -- it's in UCS-2 encoding, and is used as a 2nd stage bootloader to shim. Is it possible you've got an unsigned shim installed (that's not blessed with the microsoft cert) somehow? i.e. does it "just work" if you use Fedora 41?

>             CN=SIE_Test_PK

That's not expected either, can you attach the .bin files from `sudo cat /sys/firmware/efi/efivars/PK-8be4df61-93ca-11d2-aa0d-00e098032b8c > PK.bin` and `sudo cat /sys/firmware/efi/efivars/db-d719b2cb-3d3a-4596-a3bc-dad00e67656f > db.bin` please -- and we can add it to the checks we do.

ChristianAtSIE, 2024-12-09:
The SecureBoot keys are self created certificates. They are currently for testing only and not production use. For this reason this might seem odd. So no Microsoft Certificates are involved at all.

The shimx64.efi has to be signed correctly, as the first output of `efibootmgr` states it boots from this shim (boot index 0001):
```shell
(fwupd) phoenix@phoenix:~/fwupd$ sudo efibootmgr --verbose
[sudo] password for phoenix: 
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0002,0001
Boot0001* ubuntu	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\UBUNTU\SHIMX64.EFI..BO
Boot0002* UEFI: Corsair Voyager SliderX1000A, Partition 1	PciRoot(0x0)/Pci(0x14,0x0)/USB(12,0)/USB(2,0)/HD(1,MBR,0xb3c93838,0x800,0x39d1800)..BO
```
and the output of `mokutil --sb-state` states SecureBoot is enabled with those keys:
```shell
(fwupd) phoenix@phoenix:~/fwupd$ sudo mokutil --sb-state 
SecureBoot enabled
(fwupd) phoenix@phoenix:~/fwupd$ sudo efi-readvar 
Variable PK, length 829
PK: List 0, type X509
    Signature 0, size 801, owner 26dc4851-195f-4ae1-9a19-fbf883bbb35e
        Subject:
            CN=SIE_Test_PK
        Issuer:
            CN=SIE_Test_PK
Variable KEK, length 831
KEK: List 0, type X509
    Signature 0, size 803, owner 26dc4851-195f-4ae1-9a19-fbf883bbb35e
        Subject:
            CN=SIE_Test_KEK
        Issuer:
            CN=SIE_Test_KEK
Variable db, length 829
db: List 0, type X509
    Signature 0, size 801, owner 26dc4851-195f-4ae1-9a19-fbf883bbb35e
        Subject:
            CN=SIE_Test_DB
        Issuer:
            CN=SIE_Test_DB
Variable dbx, length -4
Variable MokList has no entries
```

I do not know how the correct `efibootmgr` syntax is, but it seems odd to me that the last call of `efibootmgr` at boot index 0000 is:
```shell
(fwupd) phoenix@phoenix:~/fwupd$ sudo efibootmgr --verbose
BootNext: 0000
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0002,0001
Boot0000* Linux Firmware Updater	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\ubuntu\shimx64.efi\.f.w.u.p.d.x.6.4...e.f.i...
Boot0001* ubuntu	HD(1,GPT,65c5253e-bb82-4439-bff8-6ddf0435352a,0x800,0x10d000)/\EFI\UBUNTU\SHIMX64.EFI..BO
Boot0002* UEFI: Corsair Voyager SliderX1000A, Partition 1	PciRoot(0x0)/Pci(0x14,0x0)/USB(12,0)/USB(2,0)/HD(1,MBR,0xb3c93838,0x800,0x39d1800)..BO
```
which displays to me that is sees shimx64.efi as an folder.
`\EFI\ubuntu\shimx64.efi\.f.w.u.p.d.x.6.4...e.f.i...`
This also differs to your quote in your comment.

hughsie, 2024-12-09:
What does `sudo fwupdtool efivar-boot` say?