fwupd/fwupd #7964: Firmware Attestation Locks Up `fwupdmgr.service`

RayJW, 2024-10-24:
**Describe the bug**
When I'm trying to verify the firmware of my GPU, the service just completely locks up and restarting it takes around 3 minutes in my case.

**Steps to Reproduce**
In GNOME Firmware, try to verify the firmware to get rid of the unknown attestation status and watch it lock up. Restarting the application will now result in an infinite spinner as the service is completely unusable it seems.

**Expected behavior**
To properly verify the firmware like on the other devices I use.

**fwupd version information**

```shell
❯ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.11.5-2-cachyos
runtime   org.freedesktop.fwupd         2.0.1
```
pacman

<details>

<summary>**fwupd device information**</summary>

```shell
❯ fwupdmgr get-devices --show-all-devices
System manufacturer System Product Name
│
├─AMD Radeon RX 6950 XT:
│ │   Device ID:          21ba7fbcdfb80c824895aad926418475726bcc79
│ │   Summary:            NAVI21 Gaming KXTX D4124100
│ │   Current version:    020.001.000.071.700002
│ │   Vendor:             Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │   GUID:               c66fc19d-6d31-5889-8f5c-e34e11caf8de ← AMD\113-EXT700
│ │   Device Flags:       • Cryptographic hash verification is available
│ │                       • Can tag for emulation
│ │ 
│ ├─Unknown Device:
│ │     Device ID:        96928709c665f953e18829015803502c7fb9a1ef
│ │     Vendor:           PNP:AOC
│ │     Serial Number:    373
│ │     GUID:             140c455a-5e93-5f01-9963-adc35eb31465 ← DRM\VEN_AOC&DEV_3520
│ │     Device Flags:     • Can tag for emulation
│ │   
│ └─Unknown Device:
│       Device ID:        a6c89aee142798063900a459b2f564f72137305e
│       Vendor:           PNP:BNQ
│       Serial Number:    L2E02370SL0
│       GUID:             6f6f7d49-8d18-5adc-8544-be0f0ca56e44 ← DRM\VEN_BNQ&DEV_7F1C
│       Device Flags:     • Can tag for emulation
│     
├─ADATA SX8200PNP:
│     Device ID:          3743975ad7f64f8d6575a9ae49fb3a8856fe186f
│     Summary:            NVM Express solid state drive
│     Current version:    32B3T8EA
│     Vendor:             ADATA Technology Co., Ltd. (PCI:0x1CC1)
│     Serial Number:      2L2329Q2QNN7
│     GUIDs:              80968304-c171-5944-8e11-18a1fb4203b9 ← NVME\VEN_1CC1&DEV_8201
│                         ae9f95e4-f502-539c-897e-84601d399a11 ← NVME\VEN_1CC1&DEV_8201&SUBSYS_1CC18201
│                         a41282e2-8560-5da8-bb61-738fd992a09f ← ADATA SX8200PNP
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Can tag for emulation
│   
├─AMD Ryzen 9 3900X 12-Core Processor:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x08701033
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              b54f3c89-8a41-5b9f-9a5c-0367c19a9451 ← CPUID\PRO_0&FAM_17&MOD_71
│ │                       d951a23c-a920-50ac-ad89-0660aa0be95c ← CPUID\PRO_0&FAM_17&MOD_71&STP_0
│ │   Device Flags:       • Internal device
│ │ 
│ └─Secure Processor:
│       Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│       Vendor:           Advanced Micro Devices, Inc. [AMD] (PCI:0x1022)
│       GUID:             bb3bed27-5f4a-554a-87c4-b52ba58e58eb ← PCI\VEN_1022&DEV_1486
│       Device Flags:     • Internal device
│                         • Can tag for emulation
│     
├─GPIO controller:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│     Device Flags:       • Can tag for emulation
│   
├─System Firmware:
│ │   Device ID:          b692fbe248e186c3c1de343500fae334146dc67d
│ │   Summary:            UEFI System Resource Table device (updated via NVRAM)
│ │   Current version:    20499
│ │   Minimum Version:    20499
│ │   Vendor:             System manufacturer (DMI:American Megatrends Inc.)
│ │   Update State:       Success
│ │   GUID:               d806a1e7-39b0-53fd-8fa0-2426bae8928e
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │   Device Requests:    • Message
│ │ 
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Summary:          UEFI revocation database
│       Current version:  371
│       Minimum Version:  371
│       Vendor:           UEFI:Linux Foundation
│       Install Duration: 1 second
│       GUIDs:            077e4c6d-e109-592a-aea2-85b1ad7884d4 ← UEFI\CRT_0D8F1EB243A4F1AA70C847EE67F1BAA2FAD22F696E8104A7D96ED61BAC38BC66&ARCH_X64
│                         f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
│       Device Flags:     • Internal device
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Only version upgrades are allowed
│                         • Signed Payload
│     
├─TPM:
│     Device ID:          1d8d50a4dbc65618f5c399c2ae827b632b3ccc11
│     Current version:    3.92.2.5
│     Vendor:             Advanced Micro Devices, Inc. (TPM:AMD)
│     GUIDs:              9305de1c-1e12-5665-81c4-37f8e51219b8 ← TPM\VEN_AMD&DEV_0001
│                         78a291ae-b499-5b0f-8f1d-74e1fefd0b1c ← TPM\VEN_AMD&MOD_AMD
│                         65a3fced-b423-563f-8098-bf5c329fc063 ← TPM\VEN_AMD&DEV_0001&VER_2.0
│                         5e704f0d-83cb-5364-8384-f46d725a23b8 ← TPM\VEN_AMD&MOD_AMD&VER_2.0
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device can recover flash failures
│                         • Full disk encryption secrets may be invalidated when updating
│                         • Signed Payload
│                         • Can tag for emulation
│   
├─UEFI Device Firmware:
│     Device ID:          8fc9414ee1b271cc8742ce1b244b9c238e6713b0
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    335609927
│     Vendor:             DMI:American Megatrends Inc.
│     Update State:       Success
│     GUID:               f766f6e6-b43d-4acd-a4bd-80ff2f0af5cc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
├─UEFI Device Firmware:
│     Device ID:          e6d62239c872f1c5f496a167fafcae04c48f377b
│     Summary:            UEFI System Resource Table device (updated via NVRAM)
│     Current version:    196864
│     Vendor:             DMI:American Megatrends Inc.
│     Update State:       Success
│     GUID:               f5536e63-e4c0-4e0d-84d4-e8e152b1ba65
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│     Device Requests:    • Message
│   
└─UEFI Platform Key:
      Device ID:          6924110cde4fa051bfdc600a60620dc7aa9d3c6a
      Summary:            Platform Key
      GUID:               8210c506-95e1-5603-bf2a-9ca139b25935 ← UEFI\CRT_8C1662434A7D021E64E40204EE137B7F3FA41C87
    
────────────────────────────────────────────────
Devices that have been updated successfully:
 • UEFI dbx (244 → 371)
```

</details>

**Additional questions**

- Operating system and version: CachyOS
- Have you tried rebooting? Yes
- Is this a regression? Not sure, looks a bit like issue #7050

superm1, 2024-10-24:
Can you please modify the daemon's `ExecStart` command to have a `-vv`and then reproduce and share the log?

RayJW, 2024-10-24:
> Can you please modify the daemon's `ExecStart` command to have a `-vv`and then reproduce and share the journal for the unit?

[This](https://github.com/user-attachments/files/17508585/tmp.log) is all the output, after that nothing else is happening. I have to attach the log as a file. Otherwise, I can't post the comment due to the 65536-character limit.

I waited for a few minutes, but nothing further was output. Then, as you can see, `sudo systemctl restart fwupd.service` takes 3 minutes for it to be killed afterwards.

superm1, 2024-10-24:
Ah yup I see that now:

```
Oct 24 16:16:31 hostname fwupd[34736]:   OpenFlags:            none
Oct 24 16:19:10 hostname systemd[1]: Stopping Firmware update daemon...
Oct 24 16:22:10 hostname systemd[1]: fwupd.service: State 'stop-sigterm' timed out. Killing.
```

I think we probably want to know what it's spinning on.  Can you attach gdb to the process and get a backtrace?

@hughsie any other ideas?

superm1, 2024-10-24:
> I only have surface level debugging knowledge, but I will look into it and try to deliver a backtrace some time soon.

Here's some basic info on how to attach to a process: https://stackoverflow.com/questions/2308653/can-i-use-gdb-to-debug-a-running-process

You can then use `bt` and if there are debugging symbols in your build we'll get the interesting stuff of what's happening.

If there are no debugging symbols, you'll need to get them from a debug symbol server if your distro provides one or recompile to get them.

hughsie, 2024-10-24:
I suspect it's causing a kernel lockup when reading the oprom data.

superm1, 2024-10-24:
> I suspect it's causing a kernel lockup when reading the oprom data.

Well not lockup persay, the system is still working. But maybe reading oprom data is not returning anything from the kernel you mean.

RayJW, 2024-10-25:
> > I only have surface level debugging knowledge, but I will look into it and try to deliver a backtrace some time soon.
> 
> Here's some basic info on how to attach to a process: [stackoverflow.com/questions/2308653/can-i-use-gdb-to-debug-a-running-process](https://stackoverflow.com/questions/2308653/can-i-use-gdb-to-debug-a-running-process)
> 
> You can then use `bt` and if there are debugging symbols in your build we'll get the interesting stuff of what's happening.
> 
> If there are no debugging symbols, you'll need to get them from a debug symbol server if your distro provides one or recompile to get them.

Sadly, I don't think this is not very helpful. I get no output for `bt` until I kill the service. This is the output (after continue I ran the attestation and then ran `bt`, if that is the wrong way to go please let me know).

```shell
(gdb) continue
(gdb) bt
(gdb) 
Thread 1 "fwupd" received signal SIGTERM, Terminated.
0x00007a74078fa89c in read () from target:/usr/lib/libc.so.6
(gdb) continue
Continuing.

Thread 3 "pool-spawner" received signal SIGCONT, Continued.
[Switching to Thread 0x7a74017ff6c0 (LWP 4145)]
0x00007a740790515d in syscall () from target:/usr/lib/libc.so.6
(gdb) bt
#0  0x00007a740790515d in syscall () from target:/usr/lib/libc.so.6
#1  0x00007a7407cd3597 in g_cond_wait () from target:/usr/lib/libglib-2.0.so.0
#2  0x00007a7407c5c48c in g_async_queue_pop_unlocked () from target:/usr/lib/libglib-2.0.so.0
#3  0x00007a7407cd9bb6 in ?? () from target:/usr/lib/libglib-2.0.so.0
#4  0x00007a7407cd73a5 in ?? () from target:/usr/lib/libglib-2.0.so.0
#5  0x00007a7407886386 in ?? () from target:/usr/lib/libc.so.6
#6  0x00007a7407907974 in clone () from target:/usr/lib/libc.so.6
```

superm1, 2024-10-25:
OK, so it's hanging on reading the stream.  I think @hughsie is right the kernel is not returning data for some reason.

Can you sprinkle some printk's all around this function in the kernel?

https://github.com/torvalds/linux/blob/v6.12-rc4/drivers/pci/pci-sysfs.c#L1295

That should hopefully gives us some hints which part of the process is failing when fwupd tries to read it.

superm1, 2024-10-25:
BTW do you have resizable BAR enabled?

RayJW, 2024-10-25:
> OK, so it's hanging on reading the stream. I think @hughsie is right the kernel is not returning data for some reason.
> 
> Can you sprinkle some printk's all around this function in the kernel?
> 
> [torvalds/linux@`v6.12-rc4`/drivers/pci/pci-sysfs.c#L1295](https://github.com/torvalds/linux/blob/v6.12-rc4/drivers/pci/pci-sysfs.c?rgh-link-date=2024-10-25T14%3A12%3A13Z#L1295)
> 
> That should hopefully gives us some hints which part of the process is failing when fwupd tries to read it.

Oh, now we're getting into recompiling the kernel territory haha, I'm sorry for my slow response times. While I do have experience coding, this is not everyday business for me, so it'll take some time to get that all running properly :D

> BTW do you have resizable BAR enabled?

I do indeed, would you like me to turn it off and see what that does?

superm1, 2024-10-25:
Heh yup rabbit holes and all 😅.

Generally you do want it enabled for performance reasons.

Sure, doesn't hurt to check as a data point if that's what causes the issue.

RayJW, 2024-10-25:
> Heh yup rabbit holes and all 😅.

Been there a lot of times recently :P

> Generally you do want it enabled for performance reasons.

Yep, also why I have it enabled. I disabled it and confirmed that the bar size is 256M again with `sudo dmesg | grep -i bar`. Still the same problem and the backtrace looks the same, it seems.

> Sure, doesn't hurt to check as a data point if that's what causes the issue.

```shell
(gdb) continue
Continuing.
^C
Thread 1 "fwupd" received signal SIGINT, Interrupt.
0x0000797887a8f89c in read () from target:/usr/lib/libc.so.6
(gdb) bt
#0  0x0000797887a8f89c in read () from target:/usr/lib/libc.so.6
#1  0x0000797887d232ea in ?? () from target:/usr/lib/libgio-2.0.so.0
#2  0x0000797887c595f5 in g_input_stream_read () from target:/usr/lib/libgio-2.0.so.0
#3  0x0000797888177e6d in ?? () from target:/usr/lib/fwupd-2.0.1/libfwupdplugin.so
#4  0x0000797888127bde in fu_device_read_firmware () from target:/usr/lib/fwupd-2.0.1/libfwupdplugin.so
#5  0x0000797888168944 in fu_plugin_runner_verify () from target:/usr/lib/fwupd-2.0.1/libfwupdplugin.so
#6  0x00007978882a81cc in fu_engine_verify () from target:/usr/lib/fwupd-2.0.1/libfwupdengine.so
#7  0x00005c2d5cc424b5 in ?? ()
#8  0x00005c2d5cc42bcf in ?? ()
#9  0x0000797887cf2cf6 in ?? () from target:/usr/lib/libgio-2.0.so.0
#10 0x0000797887e2c52a in ?? () from target:/usr/lib/libglib-2.0.so.0
#11 0x0000797887e9ff30 in ?? () from target:/usr/lib/libglib-2.0.so.0
#12 0x0000797887e34477 in g_main_loop_run () from target:/usr/lib/libglib-2.0.so.0
#13 0x00005c2d5cc44f24 in fu_daemon_start ()
#14 0x00005c2d5cc3aa1b in main ()
```

hughsie, 2024-11-15:
@RayJW does the hughsie/optionrom-with-timeout branch fix things?

RayJW, 2024-11-15:
> @RayJW does the hughsie/optionrom-with-timeout branch fix things?

I will check ASAP, I guess it's just a matter of building that branch and then using the resulting binary to run `sudo ./fwupdmgr verify c66fc19d-6d31-5889-8f5c-e34e11caf8de` or is there anything else you would like me to do?

Also, I'm sorry. It wasn't my intention to abandon this issue. I ran into trouble building the Linux kernel for my system and haven't had time since then to get that fixed to provide the logs required because it seems this fix would merely solve the symptom of the service locking up, but verification would still fail for this particular device, right?

superm1, 2024-11-15:
> @hughsie I will check ASAP, I guess it's just a matter of building that branch and then using the resulting binary to run sudo ./fwupdmgr verify c66fc19d-6d31-5889-8f5c-e34e11caf8de or is there anything else you would like me to do?

Here are the build instructions: https://fwupd.github.io/libfwupdplugin/building.html
You'll need to launch the daemon and client in separate terminal tabs, but essentially yes.

RayJW, 2024-11-15:
I think this seems to fix the issue in the sense that it doesn't lock up the daemon any more. Here are the logs 
[fwupd.log](https://github.com/user-attachments/files/17779279/fwupd.log) [fwupdmgr.log](https://github.com/user-attachments/files/17779280/fwupdmgr.log), I hope that is of help :)

RayJW, 2024-11-22:
> Can you sprinkle some printk's all around this function in the kernel? [...]

@superm1 I am confused. First off, good news, I finally got around to fix the compilation errors and I very professionally expanded `pci_read_rom` like this:

```C
static ssize_t pci_read_rom(struct file *filp, struct kobject *kobj,
			    struct bin_attribute *bin_attr, char *buf,
			    loff_t off, size_t count)
{
	pr_notice("Reading from ROM now\n");
	struct pci_dev *pdev = to_pci_dev(kobj_to_dev(kobj));
	void __iomem *rom;
	size_t size;

	if (!pdev->rom_attr_enabled) {
		pr_notice("!pdev\n");
		return -EINVAL;
	}

	pr_notice("Assigning rom now\n");
	rom = pci_map_rom(pdev, &size);	/* size starts out as PCI window size */
	pr_notice("Assigned rom\n");
	if (!rom || !size) {
		pr_notice("!rom or !size\n");
		return -EIO;
	}

	if (off >= size) {
		pr_notice("off >= size\n");
		count = 0;
	} else {
		if (off + count > size) {
			pr_notice("off + count > size\n");
			count = size - off;
		}

		pr_notice("memcpy_fromio\n");
		memcpy_fromio(buf, rom + off, count);
		pr_notice("Finished memcpy\n");
	}
	pr_notice("Unmapping ROM\n");
	pci_unmap_rom(pdev, rom);
	pr_notice("Finished unmapping\n");

	return count;
}
```

However, after booting into the kernel I watched the terminal with `sudo dmesg -w -l 7` which should be more than enough for notices and … nothing. None of my statements show up, not even “Reading from ROM now”. I tried using the terminal command `fwupdmgr verify` and the GNOME Firmware GUI. Same results. So either I did something severely wrong, or the problems occur somewhere earlier than that. This is the terminal output:

<details>
<summary>`sudo dmesg -w -l 7` Output</summary>
```shell
[    0.000000] e820: update [mem 0xbe97a018-0xbe988067] usable ==> usable
[    0.000000] e820: update [mem 0xbe95b018-0xbe979257] usable ==> usable
[    0.000000] e820: remove [mem 0xf0000000-0xf7ffffff] reserved
[    0.000000] e820: remove [mem 0xfd200000-0xfd2fffff] reserved
[    0.000000] e820: remove [mem 0xfd400000-0xfd5fffff] reserved
[    0.000000] e820: remove [mem 0xfeb80000-0xfec01fff] reserved
[    0.000000] e820: remove [mem 0xff000000-0xffffffff] reserved
[    0.000011] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000012] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000469] e820: update [mem 0xcab70000-0xcab7ffff] usable ==> reserved
[    0.000473] e820: update [mem 0xd0000000-0xffffffff] usable ==> reserved
[    0.003982] e820: update [mem 0xc9238000-0xc9238fff] usable ==> reserved
[    0.044560] e820: update [mem 0xc6a49000-0xc6b3cfff] usable ==> reserved
[    0.050265] pcpu-alloc: s237568 r8192 d28672 u524288 alloc=1*2097152
[    0.050267] pcpu-alloc: [0] 00 01 02 03 [0] 04 05 06 07 
[    0.050270] pcpu-alloc: [0] 08 09 10 11 [0] 12 13 14 15 
[    0.050273] pcpu-alloc: [0] 16 17 18 19 [0] 20 21 22 23 
[    0.538311] libata version 3.00 loaded.
[    0.542722] PCI: pci_cache_line_size set to 64 bytes
[    0.543473] e820: reserve RAM buffer [mem 0x09d1f000-0x0bffffff]
[    0.543475] e820: reserve RAM buffer [mem 0x0a200000-0x0bffffff]
[    0.543476] e820: reserve RAM buffer [mem 0xbe95b018-0xbfffffff]
[    0.543478] e820: reserve RAM buffer [mem 0xbe97a018-0xbfffffff]
[    0.543479] e820: reserve RAM buffer [mem 0xc347e000-0xc3ffffff]
[    0.543481] e820: reserve RAM buffer [mem 0xc6a49000-0xc7ffffff]
[    0.543482] e820: reserve RAM buffer [mem 0xc9238000-0xcbffffff]
[    0.543484] e820: reserve RAM buffer [mem 0xca580000-0xcbffffff]
[    0.543486] e820: reserve RAM buffer [mem 0xcd000000-0xcfffffff]
[    0.543487] e820: reserve RAM buffer [mem 0x82f300000-0x82fffffff]
[    0.546678] pnp 00:04: [dma 0 disabled]
[    0.696670] ahci 0000:07:00.0: version 3.0
[    1.185212] sd 1:0:0:0: [sda] Mode Sense: 00 3a 00 00
[    1.185668] sd 5:0:0:0: [sdb] Mode Sense: 00 3a 00 00
[    1.288042]   with arguments:
[    1.288043]     /init
[    1.288045]     splash
[    1.288046]     splash
[    1.288047]   with environment:
[    1.288048]     HOME=/
[    1.288049]     TERM=linux
[    1.288051]     cryptdevice=UUID=9710768e-c8e7-48b7-81e0-ee0a860a5d1a:luks-9710768e-c8e7-48b7-81e0-ee0a860a5d1a
[   18.514353] ieee80211 phy0: Selected rate control algorithm 'iwl-mvm-rs'
[   21.712058] rfkill: input handler disabled
[   29.249477] rfkill: input handler enabled
[   30.174824] rfkill: input handler disabled
```
</details>

I'm sorry I have no promising results to show for after the long wait, and I'm not quite sure where else to continue. If there is anything else you would like me to try, please let me know and I'll gladly try to help.

superm1, 2024-11-22:
To make sure you didn't have an error with the patch applying in the build process can you add another print right when amdgpu loads?

RayJW, 2024-11-22:
> To make sure you didn't have an error with the patch applying in the build process can you add another print right when amdgpu loads? 

I will do that, but I won't be at my desktop for at least a week so it might not be before next weekend :)

RayJW, 2024-12-04:
> To make sure you didn't have an error with the patch applying in the build process can you add another print right when amdgpu loads?

I did as you advised me to, as you can see from this `sudo dmesg -l 5` output, it doesn't seem to be a problem compiling the kernel.

```shell
[    0.000000] Linux version 6.12.1-debug (operator@cyberdyne) (gcc (GCC) 14.2.1 20240910, GNU ld (GNU Binutils) 2.43.0) #1 SMP PREEMPT_DYNAMIC Wed Dec  4 01:08:03 CET 2024
[    0.000000] random: crng init done
[    0.050348] Kernel command line: quiet zswap.enabled=0 nowatchdog splash splash rw rootflags=subvol=/@ cryptdevice=UUID=9710768e-c8e7-48b7-81e0-ee0a860a5d1a:luks-9710768e-c8e7-48b7-81e0-ee0a860a5d1a root=/dev/mapper/luks-9710768e-c8e7-48b7-81e0-ee0a860a5d1a initrd=\initramfs-6.12.1-debug.img
[    0.050428] Unknown kernel command line parameters "splash splash cryptdevice=UUID=9710768e-c8e7-48b7-81e0-ee0a860a5d1a:luks-9710768e-c8e7-48b7-81e0-ee0a860a5d1a", will be passed to user space.
[    0.471310] audit: type=2000 audit(1733274516.179:1): state=initialized audit_enabled=0 res=1
[    0.492065] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
[    0.535681] SCSI subsystem initialized
[    0.543453] VFS: Disk quotas dquot_6.6.0
[    0.604922] Initialise system trusted keyrings
[    0.604933] Key type blacklist registered
[    0.605191] integrity: Platform Keyring initialized
[    0.605195] integrity: Machine keyring initialized
[    0.615906] Key type asymmetric registered
[    0.615908] Asymmetric key parser 'x509' registered
[    0.899743] Loading compiled-in X.509 certificates
[    0.902517] Loaded X.509 cert 'Build time autogenerated kernel key: 457b61a328dd598a9a0604ffc364c27b0f07f9c7'
[    0.905017] Key type .fscrypt registered
[    0.905019] Key type fscrypt-provisioning registered
[    0.907879] integrity: Loaded X.509 cert 'Database Key: 00827627636370dc6b464489d9c61d8270'
[    0.907899] integrity: Loaded X.509 cert 'Microsoft Corporation UEFI CA 2011: 13adbf4309bd82709c8cd54f316ed522988a1bd4'
[    0.907914] integrity: Loaded X.509 cert 'Microsoft Windows Production PCA 2011: a92902398e16c49778cd90f99e4f9ae17c55af53'
[    1.174072] scsi 1:0:0:0: Direct-Access     ATA      Samsung SSD 850  3B6Q PQ: 0 ANSI: 5
[    1.174301] sd 1:0:0:0: [sda] 976773168 512-byte logical blocks: (500 GB/466 GiB)
[    1.174309] sd 1:0:0:0: [sda] Write Protect is off
[    1.174322] sd 1:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    1.174323] scsi 5:0:0:0: Direct-Access     ATA      WDC WD4004FZWX-0 0A81 PQ: 0 ANSI: 5
[    1.174594] sd 5:0:0:0: [sdb] 7814037168 512-byte logical blocks: (4.00 TB/3.64 TiB)
[    1.174599] sd 5:0:0:0: [sdb] 4096-byte physical blocks
[    1.174607] sd 5:0:0:0: [sdb] Write Protect is off
[    1.174620] sd 5:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    1.183227] sd 1:0:0:0: [sda] supports TCG Opal
[    1.183229] sd 1:0:0:0: [sda] Attached SCSI disk
[    1.222502] sd 5:0:0:0: [sdb] Attached SCSI disk
[    1.488443] ccp 0000:0d:00.1: ccp: unable to access the device: you might be running a broken BIOS.
[    1.488532] ccp 0000:0d:00.1: psp enabled
[    1.971434] usb 3-6: config 1 has an invalid interface number: 2 but max is 1
[    1.971437] usb 3-6: config 1 has no interface number 1
[    5.609062] amdgpu: Little test notice :)
[    8.029191] Key type trusted registered
[    8.158650] Key type encrypted registered
```

My test notice shows up as expected, but still, running `fwupdmgr verify c66fc19d-6d31-5889-8f5c-e34e11caf8de` or verify in the GNOME Firmware app does not trigger `pci_read_rom` at all, it seems.

superm1, 2024-12-04:
Is the dGPU actively in use?  Or could it be in runtime PM?  You can look at this to find out (update to match your device though).

```sh
❯ cat /sys/class/drm/card0//device/power/runtime_status 
suspended
```

If it's `suspended` can you try to wake it using a tool like `nvtop`?  Does reading the attribute work after that?

RayJW, 2024-12-04:
> Is the dGPU actively in use? Or could it be in runtime PM? You can look at this to find out (update to match your device though).
> 
> ```shell
> ❯ cat /sys/class/drm/card0//device/power/runtime_status 
> suspended
> ```
> 
> If it's `suspended` can you try to wake it using a tool like `nvtop`? Does reading the attribute work after that?

Nope, I checked, and it says `active` the whole time before, during, and after the verification attempt. Just to make sure, I did leave `amdgpu_top` running for another attempt though and still the same.

superm1, 2024-12-07:
@RayJW can you please test https://github.com/fwupd/fwupd/pull/8216

RayJW, 2024-12-09:
> @RayJW can you please test #8216

While this does fix the timeout issue, it doesn't successfully verify the hash yet. Here is the 
[fwupd.log](https://github.com/user-attachments/files/18065359/fwupd.log), and this is the output for the verification attempt.

```shell
❯ fwupdmgr verify c66fc19d-6d31-5889-8f5c-e34e11caf8de    
WARNING: This package has not been validated, it may not work properly.
Selected device: AMD Radeon RX 6950 XT
failed to verify AMD Radeon RX 6950 XT: failed to read firmware: dumping firmware is not supported by device
```

superm1, 2024-12-09:
Can you try just the one commit?  Drop @hughsie's second "f" commit.

RayJW, 2024-12-09:
> Can you try just the one commit? Drop @hughsie's second "f" commit.

I think, this seems to work? Not sure, my uneducated guess would be that everything seems to work fine now except that there is no known good hash for this firmware.

[fwupd.log](https://github.com/user-attachments/files/18065856/fwupd.log)

```shell
❯ fwupdmgr verify c66fc19d-6d31-5889-8f5c-e34e11caf8de    
WARNING: This package has not been validated, it may not work properly.
Selected device: AMD Radeon RX 6950 XT
failed to verify AMD Radeon RX 6950 XT: failed to find /home/operator/Documents/Projects/GitHub/fwupd/venv/bin/../dist/lib/fwupd/verify/21ba7fbcdfb80c824895aad926418475726bcc79.xml
```

closed 2024-12-09 by superm1