fwupd/fwupd #7967: UEFI F/W update failed on Thinkpad p480

adaniline, 2024-10-25:
**Describe the bug**
I have the same issue as described in #6023 with Thinkpad t480. When I installed Arch Linux, I followed an advice in the installation guide and set the logical block size of my NVE disk to 4096 for better performance - I guess this is related.
I think that `fwupd-fw` should allow configuring block size in the EFI entry options to bypass issues like this.I think that fwupd-fw should allow configuring block size in the EFI entry options to bypass issues like this.

```
fwupd-efi version 1.7
WARNING: Failed to find 'HD(1,GPTd8fa1db7-eee5-47e4-a49c-94069f529dc2)/\EFI\arch\fw\fwupd-66d6a3ef-a771-4302-9cd0-d062c79c5ef2.cap' DevicePath
WARNING: Could not locate device handle: Unsupported
```

**Steps to Reproduce**
- `fwupdmgr refresh`
- `fwupdmgr get-updates`
- `fwupdmgr update`
- Reboot when prompted.

**Expected behavior**
The cap files should be found and the system should be updated.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version

compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd         2.0.1
runtime   org.kernel                    6.11.5-arch1-1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices

LENOVO 20L5400NCD
│
├─Unknown Device:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    07
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               8b72a10c-1279-5f8e-a28a-34fb11a58240 ← PCI\VEN_8086&DEV_5917
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│   
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f6
│     Vendor:             Intel
│     GUIDs:              561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          e0057eb24a17998da8f80a0260f0c5b50656e644
[... 218 lines cut ...]
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 14 / 03 05 06 00 0b 00 / 7f ff 04 00
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b 91 af 62 59 56 44 9f 41 a7 b9 1f 4f 89 2a b0 f6 / 7f ff 04 00
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
      dp: 03 0a 24 00 d2 38 78 bc 82 0f 60 4d 83 16 c0 68 ee 79 d2 5b ae a2 09 0a df de 21 4e 8b 3a 5e 47 18 56 a3 54 / 7f ff 04 00
```

```shell
efivar -l | grep fw
<empty output>
```

```shell
tree /boot
zsh: command not found: tree
```

**Additional questions**

- Operating system and version: Arch `Linux thinkpad 6.11.5-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 22 Oct 2024 18:31:38 +0000 x86_64 GNU/Linux`
- Have you tried rebooting? yes
- Is this a regression? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled? no
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? no

hughsie, 2024-10-25:
What's the output of `udisksctl dump`?

adaniline, 2024-10-25:
```shell
udisksctl dum

/org/freedesktop/UDisks2/Manager:
  org.freedesktop.UDisks2.Manager:
    DefaultEncryptionType:      luks2
    SupportedEncryptionTypes:   luks1
                                luks2
    SupportedFilesystems:       ext2
                                ext3
                                ext4
                                xfs
                                vfat
                                ntfs
                                f2fs
                                nilfs2
                                exfat
                                btrfs
                                udf
                                swap
    Version:                    2.10.1
  org.freedesktop.UDisks2.Manager.NVMe:
    HostID:             
    HostNQN:            nqn.2014-08.org.nvmexpress:uuid:1d442401-5756-11cb-9b01-cd107b747771

/org/freedesktop/UDisks2/block_devices/nvme0n1:
  org.freedesktop.UDisks2.Block:
    Configuration:              []
    CryptoBackingDevice:        '/'
    Device:                     /dev/nvme0n1
    DeviceNumber:               66304
    Drive:                      '/org/freedesktop/UDisks2/drives/TOSHIBA_RC100_Z82PA0B7P46N'
    HintAuto:                   false
    HintIconName:               
    HintIgnore:                 false
    HintName:                   
    HintPartitionable:          true
    HintSymbolicIconName:       
    HintSystem:                 true
    Id:                         by-id-nvme-TOSHIBA-RC100_Z82PA0B7P46N
    IdLabel:                    
    IdType:                     
    IdUUID:                     
    IdUsage:                    
    IdVersion:                  
    MDRaid:                     '/'
    MDRaidMember:               '/'
    PreferredDevice:            /dev/nvme0n1
    ReadOnly:                   false
    Size:                       480103981056
    Symlinks:                   /dev/disk/by-diskseq/1
                                /dev/disk/by-id/nvme-TOSHIBA-RC100_Z82PA0B7P46N
                                /dev/disk/by-id/nvme-TOSHIBA-RC100_Z82PA0B7P46N_1
                                /dev/disk/by-id/nvme-eui.00080d040052b990
                                /dev/disk/by-path/pci-0000:02:00.0-nvme-1
    UserspaceMountOptions:      
  org.freedesktop.UDisks2.NVMe.Namespace:
    EUI64:                      00080d040052b990
    FormatPercentRemaining:     -1
    FormattedLBASize:           (4096, 0, 0x01)
    LBAFormats:                 [(4096, 0, 0x01), (512, 0, 0x04)]
    NGUID:                      
    NSID:                       1
    NamespaceCapacity:          117212886
    NamespaceSize:              117212886
    NamespaceUtilization:       117212886
    UUID:                       
    WWN:                        eui.00080d040052b990
  org.freedesktop.UDisks2.PartitionTable:
    Partitions:         /org/freedesktop/UDisks2/block_devices/nvme0n1p2
[... 294 lines cut ...]
    Serial:                     Z82PA0B7P46N
    SiblingId:                  
    Size:                       0
    SortKey:                    00coldplug/00fixed/nvme0
    TimeDetected:               1729862437861443
    TimeMediaDetected:          1729862437861443
    Vendor:                     
    WWN:                        
  org.freedesktop.UDisks2.NVMe.Controller:
    ControllerID:                       0
    FGUID:                              
    NVMeRevision:                       1.2.1
    SanitizePercentRemaining:           -1
    SanitizeStatus:                     
    SmartCriticalWarning:               
    SmartPowerOnHours:                  7129
    SmartSelftestPercentRemaining:      -1
    SmartSelftestStatus:                success
    SmartTemperature:                   307
    SmartUpdated:                       1729862437
    State:                              live
    SubsystemNQN:                       nqn.2017-03.jp.co.toshiba:TOSHIBA-RC100:Z82PA0B7P46N
    UnallocatedCapacity:                0

```

seblu, 2024-10-26:
Very similar issue  on my new x1 carbon gen12 with 4k sector size.

tblancher, 2024-11-13:
**Describe the bug**
I'm getting similar behavior on my X1 Carbon 11th Gen:

Same exact WARNINGs as original issue poster, though I didn't get a copy of the error output above the Lenovo UEFI BIOS splash screen.

**Steps to Reproduce**
Same steps as above

**Expected behavior**
Expected ThinkPad to proceed with the firmware upgrade

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
Idle…: 0%
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.11.7-zen1-1-zen
runtime   org.freedesktop.fwupd         2.0.1

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

pikaur (AUR helper, drop-in replacement unifying pacman and makepkg)

<summary>**fwupd device information**</summary>

<details>
```shell
fwupdmgr get-devices --show-all-devices
Idle…: 0%
LENOVO 21HMCTO1WW
│
├─Unknown Device:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    04
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               0e8b1f1a-8c2b-53b6-89d7-d1a05dbc039d ← PCI\VEN_8086&DEV_A7A0
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
│   
├─Unknown Device:
│     Device ID:          6b5a06645e27c4e94b6e67d5185c46d11ec9ad67
│     GUID:               c96751f4-16e8-59ea-975d-69d3dac88d7d ← GPIO\ID_INTC1055:00
│     Device Flags:       • Can tag for emulation
│   
├─13th Gen Intel Core™ i7-1370P:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00004123
│     Vendor:             Intel
│     GUIDs:              c3f97ae6-7da2-54cc-ad26-3225f9517203 ← CPUID\PRO_0&FAM_06&MOD_BA
│                         557a4f44-6981-5889-8676-684b9727b697 ← CPUID\PRO_0&FAM_06&MOD_BA&STP_2
│     Device Flags:       • Internal device
│   
├─ATNA40YK15-0:
│     Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
│     Vendor:             PNP:SDC
│     GUID:               c87d4eb6-c987-518c-9aa1-b64ea47761a3 ← DRM\VEN_SDC&DEV_4193
│     Device Flags:       • Internal device
[... 746 lines cut ...]
    WWN:                        
  org.freedesktop.UDisks2.NVMe.Controller:
    ControllerID:                       8224
    FGUID:                              
    NVMeRevision:                       1.4
    SanitizePercentRemaining:           -1
    SanitizeStatus:                     success
    SmartCriticalWarning:               
    SmartPowerOnHours:                  404
    SmartSelftestPercentRemaining:      -1
    SmartSelftestStatus:                success
    SmartTemperature:                   308
    SmartUpdated:                       1731506798
    State:                              live
    SubsystemNQN:                       nqn.2018-01.com.wdc:nguid:E8238FA6BF53-0001-001B448B4A5F14C7
    UnallocatedCapacity:                0
```
**Additional questions**

- Operating system and version:  Arch Linux `Linux fluorine 6.11.7-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Fri, 08 Nov 2024 17:57:58 +0000 x86_64 GNU/Linux`
- Have you tried rebooting?  Yes
- Is this a regression?  Yes
- Are you using an NVMe disk?  Yes
- Is secure boot enabled?  Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?  Usually yes, but in order for fwupd to work I need to disable this (discovered this the last time I tried to upgrade, see https://bbs.archlinux.org/viewtopic.php?id=298467 where I discovered the solution to this)

hughsie, 2024-11-15:
Can someone affected please run `sudo /usr/libexec/fwupd/fwupd -vv` and then schedule a firmware update please. I'm specifically looking for the line `failed to get volume block size, falling back to...` with some kind of error code.

hughsie, 2024-11-15:
Ohh and I also need `sudo fwupdtool esp-list --json` from anyone affected please.

hughsie, 2024-11-15:
This is certainly suspicious:

    Boot0000* Arch Linux			HD(1,GPT,d8fa1db7-eee5-47e4-a49c-94069f529dc2,0x100,0x80000)
    Boot0001* Linux-Firmware-Updater	HD(1,GPT,d8fa1db7-eee5-47e4-a49c-94069f529dc2,0x800,0x400000)

Does the arch linux installer use efibootmgr to install the new boot target? FuEfiHardDriveDevicePath copies the logic from libefivar (that efibootmgr uses) which also uses the BLKSSZGET ioctl to get the logical block size. It looks like fwupd is somehow detecting the logical block size is 512 rather than 4096 -- which would explain the 8x values above. The `sudo fwupdtool esp-list --json` logs might explain things some more.

adaniline, 2024-11-16:
> This is certainly suspicious:
>
> ```
> Boot0000* Arch Linux			HD(1,GPT,d8fa1db7-eee5-47e4-a49c-94069f529dc2,0x100,0x80000)
> Boot0001* Linux-Firmware-Updater [...]

Yes, I use EFI boot stub - I change boot order in BIOS to boot into `fwupd`

adaniline, 2024-11-16:
> sudo fwupdtool esp-list --json

```
{
  "Volumes" : [
    {
      "IsMounted" : true,
      "IsEncrypted" : false,
      "Size" : 2147483648,
      "BlockSize" : 4096,
      "MountPoint" : "/boot",
      "PartitionKind" : "c12a7328-f81f-11d2-ba4b-00a0c93ec93b",
      "PartitionSize" : 2147483648,
      "PartitionOffset" : 1048576,
      "PartitionNumber" : 1,
      "PartitionUuid" : "d8fa1db7-eee5-47e4-a49c-94069f529dc2"
    }
  ]
}
```

adaniline, 2024-11-16:
> sudo /usr/libexec/fwupd/fwupd -vv

```
<snipped>
0x0f00 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0f20 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0f40 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0f60 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0f80 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0fa0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0fc0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0x0fe0 | 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
13:16:28.472 FuPluginNvme         fawr: 1, nr fw slots: 1, slot1 r/o: 0
13:16:28.472 FuPluginNvme         invalid component ID, skipping
13:16:28.472 FuPluginNvme         no vendor GUID, falling back to mn
13:16:28.472 FuPlugin             emit added from nvme: 03281da317dccd2b18de2bd1cc70a782df40ed7e
13:16:28.473 FuEngine             56e9aca5-1fa8-54e8-9963-bb88d3d0d107 was not found: indexed string '56e9aca5-1fa8-54e8-9963-bb88d3d0d107' was unfound
13:16:28.473 FuEngine             b598f626-ec06-5041-94bf-5791b3a21f44 was not found: indexed string 'b598f626-ec06-5041-94bf-5791b3a21f44' was unfound
13:16:28.473 FuEngine             66224d7a-6fc5-5f31-9b58-47968bbc6dc0 was not found: indexed string '66224d7a-6fc5-5f31-9b58-47968bbc6dc0' was unfound
13:16:28.473 FuPlugin             fu_plugin_device_registered(dell_dock)
13:16:28.473 FuPlugin             fu_plugin_device_registered(dell_k2)
13:16:28.473 FuPlugin             fu_plugin_device_registered(intel_usb4)
13:16:28.473 FuPlugin             fu_plugin_device_registered(lenovo_thinklmi)
13:16:28.473 FuPlugin             fu_plugin_device_registered(logitech_tap)
13:16:28.473 FuPlugin             fu_plugin_device_registered(pci_bcr)
<snipped>
[fwupd.log](https://github.com/user-attachments/files/17784833/fwupd.log)

```

tblancher, 2024-11-29:
I tried gathering the data @hughsie was seeking, but the firmware upgrade worked just fine for me this time.  My _/var/log/pacman.log_ (filtered for `fwupd`, 
[2024-11-29_pacman.log_grep_fwupd.log](https://github.com/user-attachments/files/17963290/2024-11-29_pacman.log_grep_fwupd.log)) shows I haven't updated the package since October 19;  it wasn't until November 13 when I made my post that I ran into the problem.

I gathered the output of `fwupd -vv` three times:  
[2024-11-29_fwupd-vv_pre_firmware_upgrade.log](https://github.com/user-attachments/files/17963297/2024-11-29_fwupd-vv_pre_firmware_upgrade.log) (before the first successful run), 
[2024-11-29_fwupd-vv_post-refresh_pre_firmware_upgrade.log](https://github.com/user-attachments/files/17963300/2024-11-29_fwupd-vv_post-refresh_pre_firmware_upgrade.log) (I had forgotten to run `fwupdmgr refresh` before the first update attempt;  this second attempt succeeded as well), and then 
[2024-11-29_fwupd-vv_post_firmware_upgrade.log](https://github.com/user-attachments/files/17963302/2024-11-29_fwupd-vv_post_firmware_upgrade.log)(after the final `fwupd update` had nothing more to upgrade).

I'm also attaching 
[2024-11-29_fwupdtool_esp-list_--json.json](https://github.com/user-attachments/files/17963304/2024-11-29_fwupdtool_esp-list_--json.json) for completeness.

I'm not sure what if anything I had done to correct this, or why this error seemed to be transient.

adaniline, 2024-11-30:
@hughsie 
> Can someone affected please run `sudo /usr/libexec/fwupd/fwupd -vv` and then schedule a firmware update please. I'm specifically looking for the line `failed to get volume block size, falling back to...` with some kind of error code.

Maybe I didn't understand what needs to be captured before, but I ran the command about and scheduled the update. Did not see `Failed to get block size` message, the log is attached. 

[fwupd.log](https://github.com/user-attachments/files/17963804/fwupd.log)

adaniline, 2024-11-30:
@hughsie I tried updating again, and it succeeded now. The only different is the kernal version:
```
❯ fwupdmgr --version
❯ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.fwupd-efi     1.7
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd         2.0.1
runtime   org.kernel                    6.12.1-arch1-1
```

hughsie, 2024-12-03:
I can't explain that then :) I'll close this, please re-open if it pops up again.

epixtm, 2024-12-08:
Don't mean to take over an issue, but I've been attempting to `fwupdmgr update` on my device aswell and have been encountering the same issue as this and #6023. My device is an X13 Gen 3 and my SSD also uses a 4K logical block size.

I did manage to firmware update when this drive was in another Thinkpad ~1 yr ago, but when I moved this drive to my new device `fwupd-efi` fails to get the `.cap` files.

I suspect the EFI boot entry sector size is still being incorrectly calculated as my entry looks something like this:
```
Boot0001* Linux-Firmware-Updater	HD(1,GPT,<uuid here>,0x800,0x200000)\EFI\systemd\fwupdx64.efi
```

I may be wrong however, as `systemd-boot`'s entry also has the same calculated offsets. Can try to provide some logs in a bit.

hughsie, 2024-12-09:
@epixtm also arch linux? If so, what kernel version?

epixtm, 2024-12-09:
@hughsie 
> @epixtm also arch linux? If so, what kernel version?

Yep, here's my `uname -a`:
```
Linux teyvat 6.12.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Fri, 06 Dec 2024 11:15:43 +0000 x86_64 GNU/Linux
```

If you want me to get any logs to help debug just let me know.

bnicolae, 2024-12-09:
I am also experiencing this issue. Same kernel version.

julianbrost, 2024-12-09:
Same issue for me, also on Arch. So far, there was some focus on the offsets used in the boot entry, but fwupd-efi itself starts just fine, however, what about the EFI variable that stores the path to the cap file? Is there some documentation of that format used for the contents of that variable? At least when I hexdump it, there are two parts that look an awful lot like the wrong offsets (i.e. calculated with 512B instead of 4K blocks).

julianbrost, 2024-12-09:
Ha! I found something very interesting! Turns out that yes, I was able to just install a firmware upgrade, but the relevant change was that I wanted to try debugging it and used `fwupdtool` instead of `fwupdmgr` (which seems to be a workaround for this issue).

After some playing around trying to reinstall the firmware update, this is what I found:

The `fwupdtool esp-list` command that was used earlier for debugging actually opens the partition device:

```
# strace -f -yy fwupdtool esp-list --json |& grep nvme                                                                                                                           
[pid  3826] openat(AT_FDCWD</root>, "/dev/nvme0n1p1", O_RDONLY <unfinished ...>
[pid  3826] <... openat resumed>)       = 14</dev/nvme0n1p1<block 259:1>>
[pid  3826] ioctl(14</dev/nvme0n1p1<block 259:1>>, BLKSSZGET <unfinished ...>
[pid  3826] close(14</dev/nvme0n1p1<block 259:1>> <unfinished ...>
```

However, when I attached `strace` to the regular `fwupd` service, this is what I saw while running `fwupdmgr reinstall`:

```
2405  openat(AT_FDCWD</>, "/dev/nvme0n1p1", O_RDONLY) = -1 EPERM (Operation not permitted)
```

Hence, I believe there's a good chance that it just falls back to 512B blocks and then uses wrong values later on.

I'm not sure about the cause, maybe some change to hardening settings in the systemd unit file?

> I can't explain that then :) I'll close this, please re-open if it pops up again.

> Can someone affected please run `sudo /usr/libexec/fwupd/fwupd -vv`

If my theory is correct and that EPERM is indeed due to restrictions from the service file, running it manually would bypass the restrictions and the problem disappears.

@hughsie So please re-open this :)

hughsie, 2024-12-09:
> openat(AT_FDCWD</>, "/dev/nvme0n1p1", O_RDONLY) = -1 EPERM (Operation not permitted)

Oh good debugging! Can you narrow it down to a specific hardening option in fwupd.service?

julianbrost, 2024-12-09:
Looks like this is indeed caused by systemd restrictions (I didn't yet confirm it with the full fwupd service because I don't want to reboot all the time and I don't how to unschedule an upgrade).

The block device type for my NVMe SSD is `blkext`

```console
# ls -l /dev/nvme0n1p1  
brw-rw---- 1 root disk 259, 1 Dec  9 21:13 /dev/nvme0n1p1
# grep 259 /proc/devices
259 blkext
```

Now `fwupd.service` only specifies `DeviceAllow=block-sd rw`. If try to replicate this with a minimal example using `systemd-run`, I indeed get a permission error:

```console
# systemd-run --pipe --property=DeviceAllow='block-sd rw' blockdev --getss /dev/nvme0n1p1
Running as unit: run-u108.service; invocation ID: 6afcc5cfe5024a75b6cbb61d4a108d9d
blockdev: cannot open /dev/nvme0n1p1: Operation not permitted
```

Once I add `DeviceAllow=block-blkext rw`, I can access the block size:

```console
# systemd-run --pipe --property=DeviceAllow='block-sd rw' --property=DeviceAllow='block-blkext rw' blockdev --getss /dev/nvme0n1p1
Running as unit: run-u109.service
4096
```

<details>
<summary>Full `systemctl cat fwupd.service` on my system in case that's relevant as it seems to be dynamically generated</summary>

```ini
# /usr/lib/systemd/system/fwupd.service
[Unit]
Description=Firmware update daemon
Documentation=https://fwupd.org/
After=dbus.service
Before=display-manager.service
ConditionVirtualization=!container

[Service]
Type=dbus
TimeoutSec=180
RuntimeDirectory=motd.d
RuntimeDirectoryPreserve=yes
BusName=org.freedesktop.fwupd
ExecStart=/usr/lib/fwupd/fwupd
KeyringMode=private
LockPersonality=yes
MemoryDenyWriteExecute=yes
NoNewPrivileges=no
PrivateDevices=no
PrivateTmp=true
ProtectClock=yes
ProtectControlGroups=yes
ProtectHome=yes
ProtectHostname=yes
ProtectKernelLogs=yes
ProtectKernelModules=yes
ProtectKernelTunables=yes
ProtectProc=invisible
ProtectSystem=full
RestrictNamespaces=yes
RestrictRealtime=yes
RestrictSUIDSGID=yes
SystemCallArchitectures=native
Environment="GLIBC_TUNABLES=glibc.cpu.hwcaps=SHSTK"
ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI -/boot/grub
ConfigurationDirectory=fwupd
StateDirectory=fwupd
CacheDirectory=fwupd
RestrictAddressFamilies=AF_NETLINK AF_UNIX AF_INET AF_INET6
DeviceAllow=block-sd rw
DeviceAllow=char-aux rw
DeviceAllow=char-cpu/* rw
DeviceAllow=char-drm rw
DeviceAllow=char-gpiochip rw
DeviceAllow=char-hidraw rw
DeviceAllow=char-usb rw
DeviceAllow=char-usb_device rw
DeviceAllow=char-mem rw
DeviceAllow=char-nvme rw
DeviceAllow=char-tpm rw
DeviceAllow=char-mei rw
SystemCallFilter=@basic-io @file-system @io-event @ipc @network-io @process @sync @signal @timer @chown ioctl uname sysinfo madvise mremap splice vmsplice copy_file_range @raw-io
Environment="FWUPD_SYSCALL_FILTER=systemd"
SystemCallErrorNumber=EPERM
```
</details>

superm1, 2024-12-09:
Nice job!  Mind a PR to fixup the hardening?

julianbrost, 2024-12-09:
I can do that, but I won't have the time to properly test it today, so it might take a day or two. Any preferences regarding whether to add `block-blkext` specifically or going for `block-*`?

Also, is there a trick how I can run `fwupdmgr reinstall ...` multiple times without having to reboot in between (currently it says "No releases found" on the second attempt and `get-devices` says "An update is in progress")?

superm1, 2024-12-09:
> I can do that, but I won't have the time to properly test it today, so it might take a day or two. 

Feel free to just mark it draft until you can test it.

> Any preferences regarding whether to add block-blkext specifically or going for block-*?

I don't feel strongly.

> Also, is there a trick how I can run fwupdmgr reinstall ... multiple times without having to reboot in between (currently it says "No releases found" on the second attempt and get-devices says "An update is in progress")?

You can axe the `pending.db` file in the `venv/` directory of your checkout.
You can also remove `BootNext-*` from `/sys/firmware/efi/efivarfs` if it complaints about that.

hughsie, 2024-12-09:
>     Any preferences regarding whether to add block-blkext specifically or going for block-*?

Ideally the former please; the less the better.

epixtm, 2024-12-09:
Just wanted to confirm that adding `DeviceAllow=block-blkext rw` to the service file, restarting the service, and running `fwupdmgr update` worked perfectly! Thanks @julianbrost for figuring it out so fast lol

closed 2024-12-10 by hughsie