fwupd/fwupd #7988: Can't compile gnome-firmware and kde discover

r-vdp, 2024-10-28:
**Describe the bug**

First reported in https://gitlab.gnome.org/World/gnome-firmware/-/issues/67, but maybe it's more appropriate here.

I included #7987 but I still get the same build errors:

```
       > [40/42] Compiling C object src/gnome-firmware.p/gfu-main.c.o
       > FAILED: src/gnome-firmware.p/gfu-main.c.o
       > gcc -Isrc/gnome-firmware.p -Isrc -I../src -I. -I.. -I/nix/store/82bbzzkj4ncqg1q08f9c6f91pam9kf0n-glib-2.80.4-dev/include -I/nix/store/82bbzzkj4ncqg1q08f9c6f91pam9kf0n-glib-2.80.4-dev/include/glib-2.0 -I/nix/store/fxnchbq09sq279ssqdsgzigsyvp6hz06-glib-2.80.4/lib/glib-2.0/include -I/nix/store/4r3fsd2r23ags1gdpwz2dsrws6jy63m1-gdk-pixbuf-2.42.12-dev/include/gdk-pixbuf-2.0 -I/nix/store/r577cvxqgb2acrcv9vhw03293mwgvrq9-gtk4-4.14.5-dev/include/gtk-4.0 -I/nix/store/s9gm9vyza9yvfsz9ag01s4hw018gjhqg-cairo-1.18.2-dev/include/cairo -I/nix/store/hl14fvf6jpfv30j4p8hqgzqpmrmx2m7b-freetype-2.13.3-dev/include/freetype2 -I/nix/store/hl14fvf6jpfv30j4p8hqgzqpmrmx2m7b-freetype-2.13.3-dev/include -I/nix/store/nr8688frwk2jv3c9qwiz0ywi1979li62-graphene-1.10.8-dev/include/graphene-1.0 -I/nix/store/yvs7cfic7w4d67fpwbhgn6sxqfpzb4a1-graphene-1.10.8/lib/graphene-1.0/include -I/nix/store/na5ax7xn886i8mh91bzzda8hma313hwh-pango-1.52.2-dev/include/pango-1.0 -I/nix/store/z0g6vlvsrnsmbfq4bsp0pa4j63a69jcv-harfbuzz-9.0.0-dev/include/harfbuzz -I/nix/store/jrr54zy7h7w1lhh3rk5q4zcbfwz57r3c-vulkan-headers-1.3.290.0/include -I/nix/store/9768n9gki54vzs62p802kgjph89sxs05-fwupd-2.0.1-dev/include/fwupd-3 -I/nix/store/n5hh60n4kfkz2fzxdfgls9smzra7ldny-libxmlb-0.3.19-dev/include/libxmlb-2 -I/nix/store/5gcgm6z9k9jx2rf7r1bwngf73ma1xvcn-libadwaita-1.5.3-dev/include/libadwaita-1 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Waggregate-return -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wempty-body -Wextra -Wformat=2 -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wno-cast-function-type -Wold-style-definition -Wno-deprecated-declarations -Wno-format-nonliteral -Woverride-init -Wpacked -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wwrite-strings -mfpmath=sse -msse -msse2 -mfpmath=sse -msse -msse2 '-DG_LOG_DOMAIN="Gfu"' -MD -MQ src/gnome-firmware.p/gfu-main.c.o -MF src/gnome-firmware.p/gfu-main.c.o.d -o src/gnome-firmware.p/gfu-main.c.o -c ../src/gfu-main.c
       > In file included from /nix/store/9768n9gki54vzs62p802kgjph89sxs05-fwupd-2.0.1-dev/include/fwupd-3/fwupd.h:15,
       >                  from ../src/gfu-common.h:11,
       >                  from ../src/gfu-main.c:16:
       > /nix/store/9768n9gki54vzs62p802kgjph89sxs05-fwupd-2.0.1-dev/include/fwupd-3/libfwupd/fwupd-codec.h:10:10: fatal error: json-glib/json-glib.h: No such file or directory
       >    10 | #include <json-glib/json-glib.h>
       >       |          ^~~~~~~~~~~~~~~~~~~~~~~
       > compilation terminated.
       > ninja: build stopped: subcommand failed.
```

**Additional questions**

- Operating system and version: NixOS
- Have you tried rebooting? N/A
- Is this a regression? Yes

superm1, 2024-10-28:
I think we need to split up fwupd vs gnome-firmware to start to isolate this.  I'm able so successfully build in the fwupd development env (commit 5fe5a9e685f62763a610a16dfb15a5edc9e1b67b) when I removed libjson-glib-dev from the host OS.

My build command was:
```
build-fwupd -Djson-glib:documentation=disabled -Dintrospection=disabled -Dlibjcat:introspection=false
```

Does that work for you?

superm1, 2024-10-28:
And for gnome-firmware I did this:
```
meson build -Dlibxmlb:gtkdoc=false -Djson-glib:documentation=disabled -Dfwupd:introspection=disabled -Dlibjcat:introspection=false
```

You can see it subprojected' a bunch of them.
```
  Subprojects
    fwupd                  : YES 1 warnings
    json-glib              : YES (from fwupd => libjcat)
    libjcat                : YES 1 warnings (from fwupd)
    libxmlb                : YES 1 warnings (from fwupd)
    passim                 : NO Dependency "libsoup-3.0" not found, tried pkgconfig and cmake
```

It built just fine though.

superm1, 2024-10-28:
One idea: try deleting your `subprojects/fwupd` directory in your gnome-firmware checkout and try again.

r-vdp, 2024-10-28:
So for context, I'm trying to package fwupd 2.0.1 for nixpkgs, and the builds are running in a sandboxed environment with all dependencies provided by the build environment, there's no internet connectivity during the build.

I've got fwupd 2.0.1 to build correctly, all tests pass, and the binary works perfectly. 

I can't manage to get projects that pull in libfwupd as a dependency to build though, due to the json-glib issue. 

Concretely, I am providing libfwupd as a build dependency to gnome-firmware, which gets picked up by meson, but I'm confused as to how meson is supposed to handle the transitive dependencies, like json-glib.
I don't assume libfwupd is supposed to include its own dependencies as part of its package, so that means that meson is supposed to look for them using pkg-config?
Somehow that isn't happening though, and therefore json-glib's include dir is not added to the cc invocation and the header cannot be found.

I'm not an expert with meson, so I might actually be holding it wrong here, but I haven't had the same issue with other fwupd dependencies in previous versions.

superm1, 2024-10-28:
Can you please share the whole build log for gnome firmware?

I'm not familiar with NixOS builds but maybe I can find the needle in that haystack knowing what to look for.

And since you're doing subprojects how are you putting those two fixes into fwupd and gnome firmware? I feel like the remaining problem has to be somewhere around there.

r-vdp, 2024-10-29:
This is the full build log for gnome-firmware. There's some nix-specific output, but you can see the normal meson output as well.
It detects all the dependencies that are specified in the gnome-firmware meson build, and finds them in the environment, but there's no mention of json-glib.

```
Running phase: unpackPhase
@nix {"action":"setPhase","phase":"unpackPhase"}
unpacking source archive /nix/store/hhrh1k96x39i3psbardllz33qqh4jvws-source
source root is source
Running phase: patchPhase
@nix {"action":"setPhase","phase":"patchPhase"}
Running phase: updateAutotoolsGnuConfigScriptsPhase
@nix {"action":"setPhase","phase":"updateAutotoolsGnuConfigScriptsPhase"}
Running phase: configurePhase
@nix {"action":"setPhase","phase":"configurePhase"}
mesonConfigurePhase flags: --prefix=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0 --libdir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/lib --libexecdir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/libexec --bindir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/bin --sbindir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/sbin --includedir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/include --mandir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/share/man --infodir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/share/info --localedir=/nix/store/jd3lcn0hpfbr5ihhv7avkx7k5fif0pmm-gnome-firmware-46.0/share/locale -Dauto_features=enabled -Dwrap_mode=nodownload --buildtype=plain -Dconsolekit=false
The Meson build system
Version: 1.5.2
Source dir: /build/source
Build dir: /build/source/build
Build type: native build
Project name: gnome-firmware
Project version: 46.0
C compiler for the host machine: gcc (gcc 13.3.0 "gcc (GCC) 13.3.0")
C linker for the host machine: gcc ld.bfd 2.43.1
Host machine cpu family: x86_64
Host machine cpu: x86_64
Compiler for C supports arguments -Waggregate-return: YES 
Compiler for C supports arguments -Warray-bounds: YES 
Compiler for C supports arguments -Wcast-align: YES 
Compiler for C supports arguments -Wclobbered: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wempty-body: YES 
Compiler for C supports arguments -Wextra: YES 
meson.build:60: WARNING: Consider using the built-in warning_level option instead of using "-Wextra".
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wformat-security: YES 
Compiler for C supports arguments -Wformat-signedness: YES 
Compiler for C supports arguments -Wignored-qualifiers: YES 
Compiler for C supports arguments -Wimplicit-function-declaration: YES 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Wmissing-declarations: YES 
Compiler for C supports arguments -Wmissing-format-attribute: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wmissing-noreturn: YES 
Compiler for C supports arguments -Wmissing-parameter-type: YES 
Compiler for C supports arguments -Wmissing-prototypes: YES 
Compiler for C supports arguments -Wnested-externs: YES 
Compiler for C supports arguments -Wno-discarded-qualifiers: YES 
Compiler for C supports arguments -Wno-missing-field-initializers: YES 
Compiler for C supports arguments -Wno-strict-aliasing: YES 
Compiler for C supports arguments -Wno-suggest-attribute=format: YES 
Compiler for C supports arguments -Wno-unused-parameter: YES 
Compiler for C supports arguments -Wno-cast-function-type: YES 
Compiler for C supports arguments -Wold-style-definition: YES 
Compiler for C supports arguments -Wno-deprecated-declarations: YES 
Compiler for C supports arguments -Wno-format-nonliteral: YES 
Compiler for C supports arguments -Woverride-init: YES 
Compiler for C supports arguments -Wpacked: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Wredundant-decls: YES 
Compiler for C supports arguments -Wreturn-type: YES 
Compiler for C supports arguments -Wshadow: YES 
Compiler for C supports arguments -Wsign-compare: YES 
Compiler for C supports arguments -Wstrict-aliasing: YES 
Compiler for C supports arguments -Wstrict-prototypes: YES 
Compiler for C supports arguments -Wswitch-default: YES 
Compiler for C supports arguments -Wtype-limits: YES 
Compiler for C supports arguments -Wundef: YES 
Compiler for C supports arguments -Wuninitialized: YES 
Compiler for C supports arguments -Wunused-but-set-variable: YES 
[... 99 lines cut ...]
   10 | #include <json-glib/json-glib.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[39/42] Compiling C object src/gnome-firmware.p/gfu-common.c.o
FAILED: src/gnome-firmware.p/gfu-common.c.o 
gcc -Isrc/gnome-firmware.p -Isrc -I../src -I. -I.. -I/nix/store/82bbzzkj4ncqg1q08f9c6f91pam9kf0n-glib-2.80.4-dev/include -I/nix/store/82bbzzkj4ncqg1q08f9c6f91pam9kf0n-glib-2.80.4-dev/include/glib-2.0 -I/nix/store/fxnchbq09sq279ssqdsgzigsyvp6hz06-glib-2.80.4/lib/glib-2.0/include -I/nix/store/4r3fsd2r23ags1gdpwz2dsrws6jy63m1-gdk-pixbuf-2.42.12-dev/include/gdk-pixbuf-2.0 -I/nix/store/r577cvxqgb2acrcv9vhw03293mwgvrq9-gtk4-4.14.5-dev/include/gtk-4.0 -I/nix/store/s9gm9vyza9yvfsz9ag01s4hw018gjhqg-cairo-1.18.2-dev/include/cairo -I/nix/store/hl14fvf6jpfv30j4p8hqgzqpmrmx2m7b-freetype-2.13.3-dev/include/freetype2 -I/nix/store/hl14fvf6jpfv30j4p8hqgzqpmrmx2m7b-freetype-2.13.3-dev/i [... cut ...]
In file included from /nix/store/xns5kkmislmm5vyrgqzxzndidl30bygk-fwupd-2.0.1-dev/include/fwupd-3/fwupd.h:15,
                 from ../src/gfu-common.h:11,
                 from ../src/gfu-main.c:16:
/nix/store/xns5kkmislmm5vyrgqzxzndidl30bygk-fwupd-2.0.1-dev/include/fwupd-3/libfwupd/fwupd-codec.h:10:10: fatal error: json-glib/json-glib.h: No such file or directory
   10 | #include <json-glib/json-glib.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

r-vdp, 2024-10-29:
For the fixes, I tried both building fwupd from main directly, which should have both fixes, and patching the fwupd 2.0.1 source before building, and then in both cases I rebuild gnome-firmware using the fwupd that I obtained. In both cases I got exactly the same build errors as in the log above.

superm1, 2024-10-29:
Okay before we confirm that's the right direction to go let's double check what the pkgconfig generated by fwupd looks like.

r-vdp, 2024-10-29:
This is the .pc file that I get from my fwupd build:
```
prefix=/nix/store/n3lwq47qxkmb7kmhq605hz8a7gf4mz75-fwupd-2.0.1
includedir=/nix/store/xns5kkmislmm5vyrgqzxzndidl30bygk-fwupd-2.0.1-dev/include
libdir=/nix/store/m21xw179g2f5qwc3nbslr0blvi94n58b-fwupd-2.0.1-lib/lib

Name: fwupd
Description: fwupd is a system daemon for installing device firmware
Version: 2.0.1
Requires: gio-2.0
Requires.private: gio-unix-2.0 >=  2.72.0, jcat >=  0.2.0, json-glib-1.0 >=  1.6.0, libcurl >=  7.62.0
Libs: -L${libdir} -lfwupd
Cflags: -I${includedir}/fwupd-3
```

superm1, 2024-10-29:
So it looks like it has the right requirements then, no?

r-vdp, 2024-10-29:
I think json-glib should actually be added to `Requires` as well, in addition to `Requires.private`, shouldn't it? If I add it there manually, things seem to work.
Does that make sense?

superm1, 2024-10-29:
I just cleared libjson-glib-dev from my env, and then compiled fwupd main.  I installed it to /usr/local, and checked what fwupd.pc looked like.

```sh
❯ cat /usr/local/lib/x86_64-linux-gnu/pkgconfig/fwupd.pc
prefix=/usr/local
includedir=${prefix}/include
libdir=${prefix}/lib/x86_64-linux-gnu

Name: fwupd
Description: fwupd is a system daemon for installing device firmware
Version: 2.0.2
Requires: gio-2.0 >=  2.72.0
Requires.private: jcat, json-glib-1.0, gio-unix-2.0 >=  2.72.0, gnutls >=  3.6.0, libcurl >=  7.62.0
Libs: -L${libdir} -lfwupd
Libs.private: -lgpgme -lgpg-error
Cflags: -I${includedir}/fwupd-3
```

So then I tried to build gnome-firmware and double checked it didn't subproject fwupd and used the one from `/usr/local/`

[reproduce.txt](https://github.com/user-attachments/files/17558657/reproduce.txt)

So I really don't see what fwupd should be doing differently.

r-vdp, 2024-10-29:
But you're still compiling the subproject there. The cc invocation is not shown in the build log, but I suspect that there will be an include dir passed to cc that includes the json-glib header file, even though strictly speaking, according to the .pc file, that header is not required.

Since I am building without the subprojects, meson will only include the dependencies that are either explicitly listed by gnome-firmware, or that are listed as `Requires` by fwupd, and no others, and so I don't get the json-glib header file.

The following patch, fixes the build of gnome-firmware for me:
```patch
diff --git a/libfwupd/meson.build b/libfwupd/meson.build
index d900c6e45..3939c61eb 100644
--- a/libfwupd/meson.build
+++ b/libfwupd/meson.build
@@ -88,7 +88,7 @@ libfwupd_dep = declare_dependency(
 pkgg = import('pkgconfig')
 pkgg.generate(
   fwupd,
-  requires: [ 'gio-2.0' ],
+  requires: [ 'gio-2.0', 'json-glib-1.0' ],
   subdirs: base_dir,
   version: meson.project_version(),
   name: 'fwupd',
```

And I think it makes sense since fwupd is including the json-glib header files in fwupd.h.

PR #7994 (2024-10-29): trivial: add a dependency on json-glib to the generated pkgconfig file
This is needed when building with `-Dwrap_mode=nodownload` since fwupd.h
is including header files from json-glib.

Fixes #7988

Ping: @superm1 

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
closed 2024-10-29 by superm1