fwupd/fwupd #8005: failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: could not find EFI DP

neitsab, 2024-10-31:
**Describe the bug**

While upgrading both System Firmware and UEFI dbx on a Dell Precision 5560 using Arch Linux, I got the following error at the end for the dbx update:

```
failed to write-firmware: Blocked executable in the ESP, ensure grub and shim are up to date: could not find EFI DP
```

Upon reboot the boot process halts and displays the following error message:

```
fwupd-efi version 1.7
WARNING: Failed to find 'HD(1,GPT,LONG-UUID-SKIPPED)/\EFI\arch\fw\fwupd-some-long-string.cap' DevicePath
WARNING: Could not locate device handle: Unsupported
``` 

I have to poweroff the machine and then boot as usual. The updates haven't applied (despite GNOME Software telling me they did).

**Steps to Reproduce**

```
 % fwupdmgr get-updates
Devices with no available firmware updates: 
 • DLL0945:00 04F3:311C
 • Fingerprint Sensor
 • KXG50ZNV1T02 NVMe TOSHIBA 1024GB
 • SSD 980 PRO 2TB
 • TPM
 • UEFI Device Firmware
Dell Inc. Precision 5560
│
├─System Firmware:
│ │   Identifiant de l'appareil: 5d7db2206ec1bc7aed22f9ecc219438e13b944cb
│ │   Résumé:             UEFI System Resource Table device (updated via NVRAM)
│ │   Version actuelle:   1.28.0
│ │   Version minimum:    1.28.0
│ │   Fournisseur:        Dell (DMI:Dell Inc.)
│ │   État de mise à jour: Succès
│ │   GUID:               6e106831-11dd-4c1b-b22b-526d6f78683c
│ │   Drapeaux de périphérique: • Périphérique interne
│ │                       • Mise à jour possible
│ │                       • Le système nécessite une source d'alimentation externe
│ │                       • Supported on remote server
│ │                       • Doit redémarrer après une mise à jour
│ │                       • Cryptographic hash verification is available
│ │                       • Le périphérique est utilisable pendant la durée de la mise à jour
│ │   Device Requests:    • Message
│ │ 
│ ├─XPS 15 9510, Precision 5560 System Update:
│ │     Nouvelle version: 1.32.0
│ │     Remote ID:        lvfs
│ │     Release ID:       97366
│ │     Résumé:           Firmware for the Dell XPS 15 9510, Precision 5560
│ │     Licence:          Propriétaire
│ │     Taille:           23,6 Mo
│ │     Created:          2024-08-14
│ │     Urgence:          Haute
│ │     Fournisseur:      Dell
│ │     Release Flags:    • Trusted metadata
│ │                       • Is upgrade
│ │     Description:      
│ │     This stable release fixes the following issues:
│ │     
│ │     • Fixed the issue where the system turns on automatically after it shuts down. This issue occurs when the Dell P2422HE monitor is connected to the system through a Type-C cable.
│ │     Somme de contrôle: 51bef9d19597768d3a16099ed27d576fc4dc3350bf36589d01480ec0d7d17cbb
│ │   
│ ├─XPS 15 9510, Precision 5560 System Update:
│ │     Nouvelle version: 1.31.0
│ │     Remote ID:        lvfs
│ │     Release ID:       95824
[... 378 lines cut ...]
/boot
├── efi
│   ├── EFI
│   │   └── Linux
│   │       └── arch-linux.efi
│   └── loader
│       └── random-seed
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux

5 directories, 6 files
```

**Additional questions**

- Operating system and version: Arch Linux x86-64
- Have you tried rebooting? Yes
- Is this a regression? No idea, first time
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

Main drive is a Self-Encrypting Drive (SED) configured for OPAL management via cryptsetup, but once unlocked it remains so until a cold reboot (i.e a regular reboot doesn't re-lock the drive).

neitsab, 2024-10-31:
Okay, upon reviewing the issue I realize fwupd is trying to boot from the NVMe drive (the Toshiba), despite having been launched from the Samsung one where my Arch system is installed (the other has Bluefin on it).

Lsblk from Arch:

```
NAME        TYPE    SIZE LABEL  FSTYPE      MOUNTPOINT
nvme1n1     disk  953,9G                    
├─nvme1n1p1 part    600M        vfat        
├─nvme1n1p2 part      1G        ext4        
└─nvme1n1p3 part  952,3G        crypto_LUKS 
nvme0n1     disk    1,8T                    
├─nvme0n1p1 part    500M EFI    vfat        /boot/efi
└─nvme0n1p2 part    1,8T        crypto_LUKS 
  └─root    crypt   1,8T System btrfs       /home
```

Interestingly the string `8cc` appears in both my root FS UUID and the Toshiba boot partition PARTUUID:

```
sudo blkid | grep 8cc
/dev/mapper/root: LABEL="System" UUID="a593c2fc-2cfe-4989-8cc4-1ebd7e8c3d84" UUID_SUB="ad1b8f43-20c9-4a2c-b65c-e30bcc329ba3" BLOCK_SIZE="4096" TYPE="btrfs"
/dev/nvme1n1p1: UUID="0CC9-471D" BLOCK_SIZE="4096" TYPE="vfat" PARTLABEL="EFI System Partition" PARTUUID="8cc4a28f-6b54-471d-8ea1-c5e6cf1123aa"
```

Could fwupd drive recognition logic get confused by this?

neitsab, 2024-10-31:
I've just tried deleting the auto-created `Boot0000` entry and run `fwupdmgr update` once more but the result was the same.

Is it a case of the two drives being reversed between kernel naming/UEFI boot logic etc.?

superm1, 2024-10-31:
Can you please share `sudo fwupdtool esp-list -vv` output?  That will let us understand specifically the decisions that fwupd is making on which partition to use and we can try to match it to the code to explain it.

superm1, 2024-10-31:
Oh I see you edited the comment.  I should say my gut tells me this is the reason for the problem.

> Edited to add: I boot directly via EFISTUB (boot entry configured directly via efibootmgr) and don't use a boot loader/manager (no systemd-boot nor GRUB )

Please share that log I'm looking for, but I suspect your case we can't detect properly automatically.  It's relatively unusual.  You will probably need to use the `EspLocation` setting.
https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#daemon-parameters

neitsab, 2024-10-31:
> sudo fwupdtool esp-list -vv

First part is:

```
sudo fwupdtool esp-list -vv
[sudo] Mot de passe de bastien : 
18:31:12.276 FuDebug              verbose to debug (on console 1)
18:31:12.276 FuEngine             starting fwupd 2.0.1…
18:31:12.276 FuMain               locked /run/lock/fwupdtool
Chargement…              [ -                                     ]18:31:12.280 FuConfig             loading config /etc/fwupd/fwupd.conf
18:31:12.280 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
18:31:12.280 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
18:31:12.280 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
18:31:12.280 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
18:31:12.280 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
18:31:12.280 FuConfig             L’ouverture du fichier « /var/etc/fwupd/fwupd.conf » a échoué : Aucun fichier ou dossier de ce nom
18:31:12.280 FuConfig             ::configuration loaded
Chargement…              [                                       ]18:31:12.280 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
18:31:12.280 FuHistory            got schema version of 14
Chargement…              [*                                      ]18:31:12.284 FuPlugin             load(/usr/lib/fwupd-2.0.1/libfu_plugin_modem_manager.so)
18:31:12.284 FuQuirks             open database /var/cache/fwupd/quirks.db
18:31:12.284 FuQuirks             mtimes unchanged: quirks,pci.ids:1728109378,usb.ids:1728109378,pnp.ids:1728109378,oui.txt:1728109378, doing nothing
18:31:12.284 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
18:31:12.284 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
18:31:12.284 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
18:31:12.284 XbSilo               file: 93c26ad9-6394-1b39-9476-a6418723e5a3, current:93c26ad9-6394-1b39-9476-a6418723e5a3
18:31:12.284 XbSilo               loading silo with existing file contents
Chargement…              [**                                     ]18:31:12.285 FuEngine             ignoring: Erreur à l’ouverture du répertoire « /var/lib/fwupd/local.d » : Aucun fichier ou dossier de ce nom
18:31:12.285 FuEngine             ignoring: Erreur à l’ouverture du répertoire « /usr/share/fwupd/local.d » : Aucun fichier ou dossier de ce nom
18:31:12.285 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
18:31:12.285 XbSilo               file: 9d48ce50-cfca-3e8f-5726-d5dec357e45f, current:a560dd32-664f-a54c-ee7d-8ef384059724
18:31:12.285 XbSilo               loading silo with existing file contents
18:31:12.286 FuEngine             2795 components now in silo
Chargement…              [***                                    ]18:31:12.290 FuPlugin             constructed(flashrom)
18:31:12.290 FuPlugin             constructed(modem_manager)
18:31:12.290 FuContext            added udev subsystem watch of tty
18:31:12.290 FuContext            added udev subsystem watch of usbmisc
18:31:12.290 FuContext            added udev subsystem watch of wwan
18:31:12.290 FuPlugin             constructed(uefi_capsule)
18:31:12.290 FuContext            added udev subsystem watch of drm
18:31:12.291 FuCommon             mapped file /usr/lib/fwupd/efi/fwupdx64.efi of size 0x108a9
18:31:12.291 FuPlugin             constructed(acpi_phat)
18:31:12.291 FuPlugin             constructed(algoltek_aux)
18:31:12.291 FuContext            added udev subsystem watch of drm_dp_aux_dev
18:31:12.291 FuPlugin             constructed(algoltek_usb)
18:31:12.291 FuPlugin             constructed(algoltek_usbcr)
18:31:12.291 FuContext            added udev subsystem watch of block for plugin algoltek_usbcr
18:31:12.291 FuPlugin             constructed(amd_kria)
18:31:12.291 FuContext            added udev subsystem watch of i2c
18:31:12.291 FuPlugin             constructed(amd_pmc)
18:31:12.291 FuContext            added udev subsystem watch of platform
18:31:12.291 FuPlugin             constructed(amd_gpu)
18:31:12.291 FuContext            added udev subsystem watch of pci
18:31:12.291 FuPlugin             constructed(analogix)
18:31:12.291 FuPlugin             constructed(android_boot)
18:31:12.291 FuPlugin             constructed(ata)
18:31:12.291 FuPlugin             constructed(aver_hid)
18:31:12.291 FuPlugin             constructed(bcm57xx)
18:31:12.291 FuPlugin             constructed(ccgx)
18:31:12.291 FuPlugin             constructed(ccgx_dmc)
18:31:12.291 FuPlugin             constructed(cfu)
18:31:12.291 FuPlugin             constructed(ch341a)
18:31:12.291 FuPlugin             constructed(ch347)
18:31:12.291 FuPlugin             constructed(colorhug)
18:31:12.291 FuPlugin             constructed(corsair)
18:31:12.291 FuPlugin             constructed(cpu)
18:31:12.291 FuPlugin             constructed(cros_ec)
18:31:12.291 FuPlugin             constructed(dell)
18:31:12.291 FuPlugin             constructed(dell_dock)
[... 393 lines cut ...]
18:31:12.294 FuContext            cannot use HWIDs before calling ->load_hwinfo()
18:31:12.294 FuContext            cannot use HWIDs before calling ->load_hwinfo()
18:31:12.297 FuVolume             Looking for volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
[... cut ...]
18:31:12.308 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: 4f68bce3-e8cd-4db1-96e7-fbcaf984b709, internal: 1, fs: crypto_LUKS
18:31:12.309 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
18:31:12.309 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7

0.	Annuler
1.	/org/freedesktop/UDisks2/block_devices/nvme1n1p1
2.	/org/freedesktop/UDisks2/block_devices/nvme0n1p1
Choose volume [0-2]: 
```

And then... Should I send you both?

neitsab, 2024-10-31:
```
Choose volume [0-2]: 1
/boot/efi/EFI/Linux/arch-linux.efi
/boot/efi/loader/random-seed
18:33:09.054 FuPlugin             finalize(modem_manager)
18:33:09.054 FuPlugin             finalize(flashrom)
18:33:09.054 FuPluginTest         destroy
```

```
Choose volume [0-2]: 2
18:33:32.805 FuVolume             mounting /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/run/media/root/0CC9-471D/EFI/BOOT/BOOTX64.EFI
/run/media/root/0CC9-471D/EFI/BOOT/fbia32.efi
/run/media/root/0CC9-471D/EFI/BOOT/BOOTIA32.EFI
/run/media/root/0CC9-471D/EFI/BOOT/fbx64.efi
/run/media/root/0CC9-471D/EFI/fedora/mmia32.efi
/run/media/root/0CC9-471D/EFI/fedora/mmx64.efi
/run/media/root/0CC9-471D/EFI/fedora/grubx64.efi
/run/media/root/0CC9-471D/EFI/fedora/shim.efi
/run/media/root/0CC9-471D/EFI/fedora/shimx64.efi
/run/media/root/0CC9-471D/EFI/fedora/BOOTX64.CSV
/run/media/root/0CC9-471D/EFI/fedora/shimia32.efi
/run/media/root/0CC9-471D/EFI/fedora/grubia32.efi
/run/media/root/0CC9-471D/EFI/fedora/BOOTIA32.CSV
/run/media/root/0CC9-471D/EFI/fedora/grub.cfg
/run/media/root/0CC9-471D/EFI/arch/fwupdx64.efi
18:33:32.825 FuVolume             unmounting /org/freedesktop/UDisks2/block_devices/nvme0n1p1
18:33:32.833 FuPlugin             finalize(modem_manager)
18:33:32.833 FuPlugin             finalize(flashrom)
18:33:32.833 FuPluginTest         destroy
```

neitsab, 2024-10-31:
Oh they got reversed now! Unpredictable kernel naming will be unpredictable...

```
sblk
NAME        TYPE    SIZE LABEL  FSTYPE      MOUNTPOINT
nvme0n1     disk  953,9G                    
├─nvme0n1p1 part    600M        vfat        
├─nvme0n1p2 part      1G        ext4        
└─nvme0n1p3 part  952,3G        crypto_LUKS 
nvme1n1     disk    1,8T                    
├─nvme1n1p1 part    500M EFI    vfat        /boot/efi
└─nvme1n1p2 part    1,8T        crypto_LUKS 
  └─root    crypt   1,8T System btrfs       /home
```

neitsab, 2024-10-31:
> You will probably need to use the `EspLocation` setting.
> https://fwupd.github.io/libfwupdplugin/fwupd.conf.html#daemon-parameters

I didn't find in the docs what kind of paths/names this parameter takes, so I've just tried with adding `EspLocation=/boot/efi` but I don't see how that could help it identify the correct disk upon reboot?

neitsab, 2024-10-31:
> Well if you add it to /etc/fstab and get it mounted outside of fwupd that should make fwupd pick it.

Interesting. It's actually already defined in my fstab as follows:

```
LABEL=EFI           	/boot/efi 	vfat      	noatime,discard		0 2
```

> when we score them we'll pick the one with Fedora because it has a grub binary.

That's really interesting! I had installed this Fedora Silverblue derivative on the second drive just to test it but now it comes to bite me back 😅

I found an alternative if a bit radical option: I've just disabled the second NVMe slot altogether in the BIOS, and for sure this allowed the firmware update to proceed correctly upon reboot :+1:

I understand that my setup is unusual, it's ok that fwupdmgr cannot accommodate all topologies under the sun.

So the firmware flashing update went through, but the dbx one still doesn't, with the same error message I used in the issue title. That is despite having `EspLocation=/boot/efi` in `/etc/fwupd/fwupd.conf`, having only one drive enabled and rebooting.

Any ideas?

neitsab, 2024-10-31:
Current output of esp-list with only one drive

```
% sudo fwupdtool esp-list -vv
[sudo] Mot de passe de bastien : 
19:05:04.545 FuDebug              verbose to debug (on console 1)
19:05:04.545 FuEngine             starting fwupd 2.0.1…
19:05:04.545 FuMain               locked /run/lock/fwupdtool
Chargement…              [ -                                     ]19:05:04.548 FuConfig             loading config /etc/fwupd/fwupd.conf
19:05:04.548 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
19:05:04.548 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
19:05:04.548 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
19:05:04.548 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x49
19:05:04.548 FuConfig             setting config [fwupd] EspLocation=/boot/efi
19:05:04.548 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
19:05:04.548 FuConfig             L’ouverture du fichier « /var/etc/fwupd/fwupd.conf » a échoué : Aucun fichier ou dossier de ce nom
19:05:04.548 FuConfig             ::configuration loaded
Chargement…              [                                       ]19:05:04.548 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
19:05:04.548 FuHistory            got schema version of 14
Chargement…              [*                                      ]19:05:04.552 FuPlugin             load(/usr/lib/fwupd-2.0.1/libfu_plugin_modem_manager.so)
19:05:04.552 FuQuirks             open database /var/cache/fwupd/quirks.db
19:05:04.552 FuQuirks             mtimes unchanged: quirks,pci.ids:1728109378,usb.ids:1728109378,pnp.ids:1728109378,oui.txt:1728109378, doing nothing
19:05:04.552 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
19:05:04.552 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
19:05:04.552 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
19:05:04.552 XbSilo               file: 93c26ad9-6394-1b39-9476-a6418723e5a3, current:93c26ad9-6394-1b39-9476-a6418723e5a3
19:05:04.552 XbSilo               loading silo with existing file contents
Chargement…              [**                                     ]19:05:04.553 FuEngine             ignoring: Erreur à l’ouverture du répertoire « /var/lib/fwupd/local.d » : Aucun fichier ou dossier de ce nom
19:05:04.553 FuEngine             ignoring: Erreur à l’ouverture du répertoire « /usr/share/fwupd/local.d » : Aucun fichier ou dossier de ce nom
19:05:04.553 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
19:05:04.553 XbSilo               file: 9d48ce50-cfca-3e8f-5726-d5dec357e45f, current:a560dd32-664f-a54c-ee7d-8ef384059724
19:05:04.553 XbSilo               loading silo with existing file contents
19:05:04.554 FuEngine             2795 components now in silo
Chargement…              [***                                    ]19:05:04.558 FuPlugin             constructed(flashrom)
19:05:04.558 FuPlugin             constructed(modem_manager)
19:05:04.558 FuContext            added udev subsystem watch of tty
19:05:04.558 FuContext            added udev subsystem watch of usbmisc
19:05:04.558 FuContext            added udev subsystem watch of wwan
19:05:04.558 FuPlugin             constructed(uefi_capsule)
19:05:04.558 FuContext            added udev subsystem watch of drm
19:05:04.560 FuCommon             mapped file /usr/lib/fwupd/efi/fwupdx64.efi of size 0x108a9
19:05:04.560 FuPlugin             constructed(acpi_phat)
19:05:04.560 FuPlugin             constructed(algoltek_aux)
19:05:04.560 FuContext            added udev subsystem watch of drm_dp_aux_dev
19:05:04.560 FuPlugin             constructed(algoltek_usb)
19:05:04.560 FuPlugin             constructed(algoltek_usbcr)
19:05:04.560 FuContext            added udev subsystem watch of block for plugin algoltek_usbcr
19:05:04.560 FuPlugin             constructed(amd_kria)
19:05:04.560 FuContext            added udev subsystem watch of i2c
19:05:04.560 FuPlugin             constructed(amd_pmc)
19:05:04.560 FuContext            added udev subsystem watch of platform
19:05:04.560 FuPlugin             constructed(amd_gpu)
19:05:04.560 FuContext            added udev subsystem watch of pci
19:05:04.560 FuPlugin             constructed(analogix)
19:05:04.560 FuPlugin             constructed(android_boot)
19:05:04.560 FuPlugin             constructed(ata)
19:05:04.560 FuPlugin             constructed(aver_hid)
19:05:04.560 FuPlugin             constructed(bcm57xx)
19:05:04.560 FuPlugin             constructed(ccgx)
19:05:04.560 FuPlugin             constructed(ccgx_dmc)
19:05:04.560 FuPlugin             constructed(cfu)
19:05:04.560 FuPlugin             constructed(ch341a)
19:05:04.560 FuPlugin             constructed(ch347)
19:05:04.560 FuPlugin             constructed(colorhug)
19:05:04.560 FuPlugin             constructed(corsair)
19:05:04.560 FuPlugin             constructed(cpu)
19:05:04.560 FuPlugin             constructed(cros_ec)
19:05:04.560 FuPlugin             constructed(dell)
19:05:04.560 FuPlugin             constructed(dell_dock)
19:05:04.560 FuPlugin             constructed(dell_k2)
[... 387 lines cut ...]
  Name:                 uefi_recovery
  Flags:                require-hwid
  Order:                2

19:05:04.562 FuEngine             resetting update motd timeout
19:05:04.562 FuContext            cannot use HWIDs before calling ->load_hwinfo()
[... cut ...]
19:05:04.571 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
19:05:04.571 FuContext            no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7

Volume sélectionné: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/efi/EFI/Linux/arch-linux.efi
/boot/efi/EFI/arch/fwupdx64.efi
/boot/efi/EFI/dell/bios/recovery/BIOS_CUR.rcv
/boot/efi/loader/random-seed
19:05:04.571 FuPlugin             finalize(modem_manager)
19:05:04.571 FuPlugin             finalize(flashrom)
19:05:04.571 FuPluginTest         destroy
```

closed 2024-12-04 by superm1