fwupd/fwupd #8014: Dell WD19 upgrade issue

pecheneg11, 2024-11-03:
**Describe the bug**

Message recipient disconnected from message bus without replying 

Or

failed to write-firmware: could not write firmware to thunderbolt device at /sys/devices/pci0000:00/0000:00:01.2/0000:7c:00.0/0000:7d:00.0/0000:7e:00.0/domain1/1-0/1-1: Error opening file “/sys/devices/pci0000:00/0000:00:01.2/0000:7c:00.0/0000:7d:00.0/0000:7e:00.0/domain1/1-0/1-1/nvm_non_active0/nvmem”: Permission denied

**Steps to Reproduce**

 sudo fwupdmgr upgrade 
          fwupdmgr upgrade 

**Expected behavior**
   The firmware Thunderbolt controller on my dock is updated

**fwupd version information**

compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.0
compile   com.hughsie.libxmlb           0.3.19
compile   io.snapcraft.fwupd            6561
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.19
runtime   org.kernel                    6.8.0-48-generic
runtime   org.freedesktop.fwupd         2.0.0

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

snap install fwupd

<details>

<summary>**fwupd device information**</summary>

Apple Inc. MacBookPro16,1
│
├─WD19TB:
│ │   Device ID:          2ebc44a519ff7378e7755847a17604c4f18086de
│ │   Summary:            High performance dock
│ │   Current version:    01.01.00.07
│ │   Vendor:             Dell Inc. (USB:0x413C)
│ │   Install Duration:   1 minute
│ │   Serial Number:      B9QJ363/3143598425650012
│ │   GUID:               cd357cf1-40b2-5d87-b8df-bb2dd82774aa ← USB\VID_413C&PID_B06E&hub&embedded
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │ 
│ ├─Package level of Dell dock:
│ │     Device ID:        39fd31b8bbd5c775149831e3333e18b1dec8baca
│ │     Summary:          A representation of dock update status
│ │     Current version:  01.00.36.01
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Update State:     Success
│ │     Last modified:    2024-11-03
│ │     GUIDs:            d8927ff5-a5b2-5618-848b-8e8bfb75b66f
│ │                       8ceeeffd-51b6-580c-9b75-69143227aff8 ← USB\VID_413C&PID_B06E&hub&status
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device can recover flash failures
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │     Device Requests:  • Message (custom)
│ │   
│ ├─RTS5413 in Dell dock:
│ │     Device ID:        636a30d5264bc50db64cbc3456609e5ed544d890
│ │     Summary:          USB 3.1 Generation 1 Hub
│ │     Current version:  01.22
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 14 seconds
│ │     GUIDs:            b27d25f1-019d-5718-b41a-02ddaefe5577 ← USB\VID_413C&PID_B06F
│ │                       ac5b774c-b49d-566b-9255-85f0f7f8a4ed ← USB\VID_413C&PID_B06F&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │   
│ ├─RTS5487 in Dell dock:
│ │     Device ID:        b21f770086388c39551c731e5d4a61a8ded30774
│ │     Summary:          USB 3.1 Generation 2 Hub
│ │     Current version:  01.57
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 3 seconds
│ │     GUIDs:            acfcd89b-105d-55b9-b85b-08bf8508f38c ← USB\VID_413C&PID_B06E
│ │                       568ffa1e-a0db-5287-9ea3-872b60f7730b ← USB\VID_413C&PID_B06E&hub
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Signed Payload
│ │   
│ ├─VMM5331 in Dell dock:
│ │     Device ID:        82176e6f1f9a2463afeab0ac607ed809052ecc02
│ │     Summary:          Multi Stream Transport controller
│ │     Current version:  05.07.04
│ │     Vendor:           Dell Inc. (USB:0x413C)
│ │     Install Duration: 6 minutes
│ │     GUID:             89fec0b6-6b76-5008-b82c-5e5c6c164007 ← MST-panamera-vmm5331-259
│ │     Device Flags:     • Updatable
│ │                       • Supported on remote server
│ │                       • Device stages updates
│ │                       • Device is usable for the duration of the update
│ │                       • Unsigned Payload
│ │   
├─Unknown Device:
│     Device ID:          5f7dee69f5c6dc34f6af5a8ecff65b51ce478670
│     Vendor:             PNP:BNQ
│     Serial Number:      L5M00649019
│     GUID:               49a6f51e-b6fb-532b-abe2-8add61a83738 ← DRM\VEN_BNQ&DEV_7950
│   

</details>

**Dock SKU**
Please mention which module is installed in your WD19.

- [ ] WD19 (Single-C)
- [*] WD19TB (Thunderbolt)
- [ ] WD19DC (Dual-C)

**Peripherals connected to the dock**

BENQ [EW3270U] USB-C Monitor

**Verbose daemon logs**

Restart the daemon to make the change effective? [y|N]: y
Successfully modified configuration value

I enabled debug mode but the log entries did not appear even after running the update again.
max@max-MacBookPro:~$ journalctl -b -u fwupd.service
-- No entries --

**Additional questions**

- Operating system and version: Ubuntu 22.04.5 LTS
- Have you tried unplugging the dock or any peripherals from your machine? - Yes
- Have you tried to power cycle the dock from the AC adapter? -Yes
- Is this a regression?

superm1, 2024-11-03:
Can you please turn on dynamic debugging for thunderbolt.ko and then reproduce and share you kernel log?

pecheneg11, 2024-11-04:
Installed the version from “Ubuntu Software”. I need help I have tried every possible combination except the USB-A-USB-C adapter method. Help me, please.
[journalctl -u fwupd.service.txt](https://github.com/user-attachments/files/17616809/journalctl.-u.fwupd.service.txt)

superm1, 2024-11-04:
You didn't share the kernel log.

pecheneg11, 2024-11-04:
Ah, there he is. 
[kern.log](https://github.com/user-attachments/files/17622929/kern.log)

valentindavid, 2024-11-06:
I am not sure what causes `mknod`.

Is this what we need?
 * `/sys/devices/**/nvm_non_active*/nvmem a`
 * `/sys/devices/**/nvm_active*/nvmem r`

Or does it also read the "nvm_non_active" ones?

valentindavid, 2024-11-06:
I have opened https://github.com/canonical/snapd/pull/14693

hughsie, 2024-11-12:
Great, thanks. Lets close this downstream one as there's not much we can do.

closed 2024-11-12 by hughsie