fwupd/fwupd #8031: Add LVFS:deviceflag to set firmware chunk size

gwendalcr, 2024-11-05:
**Is your feature request related to a problem? Please describe.**
The SCSI firmware updater plugin send firmware to the SCSI device in chunk of 4KB, as described by the specification.
Some [UFS] devices do not like that and needs their chunks in 512KB increment.
It has been already reported in another project, [ufs-utils](https://github.com/westerndigitalcorporation/ufs-utils/pull/49) 

**Describe the solution you'd like**
Add a new [LVFS:deviceflags](https://lvfs.readthedocs.io/en/latest/metainfo.html#device-flags) to describe the required chunk size for that firmware image in the .cab. 
Add code in the SCSI plugin to respect that flag when setting the chunk size.

**Describe alternatives you've considered**
Adding quirks for the SCSI plugin. However, it requires to know each models (different capacity, ...) that have the issue, it will be error prone. Since the flag is generic enough, it could be applied to other plugin as well, to optimize firmware updates.

hughsie, 2024-11-05:
The way we've done this in the eMMC plugin is with a quirk entry, e.g. https://github.com/fwupd/fwupd/blob/main/plugins/emmc/README.md#emmcblocksize which looks like https://github.com/fwupd/fwupd/blob/main/plugins/emmc/emmc.quirk#L1 in practice. The reason we don't usually put this into the firmware metadata file is that we want to use the correct write size when using `fwupdtool install-blob` (which has no metadata) and also when offline.

Could we do something similar in the scsi plugin perhaps?

gwendalcr, 2024-11-06:
Yes, I could do that if it is the preferred solution.

PR #8061 (2024-11-10): scsi: Add quirk for some UFS devices
To install firmware on Samsung KLUDG4UHGC_B0E1, we need to send 512kB chuncks.

Add a quirk to alter the chunk size. This change is "inspired" from a similar change in the eMMC plugin:
commit 817d09a07bdf ("emmc: Add emmc quirk")

Check the WRITE BUFFER SCSI commands send 512kB worth of data (using kernel tracer:
WRITE_BUFFER - raw=3b 0e 00 00 00 00 08 00 00 00

Fixes #8031.

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [x] Feature
- [ ] Documentati [... cut]
commit eff3abee (2024-11-10): scsi: Add quirk for some UFS devices

To install firmware on Samsung KLUDG4UHGC_B0E1, we need to send 512kB
chuncks.

Add a quirk to alter the chunk size. This change is "inspired" from a
similar change in the eMMC plugin:
commit 817d09a07bdf ("emmc: Add emmc quirk")

Check the WRITE BUFFER SCSI commands send 512kB worth of data
(using kernel tracer:
WRITE_BUFFER - raw=3b 0e 00 00 00 00 08 00 00 00

Fixes #8031.