fwupd/fwupd #8048: update of MSI PRO Z790-P WIFI?

philfry, 2024-11-07:
Hi, I'd like to use fwupd to update a MSI PRO Z790-P mainboard.
Not sure whether or not this already should work according to https://fwupd.org/lvfs/vendors/#msi but for this board and for me, it does not.

`fwupdmgr get-devices --show-all -v`  gives me [this](https://github.com/user-attachments/files/17662297/get-devices.txt) output (stripped down to the relevant parts).

Is there anything I could try or provide to make it work?

Thanks!

philfry, 2024-11-07:
Thanks. Is there any way (for me, as an end user) to make a cabinet file from a "Intel serial flash for PCH ROM" (at least that's what `file` says) in order to use `fwupdmgr local-install`?

hughsie, 2024-11-07:
is it a capsule file? i.e. ending in .cap or having the capsule header? Maybe you could try `fwupdtool install-blob the-file.bin` but there's no support from us if you brick something. :/

philfry, 2024-11-07:
No, the file extension is part the firmware version (in this case: the firmware is 7E06vAF, the filename is E7E06IMS.AF0).
Well, I'd better upgrade using a usb stick then :laughing: 
Thanks!

closed 2024-11-07 by philfry

philfry, 2024-11-07:
Yes, I opened a support ticket for that. Let's see what happens.