fwupd/fwupd #8050: Update didn't run but fwupd still wanted to upload a "successful" report

JohnAZoidberg, 2024-11-07:
**Describe the bug**
I ran `fwupdmgr update` to apply the UEFI capsule update.
My system rebooted but didn't apply the update.
I checked with `fwupdmgr get-updates` and it prompted me to report me succesful update.
However when I checked `sudo dmidecode -t0`, as expected, the version was unchanged.

So I ran `fwupdmgr update` again and then the update was applied.
Sorry, I should've captured the ESRT state before updating again.

**Expected behavior**
I'm not sure why the update didn't apply but there definitely wasn't a successful update.
Would it make sense for fwupd to check that BIOS version was updated to verify success?

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   info.libusb                   1.0.27
compile   com.hughsie.libxmlb           0.3.20
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
compile   org.freedesktop.fwupd         2.0.1
runtime   org.freedesktop.fwupd-efi     1.6
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.0-0.rc5.20241101git6c52d4da1c74.48.fc42.x86_64
runtime   org.freedesktop.fwupd         2.0.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

It came installed with Fedora 42 

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**System UEFI configuration**
Please provide the output of the following commands:

```shell
> efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,2001,2002,2003
Boot0000* Fedora	HD(1,GPT,8952f495-9528-48cf-908f-64963417bf31,0x800,0x12c000)/\EFI\fedora\shim.efiRC
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 c0 12 00 00 00 00 00 95 f4 52 89 28 95 cf 48 90 8f 64 96 34 17 bf 31 02 02 / 04 04 2e 00 5c 00 45 00 46 00 49 00 5c 00 66 00 65 00 64 00 6f 00 72 00 61 00 5c 00 73 00 68 00 69 00 6d 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
    data: 52 43
Boot2001* EFI USB Device	RC
      dp: 7f ff 04 00
    data: 52 43
Boot2002* EFI DVD/CDROM	RC
      dp: 7f ff 04 00
    data: 52 43
Boot2003* EFI Network	RC
      dp: 7f ff 04 00
    data: 52 43
```

```shell
efivar -l | grep fw
```

```shell
> tree /boot
/boot
├── acpi_dsdt_override.cpio
├── boot -> .
├── config-6.10.0-0.rc4.20240621git50736169ecc8.42.fc41.x86_64
├── config-6.12.0-0.rc5.20241031git0fc810ae3ae1.47.fc42.x86_64
├── config-6.12.0-0.rc5.20241101git6c52d4da1c74.48.fc42.x86_64
├── efi  [error opening dir]
├── grub2  [error opening dir]
├── initramfs-0-rescue-463278ff2c4c4c9aa3970b96e79a43d6.img
├── initramfs-6.10.0-0.rc4.20240621git50736169ecc8.42.fc41.x86_64.img
├── initramfs-6.12.0-0.rc5.20241031git0fc810ae3ae1.47.fc42.x86_64.img
├── initramfs-6.12.0-0.rc5.20241101git6c52d4da1c74.48.fc42.x86_64.img
├── loader -> loader.1
├── loader.1
│   ├── entries  [error opening dir]
│   └── grub.cfg
├── lost+found  [error opening dir]
├── ostree
│   └── fedora-eca20b30e873e23c9d651351d71a6231902e8dbcc8e927893f3c8fe2bd8ab605
│       ├── initramfs-6.8.5-301.fc40.x86_64.img
│       └── vmlinuz-6.8.5-301.fc40.x86_64
├── symvers-6.10.0-0.rc4.20240621git50736169ecc8.42.fc41.x86_64.xz
├── symvers-6.12.0-0.rc5.20241031git0fc810ae3ae1.47.fc42.x86_64.xz
├── symvers-6.12.0-0.rc5.20241101git6c52d4da1c74.48.fc42.x86_64.xz
├── System.map-6.10.0-0.rc4.20240621git50736169ecc8.42.fc41.x86_64
├── System.map-6.12.0-0.rc5.20241031git0fc810ae3ae1.47.fc42.x86_64
├── System.map-6.12.0-0.rc5.20241101git6c52d4da1c74.48.fc42.x86_64
├── vmlinuz-0-rescue-463278ff2c4c4c9aa3970b96e79a43d6
├── vmlinuz-6.10.0-0.rc4.20240621git50736169ecc8.42.fc41.x86_64
├── vmlinuz-6.12.0-0.rc5.20241031git0fc810ae3ae1.47.fc42.x86_64
├── vmlinuz-6.12.0-0.rc5.20241101git6c52d4da1c74.48.fc42.x86_64
└── vmlinuz-6.9.0+
```

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?
   - I don't know
- Are you using an NVMe disk?
- Is secure boot enabled?
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?

hughsie, 2024-11-15:
>  Would it make sense for fwupd to check that BIOS version was updated to verify success?

That's exactly what we should have done. I'm a bit confused about what happened here. Would it be possible to downgrade and update again to see if it happens? And if it does, can you grab the daemon -vv startup logs on the non-successful update please.

JohnAZoidberg, 2024-11-19:
I had the same remote configured twice, so it made the update show up twice, not sure if that's related.
Me and my colleagues haven't seen it since then.

JohnAZoidberg, 2025-03-26:
I'll reopen if I see it again. Haven't seen it since then.

closed 2025-03-26 by JohnAZoidberg