fwupd/fwupd #8060: emulation: failed to load devices

CragW, 2024-11-09:
**Describe the bug**
The `eumlation-load` doesn't work for kestrel devices, no devices are loaded if multiple devices are tagged. However, if only the ec device `XXXXXXX` is tagged then `emulation-load` works as expected.

Below is the list (obscured) of the tagged devices:
```
$ fwupdmgr emulation-untag 
WARNING: This package has not been validated, it may not work properly.
0.	Cancel
1.	963961de6b2409c2b3185b18fd3d322bd136ae7c (MST VMM9)
2.	9923e66b0a6fc2a46283d59e454aa3a08a09730d (RTS0 Gen 1 USB Hub)
3.	8a345b397b1043afa96508a2808b8b23e5ab907f (RTS0 Gen 2 USB Hub)
4.	c41d6a881e088a321d4ed2cb26fa166a9c10a0be (RTS5 Gen 2 USB Hub)
5.	66e0cafaee62339dffa9af21dd6eb69d99e24c1d (XXXXXXX)
6.	287da0651bbe7af557b588ce0c9aeaa9a39487a6 (USB4 controller)
Choose device [0-6]: 
```

**Steps to Reproduce**
Preparation:
1. start daemon
2. connect kestrel usb devices
3. fwupdmgr emulation-tag (select the listed devices)
4. stop daemon
5. disconnect the kestrel usb devices

Recording:
1. start daemon
2. connect kestrel usb devices
3. fwupdmgr emulation-save all.zip
4. stop daemon
5. disconnect the kestrel usb devices

Play:
1. start daemon
2. fwupdmgr emulation-load all.zip
3. fwupdmgr get-devices --filter=emulated  ==> no devices are displayed

**Expected behavior**
The tagged devices should be listed.

**fwupd version information**
```shell
$ fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.2
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.fwupd.source  2.0.1-238-gdcf2bfafe
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.8.0-48-generic
runtime   org.freedesktop.fwupd         2.0.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): source, at dcf2bfafe25.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
$ fwupdmgr get-devices --show-all-devices
WARNING: This package has not been validated, it may not work properly.
Dell Inc. Latitude 7350
│
├─BC901 NVMe SK hynix 512GB:
│     Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
[... 297 lines cut ...]
│     Current version:    RQR12.11_B0032
│     Bootloader Version: BOT01.04_B0016
│     Vendor:             USB:0x046D|HIDRAW:0x046D
│     Install Duration:   30 seconds
│     GUID:               279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
│     Device Flags:       • Updatable
│                         • Signed Payload
│                         • Can tag for emulation
│   
└─YJ3JK:
      Device ID:          aec1a869eb0df71b7cea6b3ac71d39b830faf164
      Vendor:             PNP:AUO
      GUID:               340848af-fb81-553f-a1d4-9b81b5539548 ← DRM\VEN_AUO&DEV_CFA4
      Device Flags:       • Internal device
                          • Can tag for emulation

```

</details>

**Additional questions**

- Operating system and version: Ubuntu 24.04
- Have you tried rebooting? Yes
- Is this a regression? No

superm1, 2024-11-09:
Can up please attach an emulation with all devices tagged? This will make working out the problem in the daemon a lot easier to track down.

CragW, 2024-11-09:
> Can up please attach an emulation with all devices tagged? 

The other components are gtype devices enumerated within ec that cannot be tagged (i.e.: lack of `can tag for emulation` flag), the rest component in this plugin is the `rmm` device that is missing from kernel is due to known issues.

superm1, 2024-11-11:
> > lack of can tag for emulation flag
> 
> Can we add the flag in `_init()` for those devices? or @superm1 maybe a heuristic makes sense, e.g. if you're adding a child to a device that can tag-for-emulation maybe it makes sense to automatically tag the child with the same?

If the children are normally made automatically by the plugin I don't think they really need to be can tag; particularly in the case of a proxy device.

But to a user point of view this is probably confusing. Maybe your suggestion makes sense and we just "internally" skip anything with a proxy.