fwupd/fwupd #8083: Failed to setup backend udev on ChromeOS in 2.0.1

rishabh-chromium, 2024-11-13:
**Describe the bug**
None of the fwupdmgr commands are working (get-devices, install) because I am getting this error in fwupd.log:
```
2024-11-13T22:08:09.181680Z INFO fwupd[5217]: FuEngine             failed to setup backend udev: failed to set up netlink: bind to udev socket failed: Address already in use
```

**Steps to Reproduce**
build and deploy

**fwupd version information**
Please provide the version of the daemon and client.

```shell
fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   com.hughsie.libxmlb           0.3.19
compile   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.19
runtime   com.hughsie.libjcat           0.2.0
runtime   org.kernel                    5.15.158-22492-ga9b51c249413
runtime   org.freedesktop.fwupd         2.0.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
No detected devices
```

</details>

**Additional questions**

- Operating system and version: chromeos
- Have you tried rebooting? yes
- Is this a regression? yes

rishabh-chromium, 2024-11-13:
`fwupdtool get-devices` works as expected

hughsie, 2024-11-14:
Is this a minijail issue? i.e. if you run `sudo /usr/libexec/fwupd/fwupd -v` do you see the same failure?

hughsie, 2024-11-14:
Also, are you *sure* the daemon isn't linking against libudev? e.g. what does `ldd /usr/libexec/fwupd/fwupd` say?

rishabh-chromium, 2024-11-14:
Running `sudo /usr/libexec/fwupd/fwupd -v` doesn't show the same failure. I can see this:
```
FuUdevBackend:      
  Name:                 udev        
  Enabled:              true 
  DoneSetup:            true
  CanInvalidate:        false      
```
Which means it's working. Probably a minijail issue then.

But  `ldd /usr/libexec/fwupd/fwupd` also shows be `libudev` in the list:
```
dragonair-rev2 ~ # ldd /usr/libexec/fwupd/fwupd
        linux-vdso.so.1 (0x00007fff6d9a0000)
        libfwupdengine.so => /usr/lib64/fwupd-2.0.1/libfwupdengine.so (0x00007beb86a8f000)
        libfwupd.so.3 => /usr/lib64/libfwupd.so.3 (0x00007beb86a2d000)
        libfwupdplugin.so => /usr/lib64/fwupd-2.0.1/libfwupdplugin.so (0x00007beb86940000)
        libgio-2.0.so.0 => /usr/lib64/libgio-2.0.so.0 (0x00007beb867e9000)
        libglib-2.0.so.0 => /usr/lib64/libglib-2.0.so.0 (0x00007beb866e0000)
        libgobject-2.0.so.0 => /usr/lib64/libgobject-2.0.so.0 (0x00007beb86698000)
        libxmlb.so.2 => /usr/lib64/libxmlb.so.2 (0x00007beb86677000)
        libc.so.6 => /lib64/libc.so.6 (0x00007beb864b1000)
        libjcat.so.1 => /usr/lib64/libjcat.so.1 (0x00007beb8649e000)
        libjson-glib-1.0.so.0 => /usr/lib64/libjson-glib-1.0.so.0 (0x00007beb8647c000)
        libusb-1.0.so.0 => /usr/lib64/libusb-1.0.so.0 (0x00007beb86462000)
        libsqlite3.so.0 => /usr/lib64/libsqlite3.so.0 (0x00007beb862bb000)
        libgnutls.so.30 => /usr/lib64/libgnutls.so.30 (0x00007beb8611e000)
        libcurl.so.4 => /usr/lib64/libcurl.so.4 (0x00007beb860af000)
        libgmodule-2.0.so.0 => /usr/lib64/libgmodule-2.0.so.0 (0x00007beb860aa000)
        libz.so.1 => /usr/lib64/libz.so.1 (0x00007beb86095000)
        libarchive.so.13 => /usr/lib64/libarchive.so.13 (0x00007beb85ff9000)
        libmount.so.1 => /usr/lib64/libmount.so.1 (0x00007beb85fbe000)
        libpcre2-8.so.0 => /usr/lib64/libpcre2-8.so.0 (0x00007beb85f69000)
        libffi.so.6 => /usr/lib64/libffi.so.6 (0x00007beb85f60000)
        liblzma.so.5 => /usr/lib64/liblzma.so.5 (0x00007beb85f3b000)
        libzstd.so.1 => /usr/lib64/libzstd.so.1 (0x00007beb85e9e000)
        /lib64/ld-linux-x86-64.so.2 (0x00007beb86c43000)
        libhogweed.so.6 => /usr/lib64/libhogweed.so.6 (0x00007beb85e5b000)
        libudev.so.1 => /usr/lib64/libudev.so.1 (0x00007beb85e2b000)
        libm.so.6 => /lib64/libm.so.6 (0x00007beb85d8d000)
        libidn2.so.0 => /usr/lib64/libidn2.so.0 (0x00007beb85d6e000)
        libunistring.so.2 => /usr/lib64/libunistring.so.2 (0x00007beb85bf9000)
        libtasn1.so.6 => /usr/lib64/libtasn1.so.6 (0x00007beb85be8000)
        libnettle.so.8 => /usr/lib64/libnettle.so.8 (0x00007beb85baa000)
        libgmp.so.10 => /usr/lib64/libgmp.so.10 (0x00007beb85b18000)
        libcares.so.2 => /usr/lib64/libcares.so.2 (0x00007beb85b05000)
        libnghttp2.so.14 => /usr/lib64/libnghttp2.so.14 (0x00007beb85ae2000)
        libpsl.so.5 => /usr/lib64/libpsl.so.5 (0x00007beb85acf000)
        libssl.so.3 => /usr/lib64/libssl.so.3 (0x00007beb85a15000)
        libcrypto.so.3 => /usr/lib64/libcrypto.so.3 (0x00007beb85604000)
        libgssapi_krb5.so.2 => /usr/lib64/libgssapi_krb5.so.2 (0x00007beb855c1000)
        libbz2.so.1 => /usr/lib64/libbz2.so.1 (0x00007beb855af000)
        libxml2.so.2 => /usr/lib64/libxml2.so.2 (0x00007beb854a8000)
        libblkid.so.1 => /usr/lib64/libblkid.so.1 (0x00007beb8547a000)
        libkrb5.so.3 => /usr/lib64/libkrb5.so.3 (0x00007beb853da000)
        libk5crypto.so.3 => /usr/lib64/libk5crypto.so.3 (0x00007beb853b3000)
        libcom_err.so.2 => /usr/lib64/libcom_err.so.2 (0x00007beb853ae000)
        libkrb5support.so.0 => /usr/lib64/libkrb5support.so.0 (0x00007beb853a3000)
        libresolv.so.2 => /lib64/libresolv.so.2 (0x00007beb8538f000)
```

gwendalcr, 2024-11-14:
`fwupdtool get-devices` works fine.
`fwupdtool install --verbose ...` did not work with `libusb 1.26`: the error was identical to the one above. Using `strace`, showing with line number  we have:

```
1667 access("/run/udev/control", F_OK)       = 0
1668 socket(AF_NETLINK, SOCK_RAW|SOCK_CLOEXEC|SOCK_NONBLOCK, NETLINK_KOBJECT_UEVENT) = 12
...
3075 socket(AF_NETLINK, SOCK_RAW, NETLINK_KOBJECT_UEVENT) = 14
3076 bind(14, {sa_family=AF_NETLINK, nl_pid=16259, nl_groups=0x000002}, 12) = -1 EADDRINUSE (Address already in use)
3077 write(2, "06:38:35.704 ", 1306:38:35.704 )           = 13
3078 write(2, "FuEngine             ", 21FuEngine             )   = 21
3079 write(2, "failed to setup backend udev: fa"..., 107failed to setup backend udev: failed to set up netlink: bind to udev socket failed: Address already in use
```

The issue is before the [addition of libusb_init_context()](https://github.com/libusb/libusb/commit/6622f386f52807dac76c8a260c98aa02c311bc93), `libusb` open the netlink socket for udev event at init time, and `fwupdtool` also does it when the engine flag `FU_ENGINE_LOAD_FLAG_DEVICE_HOTPLUG` is set. So when calling `install`, we get an error.

I believe commit 5b1f8b18f6 ("Listen to the netlink udev socket rather than using GUdevClient") **only work with `libusb` that support `libusb_init_context`, when `HAVE_LIBUSB_INIT_CONTEXT` is true**.

Back to the case of fwupd at hand, it is different. When I run the command from the prompt, I don't this error, even it runs with a non-root UID. Only when I run in a [minijail](https://google.github.io/minijail/) I can repro the issue.

hughsie, 2024-11-14:
What libusb version is shipped in ChromeOS?

hughsie, 2024-11-15:
> [libudev 1.0.27](https://chromium-review.googlesource.com/c/chromiumos/overlays/chromiumos-overlay/+/6019787)

So that's new enough to have libusb_init_context() -- I still think the issue is somewhere in `minijail` I'm afraid.

rishabh-chromium, 2024-11-21:
Fixed minijail `-p` flag issue.

closed 2024-11-21 by rishabh-chromium