fwupd/fwupd #8086: Show ''failed to write-firmware: failed to read from port 0x0000: Resource temporarily unavailable"

sam412081go, 2024-11-14:
**Describe the bug**

I tried to update the BT mouse, but failed , it shows  `failed to write-firmware: failed to read from port 0x0000: Resource temporarily unavailable`.

debug.txt for all terminal message :
[debug.txt](https://github.com/user-attachments/files/17742592/debug.txt)

I found that this issue started from this commit :
```
commit b7dbc3daab99fca3d052055eb901c493f426215f
Author: Denis Pynkin <denis.pynkin@collabora.com>
Date:   Thu Sep 19 18:55:23 2024 +0200

    fix: set non-block fu-io-channel behavior by default for RDWR mode
    
    If HAVE_POLL_H is defined the non-block behaviour should be used for all
    modes by default.
```
```
------------------------ libfwupdplugin/fu-io-channel.c ------------------------
index 934164ec9..133b7d158 100644
@@ -541,7 +541,7 @@ fu_io_channel_new_file(const gchar *filename, FuIoChannelOpenFlag open_flags, GE
 #endif
 	if (open_flags & FU_IO_CHANNEL_OPEN_FLAG_READ &&
 	    open_flags & FU_IO_CHANNEL_OPEN_FLAG_WRITE) {
-		flags = O_RDWR;
+		flags |= O_RDWR;
 	} else if (open_flags & FU_IO_CHANNEL_OPEN_FLAG_READ) {
 		flags |= O_RDONLY;
 	} else if (open_flags & FU_IO_CHANNEL_OPEN_FLAG_WRITE) {

```

If I modify it back to `flags = O_RDWR;` , the update process can operating properly.

**Steps to Reproduce**
1. Connect BT mouse to PC.
2. use command` fwupdtool install --allow-older --allow-reinstall ./test.cab --verbose -v`

**Expected behavior**
Firmware update success.

**fwupd version information**

Newest fwupd source code:
```

commit number : ea006190a0ab57f7a53c5726a541d12cd6088ed6
Author: Richard Hughes <richard@hughsie.com>
Date:   Wed Nov 13 10:56:30 2024 +0000

    logitech-bulkcontroller: Add an enumeration-only test for Sight
```

```shell
fwupdmgr --version
```

```
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.2
compile   com.hughsie.libxmlb           0.3.21
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.fwupd.source  2.0.1-270-gea006190a
runtime   com.hughsie.libjcat           0.2.0
runtime   com.hughsie.libxmlb           0.3.21
runtime   org.kernel                    6.8.0-48-generic
runtime   org.freedesktop.fwupd         2.0.2
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

Install from fwupd source code

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
[all_devices.txt](https://github.com/user-attachments/files/17742587/all_devices.txt)

</details>

**Additional questions**

- Operating system and version:
- Have you tried rebooting?
- Is this a regression?

sam412081go, 2024-11-15:
[image]
@hughsie  I can't remove 'FU_IO_CHANNEL_OPEN_FLAG_NONBLOCK' flag in my plugins,  because it always open 'O_NONBLOCK' flag again 
[image]

hughsie, 2024-11-15:
@sam412081go that section:

#ifdef HAVE_POLL_H
	flags |= O_NONBLOCK;
#endif

looks completely bogus to me. I'm not sure why it's there. PR coming up.

closed 2024-11-15 by sam412081go

sam412081go, 2024-11-15:
@hughsie  Thanks for the fix.
I think the formal release version 2.0.1 has already included these issues, right?
Can you ask Google to cherry-pick these two commits to their ChromeOS to prevent our devices from failing to update? 
Thank you!
https://github.com/fwupd/fwupd/pull/8092
https://github.com/fwupd/fwupd/pull/8084

hughsie, 2024-11-15:
I'm releasing 2.0.2 on Tuesday, and I think that's the plan for the next ChromeOS release. @rishabh-chromium for visibility.

sam412081go, 2024-11-16:
Got it, thank you!