fwupd/fwupd #8090: Receiving: "WARNING: UEFI ESP partition may not be set up correctly" when trying to update

jarothenberg, 2024-11-15:
**Describe the bug**
Whenever I run "fwupdmgr update", I receive the message "WARNING: UEFI ESP partition may not be set up correctly". I have tried to go through previously issue reports and similar but have been unable to resolve the issue. Also, I am relatively new to Ubuntu and posting issues online, so I apologize if I do not understand everything or do not use the proper etiquette.

**Steps to Reproduce**
Run command "fwupdmgr update" in terminal.

**Expected behavior**
No warning to appear, list available devices and their firmware updates.

**fwupd version information**
```compile   org.freedesktop.fwupd         1.9.24
compile   com.hughsie.libxmlb           0.3.18
compile   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.fwupd-efi     1.4
compile   org.freedesktop.gusb          0.4.8
runtime   com.hughsie.libxmlb           0.3.x
runtime   com.hughsie.libjcat           0.2.0
runtime   org.freedesktop.gusb          0.4.8
runtime   org.freedesktop.fwupd         1.9.24
runtime   org.kernel                    6.8.0-48-generic```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
From terminal, using apt.

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

```WARNING: UEFI ESP partition may not be set up correctly
See https://github.com/fwupd/fwupd/wiki/PluginFlag:esp-not-valid for more information.
Dell Inc. Precision 7780
│
├─Unknown Device:
│     Device ID:          55ce9dc5944a287f47229229ad17036091ee3c32
│     GUID:               18ca092c-a388-5165-ba08-9724e965505c ← GPIO\ID_INTC1085:00
│   
├─13th Gen Intel Core™ i9-13950HX:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000129
│     Vendor:             Intel
│     GUIDs:              3d4645d8-13fe-541a-ac5c-7d9d944e4500 ← CPUID\PRO_0&FAM_06&MOD_B7
│                         988c4cf1-ace4-5eb5-b26c-905ff236f412 ← CPUID\PRO_0&FAM_06&MOD_B7&STP_1
│     Device Flags:       • Internal device
│   
├─1D8HG:
│     Device ID:          6b829906086e67fd3a1c305fa868d2cea355db96
│     GUID:               3061c373-7ab8-55dc-b3ec-11367d6790d5 ← DRM\VEN_AUO&DEV_97A2
│     Device Flags:       • Internal device
│   
├─AD103GLM [RTX 5000 Ada Generation Laptop GPU]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE, PCI:0x8086)
│     GUIDs:              1b8f96d5-8bbf-5d4e-84a7-70bc1efae59a ← PCI\VEN_10DE&DEV_2730
│                         cf87e315-6375-5097-8d1d-9c8b83b18503 ← PCI\VEN_10DE&DEV_2730&SUBSYS_10280C42
│                         ce3effe2-609e-5118-9430-9dd9c19c492a ← PCI\VEN_8086&DEV_A70D
│                         29220a8f-72d7-5627-93ea-12cb86e5095c ← PCI\VEN_8086&DEV_A70D&SUBSYS_10280C42
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Integrated Webcam FHD:
│     Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
│     Current version:    24.31
│     Vendor:             CN0D3K118LG0041OA9D6A00 (USB:0x0C45)
[... 566 lines cut ...]
├── memtest86+ia32.efi
├── memtest86+x64.bin
├── memtest86+x64.efi
├── System.map-6.8.0-47-generic
├── System.map-6.8.0-48-generic
├── vmlinuz -> vmlinuz-6.8.0-48-generic
├── vmlinuz-6.8.0-47-generic
├── vmlinuz-6.8.0-48-generic
└── vmlinuz.old -> vmlinuz-6.8.0-47-generic

6 directories, 305 files```

**Additional questions**

- Operating system and version: Ubuntu 24.04.1 LTS
- Have you tried rebooting?:
    - Yes
- Is this a regression?: 
    - No
- Are you using an NVMe disk?: 
    - Yes
- Is secure boot enabled?: 
    - No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?:
    - No

superm1, 2024-11-15:
Can you please share the output for `sudo fwupdtool esp-list -vv`?  And did you already try to fix it following https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Invalid-ESP-Partition?

jarothenberg, 2024-11-15:
```18:58:56.720 FuDebug              verbose to debug (on console 1)
18:58:56.720 FuEngine             starting fwupd 1.9.24…
18:58:56.720 FuMain               locked /run/lock/fwupdtool
18:58:56.740 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
18:58:56.740 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Loading…                 [ -                                     ]18:58:56.759 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
18:58:56.759 FuConfig             loading config /etc/fwupd/fwupd.conf
18:58:56.759 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
18:58:56.759 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
18:58:56.759 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
18:58:56.759 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
18:58:56.759 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
18:58:56.759 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
18:58:56.759 FuConfig             ::configuration loaded
Loading…                 [                                       ]18:58:56.759 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
18:58:56.760 Jcat                 loaded 1 certificates
18:58:56.760 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
18:58:56.760 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
18:58:56.760 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
18:58:56.760 Jcat                 loaded 1 certificates
18:58:56.760 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
18:58:56.760 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
18:58:56.760 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
18:58:56.760 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
18:58:56.760 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]18:58:56.760 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
18:58:56.761 FuHistory            got schema version of 12
Loading…                 [*                                      ]18:58:56.764 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-1.9.24/libfu_plugin_modem_manager.so)
18:58:56.766 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
18:58:56.766 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
18:58:56.766 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
18:58:56.766 XbSilo               file: a7f92651-5633-8ff0-1b1f-996c975ff7e6, current:a7f92651-5633-8ff0-1b1f-996c975ff7e6, cached: (null)
18:58:56.766 XbSilo               loading silo with file contents
Loading…                 [*                                      ]18:58:56.768 FuConfig             loading config /etc/fwupd/fwupd.conf
18:58:56.768 FuConfig             not loading config /var/etc/fwupd/fwupd.conf
18:58:56.768 FuConfig             skipping mode check for /var/etc/fwupd/fwupd.conf as not writable
18:58:56.768 FuConfig             trying to load config values from /etc/fwupd/fwupd.conf
18:58:56.768 FuCommon             mapped file /etc/fwupd/fwupd.conf of size 0x33
18:58:56.768 FuConfig             trying to load config values from /var/etc/fwupd/fwupd.conf
18:58:56.768 FuConfig             Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory
18:58:56.768 FuConfig             ::configuration loaded
18:58:56.768 FuStruct             SmbiosEp32:
  anchor_str: _SM_
  entry_point_csum: 0xed
  entry_point_len: 0x1f
  smbios_major_ver: 0x3
  smbios_minor_ver: 0x6
  max_structure_sz: 0x18c
  entry_point_rev: 0x0
  intermediate_anchor_str: _DMI_
  intermediate_csum: 0xc1
  structure_table_len: 0x179d
  structure_table_addr: 0x4d43d000
  number_smbios_structs: 0x5d
  smbios_bcd_rev: 0x36
18:58:56.768 FuStruct             SmbiosStructure:
  type: 0xda
  length: 0xfb
  handle: 0xda00
18:58:56.768 FuStruct             SmbiosStructure:
  type: 0xda
  length: 0xfb
  handle: 0xda01
18:58:56.768 FuStruct             SmbiosStructure:
  type: 0xda
  length: 0xfb
  handle: 0xda02
18:58:56.768 FuStruct             SmbiosStructure:
  type: 0xda
  length: 0xfb
[... 898 lines cut ...]
18:58:59.314 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: ebd0a0a2-b9e5-4433-87c0-68b6b72699c7, internal: 1, fs: ntfs
18:58:59.314 FuVolume             adding partition 'Basic data partition'
18:58:59.332 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme2n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
18:58:59.349 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: e3c9e316-0b5c-4db8-817d-f92df00215ae, internal: 1, fs: 
18:58:59.367 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
[... cut ...]
18:59:18.474 FuMain               Request canceled

18:59:18.476 FuPlugin             finalize(modem_manager)
18:59:18.476 FuPlugin             finalize(flashrom)
18:59:18.476 FuPluginTest         destroy```

I did try doing that but it did not seem to resolve the issue (assuming I did it correctly). I did it on all of my disks that have an ESP partition, which is my Windows 10 and Ubuntu 24.04 disks (I have a dual boot system). When I ran the command it gave me this message.
```Information: You may need to update /etc/fstab.```

superm1, 2024-11-15:
Have you put anything in `fwupd.conf`?

jarothenberg, 2024-11-15:
No. The file ```fwupd.conf``` located at \etc\fwupd\ only contains the following:

```
[fwupd]
# use `man 5 fwupd.conf` for documentation
```

superm1, 2024-11-15:
Can you please try to configure your preferred ESP using the `EspLocation` key?  It seems you have multiple candidates.
https://fwupd.github.io/libfwupdplugin/fwupd.conf.html

jarothenberg, 2024-11-15:
What should I set the location to? The ESP on my Ubuntu drive? Would that just be ```EspLocation = /dev/nvme2n1p1``` since my Ubuntu drive is /dev/nvme2n1 and the ESP is partition 1? Would I have to put it under any particular section, or would just the bottom of the file work?

superm1, 2024-11-15:
It needs to be in a section with [fwupd].

I don't recall for sure if we take character devices or mounted paths in that key though

jarothenberg, 2024-11-15:
I'm not particularly sure what to set it to, but I tried the following different configurations with no success:
```EspLocation = /dev/nvme2n1p1```
```EspLocation = /dev/nvme2n1```
```EspLocation = /org/freedesktop/UDisks2/block_devices/nvme2n1p1```
```EspLocation = /org/freedesktop/UDisks2/block_devices/nvme2n1```
I tried those individually, so only one of them at a time in the file, but still received the same warning. Do I have to restart in between?

jarothenberg, 2024-11-15:
I also tried rebooting my system between each change and I had the same result with the same warning.

superm1, 2024-11-15:
Try mounting it somewhere first and then putting the mount point in that key

jarothenberg, 2024-11-15:
I'm not sure what you mean.

superm1, 2024-11-16:
The ESP you want, does it get mounted automatically somewhere because you have it specified in /etc/fstab?  That's the path to try to put it there.

jarothenberg, 2024-11-16:
How do I know which one I want? I am trying to use this software to update the firmware of my system. I figured that the drive that my Ubuntu is running on is probably mounted by default, unless I'm misunderstanding something. I think I'm a bit confused.

superm1, 2024-11-16:
Probably the same one Ubuntu is on that has this boot entry:
```
Boot0005* Ubuntu	HD(1,GPT,96e051c0-4c5e-11ee-97ac-eb36cb1a70cc,0x800,0x1bd000)/File(\EFI\ubuntu\shimx64.efi)
```

jarothenberg, 2024-11-16:
So how do I find the mount point to specify in the config file?

superm1, 2024-11-16:
Usually ubuntu uses /boot/efi for that one IIRC.  You can try to look at `udisksctl dump` output to find it to confirm.

jarothenberg, 2024-11-16:
The output of that is quite large so I am attaching it as a txt file.
[dump.txt](https://github.com/user-attachments/files/17782855/dump.txt)

superm1, 2024-11-16:
Yeah put `/boot/efi` in.

jarothenberg, 2024-11-16:
Thank you so much! That seems to have resolved the issue as I am no longer receiving that warning message. It is worth noting that I had to reboot for the changes to take effect. Before I close, one last concern. I noticed that when I run ```sudo fwupdtool esp-list -vv``` there were several lines indicating that something went wrong like ```Failed to open file “/var/etc/fwupd/fwupd.conf”: No such file or directory``` and ```failed to load fdt: cannot find /sys/firmware/fdt or override /var/lib/fwupd/system.dtb```. Are these things that I should be concerned about or look into correcting or are they expected and just a part of the normal behavior for how the package runs?

superm1, 2024-11-16:
> Thank you so much! That seems to have resolved the issue as I am no longer receiving that warning message. It is worth noting that I had to reboot for the changes to take effect. Before I close, one [...]

That's good to hear it solved. Those messages are all normal.

jarothenberg, 2024-11-16:
Awesome!

During my testing of this package, I found something very odd. When I run ```fwupdmgr get-updates```, it seems to terminate somewhere in that script before it finishes running. I was trying to run ```fwupdmgr get-updates && fwupdmgr update``` so it happens in one line, but it never reaches ```fwupdmgr update```. I tested this by putting an echo between the two and the echo never runs: ```fwupdmgr get-updates && echo TEST && fwupdmgr update```. Should I make a new issue report or am I missing something?

superm1, 2024-11-16:
If you're scripting it you should use the --json argument and interpret returns.

When nothing to do the return code is specifically "2"

jarothenberg, 2024-11-16:
Understood, thanks! Adding that argument fixed it. Thank you so much for your help. Am I good to close?