fwupd/fwupd #8102: Memory leak in the Nordic plugin

hughsie, 2024-11-17:
This is what Valgrind tells me running the nrf emulation but I'm drawing a blank on the cause:

	==1273947== 227,816 (216 direct, 227,600 indirect) bytes in 1 blocks are definitely lost in loss record 11,074 of 11,075
	==1273947==    at 0x50CF03E: g_type_create_instance (in /usr/lib64/libgobject-2.0.so.0.8200.2)
	==1273947==    by 0x50B41C3: ??? (in /usr/lib64/libgobject-2.0.so.0.8200.2)
	==1273947==    by 0x50B57DD: g_object_new_with_properties (in /usr/lib64/libgobject-2.0.so.0.8200.2)
	==1273947==    by 0x50B6800: g_object_new (in /usr/lib64/libgobject-2.0.so.0.8200.2)
	==1273947==    by 0x49FBB3A: fu_nordic_hid_archive_parse_file_image_create (fu-nordic-hid-archive.c:45)
	==1273947==    by 0x49FC4C8: fu_nordic_hid_archive_parse (fu-nordic-hid-archive.c:318)
	==1273947==    by 0x4D9EC59: fu_firmware_parse_stream (fu-firmware.c:1156)
	==1273947==    by 0x4D7619D: fu_device_prepare_firmware (fu-device.c:5144)
	==1273947==    by 0x4D75D01: fu_device_write_firmware (fu-device.c:5064)
	==1273947==    by 0x4DCB464: fu_plugin_device_write_firmware (fu-plugin.c:771)
	==1273947==    by 0x4DCF492: fu_plugin_runner_write_firmware (fu-plugin.c:2236)
	==1273947==    by 0x48929B3: fu_engine_write_firmware (fu-engine.c:3327)

Any ideas?!

hughsie, 2024-11-17:
@d4s @superm1 https://github.com/fwupd/fwupd/pull/8103 didn't work so I admit defeat. Show me how dumb I am please. :)

superm1, 2024-11-17:
#8103 looks fine to me.  But I'm a bit surprised that `fu_nordic_hid_archive_parse_file_image_create` even works without `FU_FIRMWARE()` casts on the new gobject.

closed 2025-05-03 by hughsie