fwupd/fwupd #8104: Untrusted loop bound in vli plugin

hughsie, 2024-11-17:
Coverity comes up with this - i.e. a corrupt file could convince us to erase more than just the usb3 section:

	1273        /* get the size and offset of the update firmware */
	1274        buf_fw = g_bytes_get_data(fw, &buf_fwsz);
	1275        st_hd = fu_struct_vli_usbhub_hdr_parse(buf_fw, buf_fwsz, 0x0, error);
	       7. Condition st_hd == NULL, taking false branch.
	1276        if (st_hd == NULL)
	1277                return FALSE;
	1278        hd2_fw_sz = (fu_struct_vli_usbhub_hdr_get_usb3_fw_sz_high(st_hd) << 16);
	/plugins/vli/libfu_plugin_vli.a.p/fu-vli-struct.c
	385fu_struct_vli_usbhub_hdr_get_usb3_fw_sz_high(const FuStructVliUsbhubHdr *st)
	       8. tainted_return_value: Function fu_struct_vli_usbhub_hdr_get_usb3_fw_sz returns tainted data.[show details]
	       9. var_assign_var: Compound assignment involving tainted variable fu_struct_vli_usbhub_hdr_get_usb3_fw_sz(st_hd) to variable hd2_fw_sz taints hd2_fw_sz.
	1279        hd2_fw_sz += fu_struct_vli_usbhub_hdr_get_usb3_fw_sz(st_hd);
	       10. Condition hd2_fw_sz == 0, taking false branch.
	...      
	CID 1601438: (#1 of 1): Untrusted loop bound (TAINTED_SCALAR)
	11. tainted_data: Passing tainted expression hd2_fw_sz to fu_vli_device_spi_erase, which uses it as a loop boundary.[show details]
	       Ensure that tainted values are properly sanitized, by checking that their values are within a permissible range.
	1299        if (!fu_vli_device_spi_erase(FU_VLI_DEVICE(self),
	1300                                     hd2_fw_addr,
	1301                                     hd2_fw_sz,
	1302                                     fu_progress_get_child(progress),
	1303                                     error))
	1304                return FALSE;

@memily can you please tell us the **maximum** size of the USB3 firmware please. It's the thing calculated by:

	hd2_fw_sz = (fu_struct_vli_usbhub_hdr_get_usb3_fw_sz_high(st_hd) << 16);
	hd2_fw_sz += fu_struct_vli_usbhub_hdr_get_usb3_fw_sz(st_hd);

i.e. the thing found in:

	struct FuStructVliUsbhubHdr {
	    dev_id: u16be,
	    strapping1: u8,
	    strapping2: u8,
	    usb3_fw_addr: u16be,
	    usb3_fw_sz: u16be,
	    usb2_fw_addr: u16be,
	    usb2_fw_sz: u16be,
	    usb3_fw_addr_high: u8,
	    usb3_fw_sz_high: u8,

memily, 2024-11-21:
Thank you for waiting, Sherlock and I will look into it and get back to you soon.

SherlockChu219, 2024-11-22:
the maximum size of USB3 firmware is 256K-Byte ( 0x40000 ).

closed 2024-11-22 by hughsie