fwupd/fwupd #8118: troubles setting up offline lvfs mirror

CorneelSoete, 2024-11-20:
Hi good people from fwupd,

So im trying to setup an lvfs mirror in an airgapped system. I followed the steps from here: https://lvfs.readthedocs.io/en/latest/offline.html and used the sync-pulp.py script for the cloning of the repo
Although I set all the remotes to false except for my new one, when i run `fwupdmg update` i always get the error saying `Failed to download file: Could not resolve host: fwupd.org

and when running fwupdmgr update --verbose i can see that it tries to download from https://fwupd.org/ but since its offline its logical that this doesnt work.

maybe one more important piece of info:
my repo doesnt include the firmware.xml.gz which i set as MetadataBaseURI. The repo instead contains 83 firmware-xxxxx-{testing,stable}.xml.{xz,zst,gz} files

**fwupd version information**
versopm: 1.8.12

If anyone can help this would be greatly appreciated

hughsie, 2024-11-20:
> Failed to download file: Could not resolve host: fwupd.org

I think this might be the same as https://gitlab.com/fwupd/lvfs-website/-/commit/598f2b818aa5e92ec33e03087245358b1c136b00 -- can you try re-syncing with todays metadata and see if it's magically fixed?

CorneelSoete, 2025-04-02:
This was no longer a priority so i didn't look at it since until today.
I re-synced today and still no change (after running fwupdmgr update it still says Failed to download file: Could not resolve host: fwupd.org

My remote is configured as followed:
/etc/fwupd/remotes.d/myremote.conf

[fwupd Remote]
Enabled=true
Type=download
MetadataURI=http://PATH_TO_MIRROR/firmware.xml.gz
FirmwareBaseURI=http://PATH_TO_MIRROR

and all the other remotes in /etc/fwupd/remotes.d are disabled by changing the Enabled=true into Enabled=false

It seems to work for the metadata, just not for the actual download of the firmware since fwupdmgr get-updates runs into no errors and tells me what updates can be done.

hughsie, 2025-04-03:
@CorneelSoete I'm sorry you're still having problems. What does `fwupdmgr get-devices --json` output for you? Specifically the `Locations` is what I'm interested in.

closed 2025-05-03 by hughsie