fwupd/fwupd #8122: fwupdmgr get-devices --json gives file name in URI

rishabh-chromium, 2024-11-20:
Getting "Uri" : "b1275193349c3ebd1db3821462143ac859302afe3572060fab7ee4793b4892bd-DellDockFirmwareUpdateLinux_01.00.16.cab" when I did fwupdmgr get-devices --json for all releases. This should ideally be the relative URI, based on the MetadataURI in lvfs.conf

hughsie, 2024-11-20:
> Uri....

You mean Locations? or is this 1.9.x? e.g. I get: 

    {
      "Name" : "WDC PC SN720 SDAQNTW-256G-1001",
      "DeviceId" : "08e1798bf5d9cb56a0290b552cab6c1a371b5089",
      "InstanceIds" : [
        "NVME\\VEN_15B7&DEV_5002",
        "NVME\\VEN_15B7&DEV_5002&SUBSYS_15B75002",
        "WDC PC SN720 SDAQNTW-256G-1001"
      ],
      "Guid" : [
        "ff2112dc-038c-596d-90ca-d43c5077c6ec",
        "c528df4b-7972-5880-8cb1-330415e2dc6a",
        "237776ee-0bcd-5fe9-8dc8-6984a2d36ba0"
      ],
      "Serial" : "183985804591",
      "Summary" : "NVM Express solid state drive",
      "Plugin" : "nvme",
      "Protocols" : [
        "org.nvmexpress"
      ],
      "Flags" : [
        "internal",
        "updatable",
        "require-ac",
        "supported",
        "needs-reboot",
        "usable-during-update",
        "can-emulation-tag"
      ],
      "Vendor" : "Sandisk Corp",
      "VendorIds" : [
        "PCI:0x15B7"
      ],
      "Version" : "10190101",
      "VersionFormat" : "plain",
      "Icons" : [
        "drive-harddisk"
      ],
      "Created" : 1732129505,
      "Releases" : [
        {
          "AppstreamId" : "com.lenovo.PCSN720.firmware",
          "ReleaseId" : "4305",
          "RemoteId" : "lvfs",
          "Name" : "PCSN720",
          "Summary" : "WD PCSN720 Firmware Update for Lenovo PC",
          "Description" : "<p>Supported devices and firmware version: WDC PCSN720 NVMe SSD Firmware for Lenovo PC</p>",
          "Version" : "10190101",
          "Filename" : "Lenovo-WD-PCSN720-NVMe-10190101.cab",
          "Protocol" : "org.nvmexpress",
          "Categories" : [
            "X-SolidStateDrive",
            "X-Device"
          ],
          "Checksum" : [
            "aed65b1eae17a6b8dbe1441bad4c6f99fbe17c50",
            "8e2a52b74fcdbcf2f82bf2e0f6b750d6f098be887bb1e4c6acba3b63454a4f73"
          ],
          "License" : "LicenseRef-proprietary",
          "Size" : 2277376,
          "Created" : 1577426837,
          "Locations" : [
            "https://fwupd.org/downloads/8e2a52b74fcdbcf2f82bf2e0f6b750d6f098be887bb1e4c6acba3b63454a4f73-Lenovo-WD-PCSN720-NVMe-10190101.cab"
          ],
          "Homepage" : "https://www.lenovo.com",
          "Vendor" : "Lenovo",
          "Flags" : [
            "trusted-metadata"
          ]
        }
      ]
    }

rishabh-chromium, 2024-11-20:
I have both:
```
"Locations" : [
            "b1275193349c3ebd1db3821462143ac859302afe3572060fab7ee4793b4892bd-DellDockFirmwareUpdateLinux_01.00.16.cab"
          ],
          "Uri" : "b1275193349c3ebd1db3821462143ac859302afe3572060fab7ee4793b4892bd-DellDockFirmwareUpdateLinux_01.00.16.cab",

```

rishabh-chromium, 2024-11-20:
Yes, this is 1.9.22

hughsie, 2024-11-20:
Hmm, I get a full URL for 1_9_X too:

          "Locations" : [
            "https://cdn.fwupd.org/downloads/8e2a52b74fcdbcf2f82bf2e0f6b750d6f098be887bb1e4c6acba3b63454a4f73-Lenovo-WD-PCSN720-NVMe-10190101.cab"
          ],
          "Uri" : "https://cdn.fwupd.org/downloads/8e2a52b74fcdbcf2f82bf2e0f6b750d6f098be887bb1e4c6acba3b63454a4f73-Lenovo-WD-PCSN720-NVMe-10190101.cab",

One thing it might be; are you still using the gzip metadata?

hughsie, 2024-11-20:
Nope, I tried with xz and gz, and those both worked. Two things;

 * can you paste the meson configure log, i.e. the line where it has `lvfs_metadata_format`
 * can you paste your lvfs.conf please

hughsie, 2024-11-20:
It's even correct using the 1.9.22 tag. Something is weird! :)

superm1, 2024-11-20:
I think the lvfs conf in use must be missing `FirmwareBaseUri`

rishabh-chromium, 2024-11-20:
lvfs.conf:
```
[fwupd Remote]

# this remote provides metadata and firmware marked as 'stable' from the LVFS
Enabled=true
Title=Linux Vendor Firmware Service
MetadataURI=https://storage.googleapis.com/chromeos-localmirror/lvfs/firmware.xml.xz
ReportURI=https://fwupd.org/lvfs/firmware/report
SecurityReportURI=https://fwupd.org/lvfs/hsireports/upload
AutomaticReports=false
AutomaticSecurityReports=false
ApprovalRequired=false
FirmwareBaseURI=https://storage.googleapis.com/chromeos-localmirror/lvfs/

```

Correction: it is v1.9.21.
Working on getting meson config

superm1, 2024-11-20:
> I think the lvfs conf in use must be missing `FirmwareBaseUri`

Oh, based on what is shared - other way around.  It's setting `FirmwareBaseUri`.  We don't normally do that and so it goes down  a different path than we normally do in fwupd-remote.c in `fwupd_remote_build_uri`

superm1, 2024-11-20:
We dropped `FirmwareBaseUri` in main (see 861c9804a35897eda04b5d6a4918a8ca46d2f27f).  Are you guys moving up to 2.0.2 after the next release?  If so, this might not be worth changing anything in 1.9.x..

rishabh-chromium, 2024-11-20:
Removing the line `FirmwareBaseUri` fixed the issue, but based on this [pull request](https://github.com/fwupd/fwupd/pull/6877#issuecomment-1980426999), the old way should also work? Don't want older builds to start failing even after we uprev to 2.0.2

rishabh-chromium, 2024-11-20:
Confirming that 2.0.2 works even with the `FirmwareBaseUri` line. 1.9.22 breas.

superm1, 2024-11-20:
The pull request was purely to 2.0.x, not planned to backport.  So it sounds like if you move up to 2.0.2 things will be fine, and 1.9.x has this bug.  If your next release is 1.9.27 we should fix in 1.9.x, if it's 2.0.2 then I think this is "close won't fix"

rishabh-chromium, 2024-11-20:
Our next release is 2.0.1 but we still have all our devices on 1.9.22 until we release 2.0.2 to stable (~3 months out). Not sure what changed to have suddenly caused this but until a couple of days ago 1.9.22 was working, now suddenly the URIs are showing only filenames. Maybe something changed on lvfs side?

superm1, 2024-11-20:
2.0.1 should work too.  Can you have a try with that?

hughsie, 2024-11-21:
I've reverted the LVFS "fix" until we understand what's happened. It's also broken one of the ODMs too.

hughsie, 2024-11-21:
So, I dug into this a bit today. The LVFS change that made the URL into a filename was probably bogus; not because of mirroring -- but more that we started using the CDN to distribute packages.

e.g. `https://fwupd.org/downloads/fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13-DBXUpdate-20230509-x64.cab` -> `https://cdn.fwupd.org/downloads/fc3feb015df2710fcfa07583d31b5975ee398357016699cfff067f422ab91e13-DBXUpdate-20230509-x64.cab`

i.e. I think we actually do want a different URL for firmware binaries to the metadata. The alternative is doing something like `s/cdn./` but that feels like a bit of a hack. @superm1 would you be against me bringing back FirmwareBaseURI for 2.0.2?

superm1, 2024-11-21:
> .e. I think we actually do want a different URL for firmware binaries to the metadata. The alternative is doing something like s/cdn./ but that feels like a bit of a hack. @superm1 would you be against me bringing back FirmwareBaseURI for 2.0.2?

That's fine

closed 2024-11-21 by superm1