fwupd/fwupd #8136: Unknown Intel Device can't Verify Cryptographic Hash Despite Flag

RayJW, 2024-11-22:
**Describe the bug**

The following unknown Intel device has the flag set for cryptographic hash verification, despite throwing an error when trying to do so.

```shell
├─Unknown Device:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Current version:    0c
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUID:               eaad9970-8e4d-56da-88ab-41a8c1e2811f ← PCI\VEN_8086&DEV_46A6
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│                         • Can tag for emulation
...
❯ fwupdmgr verify eaad9970-8e4d-56da-88ab-41a8c1e2811f
Selected device: (null)
failed to verify (null): failed to read firmware: Error reading from file: Input/output error
```

While I did see #7909, I'm not sure if this is the same, as this device does not have a SD card reader and I do not have the same system service error.

```shell
❯ sudo systemctl status fwupd.service 
● fwupd.service - Firmware update daemon
     Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static)
     Active: active (running) since Sat 2024-11-23 00:19:40 CET; 1s ago
 Invocation: ee018e31c0ab43dfbcbd8e6bd15191b4
       Docs: https://fwupd.org/
   Main PID: 35847 (fwupd)
      Tasks: 5 (limit: 18698)
     Memory: 18.1M (peak: 19M)
        CPU: 452ms
     CGroup: /system.slice/fwupd.service
             └─35847 /usr/lib/fwupd/fwupd

Nov 23 00:19:39 host systemd[1]: Starting Firmware update daemon...
Nov 23 00:19:39 host fwupd[35847]: 23:19:39.999 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
Nov 23 00:19:39 host fwupd[35847]: 23:19:39.999 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Nov 23 00:19:40 host fwupd[35847]: 23:19:40.981 FuMain               Daemon ready for requests (locale en_GB.UTF-8)
Nov 23 00:19:40 host systemd[1]: Started Firmware update daemon.
```

**Steps to Reproduce**

Try to verify the hash of the device mentioned above.

**Expected behavior**

Either successfully verify the hash or don't mark the device as verifiable.

**fwupd version information**

```shell
❯ fwupdmgr --version                                  
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.1
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.1
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
runtime   com.hughsie.libjcat           0.2.1
runtime   org.kernel                    6.12.1-2-cachyos
runtime   org.freedesktop.fwupd         2.0.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman`

<details>

[... 252 lines cut ...]
└─WD BLACK SN850X 2000GB:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    620311WD
      Vendor:             Sandisk Corp (PCI:0x15B7)
      Serial Number:      22475Z801814
      Problems:           • Device requires AC power to be connected
      GUIDs:              69749ddc-cd7a-542e-981f-620656c59918 ← NVME\VEN_15B7&DEV_5030
                          53d3c0f5-42b2-5cc9-8a52-fa350528dc02 ← NVME\VEN_15B7&DEV_5030&SUBSYS_15B75030
                          c6686f27-9161-5f60-a26e-12268be7d913 ← WD_BLACK SN850X 2000GB
      Device Flags:       • Internal device
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Updatable
                          • Can tag for emulation
```

</details>

**Additional questions**

- Operating system and version: CachyOS fully updated
- Have you tried rebooting? Yes
- Is this a regression? Not sure

openmindead, 2024-12-05:
Can confirm a similar issue on AMD platform too. Lenovo laptop as well.
```
Loading…                 [*******************                    ]04:00:43.138 FuPluginDfu          failed to parse interface data: no image idx 33 found in firmware
04:00:43.138 FuPluginDfu          DFU version 0x0000 invalid, v1.1 assumed
Loading…                 [************************************** ]
LENOVO 21LF
│
├─AMD Ryzen 7 8845H w/ Radeon 780M Graphics:
│ │   Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│ │   Current version:    0x0a705206
│ │   Vendor:             Advanced Micro Devices, Inc.
│ │   GUIDs:              09243cb6-ee6a-50fb-9438-c5643d6302be ← CPUID\PRO_0&FAM_19&MOD_75
│ │                       88c5027f-27f8-534c-af31-4f0a21d90d5d ← CPUID\PRO_0&FAM_19&MOD_75&STP_2
│ │   Device Flags:       • Internal device
│ │ 
│ ├─AMD Radeon Graphics:
│ │ │   Device ID:        5195f11e5982a0f0c2ff0fe6cf84a36ce3dbc7c5
│ │ │   Summary:          AMD AMD_PHOENIX_GENERIC
│ │ │   Current version:  022.012.000.027.000001
│ │ │   Vendor:           Advanced Micro Devices, Inc. [AMD/ATI] (PCI:0x1002)
│ │ │   GUID:             4a1501b7-b500-5255-9d9c-41d652a4d5bc ← AMD\113-PHXGEN
│ │ │   Device Flags:     • Internal device
│ │ │                     • Can tag for emulation
│ │ │ 
│ │ ├─Unknown Device:
│ │ │     Device ID:      57e35fbedc32c4e5e69c55e3d7810f7ae6f1427a
│ │ │     Vendor:         PNP:ACI
│ │ │     Serial Number:  E3LMRS013337
│ │ │     GUID:           3597a06c-cef0-5989-92a8-e19ca7d48340 ← DRM\VEN_ACI&DEV_23E1
│ │ │     Device Flags:   • Can tag for emulation
│ │ │   
│ │ └─LEN145-3K:
│ │       Device ID:      aec1a869eb0df71b7cea6b3ac71d39b830faf164
│ │       Vendor:         PNP:LEN
│ │       GUID:           950e2a12-a8a0-5c6e-a4ac-2fea0320616c ← DRM\VEN_LEN&DEV_8AB1
│ │       Device Flags:   • Internal device
│ │                       • Can tag for emulation
│ │     
│ ├─Secure Processor:
│ │     Device ID:        c54ab0237d7a8db8c717b68e0be78e4374a2a079
│ │     Current version:  00.2d.00.7d
│ │     Bootloader Version: 00.2d.00.7d
│ │     Vendor:           Advanced Micro Devices, Inc. (PCI:0x1022)
│ │     GUID:             9eb6a793-7f97-5fb5-b49c-139d2ce3ee46 ← PCI\VEN_1022&DEV_15C7
│ │     Device Flags:     • Internal device
│ │                       • Can tag for emulation
│ │   
│ └─System Management Unit (SMU):
│       Device ID:        db0330716216c629bb2c07256e5d018f499eb6ce
│       Summary:          Microcontroller used within CPU/APU program 0
│       Current version:  76.87.0
│       Vendor:           Advanced Micro Devices, Inc.
│       GUID:             3ac3159e-6eef-5f6b-bc8d-67686b238747 ← /sys/devices/platform/AMDI0009:00
│       Device Flags:     • Internal device
│                         • Can tag for emulation
│     
├─GPIO controller:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│     Device Flags:       • Can tag for emulation
│   
├─Integrated Camera:
│     Device ID:          3ead601f43028c79656941fc3832e6b67412e2b8
│     Current version:    66.3
│     Vendor:             Chicony Electronics Co., Ltd (USB:0x04F2)
│     Serial Number:      0001
│     GUID:               9a863592-e39f-5c33-8c34-f3128bc23934 ← USB\VID_04F2&PID_B7E8
│     Device Flags:       • Updatable
│                         • Can tag for emulation
│   
[... 100 lines cut ...]
│                         b6679740-f453-56fc-aad9-5566835f7025 ← NVME\VEN_15B7&DEV_501A&SUBSYS_15B7501A
│                         ab820c0c-57bd-5021-b934-db3961830499 ← WD Blue SN570 1TB SSD
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│                         • Can tag for emulation
│   
└─YMTC YMSS2ED08D25MC:
      Device ID:          71b677ca0f1bc2c5b804fa1d59e52064ce589293
      Summary:            NVM Express solid state drive
      Current version:    LN3603D2
      Vendor:             Yangtze Memory Technologies Co.,Ltd (PCI:0x1E49)
      Serial Number:      YMB51T0JA241850Y1K
      GUIDs:              8159333a-dd9b-52d7-a304-ebbde77294ec ← NVME\VEN_1E49&DEV_1073
                          d226e2c1-69f2-5493-b13b-e28074868944 ← NVME\VEN_1E49&DEV_1073&SUBSYS_1E491073
                          d21a2f47-aa27-5644-a1c9-6627e7d6105a ← YMTC YMSS2ED08D25MC
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
                          • Can tag for emulation
```

hughsie, 2024-12-06:
Does https://github.com/fwupd/fwupd/pull/8215 help?

RayJW, 2024-12-07:
> Does #8215 help?

Not really, the output looks almost the same with `fwupdmgr v2.0.4` from the branch.

```shell
❯ fwupdmgr verify eaad9970-8e4d-56da-88ab-41a8c1e2811f
WARNING: This package has not been validated, it may not work properly.
Selected device: Alder Lake-P GT2 [Iris Xe Graphics]
failed to verify Alder Lake-P GT2 [Iris Xe Graphics]: failed to read firmware: failed to read 17: Input/output error
```

Although with the regular `fwupdmgr v2.0.3` it looks like the device is at least now recognized.

```shell
❯ fwupdmgr verify eaad9970-8e4d-56da-88ab-41a8c1e2811f
Selected device: Alder Lake-P GT2 [Iris Xe Graphics]
failed to verify Alder Lake-P GT2 [Iris Xe Graphics]: failed to read firmware: Error reading from file: Input/output error
```

Maybe you meant to comment on #7964, since this issue is not really a problem of timing out on an operation? The error is thrown basically immediately after attempting to verify.

hughsie, 2025-05-04:
@superm1 I do wonder if we ought to drop oprom support now. I don't think it's terribly useful.

superm1, 2025-05-04:
> @superm1 I do wonder if we ought to drop oprom support now. I don't think it's terribly useful.

Sure I'm fine with that.

closed 2025-05-04 by hughsie