fwupd/fwupd #8144: `fwupd-refresh.service` fails with no Internet connection resulting in degraded system

paulmenzel, 2024-11-24:
Debian sid/unstable with *fwupd* 1.9.26-2

After resuming the laptop and no Internet connection available, `systemctl status` shows the system in a degraded state, as the unit `fwupd-refresh.service` is in a failed state:

```
$ systemctl status | head -2
● abreu
    State: degraded
$ systemctl --failed
  UNIT                  LOAD   ACTIVE SUB    DESCRIPTION                           
● fwupd-refresh.service loaded failed failed Refresh fwupd metadata and update motd

Legend: LOAD   → Reflects whether the unit definition was properly loaded.
        ACTIVE → The high-level unit activation state, i.e. generalization of SUB.
        SUB    → The low-level unit activation state, values depend on unit type.

1 loaded units listed.
```

```
$ systemctl status fwupd-refresh.service
× fwupd-refresh.service - Refresh fwupd metadata and update motd
     Loaded: loaded (/usr/lib/systemd/system/fwupd-refresh.service; static)
     Active: failed (Result: exit-code) since Sun 2024-11-24 22:09:11 CET; 5min ago
 Invocation: 06d44d87462d4c0ca790180c2b5aca2b
TriggeredBy: ● fwupd-refresh.timer
       Docs: man:fwupdmgr(1)
    Process: 61600 ExecStart=/usr/bin/fwupdmgr refresh (code=exited, status=1/FAILURE)
   Main PID: 61600 (code=exited, status=1/FAILURE)
   Mem peak: 3.9M
        CPU: 98ms

Nov 24 22:09:11 abreu systemd[1]: Starting fwupd-refresh.service - Refresh fwupd metadata and update motd...
Nov 24 22:09:11 abreu fwupdmgr[61600]: Wird aktualisiert lvfs
Nov 24 22:09:11 abreu systemd[1]: fwupd-refresh.service: Main process exited, code=exited, status=1/FAILURE
Nov 24 22:09:11 abreu systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'.
Nov 24 22:09:11 abreu systemd[1]: Failed to start fwupd-refresh.service - Refresh fwupd metadata and update motd.
```

superm1, 2024-11-24:
Sounds all expected to me.  IE No internet, you can't refresh sources that need internet.  If you didn't refresh successfully a user should know.

What are you proposing _should_ happen?

paulmenzel, 2024-11-25:
> What are you proposing should happen?

If there is no Internet connection, it’s expected that downloads are going to fail. So I suggest a check, if the system is online and only fail, if it actually is. (No idea if some unit statement Condition exist for this.)

superm1, 2024-11-25:
This feels like it should be tied to `network-online.target`, but I don't really know how to use that as it pertains to a network going up and down, any ideas?

paulmenzel, 2024-11-25:
@bluca, do you have a suggestion?

superm1, 2024-11-25:
Hmm according to https://www.freedesktop.org/software/systemd/man/latest/systemd.special.html it doesn't keep tracking.  So yeah, this doesn't really seem viable to solve your complaint.

> network-online.target
Units that strictly require a configured network connection should pull in network-online.target (via a Wants= type dependency) and order themselves after it. This target unit is intended to pull in a service that delays further execution until the network is sufficiently set up. What precisely this requires is left to the implementation of the network managing service.

Note the distinction between this unit and network.target. This unit is an active unit (i.e. pulled in by the consumer rather than the provider of this functionality) and pulls in a service which possibly adds substantial delays to further execution. In contrast, network.target is a passive unit (i.e. pulled in by the provider of the functionality, rather than the consumer) that usually does not delay execution much. Usually, network.target is part of the boot of most systems, while network-online.target is not, except when at least one unit requires it. Also see [Running Services After the Network Is Up](https://systemd.io/NETWORK_ONLINE) for more information.
> All mount units for remote network file systems automatically pull in this unit, and order themselves after it. Note that networking daemons that simply provide functionality to other hosts (as opposed to consume functionality of other hosts) generally do not need to pull this in.
> systemd automatically adds dependencies of type Wants= and After= for this target unit to all SysV init script service units with an LSB header referring to the "$network" facility.
> Note that this unit is only useful during the original system start-up logic. After the system has completed booting up, it will not track the online state of the system anymore. Due to this it cannot be used as a network connection monitor concept, it is purely a one-time system start-up concept.

anarcat, 2025-01-03:
> What are you proposing _should_ happen?

i think if fwupmgr would exit with a status code specific for this, operators could deal with this accordingly.

in my case, on a laptop, it's fine if that scripts fails because of lack of network access, for example. i could add an override to the systemd unit to reflect that status code.

> I suggest a check, if the system is online and only fail, if it actually is. (No idea if some unit statement Condition exist for this.)

that that i'm aware of. i guess *maaaybe* `networkctl` could reflect that, but that assumes systemd-networkd is in operation, which is far from being granted.

If using network manager, however, it *seems* like a workaround would be to add:

```
[Service]
ExecCondition=nm-online --quiet
```

as an override.

i'll test this and report back.

anarcat, 2025-01-03:
> i'll test this and report back.

That actually works! Pretty simple fix for my problem. Just make sure network-manager properly knows about your network though, in my case `nm-online` was always succeeding because it was confused by the bridge created by libvirt. I had to tell NM to ignore that device with this:

```
root@angela:~# cat /etc/NetworkManager/conf.d/virbr.conf 
[device]
managed=0
ignore-carrier=1
match-device=interface-name:virbr0
```

superm1, 2025-01-04:
That's a pretty good workaround! Can you think of a way we can detect whether network manager is in use to make a more complicated ExecCondition? Maybe check for the existence of the binary or for any sockets it has open in `/run` or whether the service is unmasked for NM?

If we can come up with something reliable enough we can put something like that in the upstream service too.

superm1, 2025-01-06:
I've drafted a solution that could use a network monitor which was suggested at https://lists.freedesktop.org/archives/systemd-devel/2025-January/051055.html

https://github.com/fwupd/fwupd/pull/8275

Can you see if that works well for you?

anarcat, 2025-01-06:
> That's a pretty good workaround! Can you think of a way we can detect whether network manager is in use to make a more complicated ExecCondition? Maybe check for the existence of the binary or for any sockets it has open in `/run` or whether the service is unmasked for NM?

Not from the top of my head.

I'll point out that the `apt-daily.service` in Debian uses this:

```
ExecStartPre=-/usr/lib/apt/apt-helper wait-online
```

... which is essentially this chunk of C++ code:

https://sources.debian.org/src/apt/2.9.21/cmdline/apt-helper.cc/#L253-L273

which, essentially, checks network manager, networkd and conman, and fails if one of them is *active* **AND** not online.

Otherwise, as I said on the mailing list (or by email at least), I'm not sure this (#8275) is the right solution, as it allows an attacker to deny fwupd from providing updates, as it will fail silently. It would have been better to exit with a special exit code. There are errno's specifically for that kind of stuff that could be directly reused, for example:

```
$ errno -ls
[...]
ENETDOWN 100 Network is down
ENETUNREACH 101 Network is unreachable
[...]
ETIMEDOUT 110 Connection timed out
ECONNREFUSED 111 Connection refused
EHOSTDOWN 112 Host is down
EHOSTUNREACH 113 No route to host
[...]
```

... etc. Having a *single* error code would probably be better though...

closed 2025-01-07 by superm1