fwupd/fwupd #8146: Tools compiled with the latest code will fail to write-firmware

MankWang, 2024-11-25:
**Describe the question**
After compiling with the latest code, using the command fwupdmgr to upgrade the firmware will prompt 

'failed to write-firmware: failed after 30 retries: failed to open /dev/wwan0qcdm0: Operation not permitted'.

Whether entering the sudo password or switching to the root user, this error will always occur.
And when I downgraded to 1.9.26, there was no such error message. 
What could be the reason for this?

**fwupd version information**

```shell
fwupdmgr --version
compile   info.libusb                   1.0.27
compile   org.freedesktop.fwupd         2.0.2
compile   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.Passim        0.1.8
compile   com.hughsie.libjcat           0.2.2
runtime   org.freedesktop.fwupd-efi     1.9.24
runtime   org.freedesktop.Passim        0.1.8
runtime   com.hughsie.libxmlb           0.3.21
compile   org.freedesktop.fwupd.source  2.0.1-354-g88e009e76
runtime   com.hughsie.libjcat           0.2.2
runtime   org.kernel                    6.12.0-rc2
runtime   org.freedesktop.fwupd         2.0.2
```

hughsie, 2024-11-25:
> failed to open /dev/wwan0qcdm0: Operation not permitted

A permissions error perhaps? If you can get us the daemon `-vv` output that would help a lot.

lvoegl, 2024-11-25:
In [version 1.9.26](https://github.com/fwupd/fwupd/blob/21f2030b0772e392b6592671dcfa561f758ba184/plugins/modem-manager/fu-mm-device.c#L1324), we expected a `FWUPD_ERROR_INVALID_FILE` when switching to EDL mode, since the modem device changed and those ports are not available anymore. Starting from [version 2.0.0](https://github.com/fwupd/fwupd/blob/eb3bc94dfbc264c242929aca11434547eb502a75/plugins/modem-manager/fu-mm-device.c#L1363) this was removed, so instead of continuing with the update, you're seeing the error being returned here.

MankWang, 2024-11-26:
> In [version 1.9.26](https://github.com/fwupd/fwupd/blob/21f2030b0772e392b6592671dcfa561f758ba184/plugins/modem-manager/fu-mm-device.c#L1324), we expected a `FWUPD_ERROR_INVALID_FILE` when switching [...]

So what should I modify or set up now to perform a normal upgrade. This error is preventing us from using the latest version for firmware upgrades.

hughsie, 2024-11-27:
> This error is preventing us from using the latest version for firmware upgrades.

I don't have any hardware capable of being updated with the ModemManager plugin, and the [modem-manger plugin](https://app.codecov.io/gh/fwupd/fwupd/tree/main/plugins%2Fmodem-manager) currently has a 3.49% self test coverage score. I think someone with hardware is going to have to look at the code I'm afraid.

MankWang, 2024-11-28:
@hughsie @lvoegl 
I have discovered a serious bug, as shown in the screenshot below. The modification below will cause switching to EDL always fail and prevent firmware upgrade. Please help confirm this issue, as it will prevent us from proceeding with next month's testing plan. Thank you.
[image]

hughsie, 2024-11-28:
> Please help confirm this issue

I don't have any updatable modem hardware.

lvoegl, 2024-11-28:
I don't think this is an unknown bug, rather something that still has to be properly implemented as stated in fwupd#7741. I unfortunately also don't have any MHI PCI modems to test this with.

Although I haven't gotten very far yet, I've started looking into updating non-MBIM USB modems via Firehose/Sahara (which currently does not seem possible) and EDL mode switching works similarly there.

MankWang, 2024-11-29:
I see, so now it is necessary to add a notification to let fwupd know that wwan0firehose has been loaded, and then call attach() to get the edl_port information?
I'm not very familiar with the code flow in fwupd. If you have any modification suggestions, I can help test them.

lvoegl, 2024-11-29:
Basically yes. Correct me if I'm wrong, but I'd say the Sahara/Firehose update process should also follow what most other update protocols do - `->detach()` into the 'update' (EDL) mode/bootloader, load the Firehose program via Sahara if necessary, and then perform the actual Firehose update regularly in the `->write_firmware()` step.

MankWang, 2024-12-10:
Is there any modification to this now? I have tried to make some modifications on my end, but they did not take effect. I may still need your help. This problem has already hindered our testing progress.

lvoegl, 2024-12-13:
You can try if #8218 now finds your device. I didn't specifically change this for your device, but I assume it at least fixes the initial issue of switching to EDL mode.

MankWang, 2024-12-25:
When I try to install the latest compiled version of the code, I always have this error. I don't know what the problem is. Can you help me solve it?
build cmd:
meson build --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu -Dplugin_modem_manager=enabled
ninja -C build
sudo ninja -C build install
[image]

MankWang, 2024-12-25:
Previously, it was normal, but now whenever using code after 2.0.0, this problem persists. Even reinstalling the system or changing the laptop may have the same issues.

superm1, 2024-12-25:
Are you using the venv helper scripts? They should avoid this issue.

MankWang, 2024-12-25:
@superm1 
How to use the venv helper scripts? Can you provide instructions for use? Thanks!

superm1, 2024-12-25:
https://fwupd.github.io/libfwupdplugin/building.html

MankWang, 2024-12-25:
The latest code still has problem, which can cause fwupd crashes during the upgrade process, The log is as follows
[image]
[image]

superm1, 2024-12-25:
Can you please run under a debugger and get a backtrace?

lvoegl, 2024-12-25:
Your `self->port_edl` is NULL here if I look at your logs.

I've added [this](https://github.com/fwupd/fwupd/commit/c2542a0748deac2c773d82824be90d86d36f90fe#diff-5a375f230ee85cf307402aaabd8da6e6dbc8ad32e0a5e9f6d302a896a8387c4cR1038-R1042) to find the EDL port for USB devices. Something similar (or possibly the same) for PCI modems is probably missing around [here](https://github.com/fwupd/fwupd/blob/576e918b39b840d9d149f97b9a46f26925e7c35f/plugins/modem-manager/fu-mm-device.c#L1662).

lvoegl, 2024-12-25:
Nevertheless I think the final error in your log should still not happen, even if `port_edl` was `NULL`.

lvoegl, 2024-12-25:
```
fu_firehose_updater_finalize(GObject *object)
{
	...
	g_free(self->port);
	g_object_unref(self->sahara);
	...
}

fu_firehose_updater_new(const gchar *port, FuSaharaLoader *sahara)
{
	FuFirehoseUpdater *self = g_object_new(FU_TYPE_FIREHOSE_UPDATER, NULL);
	if (port != NULL)
		self->port = g_strdup(port);
	if (sahara != NULL)
		self->sahara = g_object_ref(sahara);
	return self;
}
```

We don't necessarily initialize `self->port` and `self->sahara` here but always free/unref them in finalize. That's seems like where the error came from.

MankWang, 2025-02-18:
Close this question, and continue tracking at #8382

closed 2025-02-18 by MankWang