fwupd/fwupd #8161: fwupd service dies if kernel is not configured for system suspend

jingai, 2024-11-26:
Debian sid/unstable, kernel 6.12.1, fwupd 2.0.2.

fwupd service dies if kernel is not configured for system suspend (CONFIG_SUSPEND):

```
20:29:07.310 FuPluginLinuxSleep   could not open /sys/power/mem_sleep: Error opening file /sys/power/mem_sleep: No such file or directory
20:29:07.394 GLib                 g_strsplit: assertion 'string != NULL' failed
```

I have tried disabling the linux_sleep plugin:
```
[fwupd]
DisabledPlugins=linux_sleep
```
and verified that the plugin was indeed disabled:
```
20:29:01.218 FuEngine             plugins disabled: linux_sleep, test, test_ble
```
but the service still dies there any time I try to use fwupdmgr (get-devices, refresh, etc).

This is a regression, as fwupd 1.9.9 would  complain about being unable to open /sys/power/mem_sleep, but gracefully continues on rather than crashing.

hughsie, 2024-11-26:
> g_strsplit: assertion 'string != NULL' failed

I don't think that's coming from the linux_sleep plugin. Can you do something like `sudo G_DEBUG=fatal-criticals gdb fwupdtool get-devices` and then `r` and `bt` to get a backtrace? You might need to run `fuwpdtool security` instead, or maybe even run the fwupd process too.

jingai, 2024-11-26:
`fwupdmgr get-devices` just prints to stderr:
```
Unable to connect to service
```
then exits normally (well, with error exit code), so there is no backtrace.

Here is the backtrace from fwud process:
```
# G_DEBUG=fatal-criticals gdb /usr/libexec/fwupd/fwupd
GNU gdb (Debian 15.2-1) 15.2
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/libexec/fwupd/fwupd...
(No debugging symbols found in /usr/libexec/fwupd/fwupd)
(gdb) r
Starting program: /usr/libexec/fwupd/fwupd 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff2fff6c0 (LWP 258818)]
[New Thread 0x7ffff21ff6c0 (LWP 258819)]
[New Thread 0x7ffff19fe6c0 (LWP 258820)]
[New Thread 0x7ffff11fd6c0 (LWP 258821)]
22:23:12.877 FuMain               Daemon ready for requests (locale en_US.UTF-8)
[Thread 0x7ffff19fe6c0 (LWP 258820) exited]
22:23:35.009 FuPluginLinuxSleep   could not open /sys/power/mem_sleep: Error opening file /sys/power/mem_sleep: No such file or directory
22:23:35.090 GLib                 g_strsplit: assertion 'string != NULL' failed

Thread 1 "fwupd" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff76626da in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff76626da in g_logv () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff7662943 in g_log () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff767ee25 in g_strsplit () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7d90507 in ??? () at /usr/lib/x86_64-linux-gnu/fwupd-2.0.2/libfwupdengine.so
#4  0x00007ffff7ca2f88 in ??? () at /usr/lib/x86_64-linux-gnu/fwupd-2.0.2/libfwupdengine.so
#5  0x00007ffff7cb25e6 in fu_engine_get_host_security_id () at /usr/lib/x86_64-linux-gnu/fwupd-2.0.2/libfwupdengine.so
#6  0x000055555555f34f in ??? ()
#7  0x00007ffff78c57d9 in ??? () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#8  0x00007ffff765781f in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#9  0x00007ffff7659a57 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007ffff765a4af in g_main_loop_run () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x000055555555ec95 in fu_daemon_start ()
#12 0x000055555555d497 in main ()
```

jingai, 2024-11-26:
fwupdtool security does have the same symptoms, so I suppose that means the issue is in the security plugin?  Sorry, not terribly familiar with the project.
```
# G_DEBUG=fatal-criticals gdb --args fwupdtool security
GNU gdb (Debian 15.2-1) 15.2
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fwupdtool...
(No debugging symbols found in fwupdtool)
(gdb) r
Starting program: /usr/bin/fwupdtool security
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff2fff6c0 (LWP 261273)]
[New Thread 0x7ffff27fe6c0 (LWP 261274)]
[New Thread 0x7ffff1ffd6c0 (LWP 261275)]
[New Thread 0x7ffff17fc6c0 (LWP 261276)]
Loading…                 [**                                     ][New Thread 0x7fffebfff6c0 (LWP 261277)]
Loading…                 [************************************** ]
WARNING: UEFI firmware can not be updated in legacy BIOS mode
See https://github.com/fwupd/fwupd/wiki/PluginFlag:legacy-bios for more information.
22:34:07.275 FuPluginLinuxSleep   could not open /sys/power/mem_sleep: Error opening file /sys/power/mem_sleep: No such file or directory
22:34:07.354 GLib                 g_strsplit: assertion 'string != NULL' failed

Thread 1 "fwupdtool" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff77776da in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff77776da in g_logv () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff7777943 in g_log () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff7793e25 in g_strsplit () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7d90507 in ??? () at /usr/lib/x86_64-linux-gnu/fwupd-2.0.2/libfwupdengine.so
#4  0x00007ffff7ca2f88 in ??? () at /usr/lib/x86_64-linux-gnu/fwupd-2.0.2/libfwupdengine.so
#5  0x00007ffff7cb2646 in fu_engine_get_host_security_attrs () at /usr/lib/x86_64-linux-gnu/fwupd-2.0.2/libfwupdengine.so
#6  0x000055555556539e in ??? ()
#7  0x00007ffff7fb2a15 in fu_util_cmd_array_run () at /usr/lib/x86_64-linux-gnu/fwupd-2.0.2/libfwupdutil.so
#8  0x0000555555560924 in main ()
```

hughsie, 2024-11-26:
Can you install the fwupd debug symbols and do either of those again please. Thanks!

jingai, 2024-11-26:
fwupd with debug symbols:
```
Starting program: /usr/libexec/fwupd/fwupd 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff2fff6c0 (LWP 268884)]
[New Thread 0x7ffff21ff6c0 (LWP 268885)]
[New Thread 0x7ffff19fe6c0 (LWP 268886)]
[New Thread 0x7ffff11fd6c0 (LWP 268887)]
22:47:53.744 FuMain               Daemon ready for requests (locale en_US.UTF-8)
22:48:03.013 FuPluginLinuxSleep   could not open /sys/power/mem_sleep: Error opening file /sys/power/mem_sleep: No such file or directory
22:48:03.094 GLib                 g_strsplit: assertion 'string != NULL' failed

Thread 1 "fwupd" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff76626da in g_logv () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff76626da in g_logv () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff7662943 in g_log () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff767ee25 in g_strsplit () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff7d90507 in fu_msr_plugin_add_security_attr_amd_hwcr (plugin=0x55555563c140, attrs=0x5555555b3180) at ../plugins/msr/fu-msr-plugin.c:689
#4  fu_msr_plugin_add_security_attrs (plugin=0x55555563c140, attrs=0x5555555b3180) at ../plugins/msr/fu-msr-plugin.c:734
#5  0x00007ffff7ca2f88 in fu_engine_ensure_security_attrs (self=self@entry=0x5555555b0c50) at ../src/fu-engine.c:7069
#6  0x00007ffff7cb25e6 in fu_engine_get_host_security_id (self=self@entry=0x5555555b0c50) at ../src/fu-engine.c:7102
#7  0x000055555555f34f in fu_dbus_daemon_get_property
    (connection_=<optimized out>, sender=<optimized out>, object_path=<optimized out>, interface_name=<optimized out>, property_name=0x555556d31c90 "HostSecurityId", error=0x0, user_data=0x5555555aa300) at ../src/fu-dbus-daemon.c:2610
#8  0x00007ffff78c57d9 in ??? () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#9  0x00007ffff765781f in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#10 0x00007ffff7659a57 in ??? () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007ffff765a4af in g_main_loop_run () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x000055555555ec95 in fu_daemon_start (self=self@entry=0x5555555aa300, error=error@entry=0x7fffffffe228) at ../src/fu-daemon.c:239
#13 0x000055555555d497 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:213
```

superm1, 2024-11-26:
Try the linked PR.

jingai, 2024-11-26:
@superm1 Yep, works perfectly now.  Thank you very much!

closed 2024-11-27 by hughsie