fwupd/fwupd #8169: Can not install system firmware update on Lenovo ThinkEdge SE50 system.

Richard9926, 2024-11-28:
**Describe the bug**
I can't install the updates for a Lenovo ThinkEdge SE50. After execute the commend "fwupdmgr update", we find a new FW to install, but after click y to perform an update, it has a error "no valid release URIs".
[image]

**Steps to Reproduce**
1. Copy avc-embargo.conf to /etc/fwupd/remotes.d.
2. Run fwupdmgr refresh --force.
3. Run fwupdmgr update
4. Click y to perform update.

**Expected behavior**
System can install update.

**fwupd version information**
Please provide the version of the daemon and client.
fwupdmgr --version

runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios                2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                            6.8.0-40-generic
compile   com.hughsie.libjcat             0.1.9
compile   org.freedesktop.fwupd       1.7.9
runtime   org.freedesktop.gusb          0.3.10

```shell
fwupdmgr --version
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
fwupdmgr refresh --force
fwupdmgr update

<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
```
No efibootmgr commend in UEFI shell

```shell
efivar -l | grep fw
```
No efivar commend in UEFI shell

```shell
tree /boot
```
No tree commend in UEFI shell

**Additional questions**

- Operating system and version: Ubuntu 22.04.4 LTS
- Have you tried rebooting? Yes
- Is this a regression? 
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

Richard9926, 2024-11-28:
I am not sure the error message "no valid release URIs" means MetadataURI is incorrect or not in avc-embargo.conf. 
We checked that the capsule filename is firmware-ad0dc679b22ae4f9fd59372e42cc8f1282de5bbb6d45cc15a7cb8bf70db1ffe7.xml.### **xz**.
But the filename is one word different than that in avc-embargo.conf.
After rename MetadataURI to firmware-ad0dc679b22ae4f9fd59372e42cc8f1282de5bbb6d45cc15a7cb8bf70db1ffe7.xml.**### xz** in avc-embargo.conf, we can not found out updated system fw.

[fwupd Remote]
Enabled=true
Title=Embargoed for avc
RefreshInterval=3600
MetadataURI=https://fwupd.org/downloads/firmware-ad0dc679b22ae4f9fd59372e42cc8f1282de5bbb6d45cc15a7cb8bf70db1ffe7.xml.**### gz**
ReportURI=https://fwupd.org/lvfs/firmware/report
OrderBefore=lvfs,fwupd
# ↓ MODIFY ↓ -- the email address you use to login to the LVFS
Username=
# ↓ MODIFY ↓ -- a *USER TOKEN* from https://fwupd.org/lvfs/profile
Password=

hughsie, 2024-11-28:
I don't think you can use the xz or zstd metadata; I don't  your fwupd version of  1.7.9 went EOL a few months ago: https://github.com/fwupd/fwupd/blob/main/SECURITY.md

I'd advise force refreshing the cache `fwupdmgr refresh --force` with the remote set as `.gz` and then please get us the output of `fwupdmgr get-releases --json` so we can see what the daemon thinks.

Richard9926, 2024-11-29:
Hi Sir,
Attachment is the output log after I run fwupdmgr get-releases --json. Thanks.
[get-releases.txt](https://github.com/user-attachments/files/17953530/get-releases.txt)

hughsie, 2024-11-29:
> 2210@@fwupd.org

Can you follow https://lvfs.readthedocs.io/en/latest/testing.html#embargo-remotes -- it looks like you've used your user password rather than the user token. I've reset your password for you as you made it public in the log. Newer (non-obsolete) versions of fwupd wouldn't have printed the password in the log, so that's 2 reasons why you should upgrade...

Richard9926, 2024-12-05:
It can work after using the user token in user embargo file. Thanks.

closed 2024-12-05 by hughsie